kaloz/mwlwifi #148: Complile error with openwrt

stuwilkins, 2017-04-21:
Hi, 

I got a compile error when I tried with openwrt. Anyone know what I am doing wrong?

`make[3]: Entering directory `/Volumes/OpenWrt/openwrt/package/kernel/mwlwifi'
/Library/Developer/CommandLineTools/usr/bin/make   -C "/Volumes/OpenWrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/linux-4.4.14" ARCH="arm" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" SUBDIRS="/Volumes/OpenWrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170421" NOSTDINC_FLAGS="-I/Volumes/OpenWrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170421 -I/Volumes/OpenWrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/usr/include/mac80211-backport/uapi -I/Volumes/OpenWrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/usr/include/mac80211-backport -I/Volumes/OpenWrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/usr/include/mac80211/uapi -I/Volumes/OpenWrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/usr/include/mac80211 -include backport/backport.h" modules
  CC [M]  /Volumes/OpenWrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170421/utils.o
/Volumes/OpenWrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170421/utils.c: In function 'utils_find_vif_bss':
/Volumes/OpenWrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170421/utils.c:140:7: error: implicit declaration of function 'ether_addr_equal' [-Werror=implicit-function-declaration]
   if (ether_addr_equal(bssid, mwl_vif->bssid)) {
       ^
cc1: some warnings being treated as errors
make[5]: *** [/Volumes/OpenWrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170421/utils.o] Error 1
make[4]: *** [_module_/Volumes/OpenWrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170421] Error 2
make[3]: *** [/Volumes/OpenWrt/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170421/.built] Error 2
make[3]: Leaving directory `/Volumes/OpenWrt/openwrt/package/kernel/mwlwifi'
make[2]: *** [package/kernel/mwlwifi/compile] Error 2
make[2]: Leaving directory `/Volumes/OpenWrt/openwrt'
make[1]: *** [/Volumes/OpenWrt/openwrt/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/Volumes/OpenWrt/openwrt'
make: *** [world] Error 2`

Any help appreciated

yuhhaurlin, 2017-04-22:
This version of kernel does not support "ether_addr_equal"?

stuwilkins, 2017-04-22:
@yuhhaurlin 

I found that I needed to add `#include <linux/etherdevice.h>` to some files to get it to compile with openwrt (which it has just done successfully.

Below is the diff I had to do.

```
diff -rw mwlwifi-10.3.0.17-20160601/debugfs.c /home/swilkins/mwlwifi/debugfs.c
27d26
< #include <linux/etherdevice.h>
diff -rw mwlwifi-10.3.0.17-20160601/hif/pcie/pcie.c /home/swilkins/mwlwifi/hif/pcie/pcie.c
29d28
< #include <linux/etherdevice.h>
diff -rw mwlwifi-10.3.0.17-20160601/utils.c /home/swilkins/mwlwifi/utils.c
21d20
< #include <linux/etherdevice.h>
```

yuhhaurlin, 2017-04-22:
All right. I will check and add it.

closed 2017-04-22 by yuhhaurlin