kaloz/mwlwifi #150: rng: feed some entropy to the kernel

seragh, 2017-04-26:
The new driver causes at least an order of magnitude less interrupts on a Rango (which is a good thing), the downside is not enough entropy is generated any longer to keep the pool in a healthy state without the use of haveged or a similar tool.

Extend mwlwifi to optionally feed some entropy to the kernel. As an example may serve mainline commit ed14dc0af7cc ("ath9k: feeding entropy in kernel from ADC capture")

borkra, 2017-04-26:
Armada 385 on Rango has hardware random number generator. It should generate all kind of entropy without driver help, so starving should not be an issue. My guess, you have or misconfigured kernel, or need to install rngtools. But my bet is on misconfigured kernel.

seragh, 2017-04-27:
rngd would be the alternative indeed. My first reaction was simply to set up haveged. The point is with the old driver plenty enough entropy is generated even if the device is idle without any rng daemons running. This is no longer the case.

borkra, 2017-04-27:
Sorry looks no hw rng at least no driver for it I can find.

yuhhaurlin, 2017-04-27:
I think I will close this one. Thanks.

borkra, 2017-04-27:
It seems Marvell went a different route and instead of providing general purpose HW RNG they provided HW support for generating iv. 
 
I am guessing here but adding cryptodev should resolve the issue as iv is what actually needed. I remember having entropy problems early on, but not any more even with the new driver.

seragh, 2017-04-27:
On Thu, 27 Apr 2017 01:00:19 -0700
yuhhaurlin <notifications@github.com> wrote:

> I think I will close this one. Thanks.
> 

What's the rational? There is no hurry, so could stay open for years if
you never get around to it. But unless undesirable I don't see why
you'd want to close this bug.

Neither haveged, rng-tools nor cryptodev (for some consumers) are in
OpenWrt base and as such can't simply be added to the default package
set. So I expect plenty people having their pool drained most of the
time.

The other aspect is a wifi module can reasonably be expected to be
usable as a good entropy source and there are examples of other drivers
doing the same. So it's not a far fetched idea either.

Fr3DBr, 2017-04-27:
@seragh https://github.com/openwrt/packages/tree/master/utils/rng-tools

hnyman, 2017-04-27:
+1 for this request. 
The scarcity of natural entropy sources in a headless device has always been a problem. And the ath9k solution of using wifi driver's raw ADC data as an entropy source is a nice one.

Some historical perspective: 
* in ancient times, all/most network drivers in Linux were used as entropy sources but that was dropped somewhere in 2008-2010 (didn't look up the exact commits). 
* That created a problem for embedded devices. See discussion e.g. in https://dev.openwrt.org/ticket/9631 
* Openwrt developers then designed the original ath9k solution already in 2013 and a derivative of that got mainlined in 2015.

Original version:
https://dev.openwrt.org/browser/trunk/package/kernel/mac80211/patches/550-ath9k_entropy_from_adc.patch?rev=38486

mainlined derivative:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?h=linux-4.9.y&id=ed14dc0af7ccea867b479feb88efdfe43ca2a0f9

It would be great if mwlwifi would act similarly as ath9k.

Just to prove the point, below are entropy charts from a mwlwifi WRT3200ACM with mwlwifi bd62e37 and an ath9k based router.

[image]

[image]

@Fr3DBr
rng-tools can be used to pump aadditional entropy into the kernel entropy pool, but like seragh already said, it is not in the core Openwrt (or LEDE) source repos and can not be part of the default package set.

Fr3DBr, 2017-04-27:
@hnyman Well, if this is done via hardware/drivers it's much better of course, but the question is, how long would this take to be implemented ? 👍

Fr3DBr, 2017-04-27:
@hnyman  maybe adding support to the crypto chip included in this solution, should solve the problem ?

borkra, 2017-04-27:
There is already support for crypto chip. The issue is to make hostapd to use it.

For mvebu this means use of OpenSSL and cryptodev. 

The second problem that at least lede patch mac80211 to not use kernel cryptoapi and as such not use security chip. Why is it done beats me.

Fr3DBr, 2017-04-27:
@borkra Well, this is how technology works, every day something new or new needs will show up, and then we'll must change/adapt to it. And well, since technology moves fast, or at least, most of the times it's like that, then selecting the shortest path to solve problems when possible is also cool, now which is the shortest/best path in this case, depends upon all the developers over here and their priorities. :) 👍

BrainSlayer, 2017-04-27:
@borkra you dont need hw crypto for hostapd. since all crypto is done in the wifi hw already. so this isnt a performance issue. for openssl you may use cryptodev etc. but it suffer from context switches and wont gain the same performance as in kernel space

Chadster766, 2017-04-27:
It seems to me the OS should take care of the entropy issue. 

In McDebian the below command's output is +3000:
`cat /proc/sys/kernel/random/entropy_avail`

borkra, 2017-04-27:
@BrainSlayer The issue is that Marvell CESA does not provide hardware RNG but they do provide hardware IV generation. So to tap into that and as such reduce use of RNG OpenSSL and cryptodev proposed.

BrainSlayer, 2017-04-27:
@borkra but that isnt related to hostapd. cesa is not required for any wifi authentication and encryption. this is all be done in the wifi hw itself. no RNG or IV required. you can run hostapd without crypto library support but psk etc will still work. only for wpa enterprise you need crypto support for radius authentication stuff. but at the end the the full crypto flow will be done by the wifi hw, so there is no cpu bootleneck

borkra, 2017-04-27:
@BrainSlayer Agree partially.,  We deviated from original problem. From what I recall and hostapd is the one putting out low entropy error. And it does perform few encryption operations. Mostly related to beacons. I actually traced it once.

BrainSlayer, 2017-04-27:
@borkra its not hostapd. its the kernel since hostapd is accessing dev/random and there is also a easy fix to it. just spitting for entropy into it from various sources like irq's and  whatelse is going on

Fr3DBr, 2017-04-27:
@BrainSlayer Is this a current issue on dd-wrt, or only in openwrt & other variants ?

BrainSlayer, 2017-04-27:
@Fr3DBr which issue? there is no issue. people are just discussing about low entropy in the kernel which is available for applications like hostapd and i'm trying to tell them that the wifi hw does not use any local random data or encryption code for wifi security its all enclosed in the wifi chipset

Fr3DBr, 2017-04-27:
@BrainSlayer this issue with low entropy vs hostapd, does it affect dd-wrt currently ?

BrainSlayer, 2017-04-27:
low entropy vs kernel. its not hostapd. it does affect openwrt,dd-wrt,linksys and most routers in this world without a hardware number generator which uses cosmic rays as entropy. 
but it wont decrease your security since no local entropy source is used for wpa psk encryption. the wifi chipset has a own cpu, own OS and own firmware which is provided by marvell. so forget about it.

Fr3DBr, 2017-04-27:
@BrainSlayer And how do you plan to address this @ dd-wrt ?

BrainSlayer, 2017-04-27:
i have no plan for this. see my explaination. there is no reason to increase entropy and there is no way to increase entropy in reality. it wont change the WIFI HARDWARE OS / MARVELL FIRMWARE. the marvell firmware is propertiery and provided by the wifi vendor. its a black box. beside this the wifi chipset itself is handling all crypto operations. no entropy from hostapd used. but i really dont know why i tell you this now the fourth time. you does not seem to be able to understand or you simply do not read what i wrote

Fr3DBr, 2017-04-27:
@BrainSlayer  I know, I'm not speaking about the wifi driver, but actually just in general get it ? But it seems, entropy isn't something that makes a difference. :) Unless for VPN maybe ?

yuhhaurlin, 2017-04-27:
Sorry. I will close this one.

closed 2017-04-27 by yuhhaurlin

BrainSlayer, 2017-04-27:
@Fr3DBr thats right. but entropy cannot come out of nothing. the wrt3200acm has no crypto / hw random chip as it seems unlike the older wrt models

BrainSlayer, 2017-04-27:
@yuhhaurlin thank you :-)

seragh, 2017-04-28:
On Thu, 27 Apr 2017 14:13:13 -0700
Sebastian Gottschall <notifications@github.com> wrote:

> @yuhhaurlin thank you :-)
> 

You were rather unhelpful in keeping this issue on topic.

---

Yuhhaurlin,

you were right in closing this one the first time. Once more we got a
demonstration of what is wrong with github.

Won't file any more bugs here and just wait for the driver to make it
upstream.

Sorry and thanks for your incredible patience.

---

To sum up. We lost a source of entropy, some of us need to compensate
it, I choose haveged, pick your own poison. This bug was about enabling
a cheep and good source of entropy. This is not at all a required
feature of mwlwifi, just something that would have been nice to have
had, so be it. If still not clear reread hnyman's post until it's clear.

yuhhaurlin, 2017-04-28:
Even though I close it, you can still post here. I think I will not check this one until mwlwifi for 88W8964 is ready and works well. Thanks.

Fr3DBr, 2017-04-28:
@seragh Well, if the driver generated entropy, it would help to keep the firmware as clean as possible, this is what I get from all this. In case they do it, indeed it will be nice. And as you said it's not "required" but thinking on clean things, indeed it would be nice and handy as other drivers are already doing. 👍