kaloz/mwlwifi #151: WRT3200ACM-EU new driver powertable discrepancy

inteliboy, 2017-04-29:
With LEDE and latest mwlwifi driver on WRT3200ACM-EU version the power table and available channels are different from the latest Linksys stock software. It would either suggest that the power table can be manipulated despite being fixed in EEPROM or that there is some issue with the new driver.

LEDE:
```
root@rango:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/device_pwrtbl

power table loaded from dts: no
firmware region code: 0x30
number of channel: 33
  1 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  2 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  3 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  4 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  5 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  6 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  7 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  8 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
  9 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 10 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 11 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 12 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 13 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 14 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 36 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 40 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 44 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 48 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   0  15   0
 52 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
 56 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
 60 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
 64 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
100 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
104 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
108 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
112 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
116 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
120 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
124 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
128 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
132 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
136 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
140 252 254   0   2   4   5   6   7   8   9  10  11  12  13  14  15   1  15   0
```
Linksys 1.0.5.179030 data form sysinfo.cgi diagnostics:
```
Power table for 2.4GHz
1 0 0x17 0x10 0x10 0x10 0x0F 0x0F 0x0F 0x0F 0x0E 0x0E 0x0E 0x0E 0x00 0x00 0x00 0x00 off 0x00F
2 0 0x18 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x11 0x11 0x11 0x11 0x00 0x00 0x00 0x00 off 0x00F
3 0 0x18 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x11 0x11 0x11 0x11 0x00 0x00 0x00 0x00 off 0x00F
4 0 0x18 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x11 0x11 0x11 0x11 0x00 0x00 0x00 0x00 off 0x00F
5 0 0x18 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x11 0x11 0x11 0x11 0x00 0x00 0x00 0x00 off 0x00F
6 0 0x18 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x11 0x11 0x11 0x11 0x00 0x00 0x00 0x00 off 0x00F
7 0 0x18 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x11 0x11 0x11 0x11 0x00 0x00 0x00 0x00 off 0x00F
8 0 0x18 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x11 0x11 0x11 0x11 0x00 0x00 0x00 0x00 off 0x00F
9 0 0x18 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x11 0x11 0x11 0x11 0x00 0x00 0x00 0x00 off 0x00F
10 0 0x18 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x11 0x11 0x11 0x11 0x00 0x00 0x00 0x00 off 0x00F
11 0 0x17 0x12 0x12 0x12 0x13 0x13 0x13 0x13 0x0F 0x0F 0x0F 0x0F 0x00 0x00 0x00 0x00 off 0x00F
Power table for 5GHz
36 0 0x12 0x12 0x12 0x12 0x12 0x12 0x12 0x12 0x10 0x10 0x10 0x10 0x0F 0x0F 0x0F 0x0F off 0x00F
40 0 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x0F 0x0F 0x0F 0x0F off 0x00F
44 0 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x0F 0x0F 0x0F 0x0F off 0x00F
48 0 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x0F 0x0F 0x0F 0x0F off 0x00F
149 0 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 off 0x00F
153 0 0x16 0x16 0x16 0x16 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 off 0x00F
157 0 0x16 0x16 0x16 0x16 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 off 0x00F
161 0 0x16 0x16 0x16 0x16 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 off 0x00F
165 0 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 off 0x00F
```

yuhhaurlin, 2017-04-29:
For mwlwifi：
1. It display raw data of device power table.
2. The way and the sequence to display power table is different.

yuhhaurlin, 2017-04-29:
Maybe I should check it. Thanks.

starcms, 2017-04-30:
What's up with this?  This is on a WRT1200AC V1.  Is this normal?  Everything works fine.  Driver is with all the very latest commits on @davidc502's latest build

root@WRT1200AC:/sys/kernel/debug/ieee80211/phy0/mwlwifi# cat /sys/kernel/debug/i
eee80211/phy0/mwlwifi/device_pwrtbl

power table loaded from dts: yes
firmware region code: 0x0
number of channel: 0

inteliboy, 2017-04-30:
Your WRT1200AC V1 is simply not locked to certain region and power limits and the power table is adjusted via data in DTS file.

starcms, 2017-04-30:
@inteliboy Ok, so since my model doesn't have the power table built into the eeprom, it's normal for ```
cat /sys/kernel/debug/i
eee80211/phy0/mwlwifi/device_pwrtbl
```
to show 0 which I'm guessing it's reading directly from the eeprom?

Just curious, where is the DTS file located?

Thanks!!

inteliboy, 2017-04-30:
Yes, its fine. No idea where it is in compiled image, I guess it's built into the kernel. In source for example here  `https://github.com/lede-project/source/tree/master/target/linux/mvebu/files/arch/arm/boot/dts`

yuhhaurlin, 2017-04-30:
It is only display problem. Device with power table will not be affected by driver for power setting.

inteliboy, 2017-05-01:
I still believe there might be a way to influence the power output somehow as while analyzing the stock Linksys fimrware file in a hex-editor i stumbled upon references to such file 'W8964-eeprom.bin' and in sysinfo.cgi disgnostic file 'wl0_power_table=/etc/24G_power_table_CE'. If only I knew how to decompress the image file maybe there would be more to find out.

BrainSlayer, 2017-05-01:
@inteliboy use the linux tool binwalk. it can extract almost any image type including the wrt3200acm

BrainSlayer, 2017-05-01:
bit it wont help you. the power tables are only made for devices without preprogrammed eeprom. if a preprogrammed eeprom has been found , the wifi hw will ignore any external power table. for sure there is still a possibility to modify the firmware image, but that might be very complicated and also unwanted by marvell or any of the developers here

gufus, 2017-08-03:
@inteliboy 

Mamba_FCC_v1.2_5G4TX

1 0 0x17 0x11 0x11 0x11 0x10 0x10 0x10 0x10 0x0F 0x0F 0x0F 0x0F 0x00 0x00 0x00 0x00 off 0x00F
2 0 0x17 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
3 0 0x17 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
4 0 0x17 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
5 0 0x17 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
6 0 0x17 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
7 0 0x17 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
8 0 0x17 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
9 0 0x17 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
10 0 0x17 0x16 0x16 0x16 0x16 0x16 0x16 0x14 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
11 0 0x17 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
12 0 0x17 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
13 0 0x17 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
14 0 0x17 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x10 0x10 0x10 0x10 0x00 0x00 0x00 0x00 off 0x00F
36 0 0x08 0x08 0x08 0x08 0x08 0x08 0x08 0x08 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 off 0x00F
40 0 0x08 0x08 0x08 0x08 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 off 0x00F
44 0 0x08 0x08 0x08 0x08 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 off 0x00F
48 0 0x08 0x08 0x08 0x08 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 0x09 off 0x00F
52 0 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x12 0x12 0x12 0x12 0x12 0x12 0x12 0x12 off 0x00F
56 0 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x12 0x12 0x12 0x12 0x12 0x12 0x12 0x12 off 0x00F
60 0 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x12 0x12 0x12 0x12 0x12 0x12 0x12 0x12 off 0x00F
64 0 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x0F 0x12 0x12 0x12 0x12 0x12 0x12 0x12 0x12 off 0x00F
100 0 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 off 0x00F
104 0 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 off 0x00F
108 0 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 off 0x00F
112 0 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 off 0x00F
116 0 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 off 0x00F
120 0 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 off 0x00F
124 0 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 off 0x00F
128 0 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 off 0x00F
132 0 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 off 0x00F
136 0 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 off 0x00F
140 0 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 0x14 off 0x00F
149 0 0x16 0x16 0x16 0x16 0x14 0x14 0x14 0x14 0x15 0x15 0x15 0x15 0x14 0x14 0x14 0x14 off 0x00F
153 0 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x14 0x14 0x14 0x14 off 0x00F
157 0 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x14 0x14 0x14 0x14 off 0x00F
161 0 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x15 0x14 0x14 0x14 0x14 off 0x00F
165 0 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x16 0x15 0x15 0x15 0x15 0x14 0x14 0x14 0x14 off 0x00F

ValCher1961, 2018-04-12:
That's what is in rootfs wrt32x, this is how I think loadable power tables.
```
root@wrt32x:/etc/mrvl# ls
129.ini  144.ini  145.ini  16.ini  32.ini  48.ini
```

BrainSlayer, 2018-04-22:
both are identical and from the things you found here it seems that marvell is not telling the truth at all. it seems that its possible to override the internal power table. i'm just not sure if these ini files are usually just for prototypes and if it works with production models

ValCher1961, 2018-04-22:
I think somewhere there is this point to change the power table, but where it is...

ValCher1961, 2018-04-22:
Yes, I also found this entry, but this is not enough, because you said correctly - propertary driver :) 
I can not understand why Marvell/Linksys does not allow the user to change it within the allowed boundaries, it can only be turned on and off.

lantis1008, 2018-04-23:
Slightly off-topic
@BrainSlayer have you got a working dd-wrt build for wrt32x yet? If not, what is the barrier to this?

BrainSlayer, 2018-04-23:
i have a working one. its just not yet possible to install it with web. just from commandline. the blocking point is that i was never able to test it on a original device since i dont own one and also got no usefull feedback yet from owners how to deal with it. from the firmware format it doesnt seem to be a problem. without having access to a test device its hard to deal with it

BrainSlayer, 2018-04-23:
@ValCher1961 : there is no legal reason for it. they may just handle it with low priority. unlike other vendors they just offer whats needed. who knows what else is possible with it

ValCher1961, 2018-04-23:
@BrainSlayer 
I think you're right.
-----
In the continuation of the non-main topic.
Unlike us, for many users the connection to the serial cable router and the command line does not cause much enthusiasm, they like to access only through the web-interface.
I also tried to make OpenWRT / LEDE for wrt32x / wrt3200, but not for flashing in the NAND router, but for downloading from USB-flash. Also I can not check the result, since there is no wrt32x, wrt3200 works.

BrainSlayer, 2018-04-23:
the wrt32x is identical to the wrt3200. it just uses a different partition layout and the webif flashing may use a signature check. but commandline flashing works

lantis1008, 2018-04-23:
I have a device and am willing to assist either of you. 
I would offer to send it to you if I was a little closer.

UI definitely does basic firmware magic check. I produced an image which would pass this, but not boot.

BrainSlayer, 2018-04-23:
a remote serial access and web access would be enough. so i'm able todo all the test and see also whats happening on console

ValCher1961, 2018-04-23:
Can we open a new topic or go to an already open profile topic?

lantis1008, 2018-04-23:
Ok. I will organise a serial cable and then remote access. It will take me a week or so. 

Yes, is there a better place we can discuss this?

ValCher1961, 2018-04-23:
Probably on DD-WRT, if not against @BrainSlayer.  
There's a similar theme.

BrainSlayer, 2018-04-23:
best is writing me a email

ryan-dev, 2018-04-24:
For those that have used the wrt32x, is using the proprietary driver worth
it.

Thanks to the work of Sebastian, we all know the WRT3200ACM and WRT32X are
identical just merely a different software/partitioning wrapper plus
exterior color.

I personally love both OpenWRT and DD-WRT.

>

yuhhaurlin, 2018-04-24:
I will modify mwlwifi to load power table from DTS later. However, if your device is programmed with device power table, loading of power table will not take effect. This job is tracked via issue #280.

yuhhaurlin, 2018-04-24:
The same rule is also applied for WRT32X.

BrainSlayer, 2018-04-27:
send me a direct email to s.gottschall@dd-wrt.com everything else will be received by our staff and usually forwarded to me if there any open questions

lantis1008, 2019-04-12:
Hey @BrainSlayer , i'm helping a user at the moment who has a WRT32X which is reporting the following info:
```
power table loaded from dts: no
firmware region code: 0x30
```
And when looking at mtd3 it shows `EU` as the cert region, which doesn't match up against any of the allowed values in the driver (although, 0x30 lines up with `FR`). How are you handling this in ddwrt? He's saying that its working ok in your firmware but no dfs at all in Openwrt. We've tried no countrycode, FR, EU (invalid), GB, 00 without much success.
Radio 2 (mwifiex) isn't involved here and isn't even installed.