kaloz/mwlwifi #158: WRT3200ACM mwlwifi latest hostapd invoked oom-killer

Chadster766, 2017-05-06:
May 3, 2017 mwlwifi "7e29d87ebb5ac82213a8b20677538cab5dafa20b" testing on McDebian 4.9.26:

After iperf3 transfer of a little more than 103GB WLAN to WAN this error journal entry was recorded:

```
May 05 19:33:13 MCDEBIAN kernel: hostapd invoked oom-killer: gfp_mask=0x24201ca(GFP_HIGHUSER_MOVABLE|__GFP_COLD), nodemask=0, order=0, oom_score_adj=0
May 05 19:33:17 MCDEBIAN kernel: hostapd cpuset=/ mems_allowed=0
May 05 19:33:20 MCDEBIAN kernel: CPU: 1 PID: 2760 Comm: hostapd Not tainted 4.9.26 #22
May 05 19:33:23 MCDEBIAN kernel: Hardware name: Marvell Armada 380/385 (Device Tree)
May 05 19:33:23 MCDEBIAN kernel: [<c010f658>] (unwind_backtrace) from [<c010b404>] (show_stack+0x10/0x14)
May 05 19:33:24 MCDEBIAN kernel: [<c010b404>] (show_stack) from [<c033210c>] (dump_stack+0x90/0xa4)
May 05 19:33:28 MCDEBIAN kernel: [<c033210c>] (dump_stack) from [<c01f58cc>] (dump_header+0xa0/0x1dc)
May 05 19:33:30 MCDEBIAN kernel: [<c01f58cc>] (dump_header) from [<c01b2be0>] (oom_kill_process+0x448/0x4cc)
May 05 19:33:33 MCDEBIAN kernel: [<c01b2be0>] (oom_kill_process) from [<c01b2f70>] (out_of_memory+0x118/0x330)
May 05 19:33:36 MCDEBIAN kernel: [<c01b2f70>] (out_of_memory) from [<c01b6e80>] (__alloc_pages_nodemask+0xb5c/0xcb8)
May 05 19:33:40 MCDEBIAN kernel: [<c01b6e80>] (__alloc_pages_nodemask) from [<c01b12d8>] (filemap_fault+0x470/0x614)
May 05 19:33:43 MCDEBIAN kernel: [<c01b12d8>] (filemap_fault) from [<c026ddbc>] (ext4_filemap_fault+0x2c/0x40)
May 05 19:33:46 MCDEBIAN kernel: [<c026ddbc>] (ext4_filemap_fault) from [<c01d24d8>] (__do_fault+0x6c/0xec)
May 05 19:33:48 MCDEBIAN kernel: [<c01d24d8>] (__do_fault) from [<c01d61dc>] (handle_mm_fault+0x574/0xd20)
May 05 19:33:51 MCDEBIAN kernel: [<c01d61dc>] (handle_mm_fault) from [<c0114c70>] (do_page_fault+0x25c/0x33c)
May 05 19:33:54 MCDEBIAN kernel: [<c0114c70>] (do_page_fault) from [<c01013b4>] (do_PrefetchAbort+0x34/0x98)
May 05 19:33:58 MCDEBIAN kernel: [<c01013b4>] (do_PrefetchAbort) from [<c010c508>] (ret_from_exception+0x0/0x18)
May 05 19:34:00 MCDEBIAN kernel: Exception stack(0xdc087fb0 to 0xdc087ff8)
May 05 19:34:03 MCDEBIAN kernel: 7fa0:                                     00000007 bebf8b2c 7f625024 72d77100
May 05 19:34:05 MCDEBIAN kernel: 7fc0: 00000007 bebf8b2c b6fd8f10 7f625024 7f625024 00000000 bebf8b6c 7f56f425
May 05 19:34:08 MCDEBIAN kernel: 7fe0:
```

Chadster766, 2017-05-06:
I've created a McDebian hostapd 2.6 update and will report testing results shortly.

Chadster766, 2017-05-07:
After +100GB WLAN to WAN and WLAN to LAN I was not able to reproduce the issue with kernel 4.9.26 and hostapd 2.6. It seems as though no memory leak exists in the new mwlwifi driver.

```
root@MCDEBIAN:~# uname -a
Linux MCDEBIAN 4.9.26 #22 SMP Fri May 5 10:12:37 CDT 2017 armv7l GNU/Linux
root@MCDEBIAN:~# hostapd -v
hostapd v2.6
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2016, Jouni Malinen <j@w1.fi> and contributors
root@MCDEBIAN:~#
```

yuhhaurlin, 2017-05-07:
OOM is tracking by another issue. I will close this one.

closed 2017-05-07 by yuhhaurlin

BrainSlayer, 2017-05-07:
@Chadster766 : the memory leak exists but is only present with multiple wifi cards

Chadster766, 2017-05-07:
@BrainSlayer I will continue testing by putting it into use in my house. The last time I did this the WRT3200ACM mwlwifi started failing after a couple of days.

BrainSlayer, 2017-05-07:
@Chadster766 i believe the current memoryleak makes it useless todo a long term test right now. i trigger the out of memory condition in 30 seconds with a iphone 6

Chadster766, 2017-05-07:
Ok I've put the WRT3200ACM in play. I will report results in a few days.

Chadster766, 2017-05-07:
@BrainSlayer I have a iPhone 7 how can I trigger the issue in 30 seconds? It would save me a lot of time :smiley:

Chadster766, 2017-05-07:
I'm enabling the Kernel Memory Leak Detector. I've never used this kernel hacking feature before.

BrainSlayer, 2017-05-07:
enable 2.4 ghz, enable 5 ghz. (80 or 160 mhz. whatever)
login with your iphone to the 5 ghz interface.
run speedtest app from the appstore. byebye router