kaloz/mwlwifi #159: Beacon omits MDIE when configured via hostapd

mjaustin2, 2017-05-10:
I am running the following hostapd configuration on a WRT1900ACSv2. The configuration below works fine on other models with, for instance ATH9k wireless chips, however the 88W8864 in the WRT1900ACSv2 fails to add the Mobility Domain Information Element (MDIE) to the Beacon frame when it is configured via hostapd. This cause many clients to fail their association. What needs to be done to allow hostapd to pass the beacon frame to the driver with the MDIE intact?

```
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
hw_mode=g
channel=4

ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]

interface=wlan1
ctrl_interface=/var/run/hostapd
ap_isolate=1
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=somePSKphrase
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=RoamIO
bridge=br-lan
nas_identifier=ap0wlan0
mobility_domain=beef
r0_key_lifetime=10000
r1_key_holder=deadbeefcafe
reassociation_deadline=1000
pmk_r1_push=1
r0kh=de:ad:be:ef:ca:fe ap0wlan0 b026324c6904b2a9cb4b88d6d61c81d1
r0kh=de:ad:be:ef:4d:ad ap1wlan0 b026324c6904b2a9cb4b88d6d61c81d1
r1kh=de:ad:be:ef:ca:fe de:ad:be:ef:ca:fe b026324c6904b2a9cb4b88d6d61c81d1
r1kh=de:ad:be:ef:4d:ad de:ad:be:ef:4d:ad b026324c6904b2a9cb4b88d6d61c81d1
wpa_key_mgmt=WPA-PSK FT-PSK
okc=0
disable_pmksa_caching=1
bssid=de:ad:be:ef:ca:fe
```

The beacon that is produced advertises "FT using PSK" as an available AKM, however, without the Mobility Domain Information Element, "FT using PSK" cannot be used successfully. This causes some confusion for many devices, as they will attempt to use "FT using PSK" by default and send a null MDIE in response, causing an association failure due to "Invalid MDIE."

```
Frame 1: 213 bytes on wire (1704 bits), 213 bytes captured (1704 bits)
Radiotap Header v0, Length 18
802.11 radio information
IEEE 802.11 Beacon frame, Flags: ........
IEEE 802.11 wireless LAN management frame
    Fixed parameters (12 bytes)
        Timestamp: 0x0000000ecde48006
        Beacon Interval: 0.102400 [Seconds]
        Capabilities Information: 0x0431
            .... .... .... ...1 = ESS capabilities: Transmitter is an AP
            .... .... .... ..0. = IBSS status: Transmitter belongs to a BSS
            .... ..0. .... 00.. = CFP participation capabilities: No point coordinator at AP (0x00)
            .... .... ...1 .... = Privacy: AP/STA can support WEP
            .... .... ..1. .... = Short Preamble: Allowed
            .... .... .0.. .... = PBCC: Not Allowed
            .... .... 0... .... = Channel Agility: Not in use
            .... ...0 .... .... = Spectrum Management: Not Implemented
            .... .1.. .... .... = Short Slot Time: In use
            .... 0... .... .... = Automatic Power Save Delivery: Not Implemented
            ...0 .... .... .... = Radio Measurement: Not Implemented
[... 300 lines cut ...]
            Ac Parameters ACI 2 (Video), ACM no, AIFSN 2, ECWmin/max 3/4 (CWmin/max 7/15), TXOP 94
                ACI / AIFSN Field: 0x42
                    .10. .... = ACI: Video (2)
                    ...0 .... = Admission Control Mandatory: No
                    .... 0010 = AIFSN: 2
                    0... .... = Reserved: 0
                ECW: 0x43
                    0100 .... = ECW Max: 4
                    .... 0011 = ECW Min: 3
                    CW Max: 15
                    CW Min: 7
                TXOP Limit: 94
            Ac Parameters ACI 3 (Voice), ACM no, AIFSN 2, ECWmin/max 2/3 (CWmin/max 3/7), TXOP 47
                ACI / AIFSN Field: 0x62
                    .11. .... = ACI: Voice (3)
                    ...0 .... = Admission Control Mandatory: No
                    .... 0010 = AIFSN: 2
                    0... .... = Reserved: 0
                ECW: 0x32
                    0011 .... = ECW Max: 3
                    .... 0010 = ECW Min: 2
                    CW Max: 7
                    CW Min: 3
                TXOP Limit: 47
```

mjaustin2, 2017-05-10:
Interestingly enough, I can get the MDIE to show up in the probe response, but not in the beacon.

derosier, 2017-05-10:
Without bothering to look at the current code w/re to the MDIE, I imagine this thread is relevant: [Re: [PATCH v9] Add new mac80211 driver mwlwifi.](http://marc.info/?l=linux-wireless&m=148412673828507&w=2)

In other words, it may be a consequence of the driver caring/parsing IEs instead of passing them through as a thin-mac driver should, and illustrates Johaness's concern.

But, as I said, I didn't look a the current code here so I might be wrong. But maybe it gives you an idea of a place to look.

yuhhaurlin, 2017-05-10:
Yes. I will check it later. Thanks.

mjaustin2, 2017-05-13:
I perused the code, as best I can tell, that is exactly what is happening. I don't get the response on the Linux Wireless mailing list that this is to allow using the same firmware for the proprietary and OSS driver. The MDIE is required to support 802.11r, which according the datasheet from Marvell, is supported on the 88W8864. If Marvell says it's supported, then presumably firmware would support setting an MDIE in the beacon.

The fact that the Marvell driver is hard-coding support for specific IE instead of passing them is slightly concerning. Had I been aware of that, I might have purchased devices with a different chip. It certainly wouldn't be good for the Marvell chip's reputation within the community, which is a shame, because the wrt1900ACSv2 is one of the best performing devices that I've tested.

Hopefully, this is a solvable issue without causing too much pain to the maintainers.

Is there documentation of the firmware's commands? I did some googling, but didn't find anything on first glance.

yuhhaurlin, 2017-05-13:
For next generation, the API will be changed to resolve this kind of problem.
I will modify code to support it later.

bbogey, 2017-06-10:
Ok this maybe the reason I could not get FT working. You found any workaround for this?

mjaustin2, 2017-06-12:
Unfortunately no. At this point, I'm waiting for the maintainer to update the driver.

yuhhaurlin, 2017-06-12:
I will work for it later when most remaining features are done. In fact,  you can modify mwlwifi to parse this IE and add it to proprietary IE of set IEs firmware command.

dcd, 2017-06-12:
this one is killing me too.  added an mwlwifi device to an 802.11r network and now clients drop and must reconnect frequently.  mwlwifi is unusable due to this issue.

cilix-lab, 2017-07-24:
Any news on this? Time frame? Is there a roadmap on how features will be added to mwlwifi?

dcd, 2017-07-29:
@yuhhaurlin hello?  can we get some attention on this bug?  would like to put the socs i purchased a year ago into service one of these days

yuhhaurlin, 2017-07-31:
Sorry. I will add this IE to beacon later.

seragh, 2017-08-04:
Adding this IE isn't a fix for this bug though, just a workaround. The driver really should just pass on any IE it doesn't know about / has no reason to handle.

yuhhaurlin, 2017-08-04:
This will be enhanced for next chip.

bbogey, 2017-08-07:
Workaround still will be implemented? I really liked to get 802.11r fast roaming working.

dcd, 2017-08-07:
The workaround is buy atheros...

thagabe, 2017-08-07:
@yuhhaurlin next chip?

gufus, 2017-08-07:
Sure, you know how it works, once the bugs are fixed... a V2 will be next.

yuhhaurlin, 2017-08-08:
Sorry. I mean I will add the code to support this IE. For our next generation chip, firmware API will be enhanced to avoid this problem.

thagabe, 2017-08-08:
or firmware can be fixed now... test it works, then re-implemented on future chips

yuhhaurlin, 2017-08-08:
I can't make this kind of decision.

jvicondoa, 2017-08-17:
I really need this too. I can't use 802.11r without it.

Chadster766, 2017-08-31:
I thought the wireless chipset in the WRT3200ACM supported 802.11r.

yuhhaurlin, 2017-09-01:
Yes. But it needs to information MDIE to firmware. Without this feature, AP still can work. I will add it later.

dcd, 2017-09-22:
what month/year is "later" ?  should we all just buy hardware from another vendor already?

bbogey, 2017-09-22:
Yes for people with multiple routers this is not an option...

dcd, 2017-10-22:
@yuhhaurlin can we even get a status?  this is ridiculous.  router is 2.5 years old now.

yuhhaurlin, 2017-10-23:
I will add the code later this month and let you do the test.

mjaustin2, 2017-11-06:
I was heartened to hear that this longstanding issue might be getting fixed. What update do you have on this matter? How close are we to having code that might need testing?

yuhhaurlin, 2017-11-06:
Soon.

yuhhaurlin, 2017-11-06:
Please help to check if https://github.com/kaloz/mwlwifi/commit/760f7bf1b8af6c899ccb14d47e8b987ed9a62adf resolves your problem.

yuhhaurlin, 2017-11-13:
I think MDE will be added to beacon with this commit. I close this one. If you still have problem, please let me know.

closed 2017-11-13 by yuhhaurlin

mjaustin2, 2017-11-15:
@yuhhaurlin : I was able to get this compiled over the weekend and did some testing last night. I'm glad to report that it is working successfully. With this release, 802.11r is working on as expected. Is there any more effort towards making the IE more generic, instead of hard coding them? This would allow the driver to better support new features hostapd without the kinds of changes and probably make the kernel maintainers happier, if there's any desire to get mwlwifi into the kernel by default.

yuhhaurlin, 2017-11-15:
@mjaustin2 

1. We will change the firmware API for our next generation chip. This problem has also been mentioned when I submitted the driver to upstream. However, we will not change the firmware API for current chips.
2. Yes. I have joined wireless workshop of NETDEV 2.2. I have checked with Kalle and Johannes, basically I will try to submit the driver to upstream early next year.

bbogey, 2017-11-15:
I can't get it working. It is not advertising FT.  Other router Archer C5 is ok.
Please see the screenshot:
[image]

I have full version of wapd installed.
What I should check?

bbogey, 2017-11-15:
@mjaustin2 can you please share your configs

bbogey, 2017-11-15:
Disabled other APs and now it advertises FT. It seems to work if having only one single AP enabled. 
In Archer I don't have that limitation.