kaloz/mwlwifi #160: mwlwifi crash 10.3.4.0-20170512

jacksonatl, 2017-05-13:
Using LEDE with Kernel 4.9.20 wireless crashes after a couple hours

hostapd v2.7-devel

mwlwifi driver info:
cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170512
firmware version: 0x09020105
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b9:c6:c2
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000000
radio: disable
iobase0: e0e00000
iobase1: e1080000
tx limit: 1024
rx limit: 16384

Kernel log: 
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.571042] mwifiex_sdio mmc0:0001:1: ===mwifiex driverinfo dump start===
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.578288] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.584645] mwifiex_sdio mmc0:0001:1: info: MWIFIEX VERSION: mwifiex 1.0 (15.68.7.p5) 
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.592621] mwifiex_sdio mmc0:0001:1: SDIO register dump start
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.592763] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.592841] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.592872] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.593554] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594226] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594296] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594326] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594487] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594519] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594638] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594668] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594695] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594827] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594874] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594930] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594937] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594941] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594944] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594947] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594950] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594953] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594981] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594985] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.594995] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595061] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595065] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595068] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595071] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595075] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595078] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595081] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595084] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595087] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595095] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595223] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595228] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595231] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
[... 9 lines cut ...]
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595403] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595406] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595409] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595412] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595415] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595418] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595421] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595430] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.595744] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.596398] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.596474] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.953157] mwifiex_sdio mmc0:0001:1: SDIO Func0 (0x0-0x9): 43 03 02 02 03 00 00 02 03 00 
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.961569] mwifiex_sdio mmc0:0001:1: SDIO Func1 (0x10-0x17): 00 00 00 00 00 00 00 00 
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.969683] mwifiex_sdio mmc0:0001:1: SDIO Func1: (0x8) c3 (0x58) 00 (0x5c) 08 (0x5d) 00 (0x60) 07 (0x61) 0c (0x62) 00 (0x64) 10 (0x65) 08 (0x66) 00 (0x68) 00 (0x69) 00 (0x6a) 00 
Sat May 13 10:11:14 2017 kern.info kernel: [ 6240.985880] mwifiex_sdio mmc0:0001:1: SDIO Func1 (0x90-0x9a): dc fe 6f 5a 40 04 00 e3 e3 02 00 
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.094732] mwifiex_sdio mmc0:0001:1: SDIO Func1 (0x90-0x9a): dc fe 6f 5a 40 04 00 e3 e3 02 00 
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.103534] mwifiex_sdio mmc0:0001:1: SDIO register dump end
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.109336] mwifiex_sdio mmc0:0001:1: ===mwifiex driverinfo dump end===
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.116017] mwifiex_sdio mmc0:0001:1: == mwifiex dump information to /sys/class/devcoredump start
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.125022] mwifiex_sdio mmc0:0001:1: == mwifiex dump information to /sys/class/devcoredump end
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.133786] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.140070] mwifiex_sdio mmc0:0001:1: info: shutdown mwifiex...
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.146435] mwifiex_sdio mmc0:0001:1: PREP_CMD: card is removed
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.152393] mwifiex_sdio mmc0:0001:1: Failed to delete mgmt IEs!
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.158424] mwifiex_sdio mmc0:0001:1: PREP_CMD: card is removed

Chadster766, 2017-05-13:
mwifiex_sdio is the other Blue Tooth wireless adapter in the WRT3200ACM not the mwlwifi driver.

jacksonatl, 2017-05-13:
Oh, does seem related to the wireless dropping, here is the rest of the log:
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.133786] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.140070] mwifiex_sdio mmc0:0001:1: info: shutdown mwifiex...
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.146435] mwifiex_sdio mmc0:0001:1: PREP_CMD: card is removed
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.152393] mwifiex_sdio mmc0:0001:1: Failed to delete mgmt IEs!
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.158424] mwifiex_sdio mmc0:0001:1: PREP_CMD: card is removed
Sat May 13 10:11:14 2017 daemon.notice netifd: Network device 'wlan2' link is down
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.164473] mwifiex_sdio mmc0:0001:1: Failed to stop the BSS
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.170533] br-lan: port 2(wlan2) entered disabled state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.177776] device wlan2 left promiscuous mode
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.182259] br-lan: port 2(wlan2) entered disabled state
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.187756] mwifiex_sdio mmc0:0001:1: PREP_CMD: card is removed
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.193759] mwifiex_sdio mmc0:0001:1: PREP_CMD: card is removed
Sat May 13 10:11:14 2017 kern.info kernel: [ 6241.199704] mwifiex_sdio mmc0:0001:1: PREP_CMD: card is removed
Sat May 13 10:11:15 2017 daemon.notice hostapd: wlan2: AP-STA-DISCONNECTED 8c:f5:a3:ba:41:be
Sat May 13 10:11:15 2017 daemon.notice hostapd: wlan2: AP-STA-DISCONNECTED 30:75:12:89:31:1d
Sat May 13 10:11:15 2017 daemon.notice hostapd: wlan2: AP-STA-DISCONNECTED 5c:96:9d:7f:4e:09
Sat May 13 10:11:15 2017 daemon.notice hostapd: wlan2: AP-STA-DISCONNECTED 84:16:f9:09:1a:2c
Sat May 13 10:11:15 2017 daemon.notice hostapd: wlan2: AP-STA-DISCONNECTED 3c:a9:f4:be:52:9c
Sat May 13 10:11:15 2017 daemon.notice hostapd: wlan2: AP-STA-DISCONNECTED a4:e4:b8:73:b8:fd
Sat May 13 10:11:15 2017 daemon.notice hostapd: wlan2: AP-STA-DISCONNECTED 20:c9:d0:1a:8f:5d
Sat May 13 10:11:15 2017 daemon.notice hostapd: wlan2: AP-STA-DISCONNECTED 00:24:d7:10:17:b8
Sat May 13 10:11:15 2017 daemon.notice hostapd: wlan2: INTERFACE-DISABLED 
Sat May 13 10:11:15 2017 kern.info kernel: [ 6241.355665] mwifiex_sdio mmc0:0001:1: WLAN FW already running! Skip FW dnld
Sat May 13 10:11:15 2017 kern.info kernel: [ 6241.362725] mwifiex_sdio mmc0:0001:1: WLAN FW is active
Sat May 13 10:11:15 2017 kern.info kernel: [ 6241.368172] mwifiex_sdio mmc0:0001:1: PREP_CMD: FW is in bad state
Sat May 13 10:11:15 2017 kern.info kernel: [ 6241.374516] mwifiex_sdio mmc0:0001:1: info: mwifiex_fw_dpc: unregister device

yuhhaurlin, 2017-05-13:
This is not mwlwifi, please disable radio2.

closed 2017-05-13 by yuhhaurlin

Chadster766, 2017-05-13:
You probably need new firmware files for mwifliex:

http://git.kernel.org/?p=linux/kernel/git/firmware/linux-firmware.git
git://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git