kaloz/mwlwifi #164: Missing Channels 12 and 13

starcms, 2017-05-17:
2.4GHz WiFi Channels 12 and 13 have been enabled in the default Linksys firmware and are available for legal use by the FCC in the United States.  Yet they are missing from LEDE.  I am using a 1200AC V1 which gets its power table from the firmware, not from EEPROM.  Is this a driver issue or does the problem exist elsewhere?  Thanks!

Fr3DBr, 2017-05-17:
This is not a driver issue...

starcms, 2017-05-18:
@Fr3DBr Do you know where the issue lies?  I've encountered it on every LEDE (and before then OpenWRT) build I've ever run.  @david's, @cybrnook's, etc...

cybrnook, 2017-05-18:
I believe U.S. is 1 - 11

Check out the Chart:
https://en.wikipedia.org/wiki/List_of_WLAN_channels#United_States

starcms, 2017-05-18:
@cybrnook, last time I looked at that page, the note on channels 12 and 13 were not scratched out.  Plus, it's wiki so not always the most credible source.  No reason given as to why they are now scratched out.  See firmware release notes from linksys which shows added support for channels 12 and 13 in version 1.0.4.167471. http://downloads.linksys.com/downloads/releasenotes/WRT1200AC_Customer_Release_Notes_1.0.5.177401.txt 

Lastly, that wiki article also says 5ghz wifi channel 165 is available in the US, but also had never been present in LEDE or openwrt builds.

cybrnook, 2017-05-18:
Either way, I don't think they are supported channels. And likely none of your clients will support those channels.

yuhhaurlin, 2017-05-18:
Sorry. It is not driver issue. I close this one. Thanks.

closed 2017-05-18 by yuhhaurlin

anazz20, 2017-05-18:
@starcms, Read this, it may help you understand the FCC ruling on it. https://apps.fcc.gov/edocs_public/attachmatch/FCC-16-181A1.pdf