kaloz/mwlwifi #165: 3200acm possible Transmit power issue

davidc502, 2017-05-19:
I have no proof, but wifi output seems to be stuck on 100mw for all channels no matter the transmit setting. 

The way i've tested is by using a wifi analyzer which reports signal strength.  Testing several channels a good distance away, choosing 1mw to 1Watt, signal strength appears not to change. However, when I test with the 1900acs, making the same transmit changes, does affect signal strength received on the analyzer. 

Once again, I can't tell what the transmit power actually is at the source, but only through a strength meter on my Android.

Has anyone else observed this?

yuhhaurlin, 2017-05-19:
You can't change tx power if device power table is programmed in your device. WRT3200ACM is ready after 2016/6, so every device is programmed with device power table in order to comply with FCC rules.

yuhhaurlin, 2017-05-19:
I close this one. Thanks.

closed 2017-05-19 by yuhhaurlin

ValCher1961, 2017-05-20:
Excuse me.
That's fair if someone wants to increase power, but I want to reduce it, how?

aaron1959, 2017-05-20:
Your argument is with the F.C.C.
 I recommend you give them a call and voice your opinion, I'm sure you'll get immediate results.

yuhhaurlin, 2017-05-20:
In order to make sure tx power won't be controlled by driver, firmware blocked all power setting APIs. That is there is no way to change tx power via driver no matter increasing or reducing it.

aaron1959, 2017-05-20:
The rules specifically address the fact that NO POWER levels can be manipulated. Up, Down or sideways you are not supposed to touch it!

yuhhaurlin, 2017-05-20:
This is only applied to device with device power table.

aaron1959, 2017-05-20:
Yes!

yuhhaurlin, 2017-05-20:
However, according to FCC, devices after 2016/6 must avoid tx power change for open source, so all WRT3200ACM devices won't let user change tx power even though user tries to hack mwlwifi driver. The blocking is done by the firmware of 88W8964 if device power table is detected.

ValCher1961, 2017-05-20:
Yes, you can only eat bananas of a certain length. :(

yuhhaurlin, 2017-05-20:
This is not Marvell or Belkin problem. You need to check with FCC.

yuhhaurlin, 2017-05-20:
If we did not offer this kind of solution, the only way is to aviod loading open source firmware for the device. You should thank us to come out this kind of solution.

ValCher1961, 2017-05-20:
Yes, it's probably the only solution. Thanks again.