kaloz/mwlwifi #166: 5GHz performance issue

lewisdiamond, 2017-05-25:
Tested only with 5GHz so far.
mwlwifi 6d1595e9db5dec8e61332f6be431b3bf972d6d08
lede 4b8a7c9b48c4a6cd4c7bdc92dff99cb0b2682d32

```
[  5]   0.00-1.00   sec  27.6 MBytes   232 Mbits/sec                  
[  5]   1.00-2.00   sec  29.6 MBytes   248 Mbits/sec                  
[  5]   2.00-3.00   sec  30.5 MBytes   256 Mbits/sec                  
[  5]   3.00-4.00   sec  26.4 MBytes   222 Mbits/sec                  
[  5]   4.00-5.00   sec  21.1 MBytes   177 Mbits/sec                  
[  5]   5.00-6.00   sec  23.5 MBytes   197 Mbits/sec                  
[  5]   6.00-7.00   sec  27.6 MBytes   232 Mbits/sec                  
[  5]   7.00-8.00   sec  27.9 MBytes   234 Mbits/sec                  
[  5]   8.00-9.00   sec  28.1 MBytes   235 Mbits/sec                  
[  5]   9.00-10.00  sec  28.0 MBytes   235 Mbits/sec                  
[  5]  10.00-11.00  sec  30.6 MBytes   257 Mbits/sec                  
[  5]  11.00-12.00  sec  28.9 MBytes   243 Mbits/sec                  
[  5]  12.00-13.00  sec  29.4 MBytes   246 Mbits/sec                  
[  5]  13.00-14.00  sec  29.8 MBytes   250 Mbits/sec                  
[  5]  14.00-15.00  sec  28.6 MBytes   240 Mbits/sec                  
[  5]  15.00-16.00  sec  28.7 MBytes   241 Mbits/sec                  
[  5]  16.00-17.00  sec  27.8 MBytes   233 Mbits/sec                  
[  5]  17.00-18.00  sec  27.9 MBytes   234 Mbits/sec                  
[  5]  18.00-19.00  sec  26.6 MBytes   224 Mbits/sec                  
[  5]  19.00-20.00  sec  26.9 MBytes   226 Mbits/sec                  
[  5]  20.00-21.00  sec  28.6 MBytes   240 Mbits/sec                  
[  5]  21.00-22.00  sec  27.4 MBytes   230 Mbits/sec                  
[  5]  22.00-23.00  sec  28.1 MBytes   236 Mbits/sec                  
[  5]  23.00-24.00  sec  28.8 MBytes   241 Mbits/sec                  
[  5]  24.00-25.00  sec  29.1 MBytes   245 Mbits/sec                  
[  5]  25.00-26.00  sec  28.2 MBytes   236 Mbits/sec                  
[  5]  26.00-27.00  sec  28.2 MBytes   236 Mbits/sec                  
[  5]  27.00-28.00  sec  27.8 MBytes   233 Mbits/sec                  
[  5]  28.00-29.00  sec  29.1 MBytes   244 Mbits/sec                  
[  5]  29.00-30.00  sec  28.1 MBytes   235 Mbits/sec                  
[  5]  30.00-31.00  sec  28.8 MBytes   242 Mbits/sec                  
[  5]  31.00-32.00  sec  29.1 MBytes   244 Mbits/sec                  
[  5]  32.00-33.00  sec  27.0 MBytes   226 Mbits/sec                  
[  5]  33.00-34.00  sec  25.4 MBytes   213 Mbits/sec                  
[  5]  34.00-35.00  sec  28.3 MBytes   237 Mbits/sec                  
[  5]  35.00-36.00  sec  22.1 MBytes   186 Mbits/sec                  
[  5]  36.00-37.00  sec  25.8 MBytes   217 Mbits/sec                  
[  5]  37.00-38.00  sec  26.9 MBytes   226 Mbits/sec                  
[  5]  38.00-39.00  sec  30.0 MBytes   252 Mbits/sec                  
[  5]  39.00-40.00  sec  30.3 MBytes   254 Mbits/sec                  
[  5]  40.00-41.00  sec  28.5 MBytes   239 Mbits/sec                  
[  5]  41.00-42.00  sec  28.3 MBytes   238 Mbits/sec                  
[  5]  42.00-43.00  sec  28.1 MBytes   236 Mbits/sec                  
[  5]  43.00-44.00  sec  28.0 MBytes   235 Mbits/sec                  
[  5]  44.00-45.00  sec  29.3 MBytes   246 Mbits/sec                  
[  5]  45.00-46.00  sec  29.3 MBytes   246 Mbits/sec                  
[  5]  46.00-47.00  sec  28.3 MBytes   237 Mbits/sec                  
[  5]  47.00-48.00  sec  30.0 MBytes   251 Mbits/sec                  
[  5]  48.00-49.00  sec  28.9 MBytes   242 Mbits/sec                  
[  5]  49.00-50.00  sec  30.6 MBytes   256 Mbits/sec                  
[  5]  50.00-51.00  sec  29.6 MBytes   248 Mbits/sec                  
[  5]  51.00-52.00  sec  29.3 MBytes   246 Mbits/sec                  
[  5]  52.00-53.00  sec  29.9 MBytes   251 Mbits/sec                  
[  5]  53.00-54.00  sec  28.3 MBytes   237 Mbits/sec                  
[  5]  54.00-55.00  sec  27.2 MBytes   228 Mbits/sec                  
[  5]  55.00-56.00  sec  28.0 MBytes   235 Mbits/sec                  
[  5]  56.00-57.00  sec  29.9 MBytes   251 Mbits/sec                  
[  5]  57.00-58.00  sec  28.8 MBytes   242 Mbits/sec                  
[  5]  58.00-59.00  sec  28.3 MBytes   238 Mbits/sec                  
[  5]  59.00-60.00  sec  28.6 MBytes   240 Mbits/sec                  
[  5]  60.00-61.00  sec  30.8 MBytes   259 Mbits/sec                  
[  5]  61.00-62.00  sec  28.7 MBytes   241 Mbits/sec                  
[  5]  62.00-63.00  sec  15.9 MBytes   133 Mbits/sec                  
[  5]  63.00-64.00  sec  9.10 MBytes  76.3 Mbits/sec                  
[  5]  64.00-65.00  sec  18.2 MBytes   153 Mbits/sec                  
[... 535 lines cut ...]
[  5] 358.00-359.00 sec  10.2 MBytes  85.3 Mbits/sec                  
[  5] 359.00-360.00 sec  19.0 MBytes   159 Mbits/sec                  
[  5] 360.00-361.00 sec  22.6 MBytes   189 Mbits/sec                  
[  5] 361.00-362.00 sec  22.7 MBytes   190 Mbits/sec                  
[  5] 362.00-363.00 sec  23.0 MBytes   194 Mbits/sec                  
[  5] 363.00-364.00 sec  23.8 MBytes   199 Mbits/sec                  
[  5] 364.00-365.00 sec  23.6 MBytes   198 Mbits/sec                  
[  5] 365.00-366.00 sec  24.3 MBytes   203 Mbits/sec                  
[  5] 366.00-367.00 sec  24.2 MBytes   203 Mbits/sec                  
[  5] 367.00-368.00 sec  24.8 MBytes   208 Mbits/sec                  
[  5] 368.00-369.00 sec  23.7 MBytes   199 Mbits/sec                  
[  5] 369.00-370.00 sec  24.6 MBytes   206 Mbits/sec                  
[  5] 370.00-371.00 sec  24.5 MBytes   205 Mbits/sec                  
[  5] 371.00-372.00 sec  8.64 MBytes  72.5 Mbits/sec                  
[  5] 372.00-373.00 sec  12.6 MBytes   105 Mbits/sec                  
[  5] 373.00-374.00 sec  24.4 MBytes   204 Mbits/sec                  
[  5] 374.00-375.00 sec  23.5 MBytes   197 Mbits/sec                  
[  5] 375.00-376.00 sec  24.6 MBytes   206 Mbits/sec                  
[  5] 376.00-377.00 sec  24.3 MBytes   204 Mbits/sec                  
[  5] 377.00-378.00 sec  24.8 MBytes   208 Mbits/sec                  
[  5] 378.00-379.00 sec  24.4 MBytes   205 Mbits/sec                  
[  5] 379.00-380.00 sec  24.0 MBytes   201 Mbits/sec                  
[  5] 380.00-381.00 sec  25.2 MBytes   211 Mbits/sec                  
[  5] 381.00-382.00 sec  24.8 MBytes   208 Mbits/sec 
```

Fr3DBr, 2017-05-26:
Are you running iperf in the router ?

Chadster766, 2017-05-26:
This issue doesn't exist in McDebian with that mwlwifi version. It must be LEDE\OpenWRT build.

yuhhaurlin, 2017-05-27:
I did not find this kind of problem with LEDE build. Please try last code, if you still encounter problem, please let me know your build, setting and test method.

lewisdiamond, 2017-05-31:
Tested again with the latest version
```
[  5]   0.00-1.00   sec  43.8 MBytes   368 Mbits/sec                  
[  5]   1.00-2.00   sec  35.5 MBytes   298 Mbits/sec                  
[  5]   2.00-3.00   sec  40.7 MBytes   342 Mbits/sec                  
[  5]   3.00-4.00   sec  44.3 MBytes   372 Mbits/sec                  
[  5]   4.00-5.00   sec  41.5 MBytes   348 Mbits/sec                  
[  5]   5.00-6.00   sec  35.6 MBytes   299 Mbits/sec                  
[  5]   6.00-7.00   sec  30.2 MBytes   253 Mbits/sec                  
[  5]   7.00-8.00   sec  34.0 MBytes   285 Mbits/sec                  
[  5]   8.00-9.00   sec  33.9 MBytes   284 Mbits/sec                  
[  5]   9.00-10.00  sec  39.1 MBytes   328 Mbits/sec                  
[  5]  10.00-11.00  sec  35.0 MBytes   294 Mbits/sec                  
[  5]  11.00-12.00  sec  35.7 MBytes   299 Mbits/sec                  
[  5]  12.00-13.00  sec  44.9 MBytes   376 Mbits/sec                  
[  5]  13.00-14.00  sec  39.0 MBytes   327 Mbits/sec                  
[  5]  14.00-15.00  sec  34.5 MBytes   289 Mbits/sec                  
[  5]  15.00-16.00  sec  39.0 MBytes   327 Mbits/sec                  
[  5]  16.00-17.00  sec  47.2 MBytes   396 Mbits/sec                  
[  5]  17.00-18.00  sec  41.2 MBytes   346 Mbits/sec                  
[  5]  18.00-19.00  sec  33.3 MBytes   279 Mbits/sec                  
[  5]  19.00-20.00  sec  35.5 MBytes   298 Mbits/sec                  
[  5]  20.00-21.00  sec  45.4 MBytes   381 Mbits/sec                  
[  5]  21.00-22.00  sec  45.7 MBytes   384 Mbits/sec                  
[  5]  22.00-23.00  sec  36.1 MBytes   303 Mbits/sec                  
[  5]  23.00-24.00  sec  34.0 MBytes   285 Mbits/sec                  
[  5]  24.00-25.00  sec  40.9 MBytes   343 Mbits/sec                  
[  5]  25.00-26.00  sec  43.1 MBytes   362 Mbits/sec                  
[  5]  26.00-27.00  sec  48.0 MBytes   403 Mbits/sec                  
[  5]  27.00-28.00  sec  43.1 MBytes   361 Mbits/sec                  
[  5]  28.00-29.00  sec  35.6 MBytes   298 Mbits/sec                  
[  5]  29.00-30.00  sec  39.6 MBytes   332 Mbits/sec                  
[  5]  30.00-31.00  sec  39.4 MBytes   330 Mbits/sec                  
[  5]  31.00-32.00  sec  36.8 MBytes   309 Mbits/sec                  
[  5]  32.00-33.00  sec  45.8 MBytes   384 Mbits/sec                  
[  5]  33.00-34.00  sec  42.3 MBytes   355 Mbits/sec                  
[  5]  34.00-35.00  sec  43.5 MBytes   365 Mbits/sec                  
[  5]  35.00-36.00  sec  45.2 MBytes   379 Mbits/sec                  
[  5]  36.00-37.00  sec  38.7 MBytes   325 Mbits/sec                  
[  5]  37.00-38.00  sec  45.1 MBytes   379 Mbits/sec                  
[  5]  38.00-39.00  sec  49.4 MBytes   414 Mbits/sec                  
[  5]  39.00-40.00  sec  46.0 MBytes   386 Mbits/sec                  
[  5]  40.00-41.00  sec  36.0 MBytes   302 Mbits/sec                  
[  5]  41.00-42.00  sec  37.4 MBytes   314 Mbits/sec                  
[  5]  42.00-43.00  sec  32.8 MBytes   275 Mbits/sec                  
[  5]  43.00-44.00  sec  36.4 MBytes   305 Mbits/sec                  
[  5]  44.00-45.00  sec  45.3 MBytes   380 Mbits/sec                  
[  5]  45.00-46.00  sec  48.6 MBytes   408 Mbits/sec                  
[  5]  46.00-47.00  sec  22.8 MBytes   191 Mbits/sec                  
[  5]  47.00-48.00  sec  14.7 MBytes   123 Mbits/sec                  
[  5]  48.00-49.00  sec  38.4 MBytes   322 Mbits/sec                  
[  5]  49.00-50.00  sec  35.1 MBytes   294 Mbits/sec                  
[  5]  50.00-51.00  sec  41.9 MBytes   351 Mbits/sec                  
[  5]  51.00-52.00  sec  44.8 MBytes   376 Mbits/sec                  
[  5]  52.00-53.00  sec  33.3 MBytes   279 Mbits/sec                  
[  5]  53.00-54.00  sec  31.2 MBytes   262 Mbits/sec                  
[  5]  54.00-55.00  sec  37.8 MBytes   317 Mbits/sec                  
[  5]  55.00-56.00  sec  33.9 MBytes   285 Mbits/sec                  
[  5]  56.00-57.00  sec  36.0 MBytes   302 Mbits/sec                  
[  5]  57.00-58.00  sec  36.9 MBytes   310 Mbits/sec                  
[  5]  58.00-59.00  sec  45.0 MBytes   377 Mbits/sec                  
[  5]  59.00-60.00  sec  46.2 MBytes   387 Mbits/sec                  
[  5]  60.00-61.00  sec  44.3 MBytes   372 Mbits/sec                  
[  5]  61.00-62.00  sec  39.5 MBytes   331 Mbits/sec                  
[  5]  62.00-63.00  sec  38.5 MBytes   323 Mbits/sec                  
[  5]  63.00-64.00  sec  21.0 MBytes   176 Mbits/sec                  
[  5]  64.00-65.00  sec  40.2 MBytes   337 Mbits/sec                  
[  5]  65.00-66.00  sec  46.1 MBytes   386 Mbits/sec                  
[  5]  66.00-67.00  sec  47.5 MBytes   399 Mbits/sec                  
[  5]  67.00-68.00  sec  43.8 MBytes   367 Mbits/sec                  
[... 87 lines cut ...]
[  5] 155.00-156.00 sec  31.1 MBytes   261 Mbits/sec                  
[  5] 156.00-157.00 sec  41.2 MBytes   346 Mbits/sec                  
[  5] 157.00-158.00 sec  35.3 MBytes   296 Mbits/sec                  
[  5] 158.00-159.00 sec  38.7 MBytes   324 Mbits/sec                  
[  5] 159.00-160.00 sec  39.1 MBytes   328 Mbits/sec                  
[  5] 160.00-161.00 sec  43.7 MBytes   366 Mbits/sec                  
[  5] 161.00-162.00 sec  34.8 MBytes   292 Mbits/sec                  
[  5] 162.00-163.00 sec  34.5 MBytes   290 Mbits/sec                  
[... cut ...]
[  5] 172.00-173.00 sec  41.8 MBytes   350 Mbits/sec                  
[  5] 173.00-174.00 sec  36.6 MBytes   307 Mbits/sec                  
[  5] 174.00-175.00 sec  32.9 MBytes   276 Mbits/sec                  
[  5] 175.00-176.00 sec  39.3 MBytes   330 Mbits/sec                  
[  5] 176.00-177.00 sec  34.2 MBytes   287 Mbits/sec                  
```

There's quite a lot of variance but doesn't sink down to 0 Mbits/sec. Not sure how normal that is. If that's expected you can close this issue.

closed 2017-05-31 by yuhhaurlin