kaloz/mwlwifi #168: next-20170529, aka 4.13: build failure due to type mismatch

seragh, 2017-05-29:
Commit daa2ac80834d ("kernel.h: handle pointers to arrays better in
container_of()") in next-20170529 enforces stricter type checking which trips
mwlwifi as follows:

```
  CC [M]  core.o
In file included from ./include/linux/linkage.h:4:0,
                 from ./include/linux/kernel.h:6,
                 from ./include/linux/interrupt.h:5,
                 from core.h:21,
                 from core.c:19:
core.c: In function 'mwl_chnl_switch_event':
./include/linux/compiler.h:537:38: error: call to '__compiletime_assert_553' declared with attribute error: pointer type mismatch in container_of()
  _compiletime_assert(condition, msg, __compiletime_assert_, __LINE__)
                                      ^
./include/linux/compiler.h:520:4: note: in definition of macro '__compiletime_assert'
    prefix ## suffix();    \
    ^
./include/linux/compiler.h:537:2: note: in expansion of macro '_compiletime_assert'
  _compiletime_assert(condition, msg, __compiletime_assert_, __LINE__)
  ^
./include/linux/build_bug.h:46:37: note: in expansion of macro 'compiletime_assert'
 #define BUILD_BUG_ON_MSG(cond, msg) compiletime_assert(!(cond), msg)
                                     ^
./include/linux/kernel.h:859:2: note: in expansion of macro 'BUILD_BUG_ON_MSG'
  BUILD_BUG_ON_MSG(!__same_type(*(ptr), ((type *)0)->member) && \
  ^
core.c:552:9: note: in expansion of macro 'container_of'
   vif = container_of((char *)mwl_vif, struct ieee80211_vif,
         ^
```

https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?h=next-20170529&id=daa2ac80834dc6c6752f198087aeb75fcc2afb03

yuhhaurlin, 2017-06-09:
I will check this one and previous one after all major remaining functions are done. Thanks for your information.

seragh, 2017-06-09:
On Fri, 09 Jun 2017 13:50:36 -0700
yuhhaurlin <notifications@github.com> wrote:

> I will check this one and previous one after all major remaining
> functions are done. Thanks for your information.
> 

Hi David,

Actually I fixed this one already locally, will push a PR tomorrow.

This one is far from urgent but 4.12 is around the the corner, so the
other one should be looked at "soonish". There were already others
running into it which I directed to the PR. On the other hand this one
is quite trivial to address.

Ralph

yuhhaurlin, 2017-06-09:
Thanks. I will check and merge it If it is all right.

seragh, 2017-06-10:
So submitted the PR https://github.com/kaloz/mwlwifi/pull/175

yuhhaurlin, 2017-06-12:
Done. Thanks.

closed 2017-06-12 by yuhhaurlin