kaloz/mwlwifi #169: 2.4 ghz slowdown

vit5421, 2017-05-30:
Dear Yuhhaurlin,
I have a problem with 2.4 ghz on the latest DD-wrt build (r32149, 05.27.17). Sometimes when my clients (windows, androids) little bit far form router the connection is stuck and i have around 1-5 mbs download speed for all my clients. The signal strength is around 50dbm and connection speed is still 144 mbs. The upload for my clients works fine as usual all the time. I even try to restart the router and nothing change until i am close to router. The wrt1200 and 1900 works fine in the exact same location and exact same settings.
The wifi status is still not available and i cant see the real router negotiation speed for Tx rate but i guess that something is wrong there at the driver side

yuhhaurlin, 2017-05-31:
Cat you make sure that you test the latest version of code?

vit5421, 2017-05-31:
I guess it was includes in Brainslayer latest build

yuhhaurlin, 2017-05-31:
Cat you cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info?

vit5421, 2017-05-31:
driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170512
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b2:5e:92
2g: enable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0b80000
iobase1: e0e00000
tx limit: 1024
rx limit: 16384

yuhhaurlin, 2017-05-31:
Did you encounter the same problem if you run stock firmware?

vit5421, 2017-05-31:
I newer try stock firmware but i have the same problems on 2 wrt3200

yuhhaurlin, 2017-05-31:
What did you mean?

BrainSlayer, 2017-05-31:
i add here some forum reports. one user in my forum complaints that slowdowns are happening after 7 - 8 hours of testing. another has problems with tx only performance. so downstream is full speed, but upstream is slow. i'm still trying to find out his client chipset for investigation

yuhhaurlin, 2017-05-31:
Only happened for 2.4g band?

BrainSlayer, 2017-05-31:
yes. only in 2.4 ghz band people are seeing such issues like slowdowns. the user reported its on tx only. he tested with a samsung galaxy s8 and a s8+ and with a intel client. and is operating in pure 802.11n mode . this user says the issue is present immediatly without waiting

yuhhaurlin, 2017-05-31:
Does the stock firmware have the same problem?

BrainSlayer, 2017-05-31:
i dont know. never used stock nor do i have informations about it. so this report is only related to mwlwifi so far

rs-se, 2017-05-31:
I can confirm a 2.4ghz slow down issue with the wrt3200acm, also affects stock. 5Ghz is rock solid on this driver, however DFS is broken.

yuhhaurlin, 2017-05-31:
What do you mean DFS is broken? I and our QA can test channel 36 with 160 Mhz. If DFS is broken, there is not way for this kind of setting.

yuhhaurlin, 2017-05-31:
If stock firmware has the same problem, please report this problem for stock firmware. Thanks.

rs-se, 2017-05-31:
Linksys is a lost cause for reporting errors, they have people who know less than the average open-source users.

Unknown if it is the current latest driver or lede 17.01 (will take some testing) however DFS is not working on channel 52 to 144,

yuhhaurlin, 2017-05-31:
root@LEDE:/# iw dev
phy#1
        Interface wlan1
                ifindex 8
                wdev 0x100000002
                addr 00:50:43:22:88:89
                ssid LEDE-8964-2g
                type AP
                channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
                txpower 30.00 dBm
phy#0
        Interface wlan0
                ifindex 12
                wdev 0x5
                addr 00:50:43:22:88:8a
                type AP
                channel 100 (5500 MHz), width: 80 MHz, center1: 5530 MHz
                txpower 23.00 dBm
root@LEDE:/# [ 1298.839376] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1298.845815] br-lan: port 4(wlan0) entered blocking state
[ 1298.851140] br-lan: port 4(wlan0) entered forwarding state
[ 1298.946577] IPv6: ADDRCONF(NETDEV_UP): wlan0-1: link is not ready
[ 1299.086348] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-1: link becomes ready
[ 1299.181574] IPv6: ADDRCONF(NETDEV_UP): wlan0-2: link is not ready
[ 1299.319360] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-2: link becomes ready

root@LEDE:/# iw dev
phy#1
        Interface wlan1
                ifindex 8
                wdev 0x100000002
                addr 00:50:43:22:88:89
                ssid LEDE-8964-2g
                type AP
                channel 11 (2462 MHz), width: 20 MHz, center1: 2462 MHz
                txpower 30.00 dBm
phy#0
        Interface wlan0-2
                ifindex 14
                wdev 0x7
                addr 06:50:43:22:88:8a
                ssid LEDE-8964-BSSID-2
                type AP
                channel 100 (5500 MHz), width: 80 MHz, center1: 5530 MHz
                txpower 23.00 dBm
        Interface wlan0-1
                ifindex 13
                wdev 0x6
                addr 02:50:43:22:88:8a
                ssid LEDE-8964-BSSID-1
                type AP
                channel 100 (5500 MHz), width: 80 MHz, center1: 5530 MHz
                txpower 23.00 dBm
        Interface wlan0
                ifindex 12
                wdev 0x5
                addr 00:50:43:22:88:8a
                ssid LEDE-8964-5g
                type AP
                channel 100 (5500 MHz), width: 80 MHz, center1: 5530 MHz
                txpower 23.00 dBm
root@LEDE:/#

yuhhaurlin, 2017-05-31:
You must wait for 60 seconds for the AP to work if there is no radar detected.

BrainSlayer, 2017-05-31:
??? channel 52 as well as 36 has no dfs requirement.

yuhhaurlin, 2017-05-31:
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

If you set channel 36 (160 MHz), it needs DFS working to allow this setting. You need to wait for 60 seconds to let the AP work if no radar detected.
I also tries channel 100 (80 Mhz), it can work. Please check the log I just sent out.

yuhhaurlin, 2017-05-31:
If you set DFS channels to device, you can't find SSID when you issue "iw dev" until CAC is done (60 seconds). After CAC is done, you can find SSID when you issue "iw dev". It means the AP is working.

vit5421, 2017-05-31:
Dear Yuhhaurlin,
Do you have any solution for 2.4 upstream slowdown?

vit5421, 2017-05-31:
I will flash the latest 1.0.6.181063 stock and test it during the day

tapper82, 2017-05-31:
getting 2.4 slow down on.
Firmware Version
LEDE Reboot SNAPSHOT r4290-9e1bc27 / LuCI Master (git-17.150.76278-0e9eed5) 
Kernel Version
4.9.30

vit5421, 2017-05-31:
Works great for 1 hour on stock driver. 2.4 and 5 looks very good. There is no 2.4 slowdown and 5 is even more stable.  I dont see any single slowdown  on 5 during huge file downloads/uploads but it happens on open source driver. Will tested more and report later

vit5421, 2017-05-31:
getting 2.4 slow down in 6 hours. The upstream speed is 2 mbs instead of 60 but downstream is fast as usual. So, the stock driver have the same problem as open source.

rs-se, 2017-05-31:
I was wrong about stock not being affected, I just reset wifi config on lede and still have major slowdown on 2.4ghz 20/40mhz.

WRT3200ACM - LEDE 17.01 - Latest Mwlwifi

Wan Link: 65Mbit (6400 Kbit QOS) Down / 6.5 Mbit (63750 Kbit QOS) Up

--- 2.4 Ghz --- Ch 6+10 40 Mhz
Upload > iperf server (private server over wan)
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-2.00   sec  1.62 MBytes  6.81 Mbits/sec
[  4]   2.00-4.00   sec  1.25 MBytes  5.24 Mbits/sec
[  4]   4.00-6.00   sec  1.38 MBytes  5.77 Mbits/sec
[  4]   6.00-8.00   sec   896 KBytes  3.67 Mbits/sec
[  4]   8.00-10.00  sec   256 KBytes  1.05 Mbits/sec
[  4]  10.00-12.00  sec   640 KBytes  2.62 Mbits/sec
[  4]  12.00-14.00  sec   384 KBytes  1.57 Mbits/sec
[  4]  14.00-16.00  sec   512 KBytes  2.10 Mbits/sec
[  4]  16.00-18.00  sec   384 KBytes  1.57 Mbits/sec
[  4]  18.00-20.00  sec   384 KBytes  1.57 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-20.00  sec  7.62 MBytes  3.20 Mbits/sec                  sender
[  4]   0.00-20.00  sec  7.20 MBytes  3.02 Mbits/sec                  receiver

Download (Reserve Mode)
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-2.00   sec  1.62 MBytes  6.81 Mbits/sec
[  4]   2.00-4.00   sec  1.25 MBytes  5.24 Mbits/sec
[  4]   4.00-6.00   sec  1.38 MBytes  5.77 Mbits/sec
[  4]   6.00-8.00   sec   896 KBytes  3.67 Mbits/sec
[  4]   8.00-10.00  sec   256 KBytes  1.05 Mbits/sec
[  4]  10.00-12.00  sec   640 KBytes  2.62 Mbits/sec
[  4]  12.00-14.00  sec   384 KBytes  1.57 Mbits/sec
[  4]  14.00-16.00  sec   512 KBytes  2.10 Mbits/sec
[  4]  16.00-18.00  sec   384 KBytes  1.57 Mbits/sec
[  4]  18.00-20.00  sec   384 KBytes  1.57 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-20.00  sec  7.62 MBytes  3.20 Mbits/sec                  sender
[  4]   0.00-20.00  sec  7.20 MBytes  3.02 Mbits/sec                  receiver

--- 5 Ghz --- Ch. 136 DFS / 5730-5650 80 Mhz

[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-2.00   sec  1.62 MBytes  6.81 Mbits/sec
[  4]   2.00-4.00   sec  1.25 MBytes  5.24 Mbits/sec
[  4]   4.00-6.00   sec  1.38 MBytes  5.77 Mbits/sec
[  4]   6.00-8.00   sec   896 KBytes  3.67 Mbits/sec
[  4]   8.00-10.00  sec   256 KBytes  1.05 Mbits/sec
[  4]  10.00-12.00  sec   640 KBytes  2.62 Mbits/sec
[  4]  12.00-14.00  sec   384 KBytes  1.57 Mbits/sec
[  4]  14.00-16.00  sec   512 KBytes  2.10 Mbits/sec
[  4]  16.00-18.00  sec   384 KBytes  1.57 Mbits/sec
[  4]  18.00-20.00  sec   384 KBytes  1.57 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-20.00  sec  7.62 MBytes  3.20 Mbits/sec                  sender
[  4]   0.00-20.00  sec  7.20 MBytes  3.02 Mbits/sec                  receiver

Download (Reserve Mode)
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-2.00   sec  13.3 MBytes  56.0 Mbits/sec
[  4]   2.00-4.00   sec  13.9 MBytes  58.5 Mbits/sec
[  4]   4.00-6.00   sec  14.2 MBytes  59.4 Mbits/sec
[  4]   6.00-8.00   sec  13.9 MBytes  58.3 Mbits/sec
[  4]   8.00-10.00  sec  14.1 MBytes  59.2 Mbits/sec
[  4]  10.00-12.00  sec  13.9 MBytes  58.5 Mbits/sec
[  4]  12.00-14.00  sec  14.2 MBytes  59.7 Mbits/sec
[  4]  14.00-16.00  sec  14.0 MBytes  58.7 Mbits/sec
[  4]  16.00-18.00  sec  13.9 MBytes  58.1 Mbits/sec
[  4]  18.00-20.00  sec  13.9 MBytes  58.4 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-20.00  sec   140 MBytes  58.6 Mbits/sec   27             sender
[  4]   0.00-20.00  sec   140 MBytes  58.6 Mbits/sec                  receiver

--- DFS Issue ---
I have found the issue with DFS not working, it was not caused by driver. Appears to be something LEDE may need to fix (DD-WRT also may be affected), any reconfiguration of phy#2 (third wifi) interferes with DFS/Radar checking. 

Corrected fix: LEDE
rm -f /etc/config/wireless
wifi detect > /etc/config/wireless

rs-se, 2017-05-31:
--- Sorry for last post, the iperf copy and paste got mixed up ---

I was wrong about stock not being affected, I just reset wifi config on lede and still have major slowdown on 2.4ghz 20/40mhz.

WRT3200ACM - LEDE 17.01 - Latest Mwlwifi

Wan Link: 65Mbit (63750 Kbit QOS) Down / 6.5 Mbit (6400 Kbit QOS) Up

iperf3.exe -c XXXXX -t20 -i2 -w512k
--- 2.4 Ghz --- Ch 6+10 40 Mhz
Upload > iperf server (private server over wan)
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-2.00   sec  1.62 MBytes  6.81 Mbits/sec
[  4]   2.00-4.00   sec  1.25 MBytes  5.24 Mbits/sec
[  4]   4.00-6.00   sec  1.38 MBytes  5.77 Mbits/sec
[  4]   6.00-8.00   sec   896 KBytes  3.67 Mbits/sec
[  4]   8.00-10.00  sec   256 KBytes  1.05 Mbits/sec
[  4]  10.00-12.00  sec   640 KBytes  2.62 Mbits/sec
[  4]  12.00-14.00  sec   384 KBytes  1.57 Mbits/sec
[  4]  14.00-16.00  sec   512 KBytes  2.10 Mbits/sec
[  4]  16.00-18.00  sec   384 KBytes  1.57 Mbits/sec
[  4]  18.00-20.00  sec   384 KBytes  1.57 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-20.00  sec  7.62 MBytes  3.20 Mbits/sec                  sender
[  4]   0.00-20.00  sec  7.20 MBytes  3.02 Mbits/sec                  receiver

iperf3.exe -c XXXXX -t20 -i2 -w512k -R
Download (Reserve Mode)
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-2.00   sec  11.0 MBytes  46.1 Mbits/sec
[  4]   2.00-4.00   sec  12.0 MBytes  50.2 Mbits/sec
[  4]   4.00-6.00   sec  7.88 MBytes  33.1 Mbits/sec
[  4]   6.00-8.00   sec  7.80 MBytes  32.7 Mbits/sec
[  4]   8.00-10.00  sec  7.46 MBytes  31.3 Mbits/sec
[  4]  10.00-12.00  sec  10.6 MBytes  44.6 Mbits/sec
[  4]  12.00-14.00  sec  9.43 MBytes  39.6 Mbits/sec
[  4]  14.00-16.00  sec  8.52 MBytes  35.7 Mbits/sec
[  4]  16.00-18.00  sec  8.95 MBytes  37.6 Mbits/sec
[  4]  18.00-20.00  sec  5.28 MBytes  22.1 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-20.00  sec  89.4 MBytes  37.5 Mbits/sec   16             sender
[  4]   0.00-20.00  sec  89.3 MBytes  37.5 Mbits/sec                  receiver

iperf3.exe -c XXXXX -t20 -i2 -w512k
--- 5 Ghz --- 
Ch. 136 DFS / 5730-5650 80 Mhz
Upload --
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-2.00   sec  1.75 MBytes  7.34 Mbits/sec
[  4]   2.00-4.00   sec  1.38 MBytes  5.77 Mbits/sec
[  4]   4.00-6.00   sec  1.00 MBytes  4.19 Mbits/sec
[  4]   6.00-8.00   sec  1.38 MBytes  5.77 Mbits/sec
[  4]   8.00-10.00  sec  1.38 MBytes  5.77 Mbits/sec
[  4]  10.00-12.00  sec  1.38 MBytes  5.77 Mbits/sec
[  4]  12.00-14.00  sec  1.38 MBytes  5.77 Mbits/sec
[  4]  14.00-16.00  sec  1.38 MBytes  5.77 Mbits/sec
[  4]  16.00-18.00  sec  1.38 MBytes  5.77 Mbits/sec
[  4]  18.00-20.00  sec  1.38 MBytes  5.77 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-20.00  sec  13.8 MBytes  5.77 Mbits/sec                  sender
[  4]   0.00-20.00  sec  13.3 MBytes  5.57 Mbits/sec                  receiver

iperf3.exe -c XXXXX -t20 -i2 -w512k -R
Download (Reserve Mode)
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-2.00   sec  13.3 MBytes  56.0 Mbits/sec
[  4]   2.00-4.00   sec  13.9 MBytes  58.5 Mbits/sec
[  4]   4.00-6.00   sec  14.2 MBytes  59.4 Mbits/sec
[  4]   6.00-8.00   sec  13.9 MBytes  58.3 Mbits/sec
[  4]   8.00-10.00  sec  14.1 MBytes  59.2 Mbits/sec
[  4]  10.00-12.00  sec  13.9 MBytes  58.5 Mbits/sec
[  4]  12.00-14.00  sec  14.2 MBytes  59.7 Mbits/sec
[  4]  14.00-16.00  sec  14.0 MBytes  58.7 Mbits/sec
[  4]  16.00-18.00  sec  13.9 MBytes  58.1 Mbits/sec
[  4]  18.00-20.00  sec  13.9 MBytes  58.4 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-20.00  sec   140 MBytes  58.6 Mbits/sec   27             sender
[  4]   0.00-20.00  sec   140 MBytes  58.6 Mbits/sec                  receiver

--- DFS Issue ---
I have found the issue with DFS not working, it was not caused by driver. Appears to be something LEDE may need to fix (DD-WRT also may be affected), any reconfiguration of phy#2 (third wifi) interferes with DFS/Radar checking. 

Corrected fix: LEDE
rm -f /etc/config/wireless
wifi detect > /etc/config/wireless

vit5421, 2017-05-31:
So, we have 2.4 issues on stock and open source driver. Btw, 5 ghz works more stable and faster on stock driver for sure. Unfortunately, i have to return my two 3200 units

vit5421, 2017-06-01:
when its happens then all clients like windows 7 and 10 and Androids getting the same slow speed,

yuhhaurlin, 2017-06-01:
@vit5421 Thanks. I will do tests later.

vit5421, 2017-06-01:
I would like to mention one more time that power reset will not help but when i move close to router the speed will be restored. Sometimes it works ok for hours and stuck after that.. I have to return routers by Friday. so will be nice to know

yuhhaurlin, 2017-06-01:
This issue will be checked later. Thanks.

yuhhaurlin, 2017-06-01:
@vit5421 In fact, for upload, the role of AP is receiving. Can you test these clients one by one to see if the problem only happened for specific client? Or can you use sniffer to capture traffic between AP and client?

vit5421, 2017-06-01:
I tested all clients one by one and can confirm that problem is related to all clients at the same time. The 3200 is disconnected now because of issues and wrt1900acs is handling the wifi right now. It will be nice to have a Status/Wireless info in order to see the negotiated Tx rate

rs-se, 2017-06-01:
@yuhhaurlin 

AP mode, I have been able to confirm with 2.4ghz under 20mhz speeds are unstable, after I changed it to 40mhz it was much more stable.

yuhhaurlin, 2017-06-01:
1. In real work, 11n 40 Mhz is hard to really work unless your environment is very clean, otherwise, hostapd will do overlay BSSes check to decide if 40 MHz is allowable and most of the time, 40 Mhz is not allowable and will change back to 20 Mhz by hostapd.
2. You can change interrupts of phy 5g to another core via /proc/irq/irq_num of phy 5g/smp_affinity.

vit5421, 2017-06-01:
20 is a preferable choice of course. Do you think i can try just for test to use the 20/40 or even 40 in settings?

rs-se, 2017-06-01:
In my area, 40Mhz is to put it plainly a "jerk move" due to 2.4ghz saturation. Unknown why 20mhz is unstable under AP mode.

vit5421, 2017-06-01:
My neighbours signals are at the floor level around 92-95 dbm,  so my major problem for 40 is my 2nd AP.  I can have 40 on main AP and 20 on other.  But, this  issue needs to be specified and fixed asap

yuhhaurlin, 2017-06-01:
What do you mean unstable? The same as upload slowdown for 2.4g? And from the description, it is related distance between AP and client. Can you give me detail information about the distance and test procedure?

BTW, how many users encounter this kind of problem?

vit5421, 2017-06-01:
Dear yuhhaurlin,
 I did try different settings like Mixed mode and 20/40 mbps. It worked for several hours and then upsteram only stuck as usual at 2-4mbps on all clients and i was not able to start it again even with disabling wifi on router and router restart. Another words i completely give up coz spent all week for troubleshooting and will return routers tomorrow. I hope problem will be fixed later

yuhhaurlin, 2017-06-01:
If I use one client and run iperf test, I can reproduce this problem, right? What is the distance between AP and client?

Are you sure wrt1200 and wrt1900 work well under same test?

vit5421, 2017-06-01:
I think u can reproduce it but its unpredictable. I am more than 100% sure that wrt1900acs  works well on any distance. The distance is not very big like 20-30 feet or so but it  looks like router is triggering the wrong rate and hold it until u will stay closer and will continue to be slow if you will stay far

yuhhaurlin, 2017-06-01:
If the problem happened for Tx (AP to STA ),  it could be rate adaptation problem of AP. But the problem is related to Rx (STA to AP), the only possibility is that AP does not receive packets from STA. If the distance is so close and it can run for a long time, restart can't recover the problem, it is very  strange. And there is no this kind of problem for 5g band.

Is there any other one got this kind of problem.

yuhhaurlin, 2017-06-01:
Can you try another WRT3200ACM?

vit5421, 2017-06-01:
The problem is related to 2.4 Ghz Tx only on both wrt3200 routers, 5g is fine. May be both are defected in the same way? idk

yuhhaurlin, 2017-06-01:
Tx? The problem is upload, it should be Rx of AP. When the problem happened, can you power off AP and wait for a while before restarting AP to check if the problem is still there.

If possible, please try another WRT3200ACM.

yuhhaurlin, 2017-06-02:
If the problem happen for Tx and It can't be recovered by reboot, it is not rate adaptation problem of AP.
I suggest:
1. When problem happened, power off SO for a while and restart again to see if problem is still there.
2. Try another WRT3200ACM.

vit5421, 2017-06-02:
The speed from AP to client is slow. The speed from client to AP is always fine. I guess its Tx? Sorry if i confuse you.
I did test for 2 units already and restart the power on routers and clients with no success.
I will be on business trip but i can reorder the 2 units and test it again on June 19 or 20th to make sure i am not a Don Quixote fights windmills. Lol

yuhhaurlin, 2017-06-02:
Please help to check if power off AP for a while and restart it again to see if the problem is still there.

Or can you try to capture traffic between AP and client?

vit5421, 2017-06-02:
I cant test anymore coz routers are packed for return. But what was interesting that routers always worked at the morning very well after night with no communication of course.
Again, i can reorder the routers and test it and try to capture the traffic after June 19

yuhhaurlin, 2017-06-02:
Thanks for your help.

mattbze, 2017-06-02:
@BrainSlayer Iv been following you on the DD-WRT forum, but I haven't signed up to the forums as yet. Is it possible you can attach your latest build here as you did in the previous but now closed issue on Github?

I am also having same issues across both 2.4ghz and 5ghz on a WRT3200acm with the latest LEDE build.

Reboot (SNAPSHOT, r4222-7142cb4)

root@LEDE:~# opkg list | grep mwlwifi
kmod-mwlwifi - 4.9.30+10.3.4.0-20170512-1 - Marvell 88W8864 wireless driver

Wireless on either interface drops the bandwidth to around 10kbps after about 8-10 hours of use. If not used overnight, its fine again the next morning, everything streams correctly over wifi and then towards the end of the day, drops out again.

Tested on PS4 streaming, PS4 Gaming, LGTV streaming, NOWTV, Amazon Prime TV, Netflix, Laptop, Phone, Speedtest.net. 

- Router logs dont show anything going wrong it appears and the graphs arent showing anything except the outbound packets from the router drop dramatically when this occurs during streaming content.

I.e I loaded up the lede interface, went to live graphs, looked at the bandwidth the interfaces were utlizing, starts off very high to start the content playing then a few second later drops to nothing and doesnt return, only happens towards the end of a day, never in the morning on the next day. 

Rebooting the router physically does not resolve the issue, tried that a few times now.

yuhhaurlin, 2017-06-02:
I found AMPDU still use aid instead of station ID. I will fix it and submit here when security mode of STA is done.

You can help to confirm if this problem is related to this bug. Please test one AP and one client to see if you still encounter this problem.

mattbze, 2017-06-02:
@yuhhaurlin Ok ill be able to test this out today and feed back tomorrow. 

I know youve been asked this before, but I also am thinking about returning the router unless I can get wifi stable on across all my devices as per my older router, so do you have a rough time frame you think this will be completed in. Sorry to ask.

yuhhaurlin, 2017-06-02:
I hope I can release code next week.

rs-se, 2017-06-02:
I can also confirm the consensus that it is a AP to CLIENT 2.4ghz slowness, will give DD-WRT a try to see if wifi optimizations are better.

C:\Users\abbie\Downloads\iperf-3.1.3-win64>iperf3.exe -c 10.6.55.127
Connecting to host 10.6.55.127, port 5201
[  4] local 10.6.55.174 port 1931 connected to 10.6.55.127 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  2.25 MBytes  18.8 Mbits/sec
[  4]   1.00-2.00   sec  1.50 MBytes  12.6 Mbits/sec
[  4]   2.00-3.00   sec  1.38 MBytes  11.5 Mbits/sec
[  4]   3.00-4.00   sec  1.25 MBytes  10.5 Mbits/sec
[  4]   4.00-5.00   sec  1.12 MBytes  9.44 Mbits/sec
[  4]   5.00-6.00   sec  1.38 MBytes  11.5 Mbits/sec
[  4]   6.00-7.00   sec  1.25 MBytes  10.5 Mbits/sec
[  4]   7.00-8.00   sec  1.12 MBytes  9.43 Mbits/sec
[  4]   8.00-9.00   sec  1.50 MBytes  12.6 Mbits/sec
[  4]   9.00-10.00  sec  1.62 MBytes  13.6 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  14.4 MBytes  12.1 Mbits/sec                  sender
[  4]   0.00-10.00  sec  14.4 MBytes  12.1 Mbits/sec                  receiver

C:\Users\abbie\Downloads\iperf-3.1.3-win64>iperf3.exe -c 10.6.55.127 -R
Connecting to host 10.6.55.127, port 5201
Reverse mode, remote host 10.6.55.127 is sending
[  4] local 10.6.55.174 port 1938 connected to 10.6.55.127 port 5201
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec  1.39 MBytes  11.7 Mbits/sec
[  4]   1.00-2.00   sec  1.01 MBytes  8.49 Mbits/sec
[  4]   2.00-3.00   sec   826 KBytes  6.76 Mbits/sec
[  4]   3.00-4.00   sec   358 KBytes  2.93 Mbits/sec
[  4]   4.00-5.00   sec   361 KBytes  2.95 Mbits/sec
[  4]   5.00-6.00   sec   244 KBytes  2.00 Mbits/sec
[  4]   6.00-7.00   sec  64.2 KBytes   526 Kbits/sec
[  4]   7.00-8.00   sec  65.6 KBytes   538 Kbits/sec
[  4]   8.00-9.00   sec   108 KBytes   888 Kbits/sec
[  4]   9.00-10.00  sec   252 KBytes  2.07 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec  4.97 MBytes  4.17 Mbits/sec    1             sender
[  4]   0.00-10.00  sec  4.88 MBytes  4.09 Mbits/sec                  receiver

Once new code is released, I will also help to test it. I have a iperf3 server setup on LAN under a 1gbit link.

rs-se, 2017-06-02:
@yuhhaurlin 
I missed one of your posts, I have a functioning WRT3200ACM.... Will see if I can capture packets during a iperf3 test, in case that will be of any help.

vit5421, 2017-06-02:
Dear rs-se,
U dont have to try ddwrt coz of the same issue there

[25 more replies]

closed 2017-06-11 by yuhhaurlin