kaloz/mwlwifi #170: Cannot connect ESP8266 MQTT devices

hagensieker, 2017-06-04:
I have a Linksys WRT1900ACS with DD-WRT dated 3/30 which works perfectly and controls many IoT devices in my home.  

I have a WRT3200ACM that I intended to deploy as my main router in the home.  Using the latest Brain Slayer DD-WRT versions 31 May and 1 June.  With this router, using the mwlwifi driver, and configuring the wifi identical to the previous router 10 of my devices, all of which use ESP8266 chips will not connect to the same network.  I have tried turning off the SPI firewall, port forwarding to the MQTT server, changing the wifi name entirely and reconfiguring a device to that new name...........nothing allows these devices to connect.

Out of curiosity I took an Archer C7v2 and put latest DD-WRT on it and it will connect all my ESP8266 devices with no problem. 

I posted about this issue on the DD-WRT forums and there is another person with similar problems on the forum using WRT3200ACM and Home Automation devices.

Thanks............John, New Bern, NC.

yuhhaurlin, 2017-06-04:
The client can't connect to AP? What is the client chip?

hagensieker, 2017-06-04:
ESP8266.  I have 11 of them in my home that will not connect to my wifi with the DD-WRT with latest driver.  I have confirmed now that an Archer C7v2 with latest DD-WRT will connect them and a DIR-860L router with LEDE installed will connect them.  It is only the DD-WRT with that latest mwlwifi driver installed that cannot connect the IoT devices .

mrvltest, 2017-06-05:
Can you give me more information about your problem?

yuhhaurlin, 2017-06-05:
Can you give me more information about your problem?

yuhhaurlin, 2017-06-05:
Please let me know version of your mwlwifi and firmware. Setting of your AP. If possible, please use "logread | grep hostapd" and "dmesg" to get more information for me.

hagensieker, 2017-06-05:
dmesg

`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.30 (root@seg-desktop) (gcc version 7.1.0 (LEDE GCC 7.1.0 r4105-9448e16959) ) #14 SMP Sat May 27 01:12:16 CEST 2017
[    0.000000] CPU: ARMv7 Processor [414fc091] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Linksys WRT3200ACM
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat c082b380, node_mem_map dfbf9000
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] percpu: Embedded 13 pages/cpu @dfbce000 s22860 r8192 d22196 u53248
[    0.000000] pcpu-alloc: s22860 r8192 d22196 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/mtdblock8 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,128K(u_env),256K(s_env),256K@8064K(devinfo),1920K@8320K(sysdiag),80m@10m(kernel),74m@16m(rootfs
,80m@90m(alt_kernel),74m@96m(alt_rootfs),160m@10m(ubifs),-@170m(syscfg)
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 511848K/524288K available (4096K kernel code, 176K rwdata, 912K rodata, 1024K init, 350K bss, 12440K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0500000   (5088 kB)
[    0.000000]       .init : 0xc0700000 - 0xc0800000   (1024 kB)
[    0.000000]       .data : 0xc0800000 - 0xc082c0b0   ( 177 kB)
[    0.000000]        .bss : 0xc082e000 - 0xc0885b3c   ( 351 kB)
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] mvebu_mbus: [Firmware Warn]: deprecated mbus-mvebu Device Tree, suspend/resume will not work
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 D prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 Coherent cache controller enabled, 16 ways, 1024 kB
[    0.000000] L2C-310 Coherent: CACHE_ID 0x410054c9, AUX_CTRL 0x56070001
[    0.000000] Switching to timer-based delay loop, resolution 40ns
[    0.000003] sched_clock: 32 bits at 25MHz, resolution 40ns, wraps every 85899345900ns
[    0.000008] clocksource: armada_370_xp_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 76450417870 ns
[    0.000147] Calibrating local timer... 933.15MHz.
[    0.060028] Calibrating delay loop (skipped), value calculated using timer frequency.. 50.00 BogoMIPS (lpj=250000)
[    0.060033] pid_max: default: 32768 minimum: 301
[    0.060078] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060082] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060258] CPU: Testing write buffer coherency: ok
[    0.060343] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.060354] Setting up static identity map for 0x100000 - 0x100058
[    0.060409] mvebu-soc-id: MVEBU SoC ID=0x6820, Rev=0x4
[    0.060463] mvebu-pmsu: Initializing Power Management Service Unit
[    0.060865] Booting CPU 1
[    0.061000] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.061030] Brought up 2 CPUs
[    0.061035] SMP: Total of 2 processors activated (100.00 BogoMIPS).
[    0.062666] VFP support v0.3: implementor 41 architecture 3 part 30 variant 9 rev 4
[    0.062718] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.062725] futex hash table entries: 512 (order: 3, 32768 bytes)
[    0.062757] pinctrl core: initialized pinctrl subsystem
[    0.063011] NET: Registered protocol family 16
[    0.063449] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.090040] cpuidle: using governor ladder
[    0.090161] mvebu-pmsu: CPU hotplug support is currently broken on Armada 38x: disabling
[... 290 lines cut ...]
`logread: can't find syslogd buffer: No such file or directory`

mwlwifi 

`driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170512
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b6:ee:29
2g: enable
5g: disable
antenna: 4 4
irq number: 46
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0f80000
iobase1: e1200000
tx limit: 1024
rx limit: 16384
`

hagensieker, 2017-06-05:
[image]

yuhhaurlin, 2017-06-05:
Please help to check if this problem will happen if only one client is used to connect to AP. Thanks.

hagensieker, 2017-06-05:
The AP's both 2.4 and 5Ghz will connect to clients, just not the ESP8266's.  I changed the wifi name and reprogrammed one ESP8266 and it still wouldn't connect.  Subsequently then changed the wifi connection for a raspberry pi and it connected fine.

yuhhaurlin, 2017-06-05:
What is ESP8266? 
Subsequently then changed the wifi connection for a raspberry pi and it connected fine. => So the problem is gone?

hagensieker, 2017-06-05:
ESP8266 is an Internet of Things chip found in a lot of Home Automation stuff.  I have 6 wifi outlets, a motion sensor, 2 relays and an LED strip all controlled by ESP8266 chips.  What they do is broadcast a hotspot which you connect to and then configure them to run on your local network.  These devices all work brilliantly on any SSID named "dd-wrt" with the appropriate password. 

Works with my WRT1900ACS, Archer C7v2, DIR-860L, and R7800 routers.  Those ESP8266 devices will not connect using the last two (and possibly previous) BrainSlayer DD-WRT versions using the mwlwifi version I posted above. 

Regular computers work just fine.  I can connect my laptop, my raspberry pi's, anything but the ESP8266 devices.

hagensieker, 2017-06-05:
This will give you a rough idea of the device

http://www.hagensieker.com/blog/page/?post_id=44&title=esp8266-hack-of-inexpensive-wifi-outlet

yuhhaurlin, 2017-06-05:
The latest driver can work for WRT1900ACS and WRT3200ACM, you only encounter this problem with WRT3200ACM right?

yuhhaurlin, 2017-06-05:
BTW, do you have the same problem with stock firmware of WRT3200ACM?

hagensieker, 2017-06-05:
That is correct.  Only on the 3200ACM The version of DD-WRT I'm running on the WRT1900ACS is dated 3/30 and I haven't run latest firmware yet.

Have not tried stock yet but every other dd-wrt on every other device works fine.

yuhhaurlin, 2017-06-05:
1. Please use the same driver on WRT1900ACS to see if it is working.
2. Please check if you have the same problem with stock firmware.
3. BTW, no matter open or security has the same problem, right?

hagensieker, 2017-06-05:
I'm booting to stock now will know in a few minutes.  I haven't tried to test on open network as I'd have to reprogram the devices.  I can later but not a ton of time before work this morning.   Will test stock now.

yuhhaurlin, 2017-06-05:
Thank you. These tests can help me to check this problem. If you can capture traffic between AP and ESP8266, it would be even better.

hagensieker, 2017-06-05:
Stock works.

yuhhaurlin, 2017-06-05:
Can you help just connect one ESP8266 to AP to see if it works. Thanks.

hagensieker, 2017-06-05:
I'll try to wireshark but won't be able to get to it for a couple days.  Quick work trip.

yuhhaurlin, 2017-06-05:
1. Can you help to test for one ESP8266 to see if this problem happen?
2. If you have Macbook Pro, you can install Wireshark to capture packets, you need to connect to AP first before packet capture.
3. If needed, I can build LEDE for you to get more debug information later.

hagensieker, 2017-06-05:
One ESP8266 does not connect.  I have wireshark but will have to capture later.

Consequently yesterday I installed 17.01.1 LEDE and then updated the kmod-mwlwifi driver and it also did not work

hagensieker, 2017-06-05:
This driver

https://github.com/eduperez/mwlwifi_LEDE/releases

yuhhaurlin, 2017-06-05:
If you use LEDE, you can use "logread | grep hostapd" to get more debug information.

hagensieker, 2017-06-05:
Will give that a shot later.  Out the door for work and probably won't be able to provide more info until Wed.  Many thanks for your help here this morning.

davidelang, 2017-06-05:
I think I've seen a post reporting problems with recent drivers that only 
happened with an encrypted connection, an open connection worked (using the 
esp8266)

yuhhaurlin, 2017-06-07:
Please use https://drive.google.com/open?id=0B3qLWtcWB9EdcHJvd2tIbHl3dnM to do test and use "loadread | grep hostapd" to get more information. I will check the log and update needed debug package to check this problem. Thanks.

hagensieker, 2017-06-07:
Will do tomorrow when I get back off my trip.  Thanks.

hagensieker, 2017-06-07:
Installed your LEDE and here is the logread:

`root@LEDE:~# logread | grep hostapd
Wed Jun  7 12:40:08 2017 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Wed Jun  7 12:40:09 2017 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Jun  7 12:40:09 2017 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
Wed Jun  7 12:40:09 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:09 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:09 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:09 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:09 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:09 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:09 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:11 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:11 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:11 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:11 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:11 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:11 2017 daemon.notice hostapd: handle_probe_req: send failed
Wed Jun  7 12:40:11 2017 daemon.notice hostapd: 20/40 MHz operation not permitted on channel pri=2 sec=6 based on overlapping BSSes
Wed Jun  7 12:40:11 2017 daemon.err hostapd: Using interface wlan1 with hwaddr 60:38:e0:b6:ee:29 and ssid "dd-wrt"
Wed Jun  7 12:40:11 2017 daemon.notice hostapd: wlan1: interface state HT_SCAN->ENABLED
Wed Jun  7 12:40:11 2017 daemon.notice hostapd: wlan1: AP-ENABLED 
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 80:1f:02:94:58:b1 IEEE 802.11: associated (aid 1)
Wed Jun  7 12:46:20 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 80:1f:02:94:58:b1
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 80:1f:02:94:58:b1 WPA: pairwise key handshake completed (RSN)
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 94:10:3e:4d:10:d1 IEEE 802.11: associated (aid 2)
Wed Jun  7 12:46:20 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 94:10:3e:4d:10:d1
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 94:10:3e:4d:10:d1 WPA: pairwise key handshake completed (RSN)
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:14:3f IEEE 802.11: associated (aid 3)
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA a0:20:a6:12:54:81 IEEE 802.11: associated (aid 4)
Wed Jun  7 12:46:20 2017 daemon.err hostapd: nl80211: EAPOL TX: No message of desired type
Wed Jun  7 12:46:20 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 38:2b:78:07:14:3f
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:14:3f WPA: pairwise key handshake completed (RSN)
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 80:1f:02:94:58:b1 IEEE 802.11: authenticated
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 94:10:3e:4d:10:d1 IEEE 802.11: authenticated
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:14:3f IEEE 802.11: authenticated
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA a0:20:a6:12:54:81 IEEE 802.11: authenticated
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:15:f8 IEEE 802.11: authenticated
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:49:c1 IEEE 802.11: authenticated
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:55:71 IEEE 802.11: authenticated
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:02:f0:3a IEEE 802.11: authenticated
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:15:f8 IEEE 802.11: associated (aid 5)
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:55:71 IEEE 802.11: associated (aid 6)
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:49:c1 IEEE 802.11: associated (aid 7)
Wed Jun  7 12:46:20 2017 daemon.err hostapd: nl80211: EAPOL TX: No message of desired type
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:02:f0:3a IEEE 802.11: associated (aid 8)
Wed Jun  7 12:46:20 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 38:2b:78:07:15:f8
Wed Jun  7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:15:f8 WPA: pairwise key handshake completed (RSN)
Wed Jun  7 12:46:21 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 38:2b:78:02:f0:3a
Wed Jun  7 12:46:21 2017 daemon.info hostapd: wlan1: STA 38:2b:78:02:f0:3a WPA: pairwise key handshake completed (RSN)
Wed Jun  7 12:46:21 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED a0:20:a6:12:54:81
Wed Jun  7 12:46:21 2017 daemon.info hostapd: wlan1: STA a0:20:a6:12:54:81 WPA: pairwise key handshake completed (RSN)
Wed Jun  7 12:46:21 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 38:2b:78:07:49:c1
Wed Jun  7 12:46:21 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:49:c1 WPA: pairwise key handshake completed (RSN)
Wed Jun  7 12:46:21 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 38:2b:78:07:55:71
Wed Jun  7 12:46:21 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:55:71 WPA: pairwise key handshake completed (RSN)
Wed Jun  7 12:46:21 2017 daemon.info hostapd: wlan1: STA d0:73:d5:21:7b:1f IEEE 802.11: associated (aid 9)
[... cut ...]
Wed Jun  7 12:46:39 2017 daemon.info hostapd: wlan1: STA 00:24:2c:a1:49:dd IEEE 802.11: authenticated
Wed Jun  7 12:47:12 2017 daemon.info hostapd: wlan1: STA 70:f0:87:4f:3f:f2 IEEE 802.11: associated (aid 22)
Wed Jun  7 12:47:12 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 70:f0:87:4f:3f:f2
Wed Jun  7 12:47:12 2017 daemon.info hostapd: wlan1: STA 70:f0:87:4f:3f:f2 WPA: pairwise key handshake completed (RSN)
Wed Jun  7 12:47:13 2017 daemon.info hostapd: wlan1: STA 70:f0:87:4f:3f:f2 IEEE 802.11: authenticated`

yuhhaurlin, 2017-06-07:
1. Can you just use one client?
2. It looks like client is connected to AP and completes 4 ways handshake. When the client is connected, can you use "iw wlan0 or wlan1 station dump" to dump station information for me.

hagensieker, 2017-06-07:
Not following here as the clients in question won't connect  I have a bunch of computers, devices, etc.  They all connect fine, just the ESP8266 devices won't connect.  Not sure how I can give a dump that reveals that issue without one of those clients getting rejected or something.

yuhhaurlin, 2017-06-07:
You can just use one ESP8266 to connect to AP. BTW, althoughh you set 40 MHz on 2.4g, but hostapd resets to 20 MHz due to overlapping BSS.

hagensieker, 2017-06-07:
That's just it.  The ESP8266 won't connect.

yuhhaurlin, 2017-06-07:
If you have problem to do that, please let me know MAC address of ESP8266.

yuhhaurlin, 2017-06-07:
Thanks. I will check it.

yuhhaurlin, 2017-06-08:
From the log, 38:2b:78:07:14:3f is connected and completes 4 ways handshake, I can't find AP disconnects it. The only one be disconnected is 94:10:3e:4d:10:d1. You can help to check your log again to see if I am right. Can you help to use "iw wlan1 station dump to check if 38:2b:78:07:14:3f is connected?

 <<38:2b:78:07:14:3f>>
Wed Jun 7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:14:3f IEEE 802.11: authenticated
Wed Jun 7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:14:3f IEEE 802.11: associated (aid 3)
Wed Jun 7 12:46:20 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 38:2b:78:07:14:3f
Wed Jun 7 12:46:20 2017 daemon.info hostapd: wlan1: STA 38:2b:78:07:14:3f WPA: pairwise key handshake completed (RSN)

<<e0:4f:43:0f:7d:16>>
Wed Jun 7 12:46:29 2017 daemon.info hostapd: wlan1: STA e0:4f:43:0f:7d:16 IEEE 802.11: authenticated
Wed Jun 7 12:46:29 2017 daemon.info hostapd: wlan1: STA e0:4f:43:0f:7d:16 IEEE 802.11: associated (aid 19)
Wed Jun 7 12:46:29 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED e0:4f:43:0f:7d:16
Wed Jun 7 12:46:29 2017 daemon.info hostapd: wlan1: STA e0:4f:43:0f:7d:16 WPA: pairwise key handshake completed (RSN)

<<94:10:3e:4d:10:d1>>
Wed Jun 7 12:46:26 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 94:10:3e:4d:10:d1

hagensieker, 2017-06-08:
That’s interesting.  My specific issue is that the ESP8266 devices do not respond to commands when (now I know) connected to the WRT3200ACM.  If connected to Stock, or any other router in the home these devices can be controlled from a program called Home Assistant.  One thing most of the devices have in common is they all (a couple exceptions) are also connected to my MQTT server, an Ubuntu Linux machine.

Looks like I need to look towards the Home Assistant Raspberry Pi or the MQTT server now.

Something is still not playing correctly though with this router config.  Perhaps it is not mwlwifi but some other issue.  Again though, literally every other router in my home connects and controls all these devices.  Let me dig a little deeper and I’ll get back to this here. 

Thanks.

yuhhaurlin, 2017-06-08:
1. Use "iw wlan1 station dump | grep 38:2b:78:07:14:3f" to check if station is connected. It will tell if the client is  authenticated,  associated and authorized.
2. If the station is connected, the next step should be check if data traffic is all right via ping.

root@LEDE:/# iw wlan0 station dump
Station b8:e8:56:3b:b5:34 (on wlan0)
        inactive time:  0 ms
        rx bytes:       5373259941
        rx packets:     62478303
        tx bytes:       139214342797
        tx packets:     91310113
        tx retries:     0
        tx failed:      0
        rx drop misc:   2
        signal:         0 dBm
        signal avg:     0 dBm
        tx bitrate:     6.0 MBit/s
        rx bitrate:     6.0 MBit/s
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short slot time:yes
        connected time: 3092 seconds

yuhhaurlin, 2017-06-08:
You mentioned that WRT1900ACS is all right. The same driver can work for WRT1900ACS, you can get LEDE image here:  https://drive.google.com/open?id=0B3qLWtcWB9EdeEZOT2ZTUWJkZW8.

hagensieker, 2017-06-08:
`Station 38:2b:78:07:14:3f (on wlan1)
	inactive time:	52410 ms
	rx bytes:	4623
	rx packets:	4
	tx bytes:	487
	tx packets:	4
	tx retries:	0
	tx failed:	0
	rx drop misc:	11
	signal:  	0 dBm
	signal avg:	0 dBm
	tx bitrate:	1.0 MBit/s
	rx bitrate:	1.0 MBit/s
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	short
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short preamble:	yes
	connected time:	415 seconds
`

hagensieker, 2017-06-08:
so looks like the device is connected.  May be another issue with Home Assistant or MQTT.  I'll dig deeper

yuhhaurlin, 2017-06-08:
Yes, the client is connected and it is authenticated, associated and authorized. It shows the same result as what we saw in the log. I wonder if the data traffic is all right? You can try to ping to AP first.

hagensieker, 2017-06-08:
I’m trying to Wireshark it but the odd thing is that it doesn’t show up on my network list when using an app like Fing to find all connected network devices.

davidelang, 2017-06-08:
did you try connecting with encryption disabled?

hagensieker, 2017-06-08:
Not yet because i have to reprogram the ESP8266 chip(s) which isn’t impossible to do but I haven’t gotten that far yet. 

Although my devices seem connected (under LEDE) they don’t show up on the DHCP list or on a network scan like with an app like Fing, and I can’t control the devices from Home Assistant (home automation software).

yuhhaurlin, 2017-06-09:
Yes, you can use open mode to see if problem is there. BTW, can you make sure data traffic is all right between your device and AP?

yuhhaurlin, 2017-06-12:
Please confirm data path is all right. If there is no more information from you, I will close this one.
BTW, if anyone encounters the same problem, please let me know. Connection and key is established, what we need to check is if data is all right.

hagensieker, 2017-06-12:
I’m working on this this morning. Will have some info for you in a bit.  Had to take it to work as every time I take the old router out, experiment with this one it gives me new DHCP addresses on everything and it fouls up some of my media servers and Home Automation.

yuhhaurlin, 2017-06-12:
Maybe you should disable DHCP server on WRT3200ACM? Do you have another DHCP server running on your network?
I can't completely understand what you said. I just want to know if data traffic is all right under security mode with ESP8266.

hagensieker, 2017-06-12:
Ok just for fun I set up a device that uses an MQTT server which I know doesn't exist here.  The ESP8266 connects to wifi then of course can't find the MQTT server.  Pings are fine.

I think the fix for this would be for me to rename my network and add every device one at a time.  I should add......I'm not going to do this.  I probably have 40-50 devices connected at home and it would take me a natural lifetime to redo my whole network.  

I think the more obvious approach is to use my new Synology router in this position or keep my WRT1900ACS. 

I guess you can close this.  I'm pretty sure there is some kind of bug here still but I don't want to waste your time or mine anymore with this when I have a workaround.   I appreciate all the time and help and appreciate your hard work on the Marvel OpenSource driver.

Thanks, 

John
New Bern, NC.

hagensieker, 2017-06-12:
Data seems okay.  I opened wireshark and my ESP8266 is the only device on the network besides my laptop.  The packets I'm capturing are what I would expect this device to do.   Talking to the router, looking for the MQTT server, and sending a signal on a data pin.

yuhhaurlin, 2017-06-12:
All right. It looks like data traffic is working. I close this one. If you still find any problems, you can post here or create another one to track it.

hagensieker, 2017-06-12:
You're going to hate me for this but it is a valuable clue.

I was using stable LEDE with 20170110 driver (or whatever the date is with stable)  It did not have the latest driver.  Just installed BrainSlayer DD-WRT 32270 latest and it only connects as an open network.  Will not connect with security enabled.  Sorry again.

hagensieker, 2017-06-12:
So stable LEDE 17.01.1 connects.  Your LEDE version you compiled for download earlier in the thread or LEDE or DD-WRT with latest driver does not work with encryption

yuhhaurlin, 2017-06-12:
It should not be the case. Can you just let me know if ping is all right?

hagensieker, 2017-06-12:
No ping because it does not connect and get an IP.  I am watching via Serial Monitor from the ESP8266.  I can confirm it only connect and gets IP with an Open Network

hagensieker, 2017-06-12:
but my laptop connects.  My server at work will connect, and my raspberry pi connects.  It is only the ESP8266 chips which will not connect unless the network is open.

yuhhaurlin, 2017-06-12:
Can you assign static IP to ESP8266?

hagensieker, 2017-06-12:
No you can only upload code to the chip and select the network and password.  The code is loaded via Arduino software.

[155 more replies]

closed 2017-08-21 by yuhhaurlin