kaloz/mwlwifi #171: WRT1200AC v2 WiFi 2.4GHz issue

tazbox, 2017-06-04:
Hi,

I switched currently from Linksys firmware to latest LEDE 17.01.1 r3316-7eb58cf109. I tried to enable the WiFi with the default driver and also with the latest version from this link: https://github.com/eduperez/mwlwifi_LEDE/releases.

The WiFi always gets connected and disconnected in such a way that I can't authenticate to it.

I attached some logs here .... maybe they help.

cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20170512
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x30
mac address: ##:##:##:##:##:## (it's a valid mac address here .. i just removed it)
2g: enable
5g: disable
antenna: 2 2
irq number: 105
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1100000
iobase1: e1280000
tx limit: 768
rx limit: 64
qe trigger number: 2793

dmesg:

[    8.042051] Loading modules backported from Linux version wt-2017-01-31-0-ge882dff19e7f
[    8.050091] Backport generated by backports.git backports-20160324-13-g24da7d3c
[    8.074464] <<Marvell Mac80211 Wireless PCIE Network Driver version 10.3.4.0-20170512>>
[    8.082539] pci 0000:00:01.0: enabling device (0140 -> 0142)
[    8.088364] ieee80211 phy0: priv->pcmd_buf = de334000  priv->pphys_cmd_buf = 1e334000
[    8.107501] ieee80211 phy0: fw download start
[    8.224251] ieee80211 phy0: FwSize = 118776 downloaded Size = 118776 curr_iteration 65496
[    9.181550] ieee80211 phy0: fw download complete
[    9.182703] ieee80211 phy0: pcmd = de334000
[    9.200563] ieee80211 phy0: firmware version: 0x702091a
[    9.551549] ieee80211 phy0: firmware region code: 30
[    9.581529] ieee80211 phy0: 2G disabled, 5G enabled
[    9.586460] ieee80211 phy0: 2 TX antennas, 2 RX antennas
[    9.591897] pci 0000:00:02.0: enabling device (0140 -> 0142)
[    9.597722] ieee80211 phy1: priv->pcmd_buf = dd808000  priv->pphys_cmd_buf = 1d808000
[    9.601577] ieee80211 phy1: fw download start
[    9.719594] ieee80211 phy1: FwSize = 118776 downloaded Size = 118776 curr_iteration 65511
[   10.671551] ieee80211 phy1: fw download complete
[   10.672678] ieee80211 phy1: pcmd = dd808000
[   10.690568] ieee80211 phy1: firmware version: 0x702091a
[   11.041551] ieee80211 phy1: firmware region code: 30
[   11.069511] ieee80211 phy1: 2G enabled, 5G disabled
[   11.074452] ieee80211 phy1: 2 TX antennas, 2 RX antennas
[   11.081460] ip_tables: (C) 2000-2006 Netfilter Core Team
[   11.088486] nf_conntrack version 0.5.0 (8025 buckets, 32100 max)
[   11.195122] xt_time: kernel timezone is -0000
[   11.201614] PPP generic driver version 2.4.2
[   11.206239] NET: Registered protocol family 24
[   12.399067] mvneta f1034000.ethernet eth0: configuring for fixed link mode
[   12.406263] mvneta f1034000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[   12.410840] device eth0 entered promiscuous mode
[   12.412182] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   12.424874] mvneta f1070000.ethernet eth1: configuring for fixed link mode
[   12.432362] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   12.438257] br-lan: port 1(eth0) entered forwarding state
[   12.438272] mvneta f1070000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[   12.451604] br-lan: port 1(eth0) entered forwarding state
[   12.457126] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   12.511591] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   14.130561] ieee80211 phy1: change: 0xffffffff
[   14.189820] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   14.196982] device wlan1 entered promiscuous mode
[   14.218100] pppoe-wan: renamed from ppp0
[   14.451554] br-lan: port 1(eth0) entered forwarding state
[   19.203392] ieee80211 phy1: change: 0x100
[   19.211563] ieee80211 phy1: change: 0x42
[   19.330575] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   19.337024] br-lan: port 2(wlan1) entered forwarding state
[   19.342561] br-lan: port 2(wlan1) entered forwarding state
[   21.341553] br-lan: port 2(wlan1) entered forwarding state

tazbox, 2017-06-04:
The problem seems to be related to WiFi security. I used WPA PSK and WPA2 PSK both in CCMP AES mode. If I disable the WiFi security i can connect to the router.

yuhhaurlin, 2017-06-05:
All clients or specific clients? Basically previous devices should not be affected.

yuhhaurlin, 2017-06-05:
From "dmesg", no problem found. Please use "logread | grep hostapd" to get more information.

tazbox, 2017-06-05:
Hi Yuhhaurlin,

Thank you for your quick response. The driver works fine for both 2.4 GHz and 5 GHz. It was a configuration problem (key mismatch).

Please excuse the noise. I think you can close the issue.

Regards,
Taz

closed 2017-06-05 by yuhhaurlin