kaloz/mwlwifi #172: 10.3.4.0-20170606 sta mode

BrainSlayer, 2017-06-07:
good news so far. station mode works with the new driver version. however. its only working up to vht80. it refuses to connect with vht160. i tested it with a netgear r7800. other vht160 capable clients connect fine. just the wrt3200acm only connects with vht80.

yuhhaurlin, 2017-06-07:
Our QA had done 160 MHz test for AP and STA with our own 8964 chip. It works. Can you let me know this problem is only with r7800？

BTW, how about stock firmware?

I will close another one and leave this one to track 160 MHz problem.

Thanks.

yuhhaurlin, 2017-06-07:
Please also help to check if 160 MHz of AP mode can work with other vendor chip. Thanks.

s-pimenta, 2017-06-07:
@BrainSlayer Did you build DD-WRT to the new driver release? Could you use Linux Kernel 4.9? I've been  using the 4.9 build since 12-05-2017 without issues. Thanks

BrainSlayer, 2017-06-07:
@s-pimenta  i'm only testing the latest driver using kernel 4.9. test scenario is a r7800 QCA9984 based AP. station is the WRT3200ACM. the ap has a second QCA9984 based station connected with vht160 running. the WRT3200
ACM does connect in station mode but shows only VHT80 from AP statistics on r7800 side. ah and consider the 12-05-2017 did not support station mode. it was just fixed with the last recent driverversion from 6.6.2017

BrainSlayer, 2017-06-07:
@yuhhaurlin: i tested STA mode. not AP. AP works in VHT160 likelly. but i can run another test and setup WRT3200ACM as AP with VHT160 and check if clients are able to connect in VHT160. by the way. test channel is 100 + 114

yuhhaurlin, 2017-06-07:
@BrainSlayer Our QA tests 36 (160 MHz), I will try 100 (160 MHz) with our own 8964 chip first. If needed, I will try to get r7800， but the priority will be lower.

BrainSlayer, 2017-06-07:
hostapd doesnt allow to configure vht160 at channel 36 for QCA9984. since in europe the channel specification doesnt allow vht160 at the lower band since some of them require DFS and some do not

BrainSlayer, 2017-06-07:
same for US. vht160 is not allowed for channel 36. so this isnt a good test

yuhhaurlin, 2017-06-08:
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

36 (160 MHz) is allowable for US.

BrainSlayer, 2017-06-08:
do you see the "@80" that means maximum 80 mhz the reason is that both lower bands have different properties. one is with DFS one is without

yuhhaurlin, 2017-06-08:
AUTO-BW is added.

yuhhaurlin, 2017-06-08:
If you set 36 (160 MHz), you need to wait for CAC (60 seconds) before the AP to work.

BrainSlayer, 2017-06-08:
doesnt work for QCA9984. driver refuses to work on it. i dont need to wait for CAC. hostapd exists immediatly with error

yuhhaurlin, 2017-06-08:
You can check the meaning of AUTO-BW.

BrainSlayer, 2017-06-08:
nl80211: Failed to set channel (freq=5180): -22 (Invalid argument)
Could not set channel for kernel driver
Interface initialization failed

yuhhaurlin, 2017-06-08:
phy#0
        Interface wlan0-1
                ifindex 10
                wdev 0x3
                addr 02:50:43:22:88:8a
                ssid LEDE-8964-5g-AES
                type AP
                channel 100 (5500 MHz), width: 160 MHz, center1: 5570 MHz
                txpower 23.00 dBm
        Interface wlan0
                ifindex 9
                wdev 0x2
                addr 00:50:43:22:88:8a
                ssid LEDE-8964-5g-open
                type AP
                channel 100 (5500 MHz), width: 160 MHz, center1: 5570 MHz
                txpower 23.00 dBm

yuhhaurlin, 2017-06-08:
BTW, channel 100 (160 Mhz) is also working for our own 8964. I will check the IoT problem with r7800 later with low priority. Thanks.

BrainSlayer, 2017-06-08:
i can only test with the r7800 since i dont have 2 WRT3200ACM. so there might be a vendor interoperability issue or something like that. especially with the center1 and center2 field. i can remember that the behaviour of these fields changed a while ago

yuhhaurlin, 2017-06-08:
Thanks. I will try to test with r7800 later with low priority. I think 160 Mhz (continue) is not so popular now.

BrainSlayer, 2017-06-08:
Interface ath0
        ifindex 9
        wdev 0x2
        addr 14:91:82:b9:16:d4
        type managed
        wiphy 0
        channel 100 (5500 MHz), width: 160 MHz, center1: 5570 MHz
        txpower 23.00 dBm

the client shows the same operational state as yours. but the ap shows the client associated in 80 mhz mode

BrainSlayer, 2017-06-08:
Station 14:91:82:b9:16:d4 (on ath0)
        inactive time:  22250 ms
        rx bytes:       70322179
        rx packets:     73978
        tx bytes:       6039857
        tx packets:     49381
        tx retries:     0
        tx failed:      0
        rx drop misc:   2
        signal:         -56 dBm
        signal avg:     -59 dBm
        tx bitrate:     6.0 MBit/s
        rx bitrate:     1170.0 MBit/s VHT-MCS 8 80MHz short GI VHT-NSS 3
        expected throughput:    1733.0Mbps
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short slot time:yes
        connected time: 28217 seconds

yuhhaurlin, 2017-06-08:
QA had done throughput test to make sure AP and STA are connected as 160 MHz. I think it is problem with r7800. I will check it later (I need to get r7800 first) and I think this is low priority.

yuhhaurlin, 2017-06-08:
<<Client>>
phy#0
        Interface wlan0
                ifindex 40
                wdev 0x1e
                addr 14:91:82:ba:ac:e9
                type managed
                channel 36 (5180 MHz), width: 160 MHz, center1: 5250 MHz
                txpower 20.00 dBm

<<Server>>
phy#0
        Interface wlan0-1
                ifindex 10
                wdev 0x3
                addr 02:50:43:22:88:8a
                ssid LEDE-8964-5g-AES
                type AP
                channel 36 (5180 MHz), width: 160 MHz, center1: 5250 MHz
                txpower 23.00 dBm
        Interface wlan0
                ifindex 9
                wdev 0x2
                addr 00:50:43:22:88:8a
                ssid LEDE-8964-5g-open
                type AP
                channel 36 (5180 MHz), width: 160 MHz, center1: 5250 MHz
                txpower 23.00 dBm

root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/sta

mac address: 14:91:82:ba:ac:e9
aid: 1
ampdu: true
amsdu: false
ht_cap: 0x086f, ampdu: 03, 05
rx_mask: 0xff, ff, ff, 00
vht_cap: 0x33831876, mcs: ffea, ffea
rx_bw: 3, rx_nss: 3
tdls: 0, tdls_init: 0
wme: 1, mfp: 0
IV: 000000000001

rx_bw: 3 ==> means 160 Mhz.

panda-mute, 2017-06-08:
r7800 using 80+80 or 160？wrt3200acm using continues 160.

BrainSlayer, 2017-06-08:
all plain vht160. no 80+80 tested yet

howl, 2017-06-19:
With LEDE 17.01.2 is not possible to start the wifi with 160 mhz in the WRT3200ACM, all the DFS channels fails with DFS error instantaneously.

This one trying to use channel 100 as stated before.

```
Mon Jun 19 19:20:39 2017 daemon.notice hostapd: wlan5g-ap: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=2, seg0=114, seg1=0, cac_time=60s
Mon Jun 19 19:20:39 2017 daemon.err hostapd: DFS start_dfs_cac() failed, -1
```

yuhhaurlin, 2017-06-19:
Please check issue #173.

howl, 2017-06-20:
That was the issue, anyways I get radar detected with 160 mhz quickly so I get back to be able to use the sdio wifi.

I know that it's about the other issue, but where is the problem with the global country set to 98 when multiple phy at different countries? In the linux kernel wifi stack?

yuhhaurlin, 2017-06-20:
Did you check issue #173?

howl, 2017-06-20:
Yes, but I asked here cause the other one is closed, just want to know where the bug is with the other issue, I know is not in mwlwifi.

Also in the future mwlwifi will allow selecting the country even with dts defined in the device?

yuhhaurlin, 2017-06-20:
You can't change country code for WRT3200ACM. Mwlwifi will do that for you based on region code in device power table. If you don't change country code, you will not encounter problem.

howl, 2017-06-21:
@yuhhaurlin but that could be a problem for some cases, you are limited to the zone you could use the router.

The WRT1900ACS v2 also has device power table and country can be changed with the previous mwlwifi version, haven't tested the actual one as I have it with lede 17.01.1.

In the future the driver are going to let the country be changed?

yuhhaurlin, 2017-06-22:
Device after 2016/6 won't allow user to change power and region. Driver and firmware have design to support it. If your device has device power table, power and region can't be changed.

howl, 2017-06-22:
I see, I have also a WRT1900ACS v1 that only differs from v2 that it doesn't device power table.

But anyways this behavior seems to be in conflict with the expected functionality of crda in the linux kernel. According to this documentation https://wireless.wiki.kernel.org/en/developers/regulatory/crda#changing_regulatory_domains the country can be changed always. First the device restriction is taken into account, and then in the selected country have more restrictions the are also applied.

That behavior seems to be more adequate. First because if someone has a router from x country and goes to y country where y country have more restrictions, is not possible for the consumer to operate the device illegally given that the consumer select the appropriate country code for the new place he/she brings his/her device. Second because there are devices that auto set the regulatory country based in the underlying wifi networks, for example apple laptops, and that makes that if someone with an european router goes to america could make other devices that even are not of their property to change it's regulatory country to FR in the case of WRT3200ACM.

I think that the way this works in mwlwifi should be changed at what is stated in the crda documentation of the linux kernel that doesn't goes in contradiction with fcc laws and even respect them more that the actual behavior of mwlwifi.

I suppose that also could fix the way the global country is messed up in this router when the sdio wifi card is also active.

If you agree I could open a new issue regarding this to discuss it more.

yuhhaurlin, 2017-06-22:
Sorry. This mechanism is used to comply with FCC rules. We will not change it.

howl, 2017-06-22:
@yuhhaurlin the crda proceed of the linux kernel also complain with the fcc rules and even in all scenarios, the mechanism in mwlwifi doesn't complain it in all scenarios. Anyways seems you are not open to talk about it, so agreed.

Fr3DBr, 2017-06-29:
@yuhhaurlin 

I believe that allowing raw tx power modification wouldn't comply to FCC expectations, but I'm pretty sure, that allowing customers to specify in which region, they're actually using their equipments, will match to FCC regulations, because for obvious reasons, those routers aren't used in US only, in my case as example, I'm in Brazil, but I'm using US power settings since my WRT3200ACM has been purchased in US, this means in my case, I'm not complying to my country regulation rules, and this isn't really cool.

Just my two cents on this.

howl, 2017-06-29:
@Fr3DBr that's what I pointed to @yuhhaurlin but it seems that or he didn't read the full comment, or he misunderstood it.

Also If I travel to US with my router I couldn't comply the fcc rules as I could use channels 12 and 13 where in the US are forbidden. The linux kernel rules to manage this issue making possible to comply the regulation always.

Ex: Router from the US in the US comply. Router from the US in EU comply, simply you cannot use channels 12 and 13 despite the region you are allow it because the original region of the device, but complies.

Router from the EU in EU comply. Router from the EU in the US comply, you could use channels 12 and 13 because the original region of the router allows it, but, being able to set the region to US the software will forbid the use of channels 12 and 13 at all, so we still comply, where with the actual mwlwifi behavior we weren't complying.

Also there is the mentioned problem with devices that auto set their region based in the neighbor devices. Imagine with a WRT3200ACM-EU in the US, the neighbors where you live having their devices, mostly apple devices, setting automatically their region to FR because your router.

I also think that the global region set to country 98 is fault of this behaviour, fault of mwlwifi and not fault of other linux components that respects the mentioned approach to be really fully fcc compilant.

I don't know how to make @yuhhaurlin understand it, as he understand I'm trying to ask to make the driver be able to set forbidden frequencies and powers as one desire.

BrainSlayer, 2017-06-29:
routers from US do not comply with the EU. different DFS rules and different 5 ghz frequencies

yuhhaurlin, 2017-06-29:
Please check with Belkin.

howl, 2017-06-29:
@BrainSlayer doesn't matter 2,4 GHz or 5 GHz the linux kernel approach is the same and makes comply the fcc rules always, mwlwifi doesn't.

Take your statement and apply mwlwifi approach. We go to EU with a US router, we can use the dfs rules from the US, power and frequencies in the EU.

Now take your statement with the linux kernel approach. As the US router is in the EU we set the region to the country we are, for example Germany, so we set it to DE. Now the router can't use the DFS rules from the US, can't set db power levels higher that the EU allows and so on.

The rule is, first we apply the restrictions the region the device is. Now we ALSO apply the restrictions the selected country code we specify, so both restrictions gets applied. FCC rules complained even if the costumer brings a device from one region to another.

Just please, someone read https://wireless.wiki.kernel.org/en/developers/regulatory/crda#changing_regulatory_domains is explained so well.

howl, 2017-06-29:
@yuhhaurlin the old driver previous to the hif restructuration allows, country change, belkin hasn't write it.

cilix-lab, 2017-06-29:
@howl I was just writing what it's perfectly explained in the first paragraph you see on the link you sent. That should be the approach.

First we apply the regulation from the device's region (US FCC/EU ETSI) (powertable), then we apply the country regulations (software/driver). That way, the router will always comply with the most restrictive regulation.

yuhhaurlin, 2017-06-29:
The design is requested by Belkin in order to comply with FCC rules. I am not the right one to make decision. Please focus on this issue here.

howl, 2017-06-29:
@cilix-lab  yes, and also you can't make a device from a region to work outside of it's scope, the US router will never use channels 12 and 13 even in the EU and even if you specify a EU country code.

@yuhhaurlin ok then, if you could provide a contact mail where Belkin could reply, as I know this situations, or no reply at all, or a automated one... Or if you have a contact and you could explain this yourself to him will be so appreciated. I think you can see that is contradictory that Belkin forces this behaviour to comply FCC rules where these rules makes possible to not complain FCC rules.

howl, 2017-07-05:
@yuhhaurlin please provide someone from Belkin to ask about this, will try in the following days to contact from their website.

Anyways, it's surprising that hardware being from Marvel that Belkin has so much power of decision over it's driver.

yuhhaurlin, 2017-07-05:
I have no idea about whom should be checked.

howl, 2017-07-06:
@yuhhaurlin mail send to Belkin to with the fault of their required behavior and a request to change it. Anyways I repeat is very strange that the power of decision of a Marvel product is not in the hand of Marvel itself.

howl, 2017-07-07:
@yuhhaurlin, well I recieved an automated mail the the support mail is no longer in use and alternatives to get support. I choose live chat support with Belkin. Belkin says that I have to ask with Linksys because the product is from Linksys.

As I said before this is not going anywhere, Marvel -> Belkin -> Linksys -> Pretty sure again Marvell or Belkin.

So, given that situation, could you provide evidences of what you say it's true? There should be a document or something. First, It doesn't have any sense, second Belkin seems to do not understand anything about this.

If that request from Belkin exists, perhaps not you, but someone in Marvell should know about it and whom in Belkin requested it.

howl, 2017-07-16:
@yuhhaurlin, no more info about the Belkin request to solve the multiple issues it makes? If the request is not fantasy someone have to tell you about it.

Reminder about the issues this behavior has:
1.- This behaviour doesn't complain FCC rules in some situations. (The argument about maintain this behavior is to complain FCC but it does not)
2.- Linux generic country code gets messed up because of this driver. (This makes impossible to have other wifi devices in the router, the wifi sdio also included or any other usb wifi card attached to the usb devices, without messing the dfs channels)

yuhhaurlin, 2017-07-16:
Don't change country code. Please check README. Thanks.

howl, 2017-07-16:
@yuhhaurlin thanks answering machine I didn't say anything about changing the country code, the two problems are there changing or not the country code. Is impossible to use the router with another wifi device at the same time and also because you enforce the country you can go against the FCC rules so the actual behavior is illegal. I know you understand both implications so please, don't reply with unuseful things.

At least you could say, sorry I don't want to change the erratic and problematic behavior because my own and not taking into account third parties.

howl, 2017-07-16:
I think that the only way that you are going to react to this is going to be when someone brings a EU router to US and start to transmit in channels 12 and 13 near governmental buildings because the faulty behavior in this driver.

yuhhaurlin, 2017-07-16:
AP is not client. Where you buy it where you use it. I can't answer more questions about this topic. Please focus on this issue here.

yuhhaurlin, 2017-07-16:
Unless stock firmware changes its behavior, otherwise mwlwifi won't change this mechanism.

aaron1959, 2017-07-16:
Maybe the FCC would be interested? in these texts.

howl, 2017-07-17:
@yuhhaurlin interesting, I bought it from Spain and country code is FR so still your statement is wrong. Remember the second issue I mentioned, this causes Spain Apple laptops to auto change their country code to France when they are near AP affected by this bad behavior, of course the worst thing is that this includes third people laptops. Anyways you could say that (Just because I don't want to) from the beginning instead of saying incoherent reasons.

I don't have the necessary knowledge but guess that someone fork this to follow linux kernel guidelines.

yuhhaurlin, 2017-07-17:
You can change country code to region code mapping by yourself.

howl, 2017-07-17:
The only thing I know is to define the country code and as we know that result in bogus global country code in the linux kernel, so I suppose that you mean another thing that I don't know, if you could provide more information would be appreciated.

yuhhaurlin, 2017-07-17:
There is an array in mwlwifi to do the mapping. You can modify the array to change the mapping.

[9 more replies]