kaloz/mwlwifi #173: DFS won't work with WRT3200ACM-UK

cilix-lab, 2017-06-07:
Hi! I've been trying to get DFS to work on my WRT3200ACM-UK with latest mwlwifi 10.3.4.0-20170606 with no success. I've read a lot about how DFS is supposed to work with mwlwifi, but it won't on my router. I'm thinking it might be a regulatory issue with the UK version of the router.

 My hostapd config:

```
# hostapd configuration file  
interface=wlan0
bridge=br0
driver=nl80211
ctrl_interface=/var/run/hostapd
ssid=Armada_5GHz
country_code=UK
ieee80211d=1
ieee80211h=1
hw_mode=a
channel=116
auth_algs=1
ignore_broadcast_ssid=0
wmm_enabled=1

# IEEE 802.11n
ieee80211n=1
ht_capab=[LDPC][HT40+][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]

# IEEE 802.11ac
ieee80211ac=1
vht_capab=[MAX-MPDU-11454][VHT160][RXLDPC][SHORT-GI-80][SHORT-GI-160][SU-BEAMFORMER][SU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][MAX-A-MPDU-LEN-EXP7]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=122

# WPA/IEEE 802.11i
wpa=2
wpa_passphrase=armada50
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
```

Running hostapd -d wlan0.conf will fail with:

```
phy: phy0
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Previous country code 98, new country code UK 
Continue interface setup after channel list update
(...)
nl80211: Regulatory information - country=98
nl80211: 2402-2472 @ 40 MHz 20 mBm
nl80211: 5170-5250 @ 80 MHz 17 mBm
nl80211: 5250-5330 @ 80 MHz 20 mBm (DFS)
nl80211: 5490-5710 @ 160 MHz 23 mBm (DFS)
nl80211: 57240-63720 @ 2160 MHz 40 mBm
Disable HT capability [DSSS_CCK-40] on 5 GHz band
hw vht capab: 0x33837976, conf vht capab: 0x33801976
wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Scan for neighboring BSSes prior to enabling 40 MHz channel
40 MHz affected channel range: [5570,5610] MHz
wlan0: nl80211: scan request
nl80211: Passive scan requested
Scan requested (ret=0) - scan timeout 10 seconds
Interface initialization will be completed in a callback
nl80211: Drv Event 33 (NL80211_CMD_TRIGGER_SCAN) received for wlan0
wlan0: nl80211: Scan trigger
wlan0: Event SCAN_STARTED (47) received
Unknown event 47
nl80211: Drv Event 34 (NL80211_CMD_NEW_SCAN_RESULTS) received for wlan0
wlan0: nl80211: New scan results available
nl80211: Scan included frequencies: 5580 5600
wlan0: Event SCAN_RESULTS (3) received
nl80211: Received scan results (0 BSSes)
HT40: control channel: 116  secondary channel: 120
Completing interface initialization
Mode: IEEE 802.11a  Channel: 116  Frequency: 5580 MHz
DFS 4 channels required radar detection
DFS all channels available, (SKIP CAC): no
DFS 0 chans unavailable - choose other channel: no
wlan0: interface state HT_SCAN->DFS
DFS start CAC on 5580 MHz
wlan0: DFS-CAC-START freq=5580 chan=116 sec_chan=1, width=1, seg0=122, seg1=0, cac_time=60s
nl80211: Start radar detection (CAC) 5580 MHz (ht_enabled=1, vht_enabled=1, bandwidth=80 MHz, cf1=5610 MHz, cf2=0 MHz)
  * freq=5580
  * vht_enabled=1
  * ht_enabled=1
  * bandwidth=80
  * channel_width=3
  * center_freq1=5610
  * center_freq2=0
nl80211: Failed to start radar detection: -22 (Invalid argument)
DFS start_dfs_cac() failed, -1
Interface initialization failed
wlan0: interface state DFS->DISABLED
wlan0: AP-DISABLED 
(...)
```

Doing iw reg get:

```
country 98: DFS-UNSET
    (2402 - 2472 @ 40), (N/A, 20), (N/A)
    (5170 - 5250 @ 80), (N/A, 17), (N/A)
    (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS
    (5490 - 5710 @ 160), (N/A, 23), (0 ms), DFS
    (57240 - 63720 @ 2160), (N/A, 40), (N/A)
```

cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info:

```
driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170606
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x30
mac address: (...)
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000000
radio: disable
iobase0: e1f80000
iobase1: e2200000
tx limit: 1024
rx limit: 16384
```

I've tried changing the country code, with no success whatsoever. Same config with no DFS will work.

I'm running Ubuntu 16.04 (https://github.com/cilix-lab/ubuntu-wrt) with hostapd-2.6.

aqwserf, 2017-06-07:
you're not alone, my FR version does not work either with 5GHz DFS on latest LEDE snapshot and 10.3.4.0.git-2017-05-26-1 mwlwifi driver

yuhhaurlin, 2017-06-08:
Please use https://drive.google.com/open?id=0B3qLWtcWB9EdcHJvd2tIbHl3dnM to do test and get log for me.

yuhhaurlin, 2017-06-08:
BTW, you can't change country code if your device is programmed with device power table.

cilix-lab, 2017-06-08:
Yes, I've noticed in the logs that it keeps using "country 98" which is what I see in "iw reg get", instead of an actual country. @aqwserf what do you get when you do "iw reg get"?

I'll try to test that image later today and share the results. I'm compiling my own kernel, though, since I'm running Ubuntu, so I won't be able to keep LEDE for long, just to test if it works and share the logs. Does this image's mwlwifi driver differ with the current commit?

yuhhaurlin, 2017-06-08:
It is the latest one. If you encounter problem, I can update related packages with debug messages to check it.

aqwserf, 2017-06-08:
@cilix-lab : same for me

    root@linksys:~# iw reg get
    global
    country 98: DFS-UNSET
    	(2402 - 2472 @ 40), (N/A, 20), (N/A)
    	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
    	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
    	(5490 - 5710 @ 160), (N/A, 23), (0 ms), DFS
    	(57240 - 63720 @ 2160), (N/A, 40), (N/A)

    phy#2
    country US: DFS-FCC
    	(2402 - 2472 @ 40), (N/A, 30), (N/A)
    	(5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
    	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
    	(5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
    	(5735 - 5835 @ 80), (N/A, 30), (N/A)
    	(57240 - 63720 @ 2160), (N/A, 40), (N/A)

    phy#1
    country FR: DFS-ETSI
    	(2402 - 2482 @ 40), (N/A, 20), (N/A)
    	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
    	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
    	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
    	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

    phy#0
    country FR: DFS-ETSI
    	(2402 - 2482 @ 40), (N/A, 20), (N/A)
    	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
    	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
    	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
    	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

as @yuhhaurlin said, if "power table loaded from dts" is set to "no", no chance to modify the table on the fly (if I remeber well, it's in an EEPROM):

    root@linksys:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info 

    driver name: mwlwifi
    chip type: 88W8964
    hw version: 7
    driver version: 10.3.4.0-20170512
    firmware version: 0x09030007
    power table loaded from dts: no

@yuhhaurlin : I'll do some test with your version and update when it's done

aqwserf, 2017-06-08:
@yuhhaurlin : just install your [build](https://drive.google.com/open?id=0B3qLWtcWB9EdcHJvd2tIbHl3dnM) and it seems pretty old (Reboot (SNAPSHOT, r4114-6704410):

    root@LEDE:~# uname -a
    Linux LEDE 4.9.20 #0 SMP Thu May 11 20:43:01 2017 armv7l GNU/Linux
    root@LEDE:~# opkg list-installed | grep mwl
    kmod-mwlwifi - 4.9.20+10.3.2.0-20170110-1

Am I wrong ?

yuhhaurlin, 2017-06-08:
Please use "cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info" to check version.

aqwserf, 2017-06-09:
Ok, I don't know where is the magic but with your build, everything works (almost) !! :) 

Stock configuration with @yuhhaurlin build:
* Channel 100 / Width 80MHz: OK
* Channel 120 / Width 80MHz: OK
* Channel 36 / Width 160MHz: OK
* Channel 120 / Width 160MHz: OK
* Channel auto / Width 80MHz: OK
* Channel auto / Width 160MHz: KO (the only one that don't work)

Custom configuration (SSID, security, MAC filter, DNS...) with LEDE Snapshot:
* Channel 100 / Width 80MHz: KO
* Channel 120 / Width 80MHz: KO
* Channel 36 / Width 160MHz: KO
* Channel 120 / Width 160MHz: KO
* Channel auto / Width 80MHz: KO
* Channel auto / Width 160MHz: KO

I don't know where is the difference between the two builds, except the configuration (and obviously package version, kernel...)

The wifi driver seems to be the same:

* LEDE

        root@linksys:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info 
        
        driver name: mwlwifi
        chip type: 88W8964
        hw version: 7
        driver version: 10.3.4.0-20170512
        firmware version: 0x09030007
        power table loaded from dts: no
        firmware region code: 0x30
        mac address: 60:38:e0:bc:2f:82
        2g: disable
        5g: enable
        antenna: 4 4
        irq number: 45
        ap macid support: 0000ffff
        sta macid support: 00010000
        macid used: 00000001
        radio: enable
        iobase0: e0e80000
        iobase1: e1100000
        tx limit: 1024
        rx limit: 16384

* yuhhaurlin

        root@LEDE:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info 

        driver name: mwlwifi
        chip type: 88W8964
        hw version: 7
        driver version: 10.3.4.0-20170606
        firmware version: 0x09030007
        power table loaded from dts: no
        firmware region code: 0x30
        mac address: 60:38:e0:bc:2f:82
        2g: disable
        5g: enable
        antenna: 4 4
        irq number: 45
        ap macid support: 0000ffff
        sta macid support: 00010000
        macid used: 00000000
        radio: disable
        iobase0: e0e00000
        iobase1: e1080000
        tx limit: 1024
        rx limit: 16384

Maybe in my wireless configuration...? I can post mine if you want and I also have all the "logread -f" log for every tests !

yuhhaurlin, 2017-06-09:
I get LEDE on 2017/5/21, I just build the image and replace mwlwifi with updated driver. The previous LEDE I got is around 2017/2/20. I did not encounter problem.
If you also use LEDE, compatible mac80211 package uses static regulatory database, so I think you should have the same regulatory file as I have, you can use "iw reg get" to check it.
Device with power table can't change region code. If the channel setting is allowable by your country, there should be no problem.

yuhhaurlin, 2017-06-09:
I close this one. You can still post here if needed. BTW, as I know, there are some pre-built LEDE images from community. Maybe you can try these images.

closed 2017-06-09 by yuhhaurlin

cilix-lab, 2017-06-09:
OK... so I did try @yuhhaurlin 's LEDE image and it works perfectly. I noticed that when doing "iw reg get", I get "FR" now instead of "98" for the country. I see that the problem is not mwlwifi, evidently, but a regulatory issue... something I'm missing.

I'd rather keep compiling my own kernel and stick with Ubuntu, so the question is: what am I missing?

Could you point me in the right direction to solve this issue? Is it mac80211? I don't get why compiling the latest kernel available from Ubuntu and latest mwlwifi, DFS won't work. It seems that the same thing happens to @aqwserf with LEDE Snapshot. So, what could it be?

Any help would be greatly appreciated, though I understand it's not a mwlwifi issue. I'll start googling around and post if I find any solution for reference.

yuhhaurlin, 2017-06-09:
If you use mac80211 of kernel, I suggest that you use static regulatory database. After you configure your kernel with this option, you need to get db.txt from LEDE and put it to net/wireless. 
You can try it. If you have problem, I can give you more detail information next week when I come into office.

yuhhaurlin, 2017-06-09:
Sorry. Regulatory database is controlled by cfg80211, this is option for cfg80211:
net/wireless: source code for cfg80211
net/mac80211: source code for mac80211

mac80211 Backport package used by LEDE/OpenWrt already uses static regulatory database and they will update db.txt if needed.

yuhhaurlin, 2017-06-09:
Otherwise the system must use CRDA to get regulatory database. I think maybe CRDA does not work on your image.

seragh, 2017-06-10:
cilix-lab,

your hostapd.conf says country_code=US, wouldn't be surprised for this to be the cause.

I also use crda with in kernel wireless stack on an EU model and DFS works for me.

The canonical source for the regdb is https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/

cilix-lab, 2017-06-10:
Thanks for the tip, but I've tried changing the country and it goes back to "98" since it is a WRT3200ACM and doesn't load power tables from DTS. I'm now compiling the kernel with static regulatory database from wireless-regdb. I'll report back if that ends up solving it. Thanks again!

aqwserf, 2017-06-10:
Hi !

LEDE's build:

        root@linksys:~# iw reg get
        global
        country 98: DFS-UNSET
        	(2402 - 2472 @ 40), (N/A, 20), (N/A)
        	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        	(5490 - 5710 @ 160), (N/A, 23), (0 ms), DFS
        	(57240 - 63720 @ 2160), (N/A, 40), (N/A)

        phy#2
        country US: DFS-FCC
        	(2402 - 2472 @ 40), (N/A, 30), (N/A)
        	(5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        	(5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        	(5735 - 5835 @ 80), (N/A, 30), (N/A)
        	(57240 - 63720 @ 2160), (N/A, 40), (N/A)

        phy#1
        country FR: DFS-ETSI
        	(2402 - 2482 @ 40), (N/A, 20), (N/A)
        	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

        phy#0
        country FR: DFS-ETSI
        	(2402 - 2482 @ 40), (N/A, 20), (N/A)
        	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

@yuhhaurlin 's build:

        root@LEDE:~# iw reg get
        global
        country FR: DFS-ETSI
        	(2402 - 2482 @ 40), (N/A, 20), (N/A)
        	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

        phy#1
        country FR: DFS-ETSI
        	(2402 - 2482 @ 40), (N/A, 20), (N/A)
        	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

        phy#0
        country FR: DFS-ETSI
        	(2402 - 2482 @ 40), (N/A, 20), (N/A)
        	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

As @cilix-lab said, with your build, I also have FR instead of 98 but I also have only 2 wifi interface and not 3. 
I didn't understand all you said about the mac80211 package/db.txt/CRDA but as I understand, we have to compile our build with latest sources and do some modifications to have a functionnal WIFI driver right? 
I know that some community builds are made, but I'd like to use "stock" LEDE's build as possible, cause of maintenance, evolutivity and packages compatibility. 

In other words, is it possible to modify LEDE's snapshot that works like your's for the moment ? And do you think that an upgrade will be done in the future to LEDE's repos ?

yuhhaurlin, 2017-06-10:
Can you disable phy2?

yuhhaurlin, 2017-06-10:
Or set country code to FROM. In fact mwlwifi will set country based on the region code got from your device power table. If you don't touch country code, it should be all right. Your problem is that phy0 accepts the cage of country but mwlwifi won't allow you to change it.
Even though you hack mwlwifi to change country code, firmware of 88W8964 won't accepts the change. It will still follow region code got from device power table.
The better way is that you should not change country code and let mwlwifi set it for you.

cilix-lab, 2017-06-10:
I tried compiling the kernel with cfg80211 static regulatory database (REGDB) replacing db.txt in net/wireless, but had no luck still. I'm getting country 98 again. Will keep trying to find the issue here.

yuhhaurlin, 2017-06-10:
If you don't change country code, maybe the problem is gone.

cilix-lab, 2017-06-11:
After playing around with lots of different configurations, I realized I could make DFS work on my image by modifying `/etc/default/crda` and setting `REGDOMAIN=GB`, since my router is "UK".

I'm still getting "country 98" when doing "iw reg get", but I noticed that now it says "country 98: DFS-ETSI" instead of "DFS-UNSET", and then DFS will work properly.

I wish I would get "iw reg get" to display the country code properly, but I'm happy enough with DFS working.

aqwserf, 2017-06-11:
@cilix-lab : great ;-)
@yuhhaurlin : I've tried to delete "mlan0" (phy2) with 

    root@linksys:~# iw dev mlan0 del

but "iw reg get" is the same:

        root@linksys:~# iw reg get
        global
        country 98: DFS-UNSET
        	(2402 - 2472 @ 40), (N/A, 20), (N/A)
        	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        	(5490 - 5710 @ 160), (N/A, 23), (0 ms), DFS
        	(57240 - 63720 @ 2160), (N/A, 40), (N/A)

        phy#2
        country US: DFS-FCC
        	(2402 - 2472 @ 40), (N/A, 30), (N/A)
        	(5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        	(5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        	(5735 - 5835 @ 80), (N/A, 30), (N/A)
        	(57240 - 63720 @ 2160), (N/A, 40), (N/A)

        phy#1
        country FR: DFS-ETSI
        	(2402 - 2482 @ 40), (N/A, 20), (N/A)
        	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

        phy#0
        country FR: DFS-ETSI
        	(2402 - 2482 @ 40), (N/A, 20), (N/A)
        	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

Can you tell if there is another way ? And with "iw reg set FR" don't change naything as you said...
Do you have any other ideas ?
Maybe a PR to LEDE... ?

yuhhaurlin, 2017-06-11:
Modify via /etc/config/wireless，UCI,  or GUI.

aqwserf, 2017-06-12:
Ok, tried so set in /etc/config/wireless:

         config wifi-device 'radio2'
        	option type 'mac80211'
        	option channel '36'
        	option hwmode '11a'
        	option path 'platform/soc/soc:internal-regs/f10d8000.sdhci/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'
        	option htmode 'VHT80'
        	option disable '1'
        	option country 'FR'

        config wifi-iface 'default_radio2'
        	option device 'radio2'
        	option network 'lan'
        	option ssid 'DFS'
        	option key 'iwIFGJKTAC26zEXHvDGD3NJ0IJAcj0N1tNbrRTYjJNd2teyhdizaYx4hzt921MV'
        	option mode 'monitor'
        	option disabled '1'

and the "iw reg get" is still the same. DFS still don't work, neither 160MHz...

Here is the log when change to channel 100 / width 80MHz:

        Mon Jun 12 14:15:49 2017 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
        Mon Jun 12 14:15:49 2017 daemon.notice hostapd: wlan0: AP-DISABLED 
        Mon Jun 12 14:15:49 2017 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING 
        Mon Jun 12 14:15:49 2017 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
        Mon Jun 12 14:15:49 2017 kern.info kernel: [  560.508330] device wlan0 left promiscuous mode
        Mon Jun 12 14:15:49 2017 kern.info kernel: [  560.512843] br-lan: port 3(wlan0) entered disabled state
        Mon Jun 12 14:15:49 2017 kern.debug kernel: [  560.532351] ieee80211 phy0: change: 0x40
        Mon Jun 12 14:15:50 2017 kern.debug kernel: [  560.644303] ieee80211 phy0: change: 0x100
        Mon Jun 12 14:15:50 2017 daemon.notice netifd: Network device 'wlan0' link is down
        Mon Jun 12 14:15:50 2017 daemon.notice netifd: radio0 (3688): command failed: Not supported (-95)
        Mon Jun 12 14:15:50 2017 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
        Mon Jun 12 14:15:50 2017 kern.debug kernel: [  561.098227] ieee80211 phy0: change: 0xffffffff
        Mon Jun 12 14:15:50 2017 kern.info kernel: [  561.211440] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
        Mon Jun 12 14:15:50 2017 kern.info kernel: [  561.218082] br-lan: port 3(wlan0) entered blocking state
        Mon Jun 12 14:15:50 2017 kern.info kernel: [  561.223442] br-lan: port 3(wlan0) entered disabled state
        Mon Jun 12 14:15:50 2017 kern.info kernel: [  561.228910] device wlan0 entered promiscuous mode
        Mon Jun 12 14:15:50 2017 kern.info kernel: [  561.233661] br-lan: port 3(wlan0) entered blocking state
        Mon Jun 12 14:15:50 2017 kern.info kernel: [  561.239001] br-lan: port 3(wlan0) entered forwarding state
        Mon Jun 12 14:15:50 2017 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
        Mon Jun 12 14:15:51 2017 kern.info kernel: [  561.658244] br-lan: port 3(wlan0) entered disabled state
        Mon Jun 12 14:15:55 2017 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
        Mon Jun 12 14:15:55 2017 kern.debug kernel: [  566.245026] ieee80211 phy0: change: 0x100
        Mon Jun 12 14:15:55 2017 kern.debug kernel: [  566.258099] ieee80211 phy0: change: 0x60
        Mon Jun 12 14:15:55 2017 kern.debug kernel: [  566.478183] ieee80211 phy0: change: 0x40
        Mon Jun 12 14:15:56 2017 daemon.notice hostapd: wlan0: interface state HT_SCAN->DFS
        Mon Jun 12 14:15:56 2017 daemon.notice hostapd: wlan0: DFS-CAC-START freq=5500 chan=100 sec_chan=1, width=1, seg0=106, seg1=0, cac_time=60s
        Mon Jun 12 14:15:56 2017 daemon.err hostapd: DFS start_dfs_cac() failed, -1
        Mon Jun 12 14:15:56 2017 daemon.err hostapd: Interface initialization failed
        Mon Jun 12 14:15:56 2017 daemon.notice hostapd: wlan0: interface state DFS->DISABLED
        Mon Jun 12 14:15:56 2017 daemon.notice hostapd: wlan0: AP-DISABLED 
        Mon Jun 12 14:15:56 2017 daemon.notice hostapd: wlan0: interface state DISABLED->DISABLED
        Mon Jun 12 14:15:56 2017 daemon.notice hostapd: wlan0: AP-DISABLED 
        Mon Jun 12 14:15:56 2017 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING 
        Mon Jun 12 14:15:56 2017 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
        Mon Jun 12 14:15:56 2017 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
        Mon Jun 12 14:15:56 2017 kern.debug kernel: [  566.698176] ieee80211 phy0: change: 0x60
        Mon Jun 12 14:15:56 2017 kern.info kernel: [  566.738210] device wlan0 left promiscuous mode
        Mon Jun 12 14:15:56 2017 kern.info kernel: [  566.742704] br-lan: port 3(wlan0) entered disabled state
        Mon Jun 12 14:15:56 2017 kern.debug kernel: [  566.801204] ieee80211 phy0: change: 0x100
        Mon Jun 12 14:15:56 2017 daemon.notice hostapd: ELOOP: remaining socket: sock=23 eloop_data=0xb6e89b40 user_data=0 handler=0x3545c
        Mon Jun 12 14:15:56 2017 daemon.notice netifd: radio0 (3688): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process 2416 path ()
        Mon Jun 12 14:15:56 2017 kern.debug kernel: [  566.981197] ieee80211 phy0: change: 0xffffffff
        Mon Jun 12 14:15:56 2017 kern.info kernel: [  567.094227] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
        Mon Jun 12 14:16:06 2017 daemon.notice netifd: radio0 (3988): command failed: Not supported (-95)
        Mon Jun 12 14:16:06 2017 daemon.err hostapd: Configuration file: /var/run/hostapd-phy0.conf
[... cut ...]
        Mon Jun 12 14:16:45 2017 daemon.notice netifd: radio0 (4534): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process 2416 path ()
        Mon Jun 12 14:16:45 2017 kern.debug kernel: [  616.100688] ieee80211 phy0: change: 0xffffffff
        Mon Jun 12 14:16:45 2017 kern.info kernel: [  616.213721] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

I also tried with interface phy2 up in monitor and AP mode, nothing works...
Do you see anything that could go wrong ? Any other lead ? 

Thanks

yuhhaurlin, 2017-06-12:
What do you get via "iw reg get"?

yuhhaurlin, 2017-06-12:
Sorry. If you disable radio2, you should not get information of it.

yuhhaurlin, 2017-06-12:
I can check my build tomorrow to see if I build in the support for phy2. I will reply you tomorrow.

cilix-lab, 2017-06-12:
I haven't used LEDE in a while, but I understand that only with the `config wifi-device 'radio2'` section having the `disabled` option, the radio2 shouldn't be a problem.

About DFS not working, I understand the `/etc/default/crda` is not an option since it's LEDE, but to make it work on my image, I also had to set the `CONFIG_CFG80211_INTERNAL_REGDB=y` for my kernel config and get `db.txt` from [wireless-regdb](https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/). That way I had a working static regulatory database and finally setting a European country got me from `country 98: DFS-UNSET` to `country 98: DFS-ETSI` and everything started working.

I think DFS not working in @aqwserf LEDE image might be related to a missing static regulatory database and that `DFS-UNSET` you get when doing `iw reg get`. Is there any option in LEDE's kernel config to let you set a static regulatory database?

I am also wondering, what exactly does UCI do when parsing the info in the `/etc/config/wireless` file. I mean, it creates hostapd config, but it probably does a couple more things, right?

aqwserf, 2017-06-12:
Still the same...:

    root@linksys:~# iw reg get
    global
    country 98: DFS-UNSET
    	(2402 - 2472 @ 40), (N/A, 20), (N/A)
    	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
    	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
    	(5490 - 5710 @ 160), (N/A, 23), (0 ms), DFS
    	(57240 - 63720 @ 2160), (N/A, 40), (N/A)

    phy#2
    country US: DFS-FCC
    	(2402 - 2472 @ 40), (N/A, 30), (N/A)
    	(5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
    	(5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
    	(5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
    	(5735 - 5835 @ 80), (N/A, 30), (N/A)
    	(57240 - 63720 @ 2160), (N/A, 40), (N/A)

    phy#1
    country FR: DFS-ETSI
    	(2402 - 2482 @ 40), (N/A, 20), (N/A)
    	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
    	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
    	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
    	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

    phy#0
    country FR: DFS-ETSI
    	(2402 - 2482 @ 40), (N/A, 20), (N/A)
    	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
    	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
    	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
    	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

If you can tell me how to disable/delete it, that's fine :)
Is the country code of PHY2 that is different from the other two makes the global one set to "DFS-UNSET" ?
Ok, let me know when you can !

Thanks for your investment !!

aqwserf, 2017-06-12:
@cilix-lab : sorry, I didn't see your post when writing...
For the `disable` option, you're probably right. I deleted it in `config wifi-device` and it seems disable. The only difference is that in the GUI, I can still disable it with the appropriate button.

For the kernel config, after a quick search, there seems to be this [option](http://lists.infradead.org/pipermail/lede-dev/2017-April/006983.html) but the [config.seed](https://downloads.lede-project.org/snapshots/targets/mvebu/generic/config.seed) used to compile doesn't include it.
But as I understand, even thought it'll be include, the database is an EEPROM and then can't be modifiy. Am I wrong ?

And for the last question, I would say probably :)

I'm still ok do some tests if needed !

throbulator, 2017-06-12:
I also have an EU Model WRT3200ACM bought in Germany and can confirm this.

I'm running a LEDE snapshot from friday 09 june 2017 (LEDE Reboot SNAPSHOT r4361-8b486ec) with mwlwifi 10.3.4.0-20170606.

I have only done basic wireless config via UCI/LUCI and the country codes for the wireless interfaces stay on "98/US/FR".

It doesn't matter if I change it to "DE" in UCI/LUCI for phy0/phy1.

I did some resets of the entire firmware and just the wireless part and left the country config untouched after that.

Still shows "98/US/FR" (no idea why phy2 shows up, it's disabled).

But I guess there are no differences for Europe between FR and DE regarding channels and transmit power. I still find it a bit weird that the build provided by @yuhhaurlin seems to set DFS properly for EU/UK models for the "global" part while normal LEDE snapshots don't(?).

Does the "global" part of "iw reg get" matter at all?

```
global
country 98: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5490 - 5710 @ 160), (N/A, 23), (0 ms), DFS
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

phy#2
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

phy#1
country FR: DFS-ETSI
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country FR: DFS-ETSI
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

Chadster766, 2017-06-12:
It could be that the eeprom that stores the country code has old information and you might have to customize the crda database to suit.

yuhhaurlin, 2017-06-13:
Remove kmod-mwifiex-sdio from your image:

make menuconfig
Uncheck: Kernel modules->Wireless Drivers->kmod-mwifiex-sdio
make V=s

Please remove mwifiex (phy2) first to see if your problem is gone. I will add this module and check if I can reproduce this problem later.

aqwserf, 2017-06-13:
@yuhhaurlin : you're my new god :)
You were righ. After uninstall `kmod-mwifiex-sdio` :

    global
    country FR: DFS-ETSI
    	(2402 - 2482 @ 40), (N/A, 20), (N/A)
    	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
    	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
    	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
    	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

    phy#1
    country FR: DFS-ETSI
    	(2402 - 2482 @ 40), (N/A, 20), (N/A)
    	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
    	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
    	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
    	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

    phy#0
    country FR: DFS-ETSI
    	(2402 - 2482 @ 40), (N/A, 20), (N/A)
    	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
    	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
    	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
    	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

and as expected, everything works (except `auto` and `160MHz`). 

I'll add the uninstall to my install script, except if a PR or something like that will be report to LEDE's dev... May I ? 

Your :
`make menuconfig` 
`Uncheck: Kernel modules->Wireless Drivers->kmod-mwifiex-sdio`
`make V=s` 
is a compilation of the image or the wifi driver only ? Do I still to do that or I can simply remove the package in the future?

Finally, I've still this package associate:
`root@linksys:~# opkg list-installed | grep sdio`
`mwifiex-sdio-firmware - 2016-09-21-42ad5367-1`
Can I remove it ?

And again, thanks a lot for your recommandations, your work, and the resolution of my issue !!

throbulator, 2017-06-13:
Can confirm this works on my EU model:

Removed "kmod-mwifiex-sdio" & "mwifiex-sdio-firmware"
Reboot
Reset wireless config (rm -f /etc/config/wireless => wifi config > /etc/config/wireless => reboot)

Now it shows "FR" on "global/phy0/phy1", DFS seems to be set properly and "phy2" is gone.

Haven't tried changing country code to "DE" yet via UCI/LUCI, will do that later. But even if "FR" sticks (because it is set by firmware) it shouldn't be any trouble for EU models (i guess).

Thanks for the tip!

```
global
country FR: DFS-ETSI
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#1
country FR: DFS-ETSI
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country FR: DFS-ETSI
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

throbulator, 2017-06-13:
Did set the country code to "DE" now via UCI/LUCI and global goes back to "98" but DFS stays on "ETSI".

Setting to "FR" again via UCI/LUCI does nothing. Only a wireless reset did set to "FR" for all 3 entries again.

So better not touch country code after flashing / wifi detect via UCI/LUCI for now.

FCS001FCS, 2020-09-19:
> Can confirm this works on my EU model:
> 
> Removed "kmod-mwifiex-sdio" & "mwifiex-sdio-firmware"
> Reboot
> Reset wireless config (rm -f /etc/config/wireless => wifi config > /etc/config/wireless => reboot)
> 
> Now it shows "FR" by default on "global/phy0/phy1", DFS seems to be set properly and "phy2" is gone.
> 
> Haven't tried changing country code to "DE" yet via UCI/LUCI, will do that later. But even if "FR" sticks (because it is set by firmware) it shouldn't be any trouble for EU models (i guess).
> 
> Thanks for the tip!
> 
> ```
> global
> country FR: DFS-ETSI
>         (2402 - 2482 @ 40), (N/A, 20), (N/A)
>         (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
>         (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
>         (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
>         (57000 - 66000 @ 2160), (N/A, 40), (N/A)
> 
> phy#1
> country FR: DFS-ETSI
>         (2402 - 2482 @ 40), (N/A, 20), (N/A)
>         (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
>         (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
>         (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
>         (57000 - 66000 @ 2160), (N/A, 40), (N/A)
> 
> phy#0
> country FR: DFS-ETSI
>         (2402 - 2482 @ 40), (N/A, 20), (N/A)
>         (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
>         (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
>         (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
>         (57000 - 66000 @ 2160), (N/A, 40), (N/A)
> ```

++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

This worked for me on Stable Build 19.07.4. I was stumped for a while until I came across this topic. DFS would fail and I could not use the channels between 52 and 140.

It was the interference of the Third Radio (Radio 2 - mwifiex) that was conflicting with the regulatory frequency restriction. I also have an EU model WRT3200ACM which I set to FR.

**Excellent help! Thanks to all involved.**

I used the 19.07.4 Image Builder and used the following packages, note the removal of `-mwifiex-sdio-firmware -kmod-btmrvl -kmod-mwifiex-sdio` 

`make image PROFILE=linksys_wrt3200acm PACKAGES="luci-ssl-openssl luci-app-advanced-reboot block-mount kmod-usb-storage luci-app-samba4 luci-app-uhttpd vsftpd-tls hostapd-common wpad-basic mwlwifi-firmware-88w8964 -libustream-wolfssl ppp -ppp-mod-pppoe -mwifiex-sdio-firmware -kmod-btmrvl -kmod-mwifiex-sdio"`