kaloz/mwlwifi #176: 802.1x Potentially Broken After 10.3.2

SpotComms, 2017-06-11:
I have two WRT1900AC access points running LEDE and after https://github.com/lede-project/source/commit/e783f588eba1ef56365f1293e01509140b115f7a, clients using NetworkManager were unable to connect to the access points when using 802.1x. They are able to successfully authenticate with the RADIUS server, but they get stuck on "connecting (getting IP configuration)". And according to logread on the APs the NetworkManager clients appear to successfully connect. iOS/Android devices connect fine, but NetworkManager on Linux does not. While this issue may very well be one in NetworkManager (hasn't been updated in a month) or LEDE (no commits after May 25th seem suspect), I believe its more likely to be one in mwlwifi. Any suggestions to help debug are greatly appreciated.

yuhhaurlin, 2017-06-11:
If PSK can work, I don't think 1x can't work. Please make sure you build in correct version of hostapd/wpa_supplicant. The default version is for PSK.

closed 2017-06-11 by yuhhaurlin

SpotComms, 2017-07-04:
After much more testing, it seems to only occur when 802.11w is enabled. Any insight with that?
On LEDE 17.01.1, wpad-full 802.1x + 802.11w works fine, builds after May 20ish however fail.