kaloz/mwlwifi #177: rx/tx rate, signal report

BrainSlayer, 2017-06-13:
is there any progress on this feature? i may be able to help out with some modifications but the firmware api for the 88w8964 isnt well documented (not not complete) todo this job

yuhhaurlin, 2017-06-13:
Work for this and WDS. It is not trivial and there is no specific APIs for that.

yuhhaurlin, 2017-06-13:
I close https://github.com/kaloz/mwlwifi/issues/163 and keep this one to track this job.

gufus, 2017-06-24:
@yuhhaurlin

Any update on WDS?

-Thanks

yuhhaurlin, 2017-06-24:
Sorry. WDS is still ongoing.

yuhhaurlin, 2017-07-03:
Please help to check https://github.com/kaloz/mwlwifi/commit/1cc8aa697f9d853a51d7cc60775635051b7a9302. Thanks.

yuhhaurlin, 2017-07-04:
The code works here. I close this one. If you encounter problem, please let me know.

closed 2017-07-04 by yuhhaurlin

adi-brasov, 2017-07-04:
Is the noise level implemented ? It shows constantly -114 for all devices on dd-wrt with latest mwlwifi commit,

yuhhaurlin, 2017-07-04:
No.

adi-brasov, 2017-07-04:
Will be implemented ?

yuhhaurlin, 2017-07-04:
I am not sure now. I will check it after other major features are done.

yuhhaurlin, 2017-07-07:
Noise level is done: https://github.com/kaloz/mwlwifi/commit/5599a72a5eda8fbdc4188ed3d18c7dd5fb54b4cb.

adi-brasov, 2017-07-07:
Thanks.
 I will wait for BS to build dd-wrt with this mwlwifi commit.
Have you any idea why the power is higher with stock Linksys firmware (on both 2.4 and 5gz) ? I tested with the same device (a phone) and an wifi analyzer app, at a distance between 10-15 m, and constantly (with dd-wrt installed on the router) the phone analyzer app shows much lower signal strength than stock fw, and the distance coverage is lower too.
 The router is wrt3200acm with dd-wrt.

yuhhaurlin, 2017-07-07:
Did you know what is the 88W8964 firmware version of stock firmware?

yuhhaurlin, 2017-07-07:
Basically this version of 88W8964 firmware had been verified with range vs rate test. There is no problem found for this version of firmware. 88W8964 firmware will base on device power table to control tx power, driver can't control tx power.

adi-brasov, 2017-07-07:
I dont know how to check this.
The Linksys fw is ver. 1.0.6.181063.

yuhhaurlin, 2017-07-07:
Mwlwifi and stock firmware use the same code base of 88W8964 firmware now. Power is controlled by firmware. Driver can't control it.

adi-brasov, 2017-07-07:
It is possible that some other dd-wrt config to influence this behaviour ?
I didnt tested with LEDE yet.

yuhhaurlin, 2017-07-07:
No way to change tx power via driver if device is programmed with device power table.

yuhhaurlin, 2017-07-07:
Can you check if your device is programmed with device power table? Please check README.

adi-brasov, 2017-07-07:
I cant check right now but must be programmed with device power table, since is wrt3200acm and no alteration of pwr table is posiible.
I know is possible to do that on wrt1900acs v1 (released before 2016-06 and new regulation restrictions).

yuhhaurlin, 2017-07-07:
If your device is programmed with device power table, no way to change tx power via driver.