kaloz/mwlwifi #178: lose connection to wifi as client with "cmd 0x8205=GetWatchdogBitMap timed out"

wliao229, 2017-06-14:
Router: WRT1900ACS v2
OS: Lede "Reboot"
Kernel: 4.9.30
Driver: 4.9.30+10.3.4.0-20170606-1

My router drops connection to another wifi network (as client through 2.4G) after a certain duration ranging from several hours to a few days. Simply enable/disable the wireless interface won't solve the issue, and I have to reboot it to get it back to work. This issue has been a while for me, and I found the following error messages after the issue occurs.

## kernel message ##
> Wed Jun 14 09:57:33 2017 kern.err kernel: [82876.856843] ieee80211 phy1: cmd 0x8205=GetWatchdogBitMap timed out
> Wed Jun 14 09:57:33 2017 kern.err kernel: [82876.863054] ieee80211 phy1: return code: 0x0205
> Wed Jun 14 09:57:33 2017 kern.err kernel: [82876.867612] ieee80211 phy1: timeout: 0x0205
> Wed Jun 14 09:57:38 2017 kern.err kernel: [82880.872795] ieee80211 phy1: cmd 0x8205=GetWatchdogBitMap timed out
> Wed Jun 14 09:57:38 2017 kern.err kernel: [82880.879012] ieee80211 phy1: return code: 0x0205
> Wed Jun 14 09:57:38 2017 kern.err kernel: [82880.883561] ieee80211 phy1: timeout: 0x0205
> Wed Jun 14 09:57:42 2017 kern.err kernel: [82884.912731] ieee80211 phy1: cmd 0x8205=GetWatchdogBitMap timed out
> Wed Jun 14 09:57:42 2017 kern.err kernel: [82884.918949] ieee80211 phy1: return code: 0x0205
> Wed Jun 14 09:57:42 2017 kern.err kernel: [82884.923499] ieee80211 phy1: timeout: 0x0205
> Wed Jun 14 09:57:46 2017 kern.err kernel: [82889.020668] ieee80211 phy1: cmd 0x8205=GetWatchdogBitMap timed out
> Wed Jun 14 09:57:46 2017 kern.err kernel: [82889.026886] ieee80211 phy1: return code: 0x0205
> Wed Jun 14 09:57:46 2017 kern.err kernel: [82889.031435] ieee80211 phy1: timeout: 0x0205
> Wed Jun 14 09:57:58 2017 kern.err kernel: [82901.510499] ieee80211 phy1: cmd 0x8205=GetWatchdogBitMap timed out
> Wed Jun 14 09:57:58 2017 kern.err kernel: [82901.516723] ieee80211 phy1: return code: 0x0205
> Wed Jun 14 09:57:58 2017 kern.err kernel: [82901.521272] ieee80211 phy1: timeout: 0x0205

yuhhaurlin, 2017-06-14:
Any special application to let it happen? Does this problem happen in previous version? Can you cat /sys/debug/kernel/ieee80211/phy0/mwlwifi/info?

wliao229, 2017-06-15:
I don't think there is any special application. I didn't monitor the driver's version, but I have been using [Davidc502](https://davidc502sis.dynamic-dns.net)'s LEDE built for several recent releases, and he updated the wifi drivers throughout. 

I will update once the issue to occur since I reboot my router yesterday. Usually this problem occurs after a couple days. And it occurred for two different wifi networks at my school: one requires authentication, the other does not; and both ran into the same issue.

yuhhaurlin, 2017-06-19:
It looks like the same as issue #142. I will try to build image for you to do tests later.

yuhhaurlin, 2017-06-19:
I think I close this one. Please use #142 to check your problem.

closed 2017-06-19 by yuhhaurlin