kaloz/mwlwifi #179: Compiling error when build with Openwrt 15.05.1

1plus-1, 2017-06-15:
HEAD is now at 36bc327... Change driver version to 10.3.4.0-20170606.
Packing checkout...
. /home/nick/openwrt/openwrt-15.05.1/include/shell.sh; xzcat /home/nick/openwrt/openwrt-15.05.1/dl/mwlwifi-10.3.4.0-20170606-36bc32767ed89e07c5c83036861d2fa4eb1f8629.tar.xz | tar -C /home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/.. -xf - 
touch /home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/.prepared_22eaf194d2113c0997422f6faac3c787
rm -f /home/nick/openwrt/openwrt-15.05.1/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/stamp/.mwlwifi_installed
(cd /home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/./; if [ -x ./configure ]; then find /home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/ -name config.guess | xargs -r chmod u+w; find /home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/ -name config.guess | xargs -r -n1 cp --remove-destination /home/nick/openwrt/openwrt-15.05.1/scripts/config.guess; find /home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/ -name config.sub | xargs -r chmod u+w; find /home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/ -name config.sub | xargs -r -n1 cp --remove-destination /home/nick/openwrt/openwrt-15.05.1/scripts/config.sub; AR="arm-openwrt-linux-uclibcgnueabi-gcc-ar" AS="ccache_cc -c -Os -pipe -march=armv7-a -mtune=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft" LD=arm-openwrt-linux-uclibcgnueabi-ld NM="arm-openwrt-linux-uclibcgnueabi-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="arm-openwrt-linux-uclibcgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-uclibcgnueabi-strip OBJCOPY=arm-openwrt-linux-uclibcgnueabi-objcopy OBJDUMP=arm-openwrt-linux-uclibcgnueabi-objdump SIZE=arm-openwrt-linux-uclibcgnueabi-size CFLAGS="-Os -pipe -march=armv7-a -mtune=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft " CXXFLAGS="-Os -pipe -march=armv7-a -mtune=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft " CPPFLAGS="-I/home/nick/openwrt/openwrt-15.05.1/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/include -I/home/nick/openwrt/openwrt-15.05.1/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/include -I/home/nick/openwrt/openwrt-15.05.1/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/usr/include -I/home/nick/openwrt/openwrt-15.05.1/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/include " LDFLAGS="-L/home/nick/openwrt/openwrt-15.05.1/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/lib -L/home/nick/openwrt/openwrt-15.05.1/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/lib -L/home/nick/openwrt/openwrt-15.05.1/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/usr/lib -L/home/nick/openwrt/openwrt-15.05.1/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/lib "   ./configure --target=arm-openwrt-linux --host=arm-openwrt-linux --build=x86_64-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls   ; fi; )
rm -f /home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/.configured_*
touch /home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/.configured_yyn
make   -C "/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/linux-3.18.23" ARCH="arm" CROSS_COMPILE="arm-openwrt-linux-uclibcgnueabi-" SUBDIRS="/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606" NOSTDINC_FLAGS="-I/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606 -I/home/nick/openwrt/openwrt-15.05.1/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/include/mac80211-backport/uapi -I/home/nick/openwrt/openwrt-15.05.1/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/include/mac80211-backport -I/home/nick/openwrt/openwrt-15.05.1/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/include/mac80211/uapi -I/home/nick/openwrt/openwrt-15.05.1/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/include/mac80211 -include backport/backport.h" modules
make[3]: Entering directory '/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/linux-3.18.23'
  CC [M]  /home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.o
/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.c: In function 'mwl_set_ht_caps':
/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.c:422:2: error: implicit declaration of function 'ieee80211_hw_set' [-Werror=implicit-function-declaration]
  ieee80211_hw_set(hw, AMPDU_AGGREGATION);
  ^
/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.c:422:23: error: 'AMPDU_AGGREGATION' undeclared (first use in this function)
  ieee80211_hw_set(hw, AMPDU_AGGREGATION);
                       ^
/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.c:422:23: note: each undeclared identifier is reported only once for each function it appears in
/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.c:423:23: error: 'SUPPORTS_AMSDU_IN_AMPDU' undeclared (first use in this function)
  ieee80211_hw_set(hw, SUPPORTS_AMSDU_IN_AMPDU);
                       ^
/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.c: In function 'mwl_wl_init':
/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.c:641:23: error: 'SIGNAL_DBM' undeclared (first use in this function)
  ieee80211_hw_set(hw, SIGNAL_DBM);
                       ^
/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.c:642:23: error: 'HAS_RATE_CONTROL' undeclared (first use in this function)
  ieee80211_hw_set(hw, HAS_RATE_CONTROL);
                       ^
/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.c:647:23: error: 'AP_LINK_PS' undeclared (first use in this function)
  ieee80211_hw_set(hw, AP_LINK_PS);
                       ^
/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.c:649:23: error: 'SUPPORTS_PER_STA_GTK' undeclared (first use in this function)
  ieee80211_hw_set(hw, SUPPORTS_PER_STA_GTK);
                       ^
/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/core.c:650:23: error: 'MFP_CAPABLE' undeclared (first use in this function)
  ieee80211_hw_set(hw, MFP_CAPABLE);
                       ^
cc1: some warnings being treated as errors
[... cut ...]
make[2]: *** [/home/nick/openwrt/openwrt-15.05.1/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/linux-mvebu/mwlwifi-10.3.4.0-20170606/.built] Error 2
make[2]: Leaving directory '/home/nick/openwrt/openwrt-15.05.1/package/kernel/mwlwifi'
package/Makefile:191: recipe for target 'package/kernel/mwlwifi/compile' failed
make[1]: *** [package/kernel/mwlwifi/compile] Error 2
make[1]: Leaving directory '/home/nick/openwrt/openwrt-15.05.1'
/home/nick/openwrt/openwrt-15.05.1/include/toplevel.mk:181: recipe for target 'package/kernel/mwlwifi/compile' failed
make: *** [package/kernel/mwlwifi/compile] Error 2

mrvltest, 2017-06-15:
Please move to updated mac80211 package or modify the code by yourself. For old mac80211, there is no function ieee80211_hw_set(). You can check previous code and modify mwlwifi to work with old mac80211.

yuhhaurlin, 2017-06-15:
I close this one.

closed 2017-06-15 by yuhhaurlin