kaloz/mwlwifi #182: WDS support for Rango (WRT3200ACM)

yuhhaurlin, 2017-07-03:


BrainSlayer, 2017-07-03:
does it mean wds sta / ap 4 address mode works now?

yuhhaurlin, 2017-07-03:
No. I create this one to track This job

GillyMoMo, 2017-08-01:
What's the status on this? What part of the code do I need to look at and possibly compare it to get this working right?

yuhhaurlin, 2017-08-01:
This feature won't be done quickly.

gufus, 2017-08-03:
@yuhhaurlin 

Maybe soon...

Please :-)

yuhhaurlin, 2017-08-03:
Busy on other projects. Will implement this feature later. Sorry.

gufus, 2017-08-04:
Okay.

Thanks, hopefully WDS is on top of your todo pile. :-)

- Regards

ad019, 2017-09-12:
Can we expect something will be done on WDS and MU-MIMO this year? 
Or will Marvell keep pushing these features into the future and keep dangling a carrot that they are "committed" to this product?

yuhhaurlin, 2017-09-12:
Sure.

GillyMoMo, 2017-09-12:
Like with the 1900ACv1, it'll take time before this matures. I remain in no rush for this. As before we are at the whim of Marvell as usual.

tapper82, 2017-09-21:
How long do we have to wait? Is any work being dun on this driver any more?

ad019, 2017-09-22:
Not long now I think. Now that the 32X is out and being based on OpenWRT, probably has the same driver as 3200acm, Marvell will have to resolve the pending issues.

gufus, 2017-09-22:
We'll see...

BrainSlayer, 2017-09-22:
@ad019 the wrt32x uses openwrt but does not use mwlwifi. it still uses the good old propertiery driver which has been used in the original firmware for the wrt3200acm

yuhhaurlin, 2017-09-22:
Yes. WRT32x uses proprietary driver. Mwlwifi is mac80211 driver and it can support all Marvell chips for WRT series driver. It is open source. Proprietary driver of WRT32x is dedicated driver for 88W8964. It is up to you to decide to use which driver.

yuhhaurlin, 2017-09-22:
In fact, it is the same driver used by stock firmware and Marvell ports it to work with OpenWrt.

BrainSlayer, 2017-09-22:
@yuhhaurlin i have seen that the firmware blob in this WRT32X for the chipset is a little bit newer and has slightly different code in it. is there anything changed in it which is usefull for mwlwifi? i'm testing this binary right now with mwlwifi on my wrt3200acm

yuhhaurlin, 2017-09-22:
No. Mwlwifi is mac80211 driver, it can support more features. BTW, for previous memory shortage problem,   it is fixed by mwlwifi first.

BrainSlayer, 2017-09-22:
i was talking about the W8889.bin file which is slightly different in the WRT32X version.

yuhhaurlin, 2017-09-22:
I have no idea about W8889.bin! If WRT32X uses this binary, then maybe I am wrong. Sorry.

BrainSlayer, 2017-09-22:
nono. it uses the propertiery driver, but for sure it uses the same  firmware binary format like mwlwifi, but the WRT32X has a newer version of it. but doesnt look much different from the content

yuhhaurlin, 2017-09-22:
Yes. There is new version of firmware for 88W8964.

BrainSlayer, 2017-09-22:
yea. sorry typo from my side. do you know if there is any change in this firmware? or lets say do you know whats has been changed?

yuhhaurlin, 2017-09-22:
Our QA is pretty busy. If community wants to test new version of firmware, I can do basic test next week and push it to here for you to do test.

BrainSlayer, 2017-09-22:
i already tested it. so far it works. no negative effects here in my current environment.

yuhhaurlin, 2017-09-22:
What is the version number?

BrainSlayer, 2017-09-22:
for the wifi firmware or what?

yuhhaurlin, 2017-09-22:
Firmware.

BrainSlayer, 2017-09-22:
8006.img version: B4.18.01 Jun  2 2017 04:52:13 Copyright (c) 2007, 2009-2015 by Cisco Systems, Inc

BrainSlayer, 2017-09-22:
same version string has in the previous version

BrainSlayer, 2017-09-22:
driver reports
[    4.031097] ieee80211 phy0: fw download start
[    4.031101] ieee80211 phy0: ddr init: download start
[    4.031479] ieee80211 phy0: ddr init: download complete
[    4.168238] ieee80211 phy0: FwSize = 447616 downloaded Size = 447616 curr_ite                                                                                                                                                             ration 65522
[    7.037377] ieee80211 phy0: fw download complete
[    7.050209] ieee80211 phy0: pcmd = de62c000
[    7.067420] ieee80211 phy0: firmware version: 0x9030008
[    7.295385] ieee80211 phy0: firmware region code: 10
[    7.325027] ieee80211 phy0: 2G enabled, 5G enabled
[    7.329845] ieee80211 phy0: 4 TX antennas, 4 RX antennas
[    7.335255] pci 0000:00:02.0: enabling device (0140 -> 0142)
[    7.341035] ieee80211 phy1: priv->pcmd_buf = da3fc000  priv->pphys_cmd_buf =                                                                                                                                                              1a3fc000
[    7.347674] ieee80211 phy1: fw download start
[    7.347679] ieee80211 phy1: ddr init: download start
[    7.348060] ieee80211 phy1: ddr init: download complete
[    7.487829] ieee80211 phy1: FwSize = 447616 downloaded Size = 447616 curr_ite                                                                                                                                                             ration 65527
[    8.681374] ieee80211 phy1: fw download complete
[    8.694183] ieee80211 phy1: pcmd = da3fc000
[    8.710395] ieee80211 phy1: firmware version: 0x9030008
[    8.848381] ieee80211 phy1: firmware region code: 10
[    8.870970] ieee80211 phy1: 2G enabled, 5G disabled
[    8.875894] ieee80211 phy1: 4 TX antennas, 4 RX antennas

yuhhaurlin, 2017-09-22:
cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info.

BrainSlayer, 2017-09-22:
driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x10
mac address: 14:91:82:b9:16:d4
2g: enable
5g: enable
antenna: 4 4
irq number: 46
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00010000
radio: enable
iobase0: e0c00000
iobase1: e0e80000
tx limit: 1024
rx limit: 16384

yuhhaurlin, 2017-09-22:
All right, I will push 9.3.0.8 to here next week. However, it won't be tested by our QA.

yuhhaurlin, 2017-09-25:
@BrainSlayer Please help to test version 9.3.0.8 of 88W8964 firmware via https://drive.google.com/open?id=0B3qLWtcWB9EdWU9vM09QRy1Bc00.

thagabe, 2017-09-25:
Wait, since the 32X will run on openwrt can we port the killer engine to lede on the 3200acm? Gamer here.

rs-se, 2017-09-25:
You should ask in LEDE forums, this is for the Wi-Fi driver only.

BrainSlayer, 2017-09-25:
@thagabe there is no killer engine. the so called killer feature is just a set of qos rules. i checked this already

s-pimenta, 2017-09-25:
@thagabe Marketing...

eduperez, 2017-09-25:
@yuhhaurlin : If you push the source code for 9.3.0.8, in a branch here, or even in a fork, I could make packages for the stable version of LEDE, so more people can test the new drivers, without having to reinstall the whole system from scratch.

yuhhaurlin, 2017-09-25:
9.3.0.8 is binary firmware of 88W8964. I will do basic test and push to here. LEDE is pretty stable now.

gufus, 2017-09-25:
Nice to see  work being done on this issue.

yuhhaurlin, 2017-09-25:
This is not related to WDS.

tapper82, 2017-10-06:
Is any one still working on this driver? Any news any update? the people would like to no.

nitroshift, 2017-10-07:
@yuhhaurlin 

Been running the 9.3.0.8 firmware on LEDE for over a week now without any issues. You can commit it in the official repo.

nitroshift

yuhhaurlin, 2017-10-07:
Thanks. I will check in on10/11.

spamtree, 2017-11-30:
Question:

I see this is  still listed as open. Was WDS implemented?

Thanks

yuhhaurlin, 2017-11-30:
This feature is under development now.

ad019, 2017-11-30:
Does this development also include MU-MIMO?

yuhhaurlin, 2017-12-01:
No. MU-MIMO will be done later.

adamcarter1, 2017-12-01:
Regarding mu-mimo. 

Would Be nice if it was done sooner.

Let me just ask this.  Perhaps you can help me understand.
The linksys wrt 32x uses a firmware loosely based on “open wrt”. Not a standard linksys firmware.
Open wrt pretty much gave birth to lede, which you use to test all this stuff.
The linksys 32x has mu-mimo.  Ergo Can’t any of that info from the marvell people who made the driver or firmware or whatever for 32x, be usefull in getting it to work for dd-wrt and/ or lede. 

Am i taking crazy pills?

yuhhaurlin, 2017-12-01:
Sorry. I have no idea about wrt32x.

adamcarter1, 2017-12-01:
Sorry, wasn’t trying to be rude or pushy. i was under the assumption you work for marvell. 

What i know about the 32x is this
Their processor, a marvell processor, the same processor is in both the 3200acm and the 32x.
The 32x uses an open wrt based firmware, they have things like mu-mimo working or at least thats what they claim on the box. 
If that is true, someone at marvell has already written the mu-mimo code for an open wrt based system with this processor.
Why are we reinventing the wheel here. 

If you do work for marvell. I’m not asking you to release sensitive wrt-32x information... just perhaps  send a quick internal work email to colleagues who worked on the marvell driver for the 32x to get some of this stuff working. 

I and others appreciate your time.

yuhhaurlin, 2017-12-01:
Driver of WRT32x is not mwlwifi. It takes time to port MU-MIMO to mwlwifi.

BrainSlayer, 2017-12-01:
the wrt32x and the wrt3200acm are identical devices. just firmware and price is different. mu-mimo is working for the factory firmwares for wrt3200acm and wrt32x. no difference here. but the driver used in these factory firmwares is not open source and propertiery. you cannot copy just the code to the opensource mwlwifi driver. first it would be a license problem, second the driver is totally different and does not make use of mac80211 etc. so even the code for mu-mimo would be totally different in mwlwifi once its done. the information exist, yes and marvell makes use of it to implement the same feature into mwlwifi. but its a work from scratch

yuhhaurlin, 2017-12-01:
@BrainSlayer Thanks.

ad019, 2017-12-01:
@yuhhaurlin do you have scheduled time period in mind for MU-MIMO?

adamcarter1, 2017-12-01:
Wow, so in essence  the 32x even though the firmware is quite unlike the 3200 firmware “a very resorceful person on the dd-wrt firmware who decoded it, said , looking at the files that it was loosely based on  “open wrt” so even though it is loosly based on that... the driver they use is specifically designed to work only with the propriatary firmware.

Seems to me that it is very inefficient, way to do things. 

As far as licensing issues. Isn’t marvell making the drivers for all firmwares? Who would they have to pay? themselves?

aaron1959, 2017-12-01:
The key word is loosely.

BrainSlayer, 2017-12-01:
@adamcarter1 : first i was the one who decoded the firmware and analysed it too look for the so called gaming feature which was just a few lines of qos script. and no the propertiery marvell driver can work with any firmware. it doesnt matter if its openwrt or anything else. its just a driver and its not just loosely based on openwrt, the firmware is 100% openwrt. in fact a guy known to me from the old openwrt community wrote the firmware. or lets say i found indications that he wrote parts of it

[49 more replies]

closed 2017-12-14 by yuhhaurlin