kaloz/mwlwifi #183: 5GHZ (VHT) clients cannot connect to AP (1900AC v1)

BuzzBumbleBee, 2017-07-05:
**Device :** WRT 1900ACv1
**Build :** LEDE Reboot 17.01.2
**Kernel :** 4.4.71
**Driver Version :** 4.4.71+10.3.4.0.git-2..6-1

**_Description :_** 
I have several AC / VHT clients (lots of android phones and my main laptop), non of them can connect to my 5GHz network. The logs (will paste further down) show that the WPA authentication has worked fine and (if needed) DHCP server has offed the IP. But the device doesn't get any throughput, this also affects the devices when a static IP has been set.

Switching the AC AP to N mode in the web interface / WiFi config fixes authentication for all 5Ghz AC clients (but as expected they now only connect in N mode). 

**_System Log Of Client Connecting :_**
```
Wed Jul  5 09:09:40 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 84:c7:ea:cd:5d:2f
Wed Jul  5 09:09:40 2017 daemon.info hostapd: wlan1: STA 84:c7:ea:cd:5d:2f WPA: pairwise key handshake completed (RSN)
Wed Jul  5 09:09:41 2017 daemon.info dnsmasq-dhcp[1595]: DHCPDISCOVER(br-lan) 84:c7:ea:cd:5d:2f 
Wed Jul  5 09:09:41 2017 daemon.info dnsmasq-dhcp[1595]: DHCPOFFER(br-lan) 192.168.1.76 84:c7:ea:cd:5d:2f 
Wed Jul  5 09:09:43 2017 daemon.info dnsmasq-dhcp[1595]: DHCPDISCOVER(br-lan) 84:c7:ea:cd:5d:2f 
Wed Jul  5 09:09:43 2017 daemon.info dnsmasq-dhcp[1595]: DHCPOFFER(br-lan) 192.168.1.76 84:c7:ea:cd:5d:2f 
```

**_Station Dump :_**
Note 1: 08:bd:43:c3:b9:67 is a non AC N device and that connects fine both with DHCP and static AP address setups fine.

Note 2: Soon after this log (about 35sec after starting the connection) the client gives up and disconnects totally.

```
root@1900ac:~# iw wlan1 station dump
Station 08:bd:43:c3:b9:67 (on wlan1)
	inactive time:	0 ms
	rx bytes:	7419154
	rx packets:	75332
	tx bytes:	232065158
	tx packets:	159194
	tx retries:	0
	tx failed:	0
	rx drop misc:	0
	signal:  	-72 dBm
	signal avg:	-70 dBm
	tx bitrate:	300.0 MBit/s MCS 15 40MHz short GI
	rx bitrate:	300.0 MBit/s MCS 15 40MHz short GI
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short slot time:yes
	connected time:	1601 seconds
Station 84:c7:ea:cd:5d:2f (on wlan1)
	inactive time:	790 ms
	rx bytes:	2923
	rx packets:	23
	tx bytes:	2195
	tx packets:	13
	tx retries:	0
	tx failed:	0
	rx drop misc:	0
	signal:  	-76 dBm
	signal avg:	-73 dBm
	tx bitrate:	780.0 MBit/s VHT-MCS 8 80MHz short GI VHT-NSS 2
	rx bitrate:	351.0 MBit/s VHT-MCS 4 80MHz VHT-NSS 2
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short slot time:yes
	connected time:	20 seconds
```

**_WIFI Config : _**
```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
	option country 'GB'
	option channel '2'
	option htmode 'HT20'
	option txpower '19'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option macaddr 'b4:75:0e:60:d4:42'
	option ssid 'myssid'
	option encryption 'psk2'
	option key 'MYKEY'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:03.0/0000:03:00.0'
	option htmode 'VHT80'
	option country 'GB'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option macaddr 'b4:75:0e:60:d4:43'
	option ssid 'myssid5g'
	option encryption 'psk2'
	option key 'MYKEY'
```

Any help would be greatly appreciated (Also tried a recent OpenWRT build and things did not improve)

yuhhaurlin, 2017-07-05:
1. How about open mode?
2. If your device does not have device power table, you should make sure power table is ready in DTS file.
3. Please use "cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info" to get version and device power table information.

BuzzBumbleBee, 2017-07-05:
@yuhhaurlin 

Seems power table has been loaded from DTS

```
driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20170606
firmware version: 0x0702091a
power table loaded from dts: yes
firmware region code: 0x0
mac address: 00:25:9c:<removed for issue>
2g: enable
5g: disable
antenna: 4 4
irq number: 90
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: d0e00000
iobase1: d0f80000
tx limit: 768
rx limit: 64
qe trigger number: 157165
```

Also open mode seems to have no affect.

BuzzBumbleBee, 2017-07-05:
Another point is. The following process allows for the VHT devices to work fine for a short while ~5min

1. Switch 5GHz AP to N mode
2. Switch 5Ghz AP to AC mode
3. VHT clients will now connect for a shot time fine then fail.

yuhhaurlin, 2017-07-05:
What do you mean "fail"?

BuzzBumbleBee, 2017-07-05:
If connected with an IP no throughput (ping fails ect) if a new device tried to connect again no throughput.

DHCP would result in no IP address being obtained by the client
Static would result in connection but no throughput

yuhhaurlin, 2017-07-05:
This happened after 5 min?

yuhhaurlin, 2017-07-05:
Can you use "psk2+ccmp" for encryption?

BuzzBumbleBee, 2017-07-05:
Yeah so starting in AC / 80mhz mode on channel 100. Devices would not get any throughput.

1. Switch 5GHz AP to N mode
2. Switch 5Ghz AP to AC / 80mhz mode
3. VHT clients will now connect and have throughput for a shot time fine then fail (approximately 5min).

Ill have a go at psk2+ccmp but not sure that will help as the authentication part is fine. Also after further testing forcing VHT40 mode seems fine as well.

yuhhaurlin, 2017-07-05:
Can you use "iw reg get" to get regulatory information?

BuzzBumbleBee, 2017-07-05:
```
global
country GB: DFS-ETSI
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
	(5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

yuhhaurlin, 2017-07-05:
Please don't use DFS channels for 88W8864.

BuzzBumbleBee, 2017-07-05:
@yuhhaurlin will try VHT80 on channel 60.

I take it this is a known issue with this chipset. and that the 1900ACv1 (Marvell Armada XP MV78230) is the same chip in effect ?

yuhhaurlin, 2017-07-05:
How about channel 36 (80 MHz)?

yuhhaurlin, 2017-07-05:
From your regulatory information, this is the only 80 Mhz you can use.

BuzzBumbleBee, 2017-07-05:
@yuhhaurlin right yeah that was the original channel i was having issues with.

Ill switch to that now and keep testing for the rest of the day and log when it starts to fail. Im starting to think it could just be a faulty unit :/

yuhhaurlin, 2017-07-05:
Please use channel 36 (80 Mhz). Are you sure you will encounter problem for this setting?

yuhhaurlin, 2017-07-05:
I think your device can change country code (old device without device power table). You can change country code to "US" and you can use 149 (80 MHz) for this regulatory domain.

country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

BuzzBumbleBee, 2017-07-05:
@yuhhaurlin I have now switched to AC / 80Mhz / channel 36.

It was the configuration I was running yesterday and 1st thing today and was having issues so i'm 99% it will fail again today at some point.

I will keep this issue updated.

yuhhaurlin, 2017-07-05:
O.K. You can also change country code to "US" and use 149 (80 MHz).

(5735 - 5835 @ 80), (N/A, 30), (N/A)  => 149 (80 Mhz)

yuhhaurlin, 2017-07-05:
BTW, in order to get good throughput, please make sure use AES-CCMP only.

BuzzBumbleBee, 2017-07-05:
@yuhhaurlin i didnt know the "AES-CCMP" trick, thanks for the heads up :)

yuhhaurlin, 2017-07-05:
You can check README.

BuzzBumbleBee, 2017-07-05:
So just after switching to channel 36 / VHT80 the issue happened again, log below of the switch disable -> enable to switch channels 

```
Wed Jul  5 10:25:40 2017 daemon.info hostapd: wlan1: STA 84:c7:ea:cd:5d:2f IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Jul  5 10:25:44 2017 daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
Wed Jul  5 10:25:44 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 00:0a:f5:96:a4:27
Wed Jul  5 10:25:44 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 08:bd:43:c3:b9:67
Wed Jul  5 10:25:44 2017 daemon.notice hostapd: wlan1: AP-DISABLED 
Wed Jul  5 10:25:44 2017 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING 
Wed Jul  5 10:25:44 2017 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Wed Jul  5 10:25:44 2017 kern.info kernel: [ 5961.988362] device wlan1 left promiscuous mode
Wed Jul  5 10:25:44 2017 kern.info kernel: [ 5961.992972] br-lan: port 2(wlan1) entered disabled state
Wed Jul  5 10:25:44 2017 kern.debug kernel: [ 5962.050809] ieee80211 phy1: change: 0x60
Wed Jul  5 10:25:44 2017 kern.debug kernel: [ 5962.161245] ieee80211 phy1: change: 0x100
Wed Jul  5 10:25:44 2017 daemon.notice netifd: Network device 'wlan1' link is down
Wed Jul  5 10:25:45 2017 daemon.notice netifd: radio1 (9754): command failed: Not supported (-95)
Wed Jul  5 10:25:45 2017 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Wed Jul  5 10:25:45 2017 kern.debug kernel: [ 5962.908176] ieee80211 phy1: change: 0xffffffff
Wed Jul  5 10:25:45 2017 kern.info kernel: [ 5963.027299] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Wed Jul  5 10:25:45 2017 kern.info kernel: [ 5963.038268] device wlan1 entered promiscuous mode
Wed Jul  5 10:25:45 2017 kern.info kernel: [ 5963.043131] br-lan: port 2(wlan1) entered forwarding state
Wed Jul  5 10:25:45 2017 kern.info kernel: [ 5963.048764] br-lan: port 2(wlan1) entered forwarding state
Wed Jul  5 10:25:45 2017 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Jul  5 10:25:45 2017 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
Wed Jul  5 10:25:45 2017 kern.debug kernel: [ 5963.056132] ieee80211 phy1: change: 0x100
Wed Jul  5 10:25:45 2017 kern.debug kernel: [ 5963.068315] ieee80211 phy1: change: 0x40
Wed Jul  5 10:25:45 2017 kern.info kernel: [ 5963.178180] br-lan: port 2(wlan1) entered disabled state
Wed Jul  5 10:25:46 2017 kern.debug kernel: [ 5963.288133] ieee80211 phy1: change: 0x40
Wed Jul  5 10:25:46 2017 kern.debug kernel: [ 5963.498173] ieee80211 phy1: change: 0x40
Wed Jul  5 10:25:46 2017 kern.debug kernel: [ 5963.606678] ieee80211 phy1: change: 0x100
Wed Jul  5 10:25:46 2017 daemon.err hostapd: Using interface wlan1 with hwaddr b4:75:0e:60:d4:43 and ssid "myssid5g"
Wed Jul  5 10:25:46 2017 kern.debug kernel: [ 5963.729695] ieee80211 phy1: change: 0x100
Wed Jul  5 10:25:46 2017 kern.debug kernel: [ 5963.742923] ieee80211 phy1: change: 0x42
Wed Jul  5 10:25:46 2017 kern.info kernel: [ 5963.926666] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Wed Jul  5 10:25:46 2017 kern.info kernel: [ 5963.933573] br-lan: port 2(wlan1) entered forwarding state
Wed Jul  5 10:25:46 2017 kern.info kernel: [ 5963.939204] br-lan: port 2(wlan1) entered forwarding state
Wed Jul  5 10:25:46 2017 daemon.notice hostapd: wlan1: interface state HT_SCAN->ENABLED
Wed Jul  5 10:25:46 2017 daemon.notice hostapd: wlan1: AP-ENABLED 
Wed Jul  5 10:25:46 2017 kern.debug kernel: [ 5964.101363] ieee80211 phy1: change: 0x20
Wed Jul  5 10:25:46 2017 daemon.notice netifd: Network device 'wlan1' link is up
Wed Jul  5 10:25:48 2017 kern.info kernel: [ 5965.938162] br-lan: port 2(wlan1) entered forwarding state
Wed Jul  5 10:25:48 2017 daemon.info hostapd: wlan1: STA 84:c7:ea:cd:5d:2f IEEE 802.11: authenticated
Wed Jul  5 10:25:49 2017 daemon.info hostapd: wlan1: STA 84:c7:ea:cd:5d:2f IEEE 802.11: associated (aid 1)
Wed Jul  5 10:25:49 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 84:c7:ea:cd:5d:2f
Wed Jul  5 10:25:49 2017 daemon.info hostapd: wlan1: STA 84:c7:ea:cd:5d:2f WPA: pairwise key handshake completed (RSN)
Wed Jul  5 10:25:49 2017 daemon.info dnsmasq-dhcp[1595]: DHCPDISCOVER(br-lan) 84:c7:ea:cd:5d:2f 
Wed Jul  5 10:25:49 2017 daemon.info dnsmasq-dhcp[1595]: DHCPOFFER(br-lan) 192.168.1.76 84:c7:ea:cd:5d:2f 
Wed Jul  5 10:25:50 2017 daemon.info hostapd: wlan1: STA 08:bd:43:c3:b9:67 IEEE 802.11: authenticated
Wed Jul  5 10:25:50 2017 daemon.info hostapd: wlan1: STA 08:bd:43:c3:b9:67 IEEE 802.11: associated (aid 2)
Wed Jul  5 10:25:51 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 08:bd:43:c3:b9:67
Wed Jul  5 10:25:51 2017 daemon.info hostapd: wlan1: STA 08:bd:43:c3:b9:67 WPA: pairwise key handshake completed (RSN)
```

yuhhaurlin, 2017-07-05:
From the log, I can't find problem. The two stations are connected.

BuzzBumbleBee, 2017-07-05:
Yeah, its very confusing as the client logs show the wifi connection being made fine as well but no data is passing CLIENT <-> AP while in this broken state.

Will run at VHT40 for a while and see if there are any issues with that.

yuhhaurlin, 2017-07-05:
For all clients, you encounter the same problem? Does 84:c7:ea:cd:5d:2f get the IP assign from AP?

BuzzBumbleBee, 2017-07-05:
84:C7:EA:CD:5D:2F is a Sony Xperia XZ phone with BCM (AC) WiFi
When in this broken VHT80 state it will not get an IP via DHCP (or any traffic even if IP is assigned static).

08:bd:43:c3:b9:67 is a NETGEAR WNDR3700v4 (Non AC device)
When in this broken VHT80 state it will work 100% as expected (as its does not support VHT)

For what its worth, i'm running in VHT40 mode and it hasn't had the issue yet.

yuhhaurlin, 2017-07-05:
So maybe the problem is VHT 80 MHz with Sony Xperia XZ phone.
1. Open mode is working with this client for VHT 80 Mhz?
2. Stock firmware can work with this client for VHT 80 Mhz?
3. This device can work with other AP for VHT 80 Mhz?

BuzzBumbleBee, 2017-07-05:
@yuhhaurlin i have a laptop and 5 other phones that should be VHT80 capable but will not connect while its in this state.

Later this week ill test stock firmware (and try find another VHT80 AP)

yuhhaurlin, 2017-07-05:
Basically WRT series devices are released for a long period of time. I have not yet got report about VHT 80 Mhz problem. If only specific client has problem, I can try to check it. But if all 11 ac clients has problem, I really don't know what to do. Maybe you can check stock firmware first.

yuhhaurlin, 2017-07-05:
BTW, does these devices work with previous version of mwlwifi?

BuzzBumbleBee, 2017-07-05:
@yuhhaurlin yeah ill check stock later this week and get feedback. Thanks for all the help.

I was sure it used to work back in 2016 but cant be 100% sure of what configuration was used. After testing stock ill find an old OpenWRT build to try :D

yuhhaurlin, 2017-07-13:
I put LEDE WRT1900AC images with latest driver 10.3.0.4-20170713 for you to do test. I test our RD board + 88W8864 module, I did not find problem you described. In fact, we never encounter problem with 80 Mhz bandwidth.

factory: https://drive.google.com/open?id=0B3qLWtcWB9EdZE5UX0tsQ2d0MnM
upgrade: https://drive.google.com/open?id=0B3qLWtcWB9EdSjk5WXJXTDdOMUE

root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20170713
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x10
mac address: 00:50:43:21:bd:e9
2g: disable
5g: enable
antenna: 4 4
irq number: 44
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000007
radio: enable
iobase0: f0d80000
iobase1: f0f00000
tx limit: 768
rx limit: 64
qe trigger number: 22481

yuhhaurlin, 2017-07-14:
Please help to check this version of code.

yuhhaurlin, 2017-07-14:
I close this one. If you still encounter problem, please let me know.

closed 2017-07-14 by yuhhaurlin