kaloz/mwlwifi #184: PING + Packet Loss

m4niacjp, 2017-07-05:
I'm having huge issues with the wifi , currently using a wrt1900acs router.
I've tried flashing Lede 17.01.0 as well as lastest 17.01.2 and i have 2 basic problems 

1- Around 10% packet loss
1- When opening a lot of tabs using browser ping goes up >30ms 

Connection to the LAN Wired network has no issues at all.
What is happening here?

yuhhaurlin, 2017-07-05:
Can you specify your setting?

m4niacjp, 2017-07-06:
What kind of settings you are looking for?

At the moment i'm using Lede 17.01.0 just flashed without keeping old settings ex novo and i'm seeing this issue after maybe 1h from flashing the device.
Ping is almost always fine <3ms , problem arise only when multiple devices are connected to the Wifi (i'm using only the 5Ghz radio in AC mode a 40Mhz Width ac AC, WPA2 + Force CCMP AES) that ping will arise to >50ms.

I can see immediately if only using 1pc i just open like 10 tabs on chrome at the same time, perfect ping of 1-2 immediately goes to 40-90ms and the drops again down to 1-2ms after like 10 seconds. Problem is sometimes i'm streaming over plex on the wifi from Server to Chromecast and this will cause constant >100ms with packet loss also.

yuhhaurlin, 2017-07-06:
So the application is running all right?

yuhhaurlin, 2017-07-06:
If you stop the traffic, the ping will come back?

m4niacjp, 2017-07-06:
Application are running perfectly, only issues is that when the packet loss arrives the connection becomes very unstable.

yuhhaurlin, 2017-07-06:
What do you mean unstable?

m4niacjp, 2017-07-06:
Means i will have 10/15% packet loss and since i'm under VPN Shadowosocks it will get reset and cause internet malfunction.

yuhhaurlin, 2017-07-06:
Can you do following test?
Running iperf between LAN and STA client. Ping to AP from STA client at same time.

m4niacjp, 2017-07-06:
[image]

As you can see i get 17% PL in this case.

yuhhaurlin, 2017-07-06:
It looks like the traffic stopped for a period of time. It is very strange. I will test here.

yuhhaurlin, 2017-07-06:
Can you use "cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info" to let me know the version information?

Chadster766, 2017-07-06:
Severe packet loss like this would not be a mwlwifi issue since it's be extensively tested on may platforms that do not have this issue.

IMO this indicates an issue with any of the below points:

1. A bad LEDE\OpenWRT build
1. Old mwlwifi driver
1. mwlwifi driver mismatch with mwlwifi firmware
1. radio interference with another Router\AP

m4niacjp, 2017-07-06:
driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.2.0-20170110
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x90
mac address: 00:25:9c:14:34:da
2g: disable
5g: enable
antenna: 4 4
irq number: 104
iobase0: e0e80000
iobase1: e1000000
tx limit: 768
rx limit: 64
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
qe trigger number: 2559869

I've tried also setting different radio channels, still same problem.

I'm having the same issue if i connect to the radio0 or radio1 (wifi 2.4Ghz/5Ghz).

For the LEDE Build i've tried with the latest 17.01.02 and was having issues, so i reverted to the old one 17.01.00 seems better but sometimes still having issues.

yuhhaurlin, 2017-07-06:
@Chadster766 Yes. I think so. So I want to build image and let him try the image.
@m4niacjp The driver is previous one. But I think it should be all right. I will build LEDE image based on latest mwlwifi driver. I don't have WRT1900ACS, I will check on our RD board and let you test it.

m4niacjp, 2017-07-06:
@Chadster766  you can easly remove points 4/5 as i tested with VM and without VM on 3 different laptops i'm having at home.
1 DELL XPS 15 9550, 1 Lenovo Yoga and another Samsung

Chadster766, 2017-07-06:
Good troubleshooting :smiley:

yuhhaurlin, 2017-07-06:
Please use this image https://drive.google.com/open?id=0B3qLWtcWB9EdS1dKemgzM3RrMFU to do test. I just test it on our RD AP board with 88W8864 module. It works well. The driver is the latest one from this GitHub, I just change the driver version to 10.3.4.0-20170706.

root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20170706
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x10
mac address: 00:50:43:21:bd:e9
2g: disable
5g: enable
antenna: 4 4
irq number: 44
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: f0d80000
iobase1: f0f00000
tx limit: 768
rx limit: 64
qe trigger number: 7157

m4niacjp, 2017-07-06:
Should i do some kind of hard reset or something? Or just flash over?

yuhhaurlin, 2017-07-06:
Just flash the image from uboot.

m4niacjp, 2017-07-06:
I can't flash factory, must be sysupgrade

yuhhaurlin, 2017-07-06:
Here you are: https://drive.google.com/open?id=0B3qLWtcWB9EdRmh2cU1JcnB3UWM

m4niacjp, 2017-07-06:
I'm having some issues installing some packages due to the version that is SNAPSHOT, could you kindly build using the latest stable?

https://downloads.lede-project.org/releases/17.01.2/targets/mvebu/generic/

yuhhaurlin, 2017-07-06:
Can you make sure the problem is gone with the image I sent out? Running iperf and ping at same time.
I think I am not the right one to release LEDE image. You can build the code by yourself and follow the README to replace the mwlwifi with the latest version.

m4niacjp, 2017-07-06:
Still there unfortunately,..i think my router wifi is damaged perhaps :(

yuhhaurlin, 2017-07-06:
The problem is still the same?

yuhhaurlin, 2017-07-06:
Can you check the version and show me the test result?

m4niacjp, 2017-07-06:
driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20170706
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x90
mac address: 00:25:9c:14:34:da
2g: disable
5g: enable
antenna: 4 4
irq number: 44
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0e00000
iobase1: e0f80000
tx limit: 768
rx limit: 64
qe trigger number: 89907

For some reasons is still the same driver as the one i posted earlier?

yuhhaurlin, 2017-07-06:
The version is correct. How about the test?

m4niacjp, 2017-07-06:
For the moment is stable, just higher ping but no PL. I'll do some heavy testing later after lunch.
Problem is that i should build with latest snapshot/release of LEDE as on this i cannot install several packages.

yuhhaurlin, 2017-07-06:
Do you know how to build LEDE image? If yes, you can build it by yourself. The one I gave to you is got on 2017/5/12. The default version of mwlwifi is the one that you used (10.3.2.0-20170110). You can test the image first to see the problem you encountered is due to LEDE build or mwlwifi.

Later you can follow the README to replace mwlwifi with the latest code.

yuhhaurlin, 2017-07-06:
Or you can check LEDE community to ask them to build image with latest stable version of mwlwifi (10.3.4.0-20170606). Commits after this version are specific for WRT3200ACM.

closed 2017-07-06 by yuhhaurlin

m4niacjp, 2017-07-06:
Just built with latest version and Kernel 4.4.71 seems perfect now.