kaloz/mwlwifi #185: Cannot use DFS channels on WRT3200ACM with LEDE

s-pimenta, 2017-07-05:
On a fresh LEDE (davidc502: https://davidc502sis.dynamic-dns.net/releases/#3200acm), I simply cannot enable 5GHz on DFS channels... With non DFS channels 36-48 is fine...

On Syslog DFS-CAC-START simply fails...

On DD-WRT I don't have this issue with DFS, something is wrong...

**System log**
```
Wed Jul  5 12:19:15 2017 kern.debug kernel: [   22.611343] ieee80211 phy0: change: 0x40
Wed Jul  5 12:19:15 2017 daemon.notice hostapd: wlan0: interface state HT_SCAN->DFS
Wed Jul  5 12:19:15 2017 daemon.notice hostapd: wlan0: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=1, seg0=58, seg1=0, cac_time=60s
Wed Jul  5 12:19:15 2017 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Wed Jul  5 12:19:15 2017 daemon.err hostapd: Interface initialization failed
Wed Jul  5 12:19:15 2017 daemon.notice hostapd: wlan0: interface state DFS->DISABLED
Wed Jul  5 12:19:15 2017 daemon.notice hostapd: wlan0: AP-DISABLED 
Wed Jul  5 12:19:15 2017 daemon.notice hostapd: wlan0: interface state DISABLED->DISABLED
Wed Jul  5 12:19:15 2017 daemon.notice hostapd: wlan0: AP-DISABLED 
Wed Jul  5 12:19:15 2017 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING 
Wed Jul  5 12:19:15 2017 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Wed Jul  5 12:19:15 2017 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Wed Jul  5 12:19:15 2017 kern.debug kernel: [   22.821313] ieee80211 phy0: change: 0x40
Wed Jul  5 12:19:15 2017 kern.info kernel: [   22.861342] device wlan0 left promiscuous mode
Wed Jul  5 12:19:15 2017 kern.info kernel: [   22.865853] br-lan: port 2(wlan0) entered disabled state
Wed Jul  5 12:19:15 2017 kern.debug kernel: [   22.917323] ieee80211 phy0: change: 0x100
Wed Jul  5 12:19:15 2017 daemon.notice hostapd: ELOOP: remaining socket: sock=22 eloop_data=0xa5e80 user_data=0 handler=0x3545c
Wed Jul  5 12:19:15 2017 daemon.notice netifd: radio0 (1666): cat: can't open '/var/run/wifi-phy0.pid': No such file or directory
Wed Jul  5 12:19:15 2017 daemon.notice netifd: radio0 (1666): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path ()
Wed Jul  5 12:19:15 2017 daemon.notice netifd: radio0 (1666): Command failed: Invalid argument
Wed Jul  5 12:19:15 2017 kern.debug kernel: [   23.093338] ieee80211 phy0: change: 0xffffffff
```

cilix-lab, 2017-07-05:
See what "iw reg get" shows. Maybe it's the same thing discussed in https://github.com/kaloz/mwlwifi/issues/173.

s-pimenta, 2017-07-05:
@cilix-lab Thanks!!! Finaly I've got the DFS channels working on LEDE!

Just for documentation:

Connect via ssh to Linksys WRT3200ACM and then execute the following commands:

```
opkg remove kmod-mwifiex-sdio
opkg remove mwifiex-sdio-firmware
reboot
```

Problem solved!

PS: This should be on the WIKI!

BrainSlayer, 2017-07-05:
@s-pimenta but is not real solution. it just disabled the third sd8887 based wifi interface. the real solution is to adjust the regulatory domain to the correct country subset or to fix the crda.txt database to match the channels supported by the chipset in the same country. i dont think that sd8887 belongs to that problem. at least thats the difference in dd-wrt here since dd-wrt does allow to use the sd8887 chipset

cilix-lab, 2017-07-05:
I'm not really sure what the problem is in LEDE, since I'm running Ubuntu, but I could fix it compiling the kernel with a static regulatory database (db.txt from wireless-regdb) and setting the correct regulatory domain in /etc/default/crda. 
After that, everything works including the third radio.
So, apparently it isn't a mwlwifi issue, but it might be a LEDE one?

yuhhaurlin, 2017-07-05:
1. Remove phy2.
2. Don't change country code.

closed 2017-07-05 by yuhhaurlin

yuhhaurlin, 2017-07-06:
I also add note for DFS of WRT3200ACM on README.

Emilv2, 2018-01-31:
@yuhhaurlin I changed the country code because I didn't know I wasn't supposed to. Now the only way to get DFS channelf working seems to be remove phy2.
I tried to reset from luci and remove country options from /etc/config/wireless, but I still can't get DFS channels working when kmod-mwifiex-sdio and mwifiex-sdio-firmware are installed.

How can I undo the country code change? (I think this could be a useful addition in the README)

yuhhaurlin, 2018-01-31:
If you remove the setting from /etc/config/wireless, it should work. You can use "iw reg get" to check it.

Emilv2, 2018-02-03:
Doesn't work :(
If I remove phys2 the country settings are FR for all devices and I can use DFS channels, but if I install kmod-mwifiex-sdio and mwifiex-sdio-firmware again phy2's country code is US (no matter what I put in /etc/config/wireless), the global country code is 98: unset and DFS channels don't work.

yuhhaurlin, 2018-02-03:
Mwlwifi will set country code for WRT3200ACM. If you did not change country code, you won't have this problem. BTW, you must use updated driver with firmware 9.3.2.x.

Emilv2, 2018-02-03:
Yes I know, but when I do not set any country code I have this problem too. phy2's country code is US, phy0 and phy1's are FR, global is 98: unset and DFS doesn't work.

Where do I get this updated driver/firmware? Now I run LEDE Reboot 17.01.4 r3560-79f57e422d and install packages from the default repo.

yuhhaurlin, 2018-02-05:
1. Check region code of 8964 module can be got by mwlwifi. You can also check version here. 
root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info | grep region
firmware region code: 0x10
2. Remove option country from /etc/config/wireless.

Emilv2, 2018-02-06:
This is my config file (the country code option for 'radio0' seems to have been added automatically, I removed it but it came back. If I remove it again it doesn't change anything):

```shell
root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'soc/soc:pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0'
        option htmode 'VHT80'
        option txpower '20'
        option country 'FR'
        option channel '60'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option encryption 'psk2+ccmp'
        option key 'mysecretpassword'
        option ssid 'Wifi_5.0'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
        option htmode 'HT20'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option encryption 'psk2+ccmp'
        option key 'mysecretpassword''
        option ssid 'Wifi_2.4'

config wifi-device 'radio2'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path 'platform/soc/soc:internal-regs/f10d8000.sdhci/mmc_host/mmc0/mmc0:0001/mmc0:0001:1'
        option htmode 'VHT80'

config wifi-iface 'default_radio2'
        option device 'radio2'
        option network 'lan'
        option mode 'ap'
        option ssid 'LEDE'
        option encryption 'none'
```
and this is what 'cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info' and 'iw reg get' give:

```shell
root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info
driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170606
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:c7:6a:0a
2g: disable
5g: enable
antenna: 4 4
irq number: 105
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: disable
iobase0: e0e00000
iobase1: e1080000
tx limit: 1024
rx limit: 16384
```

```shell
root@LEDE:~# iw reg get
global
country 98: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5490 - 5710 @ 160), (N/A, 23), (0 ms), DFS
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

phy#2
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)

phy#1
country FR: DFS-ETSI
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country FR: DFS-ETSI
        (2402 - 2482 @ 40), (N/A, 20), (N/A)
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW
        (5490 - 5710 @ 160), (N/A, 27), (0 ms), DFS
        (57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

I see I don't seem to have the latest firm ware you mentioned, do I have to install unstable LEDE/OpenWRT for this? I'd rather stay on a stable branch if possible.

yuhhaurlin, 2018-02-06:
"FR" is set by mwlwifi due to region code is 0x30. I don't know why phy2 will set country code to "US". I think you can remove phy2.

Emilv2, 2018-02-06:
Yes when I remove phys2 I can use DFS channels, but I would like to use phys2 :(

yuhhaurlin, 2018-02-07:
Maybe you can set country "FR" for phy2.

Emilv2, 2018-02-17:
Nope, as I said before setting country codes has no effect whatsoever

supersebbo, 2018-02-21:
The issue appears to stem from the fact that all the radios on this device (0, 1, 2) set their region automatically and will completely IGNORE the setting from wireless config.  However there seems to be a bug (by feature or design) that that the 3rd radio (phy2) always sets itself to US region.  Obviously if you are in the US, this is not an issue as all your regions will match, but if you are outside of the US then having the 3rd radio enabled creates a region mis-match and disables DFS. 

In the marketing material for the 3200ACM it is mentioned that the router uses a separate radio for DFS sensing, noting that this means that radar sensing can be done without interrupting the primary Wifi signal.  I suspect this is the purpose of the 3rd radio, hence it was only ever intended to be a receiver and not to transmit, so hard-coding US region would not be a problem.