kaloz/mwlwifi #186: 10.3.4.0-20170606 driver problem

ValCher1961, 2017-07-05:
When testing iperf with the last driver on wrt3200, I'm getting this.
```
[  4]  15.00-16.00  sec  39.3 MBytes   330 Mbits/sec    0    428 KBytes
[  4]  16.00-17.00  sec  38.5 MBytes   323 Mbits/sec    0    428 KBytes
[  109.213069] ------------[ cut here ]------------
[  109.217755] WARNING: CPU: 0 PID: 0 at net/mac80211/rx.c:4326 ieee80211_rx_napi+0x5ca/0x690 [mac80211]
[  109.227030] Modules linked in: xt_nat ipt_MASQUERADE nf_nat_masquerade_ipv4 ipt_REJECT nf_reject_ipv4 xt_TCPMSS xt_tcpudp xt_conntrack iptable_mangle iptable_filter nf_nat_ftp nf_conntrack_ftp nf_nat_proto_gre ip_gre gre iptable_nat ip_tables nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack libcrc32c x_tables bridge stp llc mwlwifi mac80211 cfg80211
[  109.259545] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.12.0 #3
[  109.265487] Hardware name: Marvell Armada 380/385 (Device Tree)
[  109.271441] [<c010c2e9>] (unwind_backtrace) from [<c0108ddb>] (show_stack+0xb/0xc)
[  109.279045] [<c0108ddb>] (show_stack) from [<c02d7691>] (dump_stack+0x71/0x80)
[  109.286300] [<c02d7691>] (dump_stack) from [<c011991d>] (__warn+0xad/0xc0)
[  109.293203] [<c011991d>] (__warn) from [<c011999b>] (warn_slowpath_null+0x13/0x18)
[  109.300832] [<c011999b>] (warn_slowpath_null) from [<bf899cdb>] (ieee80211_rx_napi+0x5ca/0x690 [mac80211])
[  109.310559] [<bf899cdb>] (ieee80211_rx_napi [mac80211]) from [<bf94f939>] (pcie_rx_recv_ndp+0x74c/0x81c [mwlwifi])
[  109.320958] [<bf94f939>] (pcie_rx_recv_ndp [mwlwifi]) from [<c011ce0b>] (tasklet_action+0x53/0xac)
[  109.329955] [<c011ce0b>] (tasklet_action) from [<c011cf43>] (__do_softirq+0x9f/0x198)
[  109.337818] [<c011cf43>] (__do_softirq) from [<c011d223>] (irq_exit+0x6f/0x94)
[  109.345071] [<c011d223>] (irq_exit) from [<c01493db>] (__handle_domain_irq+0x43/0x7c)
[  109.352934] [<c01493db>] (__handle_domain_irq) from [<c0101355>] (gic_handle_irq+0x39/0x6c)
[  109.361319] [<c0101355>] (gic_handle_irq) from [<c0109625>] (__irq_svc+0x65/0x94)
[  109.368832] Exception stack(0xc0a01f38 to 0xc0a01f80)
[  109.373902] 1f20:                                                       00000001 00000000
[  109.382113] 1f40: c0a01f90 c0113141 c0a00000 c0a03d04 c0a03cb4 c09497a8 c0a01f90 00000000
[  109.390324] 1f60: 00000000 c092f8c0 c0533cf5 c0a01f88 c01068d5 c01068d6 40070033 ffffffff
[  109.398535] [<c0109625>] (__irq_svc) from [<c01068d6>] (arch_cpu_idle+0x22/0x24)
[  109.405963] [<c01068d6>] (arch_cpu_idle) from [<c0141643>] (do_idle+0xf3/0x158)
[  109.413303] [<c0141643>] (do_idle) from [<c014186f>] (cpu_startup_entry+0x13/0x14)
[  109.420906] [<c014186f>] (cpu_startup_entry) from [<c0900a07>] (start_kernel+0x2d3/0x2dc)
[  4]  17.00-18.[  109.429135] ---[ end trace fc7fb1121835fba3 ]---
00  sec  35.6 MBytes   299 Mbits/sec    0    428 KBytes
[  4]  18.00-19.00  sec  32.7 MBytes   275 Mbits/sec    0    428 KBytes
[  4]  19.00-20.00  sec  38.3 MBytes   321 Mbits/sec    0    428 KBytes
[  4]  20.00-21.00  sec  38.7 MBytes   325 Mbits/sec    0    428 KBytes
```
The driver installed several times on KERNRL 4.11.8 and 4.12.

yuhhaurlin, 2017-07-05:
This is warning message. I will check it.

yuhhaurlin, 2017-07-06:
If the system can still work, I will check this warning later. Thanks.

ValCher1961, 2017-07-06:
For information.
This was a single warning, and when the test continues, the warnings come out more frequently and eventually merge into a continuous stream, the communication goes down, and at the end the router stops responding to commands from the serial Port console.
Tonight I'm going to try to collect kernel-4.9 x with this driver. Maybe it's incompatible with the new cores.
If you need any additional information or a test, I'm ready to provide it.
Thank you.

yuhhaurlin, 2017-07-06:
Can you give me version of mac80211 compatible module?

yuhhaurlin, 2017-07-06:
BTW, can you help to test following LEDE image to see if you will encounter this problem or not?
factory: https://drive.google.com/open?id=0B3qLWtcWB9EdTVQ1V1ExQ1JFSms
sysupgrade: https://drive.google.com/open?id=0B3qLWtcWB9EdT1NGSDlDWGRvUjg

yuhhaurlin, 2017-07-06:
The version of mac80211 compatible module of above image is 2017-01-31. If you can reproduce this problem on this image, I can check this warning message based on this version of mac80211. Otherwise, I need to know which version of mac80211 compatible module used by you to let this problem happen in order to check it.

yuhhaurlin, 2017-07-06:
I think you use updated mac80211 compatible module. File net/mac80211/rx.c of 2017-01-31 only has 4277 lines.

yuhhaurlin, 2017-07-06:
Or you use old version of mac80211 compatible module. Updated driver has problem to work with old mac80211 due to API changes.

ValCher1961, 2017-07-06:
I'm not using Lede, I'm using Mcdebian.
/net/mac80211/rc.c - https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/net/mac80211/rx.c?h=v4.12

yuhhaurlin, 2017-07-06:
Sorry. I can't check it due to I don't have updated mac80211 package.

yuhhaurlin, 2017-07-07:
The warning is related reporting wrong rate index. I will fix it. But I think this warning won't affect normal operation.

ValCher1961, 2017-07-07:
All right, thank you.

starcms, 2017-07-08:
@yuhhaurlin , as @farchord reported in #189, it can cause the router to crash and die; it's not simply just appearing in the log.  Thanks for your help!

yuhhaurlin, 2017-07-09:
I will check it ASAP.

BrainSlayer, 2017-07-09:
someone missed to add the following line. i saw this message as well now

[   85.498334] ------------[ cut here ]------------
[   85.503010] WARNING: CPU: 0 PID: 0 at /home/seg/DEV/mvebu/src/router/private/compat-wireless-2017-01-31/net/mac80211/rx.c:4221 _427+0x16c/0x94c [mac80211]
[   85.516894] Rate marked as a VHT rate but data is invalid: MCS: 6, NSS: 0

yuhhaurlin, 2017-07-10:
Please help to check https://github.com/kaloz/mwlwifi/commit/8a717138b98e38d44e71427ef9c07eabcd5d9865.

BrainSlayer, 2017-07-10:
will do asap

ValCher1961, 2017-07-10:
Unable to connect, I think clients do not receive DHCP addresses.
I'll try again tonight and let you know.

yuhhaurlin, 2017-07-10:
@ValCher1961 The latest code only changes the BSSID address assignment and bypass incorrect rx account report. I don't think original functions will be affected.

yuhhaurlin, 2017-07-10:
@BrainSlayer Thanks. I also change BSSID assignment for VAPs. The code should work even though the interface dose not allocate 8 real MAC addresses.

BrainSlayer, 2017-07-10:
no issues on my side. the new code is a little bit curious. i saw you made the local mask change for all vap's,  curious is the decreasing mac for phy0 and increasing mac for phy1 you wrote. this isnt really neccessary, but also no problem. identical mac addresses within the same device arent a problem

yuhhaurlin, 2017-07-10:
Yes. If you create more than 8 VAPs, duplicated addresses will happen.

yuhhaurlin, 2017-07-10:
Please let me know if anyone still encounters this warning message. I will wait for two more days to close this one if no problem is reported.

BrainSlayer, 2017-07-10:
my testdevice is running with it constantly. i will watch it. just as consideration. the signal report is also 0 very often (every 4 - 6 seconds of polling zero is returned (rx/tx rate looks normal))

yuhhaurlin, 2017-07-10:
It is very strange, I did not find this kind of problem. Maybe we can wait for more test results.

ValCher1961, 2017-07-10:
Everything happened again. 
`[   86.465533] WARNING: CPU: 0 PID: 0 at net/mac80211/rx.c:4301 ieee80211_rx_napi+0x686/0x688 [mac80211]`

ValCher1961, 2017-07-10:
It started with the commits # [1CC8AA6](https://github.com/kaloz/mwlwifi/commit/1cc8aa697f9d853a51d7cc60775635051b7a9302)

yuhhaurlin, 2017-07-10:
This time is 4301, I will check it. BTW, the system will be crashed later, right?

yuhhaurlin, 2017-07-10:
I will add more checks.

ValCher1961, 2017-07-10:
This time I used kernel-4.9.34, the system didn't die, but WiFi stopped joining.

ValCher1961, 2017-07-10:
The correct thing to say is that it started after the commits from 03.07.2017, because there were two of them.

yuhhaurlin, 2017-07-11:
Please help to check https://github.com/kaloz/mwlwifi/commit/53e10cfcaf8e517230b03b1bfa26a8b2dcb0c321. Thanks.

ValCher1961, 2017-07-11:
Thanks, I'll test it tonight.
Which version of kernel do you have to test the driver for?

yuhhaurlin, 2017-07-11:
Linux LEDE 4.9.20 #0 SMP Thu May 11 20:43:01 2017 armv7l GNU/Linux.

I just connect a client and ping to AP to see if I will encounter this problem. If you still encounter this problem, please let me know the way to reproduce it.
Or I can give LEDE image to you to do verification. If this problem happens on it, I can add debug messages to check it.

ValCher1961, 2017-07-11:
@yuhhaurlin Well, tonight, I'll install kernel-4.9.20 with a new driver and test it.
@All who had this problem, please report your results to the new driver test.

starcms, 2017-07-11:
@davidc502 just posted the latest driver in his thread for those using his build (LEDE snapshot r4542, so it includes kernel 4.9.34 and mac80211 module version 2017-01-31). A few people had experienced the issue originally, including @davidc502 himself. If any still experience it and post in his thread but don't report it here, I'll be sure to forward it.

ValCher1961, 2017-07-11:
As I wrote earlier I did not use lede, but I am using Mcdebian. (Is distribjutiv Debian).
To reproduce the problem, I use the IPERF3 test between the access point and the WiFi client.

BrainSlayer, 2017-07-11:
@yuhhaurlin : no negative side effects so far, but signal report still returns 0 in a very high frequency

BrainSlayer, 2017-07-11:
but not at all. so sometimes it returns 0 for a while. sometimes not

yuhhaurlin, 2017-07-11:
@BrainSlayer I run the test on 53e10cf now. I did not encounter the problem you described.

yuhhaurlin, 2017-07-11:
[image]

[image]

yuhhaurlin, 2017-07-11:
Signal and noise level are jumped within reasonable range and never become zero.

yuhhaurlin, 2017-07-11:
I change the version to 10.3.4.0-20170711.

BrainSlayer, 2017-07-11:
maybe zero values are filtered in lede. i show it plain as reported by mac80211 / netlink

BrainSlayer, 2017-07-11:
but mine is running in station mode. yours is running in ap mode. thats a big difference

yuhhaurlin, 2017-07-11:
I will check sta mode when I work for WDS.

BrainSlayer, 2017-07-11:
okay. perfect. wds is a long expected feature by users :-)

yuhhaurlin, 2017-07-11:
Yes. It is the next feature that I plan to implement. But it is not a trivial job due to special data path of 88W8964 firmware.

starcms, 2017-07-11:
@yuhhaurlin, will MU-MIMO ever be supported?  Thanks for all the hard work!

yuhhaurlin, 2017-07-11:
Yes. After WDS.

starcms, 2017-07-11:
excellent, thank you :)

ValCher1961, 2017-07-11:
It's sad, I don't know what to do.
Look at the bouquet.
```
[  4]  30.00-31.00  sec  39.6 MBytes   333 Mbits/sec    0    214 KBytes
[  133.999123] ------------[ cut here ]------------
[  134.003814] WARNING: CPU: 0 PID: 0 at net/mac80211/rx.c:4242 ieee80211_rx_napi+0x640/0x660 [mac80211]
[  134.013090] Rate marked as a VHT rate but data is invalid: MCS: 0, NSS: 0
[  134.019913] Modules linked in: xt_nat ipt_MASQUERADE nf_nat_masquerade_ipv4 ipt_REJECT nf_reject_ipv4 ipt_rpfiltebles nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack x_tables bridge stp llc mwlwifi mwifiex_sdio m
[  134.053408] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.9.20 #1
[  134.059350] Hardware name: Marvell Armada 380/385 (Device Tree)
[  134.065303] [<c010c2f9>] (unwind_backtrace) from [<c0108dc3>] (show_stack+0xb/0xc)
[  134.072907] [<c0108dc3>] (show_stack) from [<c02ca52d>] (dump_stack+0x71/0x80)
[  134.080160] [<c02ca52d>] (dump_stack) from [<c0119469>] (__warn+0xad/0xc0)
[  134.087064] [<c0119469>] (__warn) from [<c011949d>] (warn_slowpath_fmt+0x21/0x2c)
[  134.094602] [<c011949d>] (warn_slowpath_fmt) from [<bf908ad1>] (ieee80211_rx_napi+0x640/0x660 [mac80211])
[  134.104241] [<bf908ad1>] (ieee80211_rx_napi [mac80211]) from [<bf9c2a55>] (pcie_rx_recv_ndp+0x5a4/0x840 [mwlwifi]
[  134.114640] [<bf9c2a55>] (pcie_rx_recv_ndp [mwlwifi]) from [<c011ca6f>] (tasklet_action+0x53/0xac)
[  134.123637] [<c011ca6f>] (tasklet_action) from [<c011cba7>] (__do_softirq+0x9f/0x194)
[  134.131500] [<c011cba7>] (__do_softirq) from [<c011ce83>] (irq_exit+0x6f/0x94)
[  134.138752] [<c011ce83>] (irq_exit) from [<c014bb63>] (__handle_domain_irq+0x43/0x7c)
[  134.146615] [<c014bb63>] (__handle_domain_irq) from [<c0101355>] (gic_handle_irq+0x39/0x6c)
[  134.155000] [<c0101355>] (gic_handle_irq) from [<c0109625>] (__irq_svc+0x65/0x94)
[  134.162511] Exception stack(0xc0a01f48 to 0xc0a01f90)
[  134.167582] 1f40:                   00000001 00000000 c0a01fa0 c0112e61 c0a00000 c0a03088
[  134.175793] 1f60: c0a03034 00000001 00000000 00000000 c0943488 c0a03090 c0a01f78 c0a01f98
[  134.184002] 1f80: c01068d5 c01068d6 400f0033 ffffffff
[  134.189074] [<c0109625>] (__irq_svc) from [<c01068d6>] (arch_cpu_idle+0x22/0x24)
[  134.196501] [<c01068d6>] (arch_cpu_idle) from [<c0145125>] (cpu_startup_entry+0x115/0x170)
[  134.204801] [<c0145125>] (cpu_startup_entry) from [<c0900a03>] (start_kernel+0x2d3/0x2dc)
[  134.213026] ---[ end trace 8fba8380b86be708 ]---
[  4]  31.00-32.13  sec  36.6 MBytes   271 Mbits/sec    0    214 KBytes
[  134.971093] ------------[ cut here ]------------
[  134.975785] WARNING: CPU: 0 PID: 0 at net/mac80211/rx.c:4245 ieee80211_rx_napi+0x65c/0x660 [mac80211]
[  134.985056] Modules linked in: xt_nat ipt_MASQUERADE nf_nat_masquerade_ipv4 ipt_REJECT nf_reject_ipv4 ipt_rpfiltebles nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack x_tables bridge stp llc mwlwifi mwifiex_sdio m
[  135.018554] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W       4.9.20 #1
[  135.025718] Hardware name: Marvell Armada 380/385 (Device Tree)
[  135.031671] [<c010c2f9>] (unwind_backtrace) from [<c0108dc3>] (show_stack+0xb/0xc)
[  135.039275] [<c0108dc3>] (show_stack) from [<c02ca52d>] (dump_stack+0x71/0x80)
[  135.046528] [<c02ca52d>] (dump_stack) from [<c0119469>] (__warn+0xad/0xc0)
[  135.053432] [<c0119469>] (__warn) from [<c01194e7>] (warn_slowpath_null+0x13/0x18)
[  135.061059] [<c01194e7>] (warn_slowpath_null) from [<bf908aed>] (ieee80211_rx_napi+0x65c/0x660 [mac80211])
[  135.070786] [<bf908aed>] (ieee80211_rx_napi [mac80211]) from [<bf9c2a55>] (pcie_rx_recv_ndp+0x5a4/0x840 [mwlwifi]
[  135.081184] [<bf9c2a55>] (pcie_rx_recv_ndp [mwlwifi]) from [<c011ca6f>] (tasklet_action+0x53/0xac)
[  135.090182] [<c011ca6f>] (tasklet_action) from [<c011cba7>] (__do_softirq+0x9f/0x194)
[  135.098044] [<c011cba7>] (__do_softirq) from [<c011ce83>] (irq_exit+0x6f/0x94)
[  135.105296] [<c011ce83>] (irq_exit) from [<c014bb63>] (__handle_domain_irq+0x43/0x7c)
[  135.113159] [<c014bb63>] (__handle_domain_irq) from [<c0101355>] (gic_handle_irq+0x39/0x6c)
[  135.121544] [<c0101355>] (gic_handle_irq) from [<c0109625>] (__irq_svc+0x65/0x94)
[  135.129054] Exception stack(0xc0a01f48 to 0xc0a01f90)
[  135.134125] 1f40:                   00000001 00000000 c0a01fa0 c0112e61 c0a00000 c0a03088
[  135.142336] 1f60: c0a03034 00000001 00000000 00000000 c0943488 c0a03090 c0a01f78 c0a01f98
[  135.150546] 1f80: c01068d5 c01068d6 40070033 ffffffff
[  135.155618] [<c0109625>] (__irq_svc) from [<c01068d6>] (arch_cpu_idle+0x22/0x24)
[  135.163045] [<c01068d6>] (arch_cpu_idle) from [<c0145125>] (cpu_startup_entry+0x115/0x170)
[  135.171344] [<c0145125>] (cpu_startup_entry) from [<c0900a03>] (start_kernel+0x2d3/0x2dc)
[  4]  32.13-33.[  135.179573] ---[ end trace 8fba8380b86be709 ]---
00  sec  29.7 MBytes   287 Mbits/sec    0    214 KBytes
[  4]  33.00-34.00  sec  36.0 MBytes   302 Mbits/sec    0    214 KBytes
```
Maybe with my system something.
Like the others?

gufus, 2017-07-11:
------

Great news, thank you :)

ValCher1961, 2017-07-11:
Thank you, that's nice.

yuhhaurlin, 2017-07-11:
I will send LEDE image for you to do test.

ValCher1961, 2017-07-11:
Ok.

yuhhaurlin, 2017-07-11:
I will also do the test here. You only run iperf TCP  test from client to AP, right?

starcms, 2017-07-11:
@yuhhaurlin, you can use the -R command to easily make iperf3 go in the other direction, just to be sure.

yuhhaurlin, 2017-07-11:
Thanks. I know.

yuhhaurlin, 2017-07-12:
@ValCher1961 
Factory: https://drive.google.com/open?id=0B3qLWtcWB9EdZjRhb1R3a3ZGOTg
Upgrade: https://drive.google.com/open?id=0B3qLWtcWB9EdcEV3blBELWRNQ1k

root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170711
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 00:50:43:22:88:8a
2g: disable
5g: enable
antenna: 4 4
irq number: 44
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000000
radio: disable
iobase0: f0e00000
iobase1: f1080000
tx limit: 1024
rx limit: 16384

[8 more replies]

closed 2017-07-13 by yuhhaurlin