kaloz/mwlwifi #187: Broken tx rate a signal wrt3200acm

Ansuel, 2017-07-07:
With the last commit i have this problem... every device have only 6mbps or 1 mbps from luci interface. If i check the actual speed from the device, i gets the correct one. Where is the problem?

yuhhaurlin, 2017-07-07:
I don't think so. I test the code via LEDE image. WEB GUI can show tx rate without problem. You need to run traffic between AP and Client.

yuhhaurlin, 2017-07-07:
BTW, if you can get tx/rx rates via "iw wlan0/wlan1 station dump", it means the driver is all right.

yuhhaurlin, 2017-07-07:
[image]

closed 2017-07-07 by yuhhaurlin

Ansuel, 2017-07-07:
well.... if i try 
iw wlan0 station dump nothing appear
with iw wlan1 station dump
Station 84:ba:3b:07:53:07 (on wlan1)
        inactive time:  4294825206 ms
        rx bytes:       169993
        rx packets:     463
        tx bytes:       161798
        tx packets:     418
        tx retries:     0
        tx failed:      0
        rx drop misc:   0
        signal:         0 dBm
        signal avg:     0 dBm
        tx bitrate:     1.0 MBit/s
        rx bitrate:     1.0 MBit/s
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short preamble: yes
        short slot time:yes
        connected time: 132 seconds

yuhhaurlin, 2017-07-07:
[image]

Ansuel, 2017-07-07:
Will check my device sorry.

yuhhaurlin, 2017-07-07:
inactive time: 4294825206 ms => Please ping AP.

Ansuel, 2017-07-07:
Ok my problem... now it works