kaloz/mwlwifi #188: tx/rx rate + vht flag inconstistency

BrainSlayer, 2017-07-07:
following setup
wrt3200acm is in sta mode (latest driver from today)
netgear r7800 is ap running vht160 mode (qca9984)

the wrt3200acm estabilishes in vht160 mode as shown by assoc table. 
but ap side does not show the vht160 flag is set. the rx / tx rate on ap side differs also from the rx/tx rate on sta side which may be related that vht160 flag is not seen on ap side. the rate itself also look more like VHT80 than VHT160 (depending on the chains used)
i know that the QCA9984 chipset is 4x4, but runs vht160 only in 2x2. so this can be also another problem cause

AP  R7800
TX / RX
1560M | 1170M

STA WRT3200ACM
TX / RX
1170M | 1170M

sometimes RX rate on sta side is also 130M

from, my own feeling the rx / tx reporting on sta side does not seem to be correct at all and sometimes bogus

BrainSlayer, 2017-07-07:
in addition. signal is also 0 db often if i poll the report

yuhhaurlin, 2017-07-07:
I will check it later.

yuhhaurlin, 2017-07-07:
Only happened as STA mode and connected as 160 MHz?

yuhhaurlin, 2017-07-07:
I test here with 80 MHz, everything is working. I think your problem is the same as issue #172. I will check it later.

BrainSlayer, 2017-07-07:
i tested only sta mode with vht160 so far. i used fixed installed vht160 ap's in my testbed. i dont have any other vht160 client right to to test it vice versa. channel was 100 + 114

yuhhaurlin, 2017-07-07:
Thanks for your information.

yuhhaurlin, 2017-07-10:
I close this one and use #172 to check 160 MHz problem with r7800.

closed 2017-07-10 by yuhhaurlin