kaloz/mwlwifi #189: Chain Crash: Two days in a row now

farchord, 2017-07-07:
Sorry if this is a known issue, but I thought I'd post this here, those crashes have been happening twice this week now (Last was two days ago).

I have a 3200ACM on the LEDE trunk (I believe -- I use Davidc502's build)

This is the crash from two days ago:

`Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.489125] ------------[ cut here ]------------
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.493785] WARNING: CPU: 0 PID: 19536 at compat-wireless-2017-01-31/net/mac80211/rx.c:4224 ieee80211_rx_napi+0x88/0x81c [mac80211]
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.505689] Modules linked in: pppoe ppp_async pppox ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddrThu Jul  6 08:41:19 2017 kern.warn kernel: [85560.578129]  nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda mwifiex_sdio mwifiex iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables crc_ccitt fuse sch_cake em_nbyte act_ipt cls_basic sch_prio sch_pie sch_gred em_meta sch_dsmark sch_teql em_cmp act_police em_text sch_codel sch_sfq sch_fq sch_red act_connmark nf_conntrack act_skbedit act_mirredThu Jul  6 08:41:19 2017 kern.warn kernel: [85560.682698] CPU: 0 PID: 19536 Comm: kworker/u4:2 Tainted: G        W       4.9.34 #0
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.690470] Hardware name: Marvell Armada 380/385 (Device Tree)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.696421] Workqueue: phy0 mwl_account_handle [mwlwifi]
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.701768] [<c0016010>] (unwind_backtrace) from [<c0012220>] (show_stack+0x10/0x14)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.709544] [<c0012220>] (show_stack) from [<c020fb9c>] (dump_stack+0x7c/0x9c)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.716796] [<c020fb9c>] (dump_stack) from [<c0029214>] (__warn+0xbc/0xec)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.723699] [<c0029214>] (__warn) from [<c00292e8>] (warn_slowpath_null+0x1c/0x24)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.731320] [<c00292e8>] (warn_slowpath_null) from [<bf1ae250>] (ieee80211_rx_napi+0x88/0x81c [mac80211])
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.740957] [<bf1ae250>] (ieee80211_rx_napi [mac80211]) from [<bf20e454>] (pcie_rx_recv_ndp+0x92c/0xa58 [mwlwifi])
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.751357] [<bf20e454>] (pcie_rx_recv_ndp [mwlwifi]) from [<c002daf4>] (tasklet_action+0x8c/0xe8)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.760354] [<c002daf4>] (tasklet_action) from [<c002d28c>] (__do_softirq+0xd0/0x204)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.768217] [<c002d28c>] (__do_softirq) from [<c002d644>] (irq_exit+0x94/0xb8)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.775471] [<c002d644>] (irq_exit) from [<c0061d18>] (__handle_domain_irq+0x90/0xb4)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.783334] [<c0061d18>] (__handle_domain_irq) from [<c0009428>] (gic_handle_irq+0x50/0x94)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.791719] [<c0009428>] (gic_handle_irq) from [<c0012c8c>] (__irq_svc+0x6c/0x90)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.799230] Exception stack(0xd92c3e90 to 0xd92c3ed8)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.804301] 3e80:                                     d4419f94 fffffff0 00000000 00000000
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.812513] 3ea0: dd800988 dd800a7c d4419e6c 00800806 ddfa15c0 ddfa4700 0001ae95 d4419f74
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.820723] 3ec0: 00000000 d92c3ee4 00000000 c020f068 00000013 ffffffff
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.827364] [<c0012c8c>] (__irq_svc) from [<c020f068>] (__memzero+0x48/0x7c)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.834449] [<c020f068>] (__memzero) from [<bf206430>] (pcie_process_account+0x19c/0x380 [mwlwifi])
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.843540] [<bf206430>] (pcie_process_account [mwlwifi]) from [<c003d5a8>] (process_one_work+0x1d4/0x310)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.853233] [<c003d5a8>] (process_one_work) from [<c003e290>] (worker_thread+0x2d8/0x410)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.861445] [<c003e290>] (worker_thread) from [<c00421d0>] (kthread+0xd8/0xec)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.868698] [<c00421d0>] (kthread) from [<c000edf8>] (ret_from_fork+0x14/0x3c)
Thu Jul  6 08:41:19 2017 kern.warn kernel: [85560.875982] ---[ end trace 0b872fd2e1ddb361 ]---`

This is from today (I copied a couple blocks, more explanations below):

`Fri Jul  7 07:29:14 2017 kern.warn kernel: [81118.851032] ------------[ cut here ]------------
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81118.855695] WARNING: CPU: 0 PID: 6688 at compat-wireless-2017-01-31/net/mac80211/rx.c:4224 ieee80211_rx_napi+0x88/0x81c [mac80211]
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81118.867491] Modules linked in: pppoe ppp_async pppox ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddrFri Jul  7 07:29:14 2017 kern.warn kernel: [81118.939938]  nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables crc_ccitt fuse sch_cake em_nbyte act_ipt cls_basic sch_prio sch_pie sch_gred em_meta sch_dsmark sch_teql em_cmp act_police em_text sch_codel sch_sfq sch_fq sch_red act_connmark nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tFri Jul  7 07:29:14 2017 kern.warn kernel: [81119.042655] CPU: 0 PID: 6688 Comm: kworker/u4:1 Tainted: G        W       4.9.34 #0
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.050341] Hardware name: Marvell Armada 380/385 (Device Tree)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.056293] Workqueue: phy0 mwl_account_handle [mwlwifi]
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.061640] [<c0016010>] (unwind_backtrace) from [<c0012220>] (show_stack+0x10/0x14)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.069416] [<c0012220>] (show_stack) from [<c020fb9c>] (dump_stack+0x7c/0x9c)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.076668] [<c020fb9c>] (dump_stack) from [<c0029214>] (__warn+0xbc/0xec)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.083572] [<c0029214>] (__warn) from [<c00292e8>] (warn_slowpath_null+0x1c/0x24)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.091194] [<c00292e8>] (warn_slowpath_null) from [<bf1ae250>] (ieee80211_rx_napi+0x88/0x81c [mac80211])
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.100830] [<bf1ae250>] (ieee80211_rx_napi [mac80211]) from [<bf20e454>] (pcie_rx_recv_ndp+0x92c/0xa58 [mwlwifi])
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.111231] [<bf20e454>] (pcie_rx_recv_ndp [mwlwifi]) from [<c002daf4>] (tasklet_action+0x8c/0xe8)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.120228] [<c002daf4>] (tasklet_action) from [<c002d28c>] (__do_softirq+0xd0/0x204)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.128090] [<c002d28c>] (__do_softirq) from [<c002d644>] (irq_exit+0x94/0xb8)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.135344] [<c002d644>] (irq_exit) from [<c0061d18>] (__handle_domain_irq+0x90/0xb4)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.143206] [<c0061d18>] (__handle_domain_irq) from [<c0009428>] (gic_handle_irq+0x50/0x94)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.151591] [<c0009428>] (gic_handle_irq) from [<c0012c8c>] (__irq_svc+0x6c/0x90)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.159103] Exception stack(0xdf48de90 to 0xdf48ded8)
Fri Jul  7 07:29:14 2017 kern.warn kernel: [81119.164173] de80:                                     d5426794 fffffff0 00000000 00000000
[... cut ...]
Fri Jul  7 07:29:15 2017 kern.warn kernel: [81119.999685] [<c003e290>] (worker_thread) from [<c00421d0>] (kthread+0xd8/0xec)
Fri Jul  7 07:29:15 2017 kern.warn kernel: [81120.006938] [<c00421d0>] (kthread) from [<c000edf8>] (ret_from_fork+0x14/0x3c)
Fri Jul  7 07:29:15 2017 kern.warn kernel: [81120.014200] ---[ end trace f0a96ef94a0b105f ]---`

Now when those crashes happen, the router completely freezes. Looking at the kernel logs, it just... keeps on crashing non-stop until I yank power off and hard-reboot.

starcms, 2017-07-07:
@farchord explained to me that this behavior only started after having upgraded to the driver version that includes the two commits for rx/tx and signal status (up to and including commit 1cc8aa6).  Previous to that, on official driver 10.3.4.0-20170606 (up to and including commit 36bc327) he had never experienced this.

farchord, 2017-07-07:
I just wanna mention, I do not know which commit I had before. Frankly, my router was working great for probably close to one month, I just updated because, well, why not. XD So I do not know which commit I had before.

ValCher1961, 2017-07-07:
This is the same - https://github.com/kaloz/mwlwifi/issues/186

farchord, 2017-07-07:
Didn't look the same to me lol that's why I didn't add it in

starcms, 2017-07-07:
@ValCher1961, Not the same at all.  This issue causes the router to crash and die.  It's not simply a warning.

ValCher1961, 2017-07-07:
Watch warnings from Message -186
`[  109.217755] WARNING: CPU: 0 PID: 0 at net/mac80211/rx.c:4326 ieee80211_rx_napi+0x5ca/0x690 [mac80211]`

and your warning
`[85560.493785] WARNING: CPU: 0 PID: 19536 at compat-wireless-2017-01-31/net/mac80211/rx.c:4224 ieee80211_rx_napi+0x88/0x81c [mac80211]`

ValCher1961, 2017-07-07:
The mistake is the same, we have different kernels, and it also causes the system to die.

farchord, 2017-07-07:
I still don't see it XD but no need to explain I'm just not versed into Linux debug messages. :) If it needs merged or closed, I assume @yuhhaurlin will do it :)

ValCher1961, 2017-07-07:
Oh, no, I didn't even think about it. :)
Excuse me.
It's all in the hands of our revered @yuhhaurlin.

yuhhaurlin, 2017-07-07:
Yes. This is the same as issue #186. It is caused by the code to report Tx rate. If it will let device crashed, I will check it first. I close this one, due to it is duplicated.

closed 2017-07-07 by yuhhaurlin