kaloz/mwlwifi #190: With this driver : num of SSID limited to 8 instead of 16 (Linksys WRT 1900ACS)

delandtj, 2017-07-10:
Running on a linksys wrt 1900 acs, where I had 12, with the previous release.
Wireless never came back up...
removed 4 -> things worked again
Needless to say that the main reason for using these AP's is now gone :-(

BrainSlayer, 2017-07-10:
this is related to the new address fix which limited it to 8 devices. the fix for that issue is easy. let me do a short patch for you. but its up to you to compile it by yourself and try it out

BrainSlayer, 2017-07-10:
mmh. i reviewed the code. i was wrong. its limited to 16

BrainSlayer, 2017-07-10:
and i'm talking about current todays driver release

BrainSlayer, 2017-07-10:
the problem i see is that all the recent changes are not related to WRT1900ACS but only to the WRT3200ACM chipset

BrainSlayer, 2017-07-10:
so if you're using lede or openwrt, the problem source is maybe not just the driver

yuhhaurlin, 2017-07-11:
Yes. I close this one.

closed 2017-07-11 by yuhhaurlin

yuhhaurlin, 2017-07-11:
The BSSID assignment is only used for WRT3200ACM (88W8964) and it can use 16 VAPs now.

delandtj, 2017-07-11:
I could use +8 SSIDs on lede 17.01.0, but apparently the mwlwifi driver is introduced in 17.01.=>1 and then 8+ SSIDs  stopped working on WRT1900ACS ... so I would have thought it's the driver somewhere.

kylejvrsa, 2017-07-11:
@delandtj I think you should have a look at the lede changelog for the specific version.
https://lede-project.org/releases/17.01/start