kaloz/mwlwifi #191: Broken 5Ghz with version 7-11

borkra, 2017-07-13:
Commit e793246424c7767fdbab9bdce4ddb86d94301ea0 completely broke 5GHz network on WRT1900ACv2. Intel 7260 ping/pong between 5 and 2.4 GHz network. Marvell on Surface Pro ignores 5 GHz network completely. Bar indicator on Luci goes up and down.

yuhhaurlin, 2017-07-13:
If 10.3.4.0-20170606 can work, 10.3.4.0-20170711 should be no problem. BTW, another one 10.3.4.0-20170713 is ready. But the fix is specific for WRT3200ACM.

BTW, our QA had done tested on 88W8864 and 88W8964 at same time.

borkra, 2017-07-13:
10.3.4.0-20170606 works ok as well as above 0606 until commit "Added more checks for rx accounting".

yuhhaurlin, 2017-07-13:
The commits after 10.3.4.0-20170606 are specific for WRT3200ACM, these commits won't affect 88W8864 (previous devices).

yuhhaurlin, 2017-07-13:
Tx/Rx rate, signal report and BSSID addresses assignment are only worked for WRT3200ACM. In other word, these two versions of driver are almost the same  for previous devices,

borkra, 2017-07-13:
Well, I tried multiple times downgrading fixes the problem, upgrading causing 5 GHz not to work.

Reviewing code: 2 Commits for RX accounting are not specific to the WRT3200ACM and do affect all chipsets.

yuhhaurlin, 2017-07-13:
I will do test on 88W8864 and reply you.

yuhhaurlin, 2017-07-13:
I just tested 10.3.0.4-20170713 on 88W8864, I did not find problem.

[image]

yuhhaurlin, 2017-07-13:
The test is done with Macbook air (2x2).

closed 2017-07-13 by yuhhaurlin

yuhhaurlin, 2017-07-13:
I don't have WRT1900ACv2, I use our RD AP board with 88W8864 module to do the test. I put LEDE WRT1900ACv2 images for you to do test:

factory: https://drive.google.com/open?id=0B3qLWtcWB9EdTWVXb012bVBxVXM
upgrade: https://drive.google.com/open?id=0B3qLWtcWB9EddnZLQUZEZFVLNVk

root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20170713
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x10
mac address: 00:50:43:21:bd:e9
2g: disable
5g: enable
antenna: 4 4
irq number: 44
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000007
radio: enable
iobase0: f0d80000
iobase1: f0f00000
tx limit: 768
rx limit: 64
qe trigger number: 22481

borkra, 2017-07-13:
I double checked.
0711 version have an issue.
0713 seem to work ok.

yuhhaurlin, 2017-07-13:
It does not make sense. However it is good to know the problem is gone.