kaloz/mwlwifi #192: Connection regularly interrupted with 802.11g only devices and mwlwifi >= 10.3.4.0-20170606

haribert, 2017-07-13:
I summarise what I described in thread #170:

I have a [Squeezbox Radio](http://wiki.slimdevices.com/index.php/Hardware_comparison) which lose the connection to the WRT3200acm with LEDE 17.01.02 regularly. Normally a song plays for around 5 minutes, then the connection is gone and 5 minutes later the Squeezebox Radio is able to connect again. But the connection is not stable for more than around 5 minutes.

The device has 802.11g only.

This issue was not present with LEDE 17.01.01.

Updated LEDE to the version of @yuhhaurlin [from following post](https://github.com/kaloz/mwlwifi/issues/170#issuecomment-314969724). It is still present. The following information (system and log) is from the [LEDE image in this post](https://github.com/kaloz/mwlwifi/issues/170#issuecomment-314969724).

WLAN Settings:
[image]

```
root@lede:~# uname -a
Linux lede 4.9.34 #0 SMP Fri Jul 7 06:29:41 2017 armv7l GNU/Linux
root@lede:~# opkg info kmod-mwlwifi
Package: kmod-mwlwifi
Version: 4.9.34+10.3.4.0.git-2017-07-10-1
Depends: kernel (= 4.9.34-1-261bc28a665dffab13fceedc7e0a9e20), kmod-mac80211
Status: install user installed
Architecture: arm_cortex-a9_vfpv3
Installed-Time: 1499877823

root@lede:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170606
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:b6:70:da
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1080000
iobase1: e1300000
tx limit: 1024
rx limit: 16384

root@lede:~# cat /etc/banner
     _________
    /        /\      _    ___ ___  ___
   /  LE    /  \    | |  | __|   \| __|
  /    DE  /    \   | |__| _|| |) | _|
 /________/  LE  \  |____|___|___/|___|                      lede-project.org
 \        \   DE /
  \    LE  \    /  -----------------------------------------------------------
   \  DE    \  /    Reboot (SNAPSHOT, r4542-0b493ed)
    \________\/    -----------------------------------------------------------

root@lede:~#
```

Corresponding Log:

```
root@lede:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.34 (ag33e33@ubuntu) (gcc version 5.4.0 (LEDE GCC 5.4.0 r4542-0b493ed) ) #0 SMP Fri Jul 7 06:29:41 2017
[    0.000000] CPU: ARMv7 Processor [414fc091] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Linksys WRT3200ACM
[... 568 lines cut ...]
Thu Jul 13 21:27:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: authenticated
Thu Jul 13 21:32:54 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 00:04:20:29:87:37
Thu Jul 13 21:32:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: associated (aid 3)
Thu Jul 13 21:32:54 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 00:04:20:29:87:37
Thu Jul 13 21:32:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 WPA: pairwise key handshake completed (RSN)
Thu Jul 13 21:33:02 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: authenticated
Thu Jul 13 21:37:54 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 00:04:20:29:87:37
Thu Jul 13 21:37:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: disassociated due to inactivity
Thu Jul 13 21:37:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: associated (aid 3)
Thu Jul 13 21:37:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: authenticated
Thu Jul 13 21:37:54 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 00:04:20:29:87:37
Thu Jul 13 21:37:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 WPA: pairwise key handshake completed (RSN)
Thu Jul 13 21:42:54 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 00:04:20:29:87:37
Thu Jul 13 21:42:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: associated (aid 3)
Thu Jul 13 21:42:54 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 00:04:20:29:87:37
Thu Jul 13 21:42:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 WPA: pairwise key handshake completed (RSN)
Thu Jul 13 21:43:09 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: authenticated
Thu Jul 13 21:47:54 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 00:04:20:29:87:37
Thu Jul 13 21:47:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: disassociated due to inactivity
Thu Jul 13 21:47:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: associated (aid 3)
Thu Jul 13 21:47:54 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 00:04:20:29:87:37
Thu Jul 13 21:47:54 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 WPA: pairwise key handshake completed (RSN)
Thu Jul 13 21:47:55 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: authenticated
```
Is any additional log needed? Is there anything suspicious?

yuhhaurlin, 2017-07-14:
Can you make sure open mode (without security) is all right? If yes, I will update hostapd to you to dump more debug information.

yuhhaurlin, 2017-07-14:
Can you let me know version of mwlwifi for LEDE.01.01？

haribert, 2017-07-14:
I will check later today with the open mode and let you know about the outcome.

[Looks like LEDE 17.01.1 had mwlwifi version 4.4.61+10.3.2.0-20170110-1.](http://downloads.lede-project.org/releases/17.01.1/targets/mvebu/generic/packages/).

haribert, 2017-07-15:
There is no interruption In open mode without encryption. Connection is great.

There is something else I do not understand. If I flash your LEDE sysupgrade and do a restart (may be combined with `opkg update/install packages`), the system is back to the previous one from davidc502sis. Is your sysupgrade somehow marked as non-permanent?

yuhhaurlin, 2017-07-15:
No. These images are got from directory "bin". I will try to build hostapd with more debug information for you to do test. But you should make sure you can run the LEDE image I sent out.

haribert, 2017-07-15:
I can run your LEDE images.

The `opkg update/upgrade/install` has probably overwritten my packages. So I will test without opkg update.

Thanks!

yuhhaurlin, 2017-07-18:
Please use following ways to get more debug information for me:

grep _level /var/run/hostapd-phy1.conf 
    logger_syslog_level=2
    logger_stdout_level=2
uci set wireless.radio1.log_level=1
uci commit wireless
wifi up
grep _level /var/run/hostapd-phy1.conf 
    logger_syslog_level=1
    logger_stdout_level=1

yuhhaurlin, 2017-07-19:
Or You can get LEDE image with 612df7c from:

factory: https://drive.google.com/open?id=0B3qLWtcWB9EdbklyWUowRkJuT1U
upgrade: https://drive.google.com/open?id=0B3qLWtcWB9EdTUlnU0hsYjlCd00

root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170713-debug
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 00:50:43:22:88:8a
2g: disable
5g: enable
antenna: 4 4
irq number: 44
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: f0e00000
iobase1: f1080000
tx limit: 1024
rx limit: 16384

yuhhaurlin, 2017-07-24:
No reply for this issue. If needed, you can request to reopen it or create another issue. Thanks.

haribert, 2017-07-24:
I had no time to test this in the last week.
I will do ASAP.

haribert, 2017-07-25:
Here is the result with the sysupgrade and debug level = 1

```
root@lede:~# grep _level /var/run/hostapd-phy1.conf
logger_syslog_level=1
logger_stdout_level=1
root@lede:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170713-debug
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:b6:70:da
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0e00000
iobase1: e1080000
tx limit: 1024
rx limit: 16384

root@lede:~#
```

Connection interrupted at 20:42

```
root@lede:~# dmesg
[...]
[   84.961240] ieee80211 phy0: change: 0x40
[   85.060216] ieee80211 phy0: change: 0x100
[   85.119566] ieee80211 phy0: change: 0x100
[   85.131203] ieee80211 phy0: change: 0x42
[   85.227214] ieee80211 phy0: change: 0x40
[   85.408219] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   85.414665] br-xxx: port 3(wlan0) entered blocking state
[   85.420522] br-xxx: port 3(wlan0) entered forwarding state
[  128.032508] ieee80211 phy1: staid 1 deleted
[  141.378063] Rx EAPOL
[  141.387353] Rx EAPOL
[  634.811932] Rx EAPOL
[  635.542546] Rx EAPOL
[  638.303622] ieee80211 phy1: staid 1 deleted
[  638.329089] ieee80211 phy1: staid 1 deleted
[  639.380059] ieee80211 phy1: staid 5 deleted
[  642.459230] ieee80211 phy1: staid 1 deleted
[  643.285871] Rx EAPOL
[  643.310284] Rx EAPOL

root@lede:~# logread | grep hostapd | grep 00:04:20:29:87:37
Tue Jul 25 20:32:35 2017 daemon.info hostapd: wlan1-3: STA 00:04:20:29:87:37 IEEE 802.11: associated (aid 1)
Tue Jul 25 20:32:35 2017 daemon.notice hostapd: wlan1-3: AP-STA-CONNECTED 00:04:20:29:87:37
Tue Jul 25 20:32:35 2017 daemon.info hostapd: wlan1-3: STA 00:04:20:29:87:37 IEEE 802.11: authenticated
Tue Jul 25 20:34:02 2017 daemon.notice hostapd: wlan1-3: AP-STA-DISCONNECTED 00:04:20:29:87:37
Tue Jul 25 20:34:02 2017 daemon.info hostapd: wlan1-3: STA 00:04:20:29:87:37 IEEE 802.11: disassociated
Tue Jul 25 20:34:07 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: authentication OK (open system)
Tue Jul 25 20:34:07 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 MLME: MLME-AUTHENTICATE.indication(00:04:20:29:87:37, OPEN_SYSTEM)
Tue Jul 25 20:34:07 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 MLME: MLME-DELETEKEYS.request(00:04:20:29:87:37)
Tue Jul 25 20:34:07 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: association OK (aid 2)
Tue Jul 25 20:34:07 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: associated (aid 2)
Tue Jul 25 20:34:07 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 MLME: MLME-ASSOCIATE.indication(00:04:20:29:87:37)
Tue Jul 25 20:34:07 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 MLME: MLME-DELETEKEYS.request(00:04:20:29:87:37)
[... 51 lines cut ...]
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 WPA: event 0 notification
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 MLME: MLME-AUTHENTICATE.indication(00:04:20:29:87:37, OPEN_SYSTEM)
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 MLME: MLME-DELETEKEYS.request(00:04:20:29:87:37)
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: association OK (aid 2)
Tue Jul 25 20:42:36 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: associated (aid 2)
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 MLME: MLME-REASSOCIATE.indication(00:04:20:29:87:37)
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 MLME: MLME-DELETEKEYS.request(00:04:20:29:87:37)
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: binding station to interface 'wlan1'
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 WPA: event 1 notification
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 WPA: sending 1/4 msg of 4-Way Handshake
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 WPA: received EAPOL-Key frame (2/4 Pairwise)
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 WPA: sending 3/4 msg of 4-Way Handshake
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 WPA: received EAPOL-Key frame (4/4 Pairwise)
Tue Jul 25 20:42:36 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 00:04:20:29:87:37
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.1X: authorizing port
Tue Jul 25 20:42:36 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 WPA: pairwise key handshake completed (RSN)
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 MLME: MLME-DEAUTHENTICATE.indication(00:04:20:29:87:37, 2)
Tue Jul 25 20:42:36 2017 daemon.debug hostapd: wlan1: STA 00:04:20:29:87:37 MLME: MLME-DELETEKEYS.request(00:04:20:29:87:37)
Tue Jul 25 20:42:36 2017 daemon.info hostapd: wlan1: STA 00:04:20:29:87:37 IEEE 802.11: authenticated
root@lede:~#
```

Any additional information needed?

Can you please reopen the issue. Thanks.

yuhhaurlin, 2017-07-25:
Thanks. I will prepare another debug image.

yuhhaurlin, 2017-07-26:
It looks like your problem is related to group rekey. I will update image to dump EAPOL packets received by driver later.
You can add "option wpa_group_rekey '0'" to /etc/config/wireless to disable group rekey to work around this problem first.

yuhhaurlin, 2017-08-15:
Since 11g only client hard to find now. If disable group rekey can resolve your problem, I won't take time to check it. I close this one. Thanks.

haribert, 2017-08-15:
I am a little bit confused and disappointed. Is this really how open and confirmed issues are handled?

I did not yet test if `wpa_group_rekey` works (because I thought you will solve this issue the clean way), therefore it is not clear if there is a running workaround. Furthermore, with the previous version of mwlwifi the connection was functional. So you removed functionality from the driver and argue that there are too less devices which are affected?

I thought the opensource version of WRT3200acm should have the same functionality as the original firmware. At least this is always your argument. If you close issues without a proper solution this is not given.

yuhhaurlin, 2017-08-15:
All right. This issue will be low priority.

haribert, 2017-08-27:
For your information, the option `option wpa_group_rekey '0'` is not a working workaround. I still have the same issue.

I have currently a wlan without encryption and mac filter to use this device.

yuhhaurlin, 2017-08-27:
Can you give me the log you got?

haribert, 2017-08-29:
@yuhhaurlin I can give you the logs next week. I am currently off site.

yuhhaurlin, 2017-10-31:
11g only device is pretty rare now. I close this one.

closed 2017-10-31 by yuhhaurlin

thagabe, 2017-11-01:
AFAIK I do not have any 11g devices. Yet my logs seem to imply that I do. Here are my kernel logs:
`[   23.098366] device wlan0 entered promiscuous mode
[   23.103137] br-lan: port 2(wlan0) entered blocking state
[   23.108470] br-lan: port 2(wlan0) entered forwarding state
[   23.114102] br-lan: port 2(wlan0) entered disabled state
[   23.120240] ieee80211 phy0: change: 0x100
[   23.133331] ieee80211 phy0: change: 0x40
[   23.133619] br-lan: port 3(wlan1) entered blocking state
[   23.133620] br-lan: port 3(wlan1) entered disabled state
[   23.133755] device wlan1 entered promiscuous mode
[   23.182084] ieee80211 phy1: change: 0x100
[   23.195189] ieee80211 phy1: change: 0x42
[   23.381326] ieee80211 phy0: change: 0x40
[   23.384894] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   23.384946] br-lan: port 3(wlan1) entered blocking state
[   23.384948] br-lan: port 3(wlan1) entered forwarding state
[   23.681353] ieee80211 phy0: change: 0x40
[   23.808361] ieee80211 phy0: change: 0x100
[   23.869374] ieee80211 phy0: change: 0x100
[   23.881307] ieee80211 phy0: change: 0x42
[   24.093639] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   24.100074] br-lan: port 2(wlan0) entered blocking state
[   24.105418] br-lan: port 2(wlan0) entered forwarding state
[  386.366823] ieee80211 phy1: staid 2 deleted
[11605.476027] ieee80211 phy1: staid 1 deleted
[21562.009382] ieee80211 phy1: staid 2 deleted
[28127.002658] ieee80211 phy1: staid 4 deleted
[32034.570034] ieee80211 phy1: staid 4 deleted
[32736.438754] ieee80211 phy1: staid 5 deleted
[33573.286541] ieee80211 phy1: staid 5 deleted
[39191.974842] ieee80211 phy1: staid 3 deleted
[41761.287955] ieee80211 phy1: staid 2 deleted
[42003.089047] ieee80211 phy1: staid 6 deleted
[42174.726982] ieee80211 phy1: staid 3 deleted
[44717.633854] ieee80211 phy1: staid 5 deleted
[49551.832932] ieee80211 phy1: staid 5 deleted
[51516.761526] ieee80211 phy1: staid 4 deleted
[56307.680597] ieee80211 phy1: staid 3 deleted
[56623.337071] ieee80211 phy0: staid 1 deleted
[56632.094139] ieee80211 phy0: staid 2 deleted
[59253.149762] ieee80211 phy1: staid 4 deleted
[68749.858864] ieee80211 phy1: staid 2 deleted
[69879.678936] ieee80211 phy1: staid 2 deleted
[70585.555910] ieee80211 phy1: staid 5 deleted
[73794.034001] ieee80211 phy1: staid 2 deleted
[74221.957404] ieee80211 phy0: staid 3 deleted
[74889.059512] ieee80211 phy0: staid 3 deleted
[75726.923669] ieee80211 phy1: staid 4 deleted`

Here is a more comprehensive system log:
`Tue Oct 31 16:43:18 2017 daemon.info dnsmasq[1914]: read /etc/hosts - 4 addresses
Tue Oct 31 16:43:18 2017 daemon.info dnsmasq[1914]: read /tmp/hosts/odhcpd - 0 addresses
Tue Oct 31 16:43:18 2017 daemon.info dnsmasq[1914]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Tue Oct 31 16:43:18 2017 daemon.info dnsmasq-dhcp[1914]: read /etc/ethers - 0 addresses
Tue Oct 31 16:43:19 2017 daemon.warn odhcpd[1246]: DHCPV6 REQUEST IA_NA from 000100011f8307a6f832e42b489b on br-lan: ok 2601:646:c200:74d4::f89/128 fd7c:8024:9135::f89/128 
Tue Oct 31 16:43:19 2017 daemon.info dnsmasq[1914]: read /etc/hosts - 4 addresses
Tue Oct 31 16:43:19 2017 daemon.info dnsmasq[1914]: read /tmp/hosts/odhcpd - 2 addresses
Tue Oct 31 16:43:19 2017 daemon.info dnsmasq[1914]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Tue Oct 31 16:43:19 2017 daemon.info dnsmasq-dhcp[1914]: read /etc/ethers - 0 addresses
Tue Oct 31 16:44:39 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 5c:a3:9d:68:bd:53
Tue Oct 31 16:45:09 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 54:60:09:05:2c:16
Tue Oct 31 16:49:42 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 5c:a3:9d:68:bd:53
Tue Oct 31 16:50:12 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 54:60:09:05:2c:16
Tue Oct 31 16:54:45 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 5c:a3:9d:68:bd:53
Tue Oct 31 16:55:15 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 54:60:09:05:2c:16
Tue Oct 31 16:59:48 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 5c:a3:9d:68:bd:53
Tue Oct 31 17:00:18 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 54:60:09:05:2c:16
Tue Oct 31 17:04:51 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 5c:a3:9d:68:bd:53
Tue Oct 31 17:05:21 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 54:60:09:05:2c:16
Tue Oct 31 17:09:54 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 5c:a3:9d:68:bd:53
[... 407 lines cut ...]
Tue Oct 31 23:47:36 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK 40:4e:36:21:6a:f7
Tue Oct 31 23:47:47 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK 68:db:ca:8f:11:91
Tue Oct 31 23:48:47 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 54:60:09:05:2c:16
Tue Oct 31 23:50:12 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 5c:a3:9d:68:bd:53
Tue Oct 31 23:52:50 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK 68:db:ca:8f:11:91
Tue Oct 31 23:53:50 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 54:60:09:05:2c:16
Tue Oct 31 23:55:15 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 5c:a3:9d:68:bd:53
Tue Oct 31 23:57:53 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK 68:db:ca:8f:11:91
Tue Oct 31 23:58:53 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 54:60:09:05:2c:16
Tue Oct 31 23:59:29 2017 kern.info kernel: [75726.923669] ieee80211 phy1: staid 4 deleted
Tue Oct 31 23:59:29 2017 daemon.info hostapd: wlan1: STA 68:db:ca:8f:11:91 IEEE 802.11: associated (aid 4)
Tue Oct 31 23:59:29 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 68:db:ca:8f:11:91
[... cut ...]
Wed Nov  1 00:00:18 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 5c:a3:9d:68:bd:53
Wed Nov  1 00:03:56 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 54:60:09:05:2c:16
Wed Nov  1 00:05:21 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 5c:a3:9d:68:bd:53
Wed Nov  1 00:08:59 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 54:60:09:05:2c:16
Wed Nov  1 00:10:24 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 5c:a3:9d:68:bd:53`

Mind you I am running LEDE master branch with the latest commit from from mwlwifi.

haribert, 2017-11-03:
I know it does not bother you, because you are the developer and should not concern about marketing stuff. But perhaps you can forward this to your marketing department.

I decided to sell my WRT3200acm router and buy something which I call stable and take the open issues serious. I am a little bit disappointed about the phrase "open source ready", because the documentation of the driver is not open source. And without any information, it is not possible to develop anything for this device. Therefore @yuhhaurlin is the only person who has the documentation and is able to implement anything.
I own this device now since about a year and still have some serious workarounds in place and have to reboot the device every week. I spent to much time debugging my environment and device without any improvements in the last months.

So Long, and Thanks for All the Fish

yuhhaurlin, 2017-11-03:
You did not offer log to me for a long period of time. I think maybe you do not use 11g only device now. That is the reason I close it.

yuhhaurlin, 2017-11-03:
The remaining severe one is host command timeout, we will fix it.

pilotmattk, 2017-11-10:
Wait..  What the actual fuck?   There are a whole pile of closed issues on this very topic and you continue to close them without providing a fix?  Obviously people are still using 11g and this needs to be fixed, disabling group rekey isn't a real fix.   Either open source the docs to go with the "opensource" driver or actually fix something for once.
[image]

yuhhaurlin, 2017-11-10:
Can you help me to check this problem?

yuhhaurlin, 2017-11-10:
Please follow the description above and give me log with group key disable. Thanks.

pilotmattk, 2017-11-10:
I thought root cause was found above when you said it would be fixed with low priority?   Do I need a debug build of hostapd?   Additional debug logging turned on?

I can get logs, however I'm now relying on the router to be stable (with the rekey off hack).   I decommissioned my backup router due to KRACK.

yuhhaurlin, 2017-11-10:
Did you encounter problem?

pilotmattk, 2017-11-15:
```[57624.519201] ieee80211 phy1: staid 3 deleted
[58224.386900] ieee80211 phy0: staid 1 deleted
[58836.254648] ieee80211 phy1: staid 2 deleted
[60024.336897] ieee80211 phy0: staid 1 deleted
[61823.047731] ieee80211 phy1: staid 1 deleted
[61824.338455] ieee80211 phy1: staid 1 deleted
[69267.112682] ieee80211 phy1: create ba result error 1
[69267.126694] ieee80211 phy1: ampdu operation error code: -22
[69303.582344] ieee80211 phy1: create ba result error 1
[69303.595442] ieee80211 phy1: ampdu operation error code: -22
[69314.676467] ieee80211 phy0: staid 1 deleted
[69320.922204] ieee80211 phy1: create ba result error 1
[69320.936206] ieee80211 phy1: ampdu operation error code: -22
[69354.171877] ieee80211 phy0: staid 1 deleted
[69367.471527] ieee80211 phy1: create ba result error 1
[69367.485553] ieee80211 phy1: ampdu operation error code: -22
[69393.755328] ieee80211 phy0: staid 1 deleted
[69399.980546] ieee80211 phy1: create ba result error 1
[69399.994559] ieee80211 phy1: ampdu operation error code: -22
[69433.223132] ieee80211 phy0: staid 1 deleted
[69469.417148] ieee80211 phy0: staid 1 deleted
[70025.268952] ieee80211 phy1: create ba result error 1
[70025.281938] ieee80211 phy1: ampdu operation error code: -22
[70146.375311] ieee80211 phy0: staid 1 deleted
[70152.580084] ieee80211 phy1: create ba result error 1
[70152.594089] ieee80211 phy1: ampdu operation error code: -22
[70224.589150] ieee80211 phy0: staid 1 deleted
[70521.954096] ieee80211 phy1: create ba result error 1
[70521.968102] ieee80211 phy1: ampdu operation error code: -22
[70595.511792] ieee80211 phy0: staid 1 deleted
[70695.281071] ieee80211 phy1: create ba result error 1
[70695.294189] ieee80211 phy1: ampdu operation error code: -22
[70737.099726] ieee80211 phy0: staid 1 deleted
[70774.289988] ieee80211 phy1: create ba result error 1
[70774.304000] ieee80211 phy1: ampdu operation error code: -22
[70797.410683] ieee80211 phy1: create ba result error 1
[70797.424685] ieee80211 phy1: ampdu operation error code: -22
[70816.417558] ieee80211 phy0: staid 1 deleted
[70920.654908] ieee80211 phy1: create ba result error 1
[70920.668908] ieee80211 phy1: ampdu operation error code: -22
[70962.993428] ieee80211 phy0: staid 1 deleted
[71044.386234] ieee80211 phy0: staid 1 deleted
[71051.596661] ieee80211 phy1: create ba result error 1
[71051.610744] ieee80211 phy1: ampdu operation error code: -22
[71083.770675] ieee80211 phy0: staid 1 deleted
root@ROOter:~# ```

```Tue Nov 14 18:42:09 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK 50:c7:bf:05:8d:56
Tue Nov 14 18:42:20 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK 50:f5:da:df:e3:da
Tue Nov 14 18:42:52 2017 daemon.info hostapd: wlan0: STA 58:00:e3:41:f8:57 IEEE 802.11: associated (aid 2)
Tue Nov 14 18:42:52 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 58:00:e3:41:f8:57
Tue Nov 14 18:42:52 2017 daemon.info hostapd: wlan0: STA 58:00:e3:41:f8:57 WPA: pairwise key handshake completed (RSN)
Tue Nov 14 18:42:52 2017 daemon.info dnsmasq-dhcp[2082]: DHCPREQUEST(br-lan) 192.168.1.112 58:00:e3:41:f8:57 
Tue Nov 14 18:42:52 2017 daemon.info dnsmasq-dhcp[2082]: DHCPACK(br-lan) 192.168.1.112 58:00:e3:41:f8:57 arch_klap
Tue Nov 14 18:42:52 2017 daemon.warn dnsmasq-dhcp[2082]: not giving name Ooma.lan to the DHCP lease of 192.168.1.245 because the name exists in /tmp/hosts/dhcp.cfg02411c with address 192.168.1.244
Tue Nov 14 18:42:53 2017 daemon.info hostapd: wlan0: STA 58:00:e3:41:f8:57 IEEE 802.11: authenticated
Tue Nov 14 18:42:53 2017 daemon.info odhcpd[1283]: Using a RA lifetime of 1800 seconds on br-lan
Tue Nov 14 18:42:57 2017 daemon.info odhcpd[1283]: Using a RA lifetime of 1800 seconds on br-lan
Tue Nov 14 18:42:57 2017 daemon.notice odhcpd[1283]: Got DHCPv6 request
Tue Nov 14 18:42:57 2017 daemon.warn odhcpd[1283]: DHCPV6 SOLICIT IA_NA from 0004b89b93de93b897fe6197eab637baccb9 on br-lan: ok fd3b:2448:f853::8cd/128 
Tue Nov 14 18:42:58 2017 daemon.notice odhcpd[1283]: Got DHCPv6 request
Tue Nov 14 18:42:58 2017 daemon.warn odhcpd[1283]: DHCPV6 REQUEST IA_NA from 0004b89b93de93b897fe6197eab637baccb9 on br-lan: ok fd3b:2448:f853::8cd/128 
Tue Nov 14 18:42:59 2017 authpriv.info dropbear[7963]: Child connection from 192.168.1.112:39222
Tue Nov 14 18:43:01 2017 authpriv.notice dropbear[7963]: Pubkey auth succeeded for 'root' with key md5 26:30:29:1e:95:40:33:5c:1f:95:21:17:8a:d9:0e:23 from 192.168.1.112:39222
Tue Nov 14 18:43:44 2017 daemon.debug hostapd: wlan1: STA 84:10:0d:63:dc:fe IEEE 802.11: authentication OK (open system)
Tue Nov 14 18:43:44 2017 daemon.debug hostapd: wlan1: STA 84:10:0d:63:dc:fe MLME: MLME-AUTHENTICATE.indication(84:10:0d:63:dc:fe, OPEN_SYSTEM)
Tue Nov 14 18:43:44 2017 daemon.debug hostapd: wlan1: STA 84:10:0d:63:dc:fe MLME: MLME-DELETEKEYS.request(84:10:0d:63:dc:fe)
Tue Nov 14 18:43:44 2017 daemon.debug hostapd: wlan1: STA 84:10:0d:63:dc:fe IEEE 802.11: association OK (aid 7)
Tue Nov 14 18:43:44 2017 daemon.info hostapd: wlan1: STA 84:10:0d:63:dc:fe IEEE 802.11: associated (aid 7)
Tue Nov 14 18:43:44 2017 daemon.debug hostapd: wlan1: STA 84:10:0d:63:dc:fe MLME: MLME-REASSOCIATE.indication(84:10:0d:63:dc:fe)
[... 13 lines cut ...]
Tue Nov 14 18:43:49 2017 kern.err kernel: [71222.690613] ieee80211 phy1: ampdu operation error code: -22
Tue Nov 14 18:43:53 2017 daemon.info hostapd: wlan1: STA 84:10:0d:63:dc:fe IEEE 802.11: authenticated
Tue Nov 14 18:44:24 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 84:10:0d:63:dc:fe
[... cut ...]
Tue Nov 14 18:44:33 2017 daemon.info dnsmasq-dhcp[2082]: DHCPOFFER(br-lan) 192.168.1.154 84:10:0d:63:dc:fe 
Tue Nov 14 18:44:36 2017 daemon.info odhcpd[1283]: Using a RA lifetime of 1800 seconds on br-lan
Tue Nov 14 18:44:41 2017 daemon.info dnsmasq-dhcp[2082]: DHCPDISCOVER(br-lan) 84:10:0d:63:dc:fe 
Tue Nov 14 18:44:41 2017 daemon.info dnsmasq-dhcp[2082]: DHCPOFFER(br-lan) 192.168.1.154 84:10:0d:63:dc:fe 
Tue Nov 14 18:44:43 2017 daemon.notice hostapd: wlan1-1: AP-STA-POLL-OK f8:e0:79:da:f3:35
root@ROOter:~# 
```

yuhhaurlin, 2017-11-15:
It looks like Block ACK stream can't be established with that client.