kaloz/mwlwifi #194: Connection drops during WPA: group key handshake completed

ajmgh, 2017-07-14:
On my WRT3200ACM and I'm seeing this behavior with 802.11ac devices with VHT:

```
Fri Jul 14 12:14:36 2017 daemon.info hostapd: wlan1: STA b8:27:eb:b6:2f:94 WPA: group key handshake completed (RSN)
Fri Jul 14 12:14:36 2017 daemon.info hostapd: wlan1: STA ac:d1:b8:55:84:28 WPA: group key handshake completed (RSN)
Fri Jul 14 12:14:37 2017 daemon.info hostapd: wlan1: STA 50:f5:da:06:de:44 WPA: group key handshake completed (RSN)
Fri Jul 14 12:14:37 2017 daemon.info hostapd: wlan1: STA bc:f5:ac:f7:31:f1 WPA: group key handshake completed (RSN)
Fri Jul 14 12:14:39 2017 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 6c:72:20:03:f8:41
Fri Jul 14 12:14:41 2017 kern.info kernel: [79840.065438] ieee80211 phy0: staid 1 deleted
Fri Jul 14 12:14:43 2017 daemon.info hostapd: wlan0: STA 6c:72:20:03:f8:41 IEEE 802.11: associated (aid 1)
```

This seems to repeat whenever there is a WPA group key handshake.

```
Fri Jul 14 12:04:36 2017 daemon.info hostapd: wlan1: STA ac:d1:b8:55:84:28 WPA: group key handshake completed (RSN)
Fri Jul 14 12:04:36 2017 daemon.info hostapd: wlan1: STA b8:27:eb:b6:2f:94 WPA: group key handshake completed (RSN)
Fri Jul 14 12:04:37 2017 daemon.info hostapd: wlan1: STA 50:f5:da:06:de:44 WPA: group key handshake completed (RSN)
Fri Jul 14 12:04:37 2017 daemon.info hostapd: wlan1: STA bc:f5:ac:f7:31:f1 WPA: group key handshake completed (RSN)
Fri Jul 14 12:04:39 2017 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 6c:72:20:03:f8:41
Fri Jul 14 12:04:40 2017 kern.info kernel: [79238.241496] ieee80211 phy0: staid 1 deleted
```

It causes a drop but the client reconnects. However it interrupts any connections or downloads happening at the time.
Using latest custom-built LEDE. Driver version: 10.3.4.0-20170711.

Like issue #192 this only happens mwlwifi >= 10.3.4.0-20170606.

```
root@MooRouter:~# uname -a
Linux MooRouter 4.9.30 #0 SMP Sat Jun 3 15:48:57 2017 armv7l GNU/Linux
root@MooRouter:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170713
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:b6:ab:c2
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0d00000
iobase1: e0f80000
tx limit: 1024
rx limit: 16384

root@MooRouter:~#
```
Client device is a DWA-192, on Mac OS.

ajmgh, 2017-07-14:
Some more logging:
```
Fri Jul 14 13:16:25 2017 daemon.debug hostapd: wlan0: WPA rekeying GTK
Fri Jul 14 13:16:25 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 WPA: sending 1/2 msg of Group Key Handshake
Fri Jul 14 13:16:26 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 WPA: EAPOL-Key timeout
Fri Jul 14 13:16:26 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 WPA: sending 1/2 msg of Group Key Handshake
Fri Jul 14 13:16:26 2017 daemon.info hostapd: wlan1: STA b8:27:eb:b6:2f:94 WPA: group key handshake completed (RSN)
Fri Jul 14 13:16:26 2017 daemon.info hostapd: wlan1: STA ac:d1:b8:55:84:28 WPA: group key handshake completed (RSN)
Fri Jul 14 13:16:26 2017 daemon.info hostapd: wlan1: STA 50:f5:da:06:de:44 WPA: group key handshake completed (RSN)
Fri Jul 14 13:16:27 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 WPA: EAPOL-Key timeout
Fri Jul 14 13:16:27 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 WPA: sending 1/2 msg of Group Key Handshake
Fri Jul 14 13:16:28 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 WPA: EAPOL-Key timeout
Fri Jul 14 13:16:28 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 WPA: sending 1/2 msg of Group Key Handshake
Fri Jul 14 13:16:29 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 WPA: EAPOL-Key timeout
Fri Jul 14 13:16:29 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 WPA: WPA_PTK: sm->Disconnect
Fri Jul 14 13:16:29 2017 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 6c:72:20:03:f8:41
Fri Jul 14 13:16:29 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 WPA: event 3 notification
Fri Jul 14 13:16:29 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 IEEE 802.1X: unauthorizing port
Fri Jul 14 13:16:31 2017 kern.info kernel: [83549.415385] ieee80211 phy0: staid 1 deleted
Fri Jul 14 13:16:31 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 MLME: MLME-DEAUTHENTICATE.indication(6c:72:20:03:f8:41, 2)
Fri Jul 14 13:16:31 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 MLME: MLME-DELETEKEYS.request(6c:72:20:03:f8:41)
Fri Jul 14 13:16:32 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 IEEE 802.11: authentication OK (open system)
Fri Jul 14 13:16:32 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 WPA: event 0 notification
Fri Jul 14 13:16:32 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 MLME: MLME-AUTHENTICATE.indication(6c:72:20:03:f8:41, OPEN_SYSTEM)
Fri Jul 14 13:16:32 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 MLME: MLME-DELETEKEYS.request(6c:72:20:03:f8:41)
Fri Jul 14 13:16:32 2017 daemon.debug hostapd: wlan0: STA 6c:72:20:03:f8:41 IEEE 802.11: association OK (aid 1)
Fri Jul 14 13:16:32 2017 daemon.info hostapd: wlan0: STA 6c:72:20:03:f8:41 IEEE 802.11: associated (aid 1)
```

yuhhaurlin, 2017-07-14:
From the log, wlan1 group handshake is all right. wlan0 the group rekey is sending out but no reply.  Can you help to capture packet on air to see if EAPOL packet is sent out?

BTW, if you want to disable group rekey, you can edit /etc/config/wireless and add "option wpa_group_rekey "0"" to disable it.

yuhhaurlin, 2017-07-14:
Can you check if other clients on wlan0 can complete group rekey?

yuhhaurlin, 2017-07-14:
From the log, I just can make sure hostapd sent out group rekey EAPOL packet to client 6c:72:20:03:f8:41, but no reply from the client. If no more information, I can't tell what is going wrong. Maybe you can follow the way I described to disable group rekey.

max2344, 2017-07-17:
We have a similar problem since longer time and debugging this with driver version 10.3.4.0-20170713 on latest LEDE release from davidc502. We run EAP-TLS with enterprise auth (RADIUS). We noticed that the GTK rekeying works without problems, however the connections stop immediately after the 4 step handshake, of PTK exchange. The timeout for GTK rekeying is 600s and this falls in the same time for PTK rekey 3600s after 1 hour (default settings of hostap/wpa_supplicant). We're guessing that it is the same problem here, describred by ajmgh. The RADIUS server accepts the rekeying. We're out of ideas how to debug this further. Do you have any ideas?

yuhhaurlin, 2017-07-18:
Can you help to check https://github.com/kaloz/mwlwifi/commit/612df7ce8d02a45e78d24f2a5ab981ae6fc1594f?

yuhhaurlin, 2017-07-18:
You can get LEDE image from:

factory: https://drive.google.com/open?id=0B3qLWtcWB9EdbklyWUowRkJuT1U
upgrade: https://drive.google.com/open?id=0B3qLWtcWB9EdTUlnU0hsYjlCd00

root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170713-debug
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 00:50:43:22:88:8a
2g: disable
5g: enable
antenna: 4 4
irq number: 44
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: f0e00000
iobase1: f1080000
tx limit: 1024
rx limit: 16384

max2344, 2017-07-18:
Hey yuhhaurlin,
we'll retest with the patch you provided, however we can't use the image you prepared cause we're missing a few packages which are required for our setup to function. We are currently rebuilding LEDE git with the commit 612df7c and will report back asap.

ajmgh, 2017-07-18:
Setting
```
option wpa_group_rekey "0"
```
Fixed this for now, however will test the patch shortly

BrainSlayer, 2017-07-19:
this will disable rekeying and weak the encryption strength

yuhhaurlin, 2017-07-19:
Yes. So it is better to test the patch to see if the problem is fixed.

max2344, 2017-07-19:
tests were run todays but that change 612df7c didn't work. We're testing an alternative variant, will report back in a few hrs.

yuhhaurlin, 2017-07-19:
Maybe your problem is not related to this issue. This issue is that group key sending out and timeout from the log. The LEDE image will print out EAPOL if the reply packet is received.

max2344, 2017-07-21:
We activated debug log on the AP pls take a look. The STA completes the PTK EAPOL but all traffic stops immediately afterwards. We've ran a tcpdump of this on the STA to validate. The STA remains associated until the sessions time out, then it drops out of the connection and reassociates. Everything starts again. Also we've reviewed this patch 612df7c . Thats right, it fixes the fast path for EAPOL packets, new offsets are the right ones.

[eapol-debug-wrt3200acm-20170721-14.txt](https://github.com/kaloz/mwlwifi/files/1166535/eapol-debug-wrt3200acm-20170721-14.txt)

max2344, 2017-07-22:
We also have that pcap capture from the STA EAPOL for the last part of the log in this debugging session. But will send only on private mail address.

max2344, 2017-07-22:
view on the AP log from the STA:
```
Fri Jul 21 21:00:28 2017 daemon.debug hostapd: wlan0: STA 00:13:e8:d9:8a:27 IEEE 802.1X: Sending EAP Packet (identifier 109)
Fri Jul 21 21:00:28 2017 daemon.debug hostapd: wlan0: STA 00:13:e8:d9:8a:27 IEEE 802.1X: authorizing port
Fri Jul 21 21:00:28 2017 daemon.info hostapd: wlan0: STA 00:13:e8:d9:8a:27 IEEE 802.1X: authenticated - EAP type: 13 (TLS)
Fri Jul 21 21:00:28 2017 daemon.debug hostapd: wlan0: STA 00:13:e8:d9:8a:27 WPA: sending 1/4 msg of 4-Way Handshake
Fri Jul 21 21:00:28 2017 daemon.debug hostapd: wlan0: STA 00:13:e8:d9:8a:27 WPA: received EAPOL-Key frame (2/4 Pairwise)
Fri Jul 21 21:00:28 2017 daemon.debug hostapd: wlan0: STA 00:13:e8:d9:8a:27 WPA: sending 3/4 msg of 4-Way Handshake
Fri Jul 21 21:00:28 2017 daemon.debug hostapd: wlan0: STA 00:13:e8:d9:8a:27 WPA: received EAPOL-Key frame (4/4 Pairwise)
Fri Jul 21 21:00:28 2017 daemon.info hostapd: wlan0: STA 00:13:e8:d9:8a:27 WPA: pairwise key handshake completed (RSN)
Write failed: Broken pipe
max2344@STA:~$
```

max2344, 2017-07-22:
We'd like to revise our last statement concerning independent disassociation. While it is correct that general communication between the STA and AP stops immediately after 4th step of the pairwise handshake our detailed analysis outlined that the STA still receives the following round of regular EAPOL GTK reqs from the AP and it sends out replies. These replies don't arrive at the AP as outlined by the debug log file of hostap. At this point the AP concludes the STA went down silently and deletes the association.

 Here is the transcript of the pcap dump from the STA to the corresponding log session

```
No.     Time                          Source                Destination           Protocol Info
      1 2017-07-21 20:30:18.092339000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
      2 2017-07-21 20:30:18.094936000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
      3 2017-07-21 20:40:18.092486000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
      4 2017-07-21 20:40:18.095173000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
      5 2017-07-21 20:50:18.093396000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
      6 2017-07-21 20:50:18.096244000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
      7 2017-07-21 21:00:18.093560000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
      8 2017-07-21 21:00:18.096211000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
      9 2017-07-21 21:00:28.893813000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAP      Request, Identity
     10 2017-07-21 21:00:28.893987000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAP      Response, Identity
     11 2017-07-21 21:00:28.897282000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAP      Request, TLS EAP (EAP-TLS)
     12 2017-07-21 21:00:28.897455000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     TLSv1    Client Hello
     13 2017-07-21 21:00:28.902531000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     TLSv1    Server Hello, Change Cipher Spec, Encrypted Handshake Message
     14 2017-07-21 21:00:28.903288000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     TLSv1    Change Cipher Spec, Encrypted Handshake Message
     15 2017-07-21 21:00:28.907125000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAP      Success
     16 2017-07-21 21:00:28.907458000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Message 1 of 4)
     17 2017-07-21 21:00:28.907894000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Message 2 of 4)
     18 2017-07-21 21:00:28.910395000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Message 3 of 4)
     19 2017-07-21 21:00:28.910531000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Message 4 of 4)
     20 2017-07-21 21:10:18.093893000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
     21 2017-07-21 21:10:18.096454000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
     22 2017-07-21 21:10:18.594371000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
     23 2017-07-21 21:10:18.599440000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
     24 2017-07-21 21:10:19.594545000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
     25 2017-07-21 21:10:19.597219000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
     26 2017-07-21 21:10:20.594736000 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
     27 2017-07-21 21:10:20.599320000 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
```
After the packet Nr 19 immediately the STA can't communicate over the existing link. We see lots of ICMP, ND and tcp connection retries for the active sessions.

max2344, 2017-07-22:
Also we want to note that other concurrent associations are unaffected. However we see same behavior. The STAs associate, the sessions run for 60 mins (end of PTK rekey) and after 70 mins (GTK timeout) the STAs start to reconnect the link. In the reports #197 and #192 we recognize the same problem as we describe here.

max2344, 2017-07-23:
Heres the view on the comm from AP point of view. This is from another session but you get the idea. Here we see packets nr 1-12 belonging to the running association of STA about the regular GTK rekey each 600s. Packets 13-23 belong to the PTK exchange. We see that the step 4 exchange in 23 completes. This is the same cue point in the previous log above with packet nr 19. After 600s we see another GTK round (nr 24-27), which remains without reply. However we see from the previous log of STA that the STA sees this and sends replies. From packet 29 onwards the STA noticed that the connection was dropped and reassociates. We see the corresponding exchange.

```
No.     Time                       Source                Destination           Protocol Info
      1 2017-07-23 05:20:18.513945 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
      2 2017-07-23 05:20:18.519861 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
      3 2017-07-23 05:30:18.514314 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
      4 2017-07-23 05:30:18.518711 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
      5 2017-07-23 05:40:18.515182 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
      6 2017-07-23 05:40:18.519540 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
      7 2017-07-23 05:50:18.515474 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
      8 2017-07-23 05:50:18.520152 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
      9 2017-07-23 06:00:18.516200 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
     10 2017-07-23 06:00:18.521369 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
     11 2017-07-23 06:10:18.517306 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
     12 2017-07-23 06:10:18.522044 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Group Message 2 of 2)
     13 2017-07-23 06:13:42.733953 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAP      Request, Identity
     14 2017-07-23 06:13:42.735869 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAP      Response, Identity
     15 2017-07-23 06:13:42.737501 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAP      Request, TLS EAP (EAP-TLS)
     16 2017-07-23 06:13:42.740526 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     TLSv1    Client Hello
     17 2017-07-23 06:13:42.741827 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     TLSv1    Server Hello, Change Cipher Spec, Encrypted Handshake Message
     18 2017-07-23 06:13:42.744168 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     TLSv1    Change Cipher Spec, Encrypted Handshake Message
     19 2017-07-23 06:13:42.745552 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAP      Success
     20 2017-07-23 06:13:42.745675 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Message 1 of 4)
     21 2017-07-23 06:13:42.748782 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Message 2 of 4)
     22 2017-07-23 06:13:42.748972 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Message 3 of 4)
     23 2017-07-23 06:13:42.752245 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Message 4 of 4)
     24 2017-07-23 06:20:18.517919 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
     25 2017-07-23 06:20:19.018543 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
     26 2017-07-23 06:20:20.019596 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
     27 2017-07-23 06:20:21.020654 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Group Message 1 of 2)
     28 2017-07-23 06:20:24.428027 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAP      Request, Identity
     29 2017-07-23 06:22:12.574331 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAP      Request, Identity
     30 2017-07-23 06:22:14.616936 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Start
     31 2017-07-23 06:22:14.617161 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAP      Request, Identity
     32 2017-07-23 06:22:14.618083 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAP      Response, Identity
     33 2017-07-23 06:22:14.619646 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAP      Request, TLS EAP (EAP-TLS)
     34 2017-07-23 06:22:14.621638 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     TLSv1    Client Hello
     35 2017-07-23 06:22:14.625397 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     TLSv1    Server Hello, Certificate, Server Key Exchange, Certificate Request, Server Hello Done
     36 2017-07-23 06:22:14.627418 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAP      Response, TLS EAP (EAP-TLS)
     37 2017-07-23 06:22:14.628595 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     TLSv1    Server Hello, Certificate, Server Key Exchange, Certificate Request, Server Hello Done
     38 2017-07-23 06:22:14.630636 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAP      Response, TLS EAP (EAP-TLS)
     39 2017-07-23 06:22:14.631733 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     TLSv1    Server Hello, Certificate, Server Key Exchange, Certificate Request, Server Hello Done
[... cut ...]
     47 2017-07-23 06:22:14.850419 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAP      Success
     48 2017-07-23 06:22:14.850482 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Message 1 of 4)
     49 2017-07-23 06:22:14.851655 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Message 2 of 4)
     50 2017-07-23 06:22:14.851839 60:38:e0:b9:07:6a     00:13:e8:d9:8a:27     EAPOL    Key (Message 3 of 4)
     51 2017-07-23 06:22:14.852779 00:13:e8:d9:8a:27     60:38:e0:b9:07:6a     EAPOL    Key (Message 4 of 4)
```

mrvltest, 2017-07-24:
1. If this is specific to one client, please check if the client has the same problem with other AP.
2. Under this case, maybe you can disable group rekey.

yuhhaurlin, 2017-07-24:
The fix for EAPOL packet with high priority is done. If you still find group rekey won't get reply from client, you can:
1. Check if this is specific problem with the client and if the client has the same problem with other AP.
2. Disable group rekey to work around it.
3. Use sniffer to get packets on air and send it to me for analysis.
I close this one first. Thanks.

closed 2017-07-24 by yuhhaurlin

max2344, 2017-08-12:
Hi,
we've verified this with a bunch of other access points from mostly tp-link, cisco with OEM and LEDE firmware. PTK and GTK working correctly there. no issues or smth. we will mark this chipset as unsupported due to the issue in this and theother open tickets currently. ATM we're short on time but we'll provide some more infoon chance to test. currently much on the way/out of office so cant test. but the issues been noticed on the openwrt 3200acm page of davidc. also, amsdu disabling indeedfixes the latency problems. but not the rekey of PTK. our 2c on this issue. tyvm.

max2344, 2017-08-12:
also disabling GTK doesnt fix. and theres currently no way to disable PTK… but thats smth for LEDE/openwrt.

yuhhaurlin, 2017-08-12:
I don't know what you mean. There is no problem for PTK and GTK.  Let me know which client you use. If you have problem for all connected clients, you need to check what the problem is.

yuhhaurlin, 2017-08-12:
From the log you sent out I can't find problem related to 4 ways handshake and group rekey. From the log, they are completed without problem.

yuhhaurlin, 2017-08-12:
About the one, no reply for Group rekey sometimes, does it happen for specific client? Please reproduce It with latest driver. If you can reproduce it, I will prepare LEDE image to debug the problem.

yuhhaurlin, 2017-08-12:
I had set group rekey as 60 seconds and tested with Macbook Air (running iPerf test). I did not find problem. However, the test is done with PSK.
I wonder if you can check PSK is all right or not.