kaloz/mwlwifi #196: phy1: buffer is NULL for tx done ring, phy1: adapter does not exist

s-pimenta, 2017-07-16:
Hardware: Linksys WRT3200ACM-EU
Software: davidc502 2017-07-15 (Lede Reboot SNAPSHOT r4576-9fe9175)
Kernel: 4.9.37
Wifi driver: 10.3.4.0-20170713

Fresh flash (from Linksys OEM firmware), and just configured the Wireless interfaces:
5GHz 160MHz, Channel 100, WPA2-PSK (CCMP)

And 2.4GHz 20MHz, Channel 13 (Europe), WPA2-PSK (CCMP)

Then tested to connect various devices to 5GHz and everything fine...

Then switched the same devices from 5GHz to 2.4GHz and cannot connect...

After some troubleshooting... realized the 2.4GHz is gone (no ESSID broadcast)

Then... checked the kernel log.... and I saw this _(Log below)_

.....BUT!!! **_After rebooting_** the router the problem **_is GONE_**! Connecting to both 2.4GHz and 5GHz... everything fine....

I don't know what happened, **I created this issue just for the record.**.. if this problem appears again...

**Log:**
```
[ 1186.217737] ieee80211 phy0: change: 0xffffffff
[ 1186.321951] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 1186.329831] br-lan: port 2(wlan0) entered blocking state
[ 1186.335176] br-lan: port 2(wlan0) entered disabled state
[ 1186.340663] device wlan0 entered promiscuous mode
[ 1186.345861] ieee80211 phy0: change: 0x100
[ 1186.354767] ieee80211 phy0: change: 0x60
[ 1186.567763] ieee80211 phy0: change: 0x40
[ 1186.777744] ieee80211 phy0: change: 0x60
[ 1186.873740] ieee80211 phy0: change: 0x100
[ 1186.882759] ieee80211 phy0: change: 0x100
[ 1186.891775] ieee80211 phy0: change: 0x62
[ 1186.987724] ieee80211 phy0: change: 0x40
[ 1247.197020] ieee80211 phy0: change: 0x40
[ 1247.295043] ieee80211 phy0: change: 0x100
[ 1247.350464] ieee80211 phy0: change: 0x100
[ 1247.359485] ieee80211 phy0: change: 0x42
[ 1247.454010] ieee80211 phy0: change: 0x40
[ 1247.632019] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1247.638469] br-lan: port 2(wlan0) entered blocking state
[ 1247.643803] br-lan: port 2(wlan0) entered forwarding state
[ 1253.581948] ieee80211 phy1: change: 0xffffffff
[ 1253.669193] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[ 1253.675949] br-lan: port 3(wlan1) entered blocking state
[ 1253.681327] br-lan: port 3(wlan1) entered disabled state
[ 1253.686801] device wlan1 entered promiscuous mode
[ 1253.741769] ieee80211 phy1: change: 0x100
[ 1253.750804] ieee80211 phy1: change: 0x42
[ 1253.940361] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[ 1253.946808] br-lan: port 3(wlan1) entered blocking state
[ 1253.952154] br-lan: port 3(wlan1) entered forwarding state
[ 1513.253479] ieee80211 phy1: buffer is NULL for tx done ring
[ 1513.333477] ieee80211 phy1: buffer is NULL for tx done ring
[ 1513.493474] ieee80211 phy1: buffer is NULL for tx done ring
[ 1513.573472] ieee80211 phy1: buffer is NULL for tx done ring
[ 1513.723475] ieee80211 phy1: buffer is NULL for tx done ring
[ 1513.783468] ieee80211 phy1: buffer is NULL for tx done ring
[ 1513.923468] ieee80211 phy1: buffer is NULL for tx done ring
[ 1514.003467] ieee80211 phy1: buffer is NULL for tx done ring
[ 1514.203487] ieee80211 phy1: buffer is NULL for tx done ring
[ 1514.263466] ieee80211 phy1: buffer is NULL for tx done ring
[ 1514.403473] ieee80211 phy1: buffer is NULL for tx done ring
[ 1514.473460] ieee80211 phy1: buffer is NULL for tx done ring
[ 1514.493003] ieee80211 phy1: adapter does not exist
[ 1514.498173] ieee80211 phy1: adapter does not exist
[ 1514.503331] ieee80211 phy1: adapter does not exist
[ 1514.508167] ieee80211 phy1: adapter does not exist
[ 1514.512979] ieee80211 phy1: adapter does not exist
[ 1514.517793] ieee80211 phy1: adapter does not exist
[ 1514.524117] ieee80211 phy1: adapter does not exist
[ 1514.529103] ieee80211 phy1: adapter does not exist
[ 1514.534101] ieee80211 phy1: adapter does not exist
[ 1514.538914] ieee80211 phy1: adapter does not exist
[ 1514.543742] ieee80211 phy1: adapter does not exist
[ 1514.548551] ieee80211 phy1: adapter does not exist
[ 1514.653457] ieee80211 phy1: buffer is NULL for tx done ring
[ 1514.713454] ieee80211 phy1: buffer is NULL for tx done ring
[ 1514.883451] ieee80211 phy1: buffer is NULL for tx done ring
[ 1514.953449] ieee80211 phy1: buffer is NULL for tx done ring
[ 1515.113447] ieee80211 phy1: buffer is NULL for tx done ring
[ 1515.173444] ieee80211 phy1: buffer is NULL for tx done ring
[ 1515.323445] ieee80211 phy1: buffer is NULL for tx done ring
[ 1515.393442] ieee80211 phy1: buffer is NULL for tx done ring
[ 1515.543441] ieee80211 phy1: buffer is NULL for tx done ring
[ 1515.603438] ieee80211 phy1: buffer is NULL for tx done ring
[ 1515.743438] ieee80211 phy1: buffer is NULL for tx done ring
[ 1515.803436] ieee80211 phy1: buffer is NULL for tx done ring
[ 1515.953435] ieee80211 phy1: buffer is NULL for tx done ring
[ 1516.023433] ieee80211 phy1: buffer is NULL for tx done ring
[ 1516.163432] ieee80211 phy1: buffer is NULL for tx done ring
[ 1516.243429] ieee80211 phy1: buffer is NULL for tx done ring
[ 1516.403424] ieee80211 phy1: buffer is NULL for tx done ring
[ 1516.473428] ieee80211 phy1: buffer is NULL for tx done ring
[ 1516.653421] ieee80211 phy1: buffer is NULL for tx done ring
[ 1516.713418] ieee80211 phy1: buffer is NULL for tx done ring
[ 1516.892549] ieee80211 phy0: staid 1 deleted
[ 1516.893415] ieee80211 phy1: buffer is NULL for tx done ring
[ 1516.943456] ieee80211 phy1: buffer is NULL for tx done ring
[ 1517.083414] ieee80211 phy1: buffer is NULL for tx done ring
[ 1517.143410] ieee80211 phy1: buffer is NULL for tx done ring
[ 1517.313412] ieee80211 phy1: buffer is NULL for tx done ring
[ 1517.373406] ieee80211 phy1: buffer is NULL for tx done ring
[ 1517.523405] ieee80211 phy1: buffer is NULL for tx done ring
[ 1517.603410] ieee80211 phy1: buffer is NULL for tx done ring
[ 1517.753430] ieee80211 phy1: buffer is NULL for tx done ring
[ 1517.833402] ieee80211 phy1: buffer is NULL for tx done ring
[ 1518.003399] ieee80211 phy1: buffer is NULL for tx done ring
[ 1518.073394] ieee80211 phy1: buffer is NULL for tx done ring
[ 1518.223397] ieee80211 phy1: buffer is NULL for tx done ring
[ 1518.293392] ieee80211 phy1: buffer is NULL for tx done ring
[ 1518.443392] ieee80211 phy1: buffer is NULL for tx done ring
[ 1518.503388] ieee80211 phy1: buffer is NULL for tx done ring
[ 1518.653386] ieee80211 phy1: buffer is NULL for tx done ring
[ 1518.713424] ieee80211 phy1: buffer is NULL for tx done ring
[ 1518.853384] ieee80211 phy1: buffer is NULL for tx done ring
[ 1518.923381] ieee80211 phy1: buffer is NULL for tx done ring
[ 1519.093382] ieee80211 phy1: buffer is NULL for tx done ring
[ 1519.163379] ieee80211 phy1: buffer is NULL for tx done ring
[ 1519.313381] ieee80211 phy1: buffer is NULL for tx done ring
[ 1519.373395] ieee80211 phy1: buffer is NULL for tx done ring
[ 1519.456159] ieee80211 phy1: adapter does not exist
[ 1519.460974] ieee80211 phy1: adapter does not exist
[ 1519.465807] ieee80211 phy1: adapter does not exist
[ 1519.470620] ieee80211 phy1: adapter does not exist
[ 1519.475437] ieee80211 phy1: adapter does not exist
[ 1519.480246] ieee80211 phy1: adapter does not exist
[ 1519.485343] ieee80211 phy1: adapter does not exist
[ 1519.490198] ieee80211 phy1: adapter does not exist
[ 1519.495048] ieee80211 phy1: adapter does not exist
[ 1519.499905] ieee80211 phy1: adapter does not exist
[ 1519.504752] ieee80211 phy1: adapter does not exist
[ 1519.509562] ieee80211 phy1: adapter does not exist
[ 1519.523387] ieee80211 phy1: buffer is NULL for tx done ring
[ 1519.603371] ieee80211 phy1: buffer is NULL for tx done ring
[ 1519.763372] ieee80211 phy1: buffer is NULL for tx done ring
[ 1519.823370] ieee80211 phy1: buffer is NULL for tx done ring
[ 1519.983364] ieee80211 phy1: buffer is NULL for tx done ring
[ 1520.053363] ieee80211 phy1: buffer is NULL for tx done ring
[ 1520.213362] ieee80211 phy1: buffer is NULL for tx done ring
[ 1520.283360] ieee80211 phy1: buffer is NULL for tx done ring
[ 1520.433358] ieee80211 phy1: buffer is NULL for tx done ring
```

yuhhaurlin, 2017-07-16:
The log shows driver has problem to access registers of phy1 (2.4 ghz) at that time. However it is all right after reboot. I close this one first, you can check if it will happen again.

closed 2017-07-16 by yuhhaurlin

s-pimenta, 2017-07-17:
Unfortunately the problem appeared again... don't know what is triggering this issue...

First the 5GHz (phy0) disapeared....

And 5 min later the 2.4GHz (phy1) also disapeared...

Log:
```
Mon Jul 17 13:38:28 2017 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED
Mon Jul 17 13:38:28 2017 kern.info kernel: [ 4712.462900] ieee80211 phy0: staid 3 deleted
Mon Jul 17 13:39:24 2017 kern.err kernel: [ 4767.788387] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:24 2017 kern.err kernel: [ 4767.899629] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:24 2017 kern.err kernel: [ 4767.959283] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:24 2017 kern.err kernel: [ 4768.019283] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:24 2017 kern.err kernel: [ 4768.079265] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:24 2017 kern.err kernel: [ 4768.139261] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:24 2017 kern.err kernel: [ 4768.259261] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:24 2017 kern.err kernel: [ 4768.379260] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:24 2017 kern.err kernel: [ 4768.439257] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.559257] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.619256] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.739283] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.849266] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.897477] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.902367] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.907187] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.912021] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.916838] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.921689] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.927104] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.931962] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.936776] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.941607] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.946423] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.951246] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4768.959268] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4769.019259] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4769.189262] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4769.299598] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:25 2017 kern.err kernel: [ 4769.419585] ieee80211 phy0: buffer is NULL for tx done ring
```
After 5GHz disappeared the device automatically connects to the 2.4GHz

```
Mon Jul 17 13:39:42 2017 kern.err kernel: [ 4785.499432] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:42 2017 kern.err kernel: [ 4785.609113] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:42 2017 kern.err kernel: [ 4785.719112] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:42 2017 daemon.info hostapd: wlan1: STA ff:f0:05:08:ad:ff IEEE 802.11: associated (aid 1)
Mon Jul 17 13:39:42 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 
Mon Jul 17 13:39:42 2017 daemon.info hostapd: wlan1: STA  WPA: pairwise key handshake completed (RSN)
Mon Jul 17 13:39:42 2017 kern.err kernel: [ 4785.779111] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:42 2017 daemon.info dnsmasq-dhcp[3467]: DHCPDISCOVER(br-lan) 192.168.1.115 
Mon Jul 17 13:39:42 2017 daemon.info dnsmasq-dhcp[3467]: DHCPOFFER(br-lan) 192.168.1.115 
Mon Jul 17 13:39:42 2017 daemon.info dnsmasq-dhcp[3467]: DHCPREQUEST(br-lan) 192.168.1.115 
Mon Jul 17 13:39:42 2017 daemon.info dnsmasq-dhcp[3467]: DHCPACK(br-lan) 192.168.1.115
Mon Jul 17 13:39:42 2017 kern.err kernel: [ 4785.830051] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:42 2017 kern.err kernel: [ 4785.839115] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:42 2017 kern.err kernel: [ 4785.949110] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:42 2017 kern.err kernel: [ 4786.049111] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:42 2017 kern.err kernel: [ 4786.159109] ieee80211 phy0: buffer is NULL for tx done ring
Mon Jul 17 13:39:42 2017 kern.err kernel: [ 4786.269107] ieee80211 phy0: buffer is NULL for tx done ring
```
Then the 2.4GHz died too
```
Mon Jul 17 13:51:06 2017 kern.err kernel: [ 5469.555386] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:51:06 2017 kern.err kernel: [ 5469.560195] ieee80211 phy0: adapter does not exist
Mon Jul 17 13:51:06 2017 kern.err kernel: [ 5469.565033] ieee80211 phy0: adapter does not exist
[... cut ...]
[ 6115.247869] ieee80211 phy1: buffer is NULL for tx done ring
[ 6115.307869] ieee80211 phy1: buffer is NULL for tx done ring
[ 6115.367868] ieee80211 phy0: buffer is NULL for tx done ring
[ 6115.367871] ieee80211 phy1: buffer is NULL for tx done ring
[ 6115.368132] mmc0: card 0001 removed
[ 6115.427871] ieee80211 phy1: buffer is NULL for tx done ring
[ 6115.487868] ieee80211 phy1: buffer is NULL for tx done ring
[ 6115.547868] ieee80211 phy1: buffer is NULL for tx done ring
[ 6115.547870] ieee80211 phy0: buffer is NULL for tx done ring
```

yuhhaurlin, 2017-07-17:
I suspect that your device has problem. I have no idea how to check it.

yuhhaurlin, 2017-07-17:
From the log, it looks like SDIO has problem too!

s-pimenta, 2017-07-17:
Last time happened the problem, I was changing the behaviour of the LED (ON LED, Wan, 2.4GHz LED, 5GHz LED, WPS LED)