kaloz/mwlwifi #197: Crash during streaming audio

jacksonatl, 2017-07-17:
This may be similar to #193 but using 20170713 driver. 

cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info
driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170713
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: <>
2g: disable
5g: enable
antenna: 4 4
irq number: 105
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0e00000
iobase1: e1080000
tx limit: 1024
rx limit: 16384

logread |grep hostapd | sed -E  "s/[0-9a-fA-F:]{17}/11:22:33:44:55:66/" 
Mon Jul 17 15:14:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:14:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:14:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:14:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:14:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:14:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:14:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:14:12 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:24:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:24:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:24:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:24:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:24:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:24:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:24:11 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:24:12 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:34:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:34:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:34:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:34:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:34:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:34:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:34:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:34:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:44:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:44:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:44:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:44:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:44:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:44:10 2017 daemon.err hostapd: nl80211: EAPOL TX: No message of desired type
Mon Jul 17 15:44:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:44:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:44:12 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:54:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:54:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:54:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:54:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:54:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:54:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:54:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 15:54:11 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 16:04:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 16:04:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 16:04:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
Mon Jul 17 16:04:10 2017 daemon.info hostapd: wlan0: STA 11:22:33:44:55:66 WPA: group key handshake completed (RSN)
[... 467 lines cut ...]
[   15.000891] br-lan: port 1(eth0.1) entered forwarding state
[   15.006515] br-lan: port 1(eth0.1) entered forwarding state
[   15.020833] mvneta f1070000.ethernet eth1: configuring for fixed link mode
[   15.028127] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[   15.034324] mvneta f1070000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[   15.043161] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[   15.949046] ieee80211 phy0: change: 0xffffffff
[   16.091682] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   16.098663] device wlan0 entered promiscuous mode
[   16.104480] ieee80211 phy0: change: 0x100
[   16.117951] ieee80211 phy0: change: 0x60
[   16.366198] ieee80211 phy0: change: 0x40
[   16.581409] ieee80211 phy0: change: 0x60
[   16.683518] ieee80211 phy0: change: 0x100
[   16.759321] ieee80211 phy0: change: 0x100
[   16.780059] ieee80211 phy0: change: 0x62
[   16.983687] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   16.990122] br-lan: port 2(wlan0) entered forwarding state
[   16.995663] br-lan: port 2(wlan0) entered forwarding state
[   17.001269] br-lan: port 1(eth0.1) entered forwarding state
[   18.991227] br-lan: port 2(wlan0) entered forwarding state
[   42.801438] ieee80211 phy0: staid 4 deleted
[ 2266.190370] ieee80211 phy0: staid 2 deleted
[ 2295.533006] ieee80211 phy0: staid 2 deleted
[ 3866.769359] ieee80211 phy0: staid 6 deleted

yuhhaurlin, 2017-07-17:
I can't see log related to mwlwifi. Did you get kernel OOPS?

yuhhaurlin, 2017-07-17:
BTW, why did you use multicast MAC address for your client?

jacksonatl, 2017-07-17:
No, no kernel oops but will keep trying to replicate. MAC address were changed via sed script.

yuhhaurlin, 2017-07-18:
Thanks. I will also do stress test to see if I will encounter problem.

yuhhaurlin, 2017-07-31:
Close this one and use #198 to check related problems.

closed 2017-07-31 by yuhhaurlin