kaloz/mwlwifi #198: Router reboot

s-pimenta, 2017-07-29:
After leaving the OEM kernel again, installed the LEDE image with the latest wifi driver and the same issue happened again...

I will try to give all possible information as I can...

Router: Linksys WRT3200ACM-EU
Image: Lede Reboot SNAPSHOT r4576-9fe9175 (Davidc502c)
Linux: 4.9.37
Wifi driver version: 10.3.4.0-20170727

Wifi config:
5GHz (channel 100 VHT 80MHz)
2.4GHz (Channel 13 40MHz)
Security: WPA2 CCMP

I have an Android tablet (able to connect to 5GHz and 2.4GHz 802.11N) LG G Pad 8.3 Android 5.1.1 (MAC ADDRESS: 8c:3a:e3:56:71:77)

This issue never happened with Wifi driver version: 10.3.4.0-20170606 with this particular Android tablet

But with this new wifi driver 10.3.4.0-20170713 and 10.3.4.0-20170727 this problem appeared.

The tablet was an uptime about 28 days... and I was only able to connect to 2.4GHz (and work normally), when I tried to connect to 5GHz (android says wifi connection problem), other devices can connect to 5GHz and 2.4GHz with no problem...

After rebooting the tablet (showing the lockscreen), without even touching anything... the router rebooted... Maybe the tablet was automatically to connect to the router (don't know if was trying to connect to 5GHz or 2.4GHz)

And since I was saving the logs to and USB thumb drive I have the following log (maybe not very helpfull but this is what was saved in the file...) Maybe someone can help me to have an more verbose log?!?!

```
Fri Jul 28 14:56:43 2017 daemon.info hostapd: wlan0: STA c0:ee:fb:20:60:c0 IEEE 802.11: authenticated
Fri Jul 28 14:56:46 2017 daemon.notice hostapd: handle_auth_cb: STA c0:ee:fb:20:60:c0 not found
Fri Jul 28 15:02:46 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 WPA: group key handshake completed (RSN)
Fri Jul 28 15:05:03 2017 daemon.info hostapd: wlan0: STA c0:ee:fb:20:60:c0 WPA: group key handshake completed (RSN)
Fri Jul 28 15:07:26 2017 kern.debug kernel: [ 5213.364720] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:26 2017 kern.debug kernel: [ 5213.512107] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:27 2017 kern.debug kernel: [ 5213.662094] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:27 2017 kern.debug kernel: [ 5213.812099] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:27 2017 kern.debug kernel: [ 5213.962089] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:27 2017 kern.debug kernel: [ 5214.112128] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:27 2017 kern.debug kernel: [ 5214.262108] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:27 2017 kern.debug kernel: [ 5214.422102] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:27 2017 kern.debug kernel: [ 5214.582082] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:28 2017 kern.debug kernel: [ 5214.732076] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:28 2017 kern.debug kernel: [ 5214.882079] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:28 2017 kern.debug kernel: [ 5215.032077] ieee80211 phy1: change: 0x40
Fri Jul 28 15:07:28 2017 kern.debug kernel: [ 5215.192071] ieee80211 phy1: change: 0x42
Fri Jul 28 15:09:31 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 8c:3a:e3:56:71:77
Fri Jul 28 15:09:31 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 IEEE 802.11: disassociated
Fri Jul 28 15:09:31 2017 kern.info kernel: [ 5338.597510] ieee80211 phy1: staid 1 deleted
Fri Jul 28 15:09:32 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jul 28 15:10:10 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 IEEE 802.11: associated (aid 1)
Fri Jul 28 15:10:10 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 8c:3a:e3:56:71:77
Fri Jul 28 15:10:10 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 WPA: pairwise key handshake completed (RSN)
Fri Jul 28 15:10:11 2017 daemon.info dnsmasq-dhcp[3867]: DHCPREQUEST(br-lan) 192.168.1.183 8c:3a:e3:56:71:77 
Fri Jul 28 15:10:11 2017 daemon.info dnsmasq-dhcp[3867]: DHCPACK(br-lan) 192.168.1.183 8c:3a:e3:56:71:77 android-lgv500-spimenta
Fri Jul 28 15:10:14 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 IEEE 802.11: authenticated
Fri Jul 28 15:10:50 2017 daemon.notice netifd: wan (1863): udhcpc: sending renew
Fri Jul 28 15:10:50 2017 daemon.notice netifd: wan (1863): udhcpc: lease of 2.80.197.105 obtained, lease time 3600
Fri Jul 28 15:12:45 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 WPA: group key handshake completed (RSN)
Fri Jul 28 15:15:03 2017 daemon.info hostapd: wlan0: STA c0:ee:fb:20:60:c0 WPA: group key handshake completed (RSN)
Fri Jul 28 15:19:27 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK 8c:3a:e3:56:71:77
Fri Jul 28 15:22:44 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 8c:3a:e3:56:71:77
Fri Jul 28 15:22:44 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 IEEE 802.11: disassociated
Fri Jul 28 15:22:44 2017 kern.info kernel: [ 6131.561992] ieee80211 phy1: staid 1 deleted
Fri Jul 28 15:22:45 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jul 28 15:22:48 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 IEEE 802.11: associated (aid 1)
Fri Jul 28 15:22:48 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 8c:3a:e3:56:71:77
Fri Jul 28 15:22:48 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 WPA: pairwise key handshake completed (RSN)
Fri Jul 28 15:22:48 2017 daemon.info dnsmasq-dhcp[3867]: DHCPREQUEST(br-lan) 192.168.1.183 8c:3a:e3:56:71:77 
Fri Jul 28 15:22:48 2017 daemon.info dnsmasq-dhcp[3867]: DHCPACK(br-lan) 192.168.1.183 8c:3a:e3:56:71:77 android-lgv500-spimenta
Fri Jul 28 15:22:50 2017 daemon.info hostapd: wlan1: STA 8c:3a:e3:56:71:77 IEEE 802.11: authenticated
Fri Jul 28 15:25:03 2017 daemon.info hostapd: wlan0: STA c0:ee:fb:20:60:c0 WPA: group key handshake completed (RSN)
[... 157 lines cut ...]
Fri Jul 28 21:59:11 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 WPA: group key handshake completed (RSN)
Fri Jul 28 21:59:11 2017 daemon.info hostapd: wlan0: STA c0:ee:fb:20:60:c0 WPA: group key handshake completed (RSN)
Fri Jul 28 22:00:11 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED f8:32:e4:0f:6c:c9
Fri Jul 28 22:00:11 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 IEEE 802.11: disassociated
Fri Jul 28 22:00:11 2017 kern.info kernel: [19340.435958] ieee80211 phy1: staid 2 deleted
Fri Jul 28 22:00:12 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jul 28 22:00:14 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 IEEE 802.11: associated (aid 1)
Fri Jul 28 22:00:14 2017 daemon.notice hostapd: wlan1: AP-STA-CONNECTED f8:32:e4:0f:6c:c9
Fri Jul 28 22:00:14 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 WPA: pairwise key handshake completed (RSN)
Fri Jul 28 22:00:14 2017 daemon.info dnsmasq-dhcp[3421]: DHCPREQUEST(br-lan) 192.168.1.182 f8:32:e4:0f:6c:c9 
Fri Jul 28 22:00:14 2017 daemon.info dnsmasq-dhcp[3421]: DHCPACK(br-lan) 192.168.1.182 f8:32:e4:0f:6c:c9 android-b430e326f06bf6a7
Fri Jul 28 22:00:16 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 IEEE 802.11: authenticated
Fri Jul 28 22:01:56 2017 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED f8:32:e4:0f:6c:c9
Fri Jul 28 22:01:56 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 IEEE 802.11: disassociated
Fri Jul 28 22:01:56 2017 kern.info kernel: [19445.149086] ieee80211 phy1: staid 1 deleted
Fri Jul 28 22:01:57 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jul 28 22:02:17 2017 user.warn igmpproxy[3014]: MRT_DEL_MFC; Errno(2): No such file or directory
```
(the router stopped working 30min before... so the time between newtork lockup and the reboot nothing was recorded to USB thumb drive)

Using: **logread -f >> /mnt/usb/wrt3200acm-log.txt**

**Maybe someone can help me to have an more verbose log?!?!**

I don't know if this issue is only LEDE related because I did not try DD-WRT yet...

BrainSlayer, 2017-07-29:
the only way here is to attach a serial console to log kernel crashes. that might help. since i'm using dd-wrt (and developing it) i can just say that its rock stable right now for days. but that doesnt mean anything since the router isnt used that much in by wifi clients. its just a test device

s-pimenta, 2017-07-29:
@BrainSlayer But a serial console requires to open the router? (voiding the warranty?)

aaron1959, 2017-07-29:
I have an Open one (WRT3200ACM) and can reproduce that issue I also have the hardware for serial connection and have successfully connected that way can you tell me how to log kernel that way? I know virtually nothing about Linux but understand the basic syntax.
This Hardware board is marked 2016.05.18   and 48SAM203.SGD This was obtained in the U.S.
I'd be happy to try to log when it "locks" up.

s-pimenta, 2017-07-29:
@BrainSlayer Do you have the DD-WRT with the lastest commit? I want to try if this problem also occurs on DD-WRT

BrainSlayer, 2017-07-29:
yes .dd-wrt uses the latest commit. you can download the current beta from the ftp server

BrainSlayer, 2017-07-29:
or just go to other download on our website

BrainSlayer, 2017-07-29:
@aaron1959 : if you attach the serial you will see all crashes before the kernel reboots which isnt possible with a shell login since crashes stop networking as well

aaron1959, 2017-07-30:
So I log in on serial with putty in my case and once I'm on the DD-WRT command prompt do I just let it sit there or is there some command needed to get it to start outputting any data? Perhaps some script?

s-pimenta, 2017-07-30:
The problem happened again...
Leaved the router turn on overnight and this morning pickup my tablet (the only device I have that causes the problem), trying to connect to the router... and locks up immediately...

Loging to the USB thumb drive, again not very helpfull...

```
Sun Jul 30 06:58:34 2017 daemon.info dnscrypt-proxy[1778]: dnscrypt-proxy Refetching server certificates
Sun Jul 30 06:58:49 2017 daemon.err dnscrypt-proxy[1778]: dnscrypt-proxy Unable to retrieve server certificates
Sun Jul 30 06:59:39 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK 2c:59:8a:63:08:34
Sun Jul 30 06:59:52 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 WPA: group key handshake completed (RSN)
Sun Jul 30 06:59:53 2017 daemon.info hostapd: wlan1: STA 2c:59:8a:63:08:34 WPA: group key handshake completed (RSN)
Sun Jul 30 06:59:55 2017 daemon.info hostapd: wlan1: STA c4:b3:01:03:9d:66 WPA: group key handshake completed (RSN)
Sun Jul 30 07:01:30 2017 daemon.info hostapd: wlan0: STA 30:c7:ae:02:a8:32 IEEE 802.11: associated (aid 1)
Sun Jul 30 07:01:30 2017 kern.info kernel: [46914.114143] ieee80211 phy0: staid 1 deleted
Sun Jul 30 07:02:07 2017 daemon.notice netifd: wan (1631): udhcpc: sending select for 2.80.197.105
Sun Jul 30 07:02:07 2017 daemon.notice netifd: wan (1631): udhcpc: lease of 2.80.197.105 obtained, lease time 3600
Sun Jul 30 07:02:07 2017 daemon.notice netifd: Interface 'wan' is now up
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: reading /tmp/resolv.conf.auto
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using local addresses only for domain test
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using local addresses only for domain onion
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using local addresses only for domain localhost
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using local addresses only for domain local
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using local addresses only for domain invalid
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using local addresses only for domain example.net
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using local addresses only for domain example.org
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using local addresses only for domain example.com
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using nameserver 212.55.154.174#53
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using nameserver 212.55.154.190#53
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using nameserver 2001:8a0:2106:5:212:55:154:174#53
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using nameserver 2001:8a0:2206:5:212:55:154:190#53
Sun Jul 30 07:02:07 2017 daemon.info dnsmasq[6858]: using 3 more local addresses
Sun Jul 30 07:02:07 2017 user.notice firewall: Reloading firewall due to ifup of wan (eth1.2)
Sun Jul 30 07:02:07 2017 user.notice ddns-scripts[7167]: myddns_ipv4: PID '7167' started at 2017-07-30 07:02
Sun Jul 30 07:02:07 2017 user.warn ddns-scripts[7167]: myddns_ipv4: Service section disabled! - TERMINATE
Sun Jul 30 07:02:07 2017 user.warn ddns-scripts[7167]: myddns_ipv4: PID '7167' exit WITH ERROR '1' at 2017-07-30 07:02
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[6858]: exiting on receipt of SIGTERM
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: started, version 2.77 cachesize 150
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: DNS service limited to local subnets
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-DNSSEC no-ID loop-detect inotify
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq-dhcp[7341]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain test
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain onion
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain localhost
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain local
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain invalid
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain example.net
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain example.org
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain example.com
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using 3 more local addresses
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: reading /tmp/resolv.conf.auto
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain test
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain onion
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain localhost
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain local
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain invalid
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain example.net
Sun Jul 30 07:02:09 2017 daemon.info dnsmasq[7341]: using local addresses only for domain example.org
[... cut ...]
Sun Jul 30 07:09:53 2017 daemon.info hostapd: wlan1: STA 2c:59:8a:63:08:34 WPA: group key handshake completed (RSN)
Sun Jul 30 07:09:53 2017 daemon.info hostapd: wlan1: STA c4:b3:01:03:9d:66 WPA: group key handshake completed (RSN)
Sun Jul 30 07:10:57 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 WPA: group key handshake completed (RSN)
```

Bought my Linksys WRT3200ACM since March, 4 months passed and still not a stable driver.... I've being frustrated of always debugging an expensive router and not having a reliable router...

Anyway.... I will now try DD-WRT if the problem happens too...

BrainSlayer, 2017-07-30:
the kernel logs always to serial. you dont need todo anything. just wait and watch the messages. but this is clearly obvious since you have seen all the kernel log messages already at booting

aaron1959, 2017-07-30:
Thank you! Yes I'm not very sharp lately.  I assumed that I would get something, so I'll just leave a laptop plugged into the serial and just let it sit there in the future. I'm much more comfortable with working with the hardware, I'm an old school Electrical Engineer from university days.

yuhhaurlin, 2017-07-31:
I will try to check if I can reproduce it.

wongsyrone, 2017-07-31:
I got these from my router

```
Mon Jul 31 11:30:59 2017 kern.err kernel: [139524.908468] ieee80211 phy1: cmd 0x8050=broadcast_ssid_enable timed out
Mon Jul 31 11:30:59 2017 kern.err kernel: [139524.915114] ieee80211 phy1: return code: 0x0050
Mon Jul 31 11:30:59 2017 kern.err kernel: [139524.919760] ieee80211 phy1: timeout: 0x0050
Mon Jul 31 11:31:03 2017 kern.err kernel: [139528.924317] ieee80211 phy1: cmd 0x8127=SetInformationElements timed out
Mon Jul 31 11:31:03 2017 kern.err kernel: [139528.931057] ieee80211 phy1: return code: 0x0127
Mon Jul 31 11:31:03 2017 kern.err kernel: [139528.935692] ieee80211 phy1: timeout: 0x0127
Mon Jul 31 11:31:08 2017 kern.err kernel: [139532.942380] ieee80211 phy1: cmd 0x801c=80211RadioControl timed out
Mon Jul 31 11:31:08 2017 kern.err kernel: [139532.948718] ieee80211 phy1: return code: 0x001c
Mon Jul 31 11:31:08 2017 kern.err kernel: [139532.953358] ieee80211 phy1: timeout: 0x001c
Mon Jul 31 11:31:12 2017 kern.err kernel: [139536.958237] ieee80211 phy1: cmd 0x8126=SetFixedRate timed out
Mon Jul 31 11:31:12 2017 kern.err kernel: [139536.964105] ieee80211 phy1: return code: 0x0126
Mon Jul 31 11:31:12 2017 kern.err kernel: [139536.968748] ieee80211 phy1: timeout: 0x0126
Mon Jul 31 11:31:16 2017 kern.err kernel: [139540.976233] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Mon Jul 31 11:31:16 2017 kern.err kernel: [139540.982188] ieee80211 phy1: return code: 0x001d
Mon Jul 31 11:31:16 2017 kern.err kernel: [139540.986823] ieee80211 phy1: timeout: 0x001d
Mon Jul 31 11:31:20 2017 kern.err kernel: [139544.990275] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Mon Jul 31 11:31:20 2017 kern.err kernel: [139544.996223] ieee80211 phy1: return code: 0x001d
Mon Jul 31 11:31:20 2017 kern.err kernel: [139545.000873] ieee80211 phy1: timeout: 0x001d
Mon Jul 31 11:31:24 2017 kern.err kernel: [139549.004198] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Mon Jul 31 11:31:24 2017 kern.err kernel: [139549.010159] ieee80211 phy1: return code: 0x001d
Mon Jul 31 11:31:24 2017 kern.err kernel: [139549.014800] ieee80211 phy1: timeout: 0x001d
Mon Jul 31 11:31:28 2017 kern.err kernel: [139553.020148] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Mon Jul 31 11:31:28 2017 kern.err kernel: [139553.026097] ieee80211 phy1: return code: 0x001d
Mon Jul 31 11:31:28 2017 kern.err kernel: [139553.030746] ieee80211 phy1: timeout: 0x001d
Mon Jul 31 11:31:32 2017 kern.err kernel: [139557.034108] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Mon Jul 31 11:31:32 2017 kern.err kernel: [139557.040064] ieee80211 phy1: return code: 0x001d
Mon Jul 31 11:31:32 2017 kern.err kernel: [139557.044700] ieee80211 phy1: timeout: 0x001d
Mon Jul 31 11:31:36 2017 kern.err kernel: [139561.048054] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Mon Jul 31 11:31:36 2017 kern.err kernel: [139561.054036] ieee80211 phy1: return code: 0x001d
Mon Jul 31 11:31:36 2017 kern.err kernel: [139561.058688] ieee80211 phy1: timeout: 0x001d
Mon Jul 31 11:31:40 2017 kern.err kernel: [139565.068009] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Mon Jul 31 11:31:40 2017 kern.err kernel: [139565.073964] ieee80211 phy1: return code: 0x001d
Mon Jul 31 11:31:40 2017 kern.err kernel: [139565.078615] ieee80211 phy1: timeout: 0x001d
```

yuhhaurlin, 2017-07-31:
For 88W8964?

wongsyrone, 2017-07-31:
WRT3200ACM. It looks all mac80211 commands timed out. I have to execute `reboot`, and all looks good after rebooting. The same issue appears randomly.

yuhhaurlin, 2017-07-31:
I will try to reproduce the problem first.

yuhhaurlin, 2017-07-31:
BTW, is it all right for 10.3.4.0-20170606?

wongsyrone, 2017-07-31:
Thank you in advance.

I don't know whether it's related to compat-wireless(mac80211) package in OpenWrt/LEDE or not.
It looks like https://github.com/lede-project/source/tree/master/package/kernel/mac80211 hasn't been updated for a long time.

> is it all right for 10.3.4.0-20170606

It works well before, but I'm not sure since both mwlwifi and kernel got updated. It's really difficult to tell which part went wrong.

yuhhaurlin, 2017-07-31:
You can follow the way described in README to replace your mwlwifi driver on your current LEDE image.

wongsyrone, 2017-07-31:
I will try 10.3.4.0-20170606 and report back.

yuhhaurlin, 2017-07-31:
Thanks. I try to do stress test on 10.3.4.0-20170713 now.

s-pimenta, 2017-08-01:
I can confim that this problem also happen 2 times in DD-WRT

This last time a simple check in the tablet (turn on the screen), and because the wifi is suspended when the tablet is in standby, and when I turn on the screen the tablet try to connect to the router and instantly the router **locks up everything** (other wifi devices, and ethernet wired devices)

If I wait (about 30min to 1 hour after the lockup) the router reboots by itself...

Syslog of the tablet:
```
08-01 13:14:53.179   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD

08-01 13:14:53.271   856   856 I wpa_supplicant: wlan0: Trying to associate with SSID 'dd-wrt 5GHz'

08-01 13:14:54.395   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=60:38:e0:b9:0b:2a status_code=1

08-01 13:14:56.792   856   856 I wpa_supplicant: wlan0: Trying to associate with SSID 'dd-wrt 5GHz'

08-01 13:14:57.878   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=60:38:e0:b9:0b:2a status_code=1

08-01 13:15:00.790   856   856 I wpa_supplicant: wlan0: Trying to associate with SSID 'dd-wrt 5GHz'

08-01 13:15:01.865   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=60:38:e0:b9:0b:2a status_code=1

08-01 13:15:05.375   856   856 I wpa_supplicant: wlan0: Trying to associate with SSID 'dd-wrt 5GHz'

08-01 13:15:06.477   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=60:38:e0:b9:0b:2a status_code=1

08-01 13:15:06.477   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=40 ssid="dd-wrt 5GHz" auth_failures=1 duration=10 reason=CONN_FAILED

08-01 13:15:14.022   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-SSID-REENABLED id=40 ssid="dd-wrt 5GHz"

08-01 13:15:14.023   856   856 I wpa_supplicant: wlan0: Trying to associate with SSID 'dd-wrt 5GHz'

08-01 13:15:15.075   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=60:38:e0:b9:0b:2a status_code=1

08-01 13:15:15.076   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=40 ssid="dd-wrt 5GHz" auth_failures=1 duration=10 reason=CONN_FAILED

08-01 13:15:27.572   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-SSID-REENABLED id=40 ssid="dd-wrt 5GHz"

08-01 13:15:27.591   856   856 I wpa_supplicant: wlan0: Trying to associate with SSID 'dd-wrt 5GHz'

08-01 13:15:28.697   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=60:38:e0:b9:0b:2a status_code=1

08-01 13:15:28.697   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=40 ssid="dd-wrt 5GHz" auth_failures=1 duration=10 reason=CONN_FAILED

08-01 13:15:41.051   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-SSID-REENABLED id=40 ssid="dd-wrt 5GHz"

08-01 13:15:41.069   856   856 I wpa_supplicant: wlan0: Trying to associate with SSID 'dd-wrt 5GHz'

08-01 13:15:42.115   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=60:38:e0:b9:0b:2a status_code=1

08-01 13:15:42.115   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=40 ssid="dd-wrt 5GHz" auth_failures=1 duration=10 reason=CONN_FAILED

08-01 13:15:54.443   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-SSID-REENABLED id=40 ssid="dd-wrt 5GHz"

08-01 13:15:54.455   856   856 I wpa_supplicant: wlan0: Trying to associate with SSID 'dd-wrt 5GHz'

08-01 13:15:55.528   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=60:38:e0:b9:0b:2a status_code=1

08-01 13:15:55.528   856   856 I wpa_supplicant: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=40 ssid="dd-wrt 5GHz" auth_failures=1 duration=10 reason=CONN_FAILED
```

Going back to 10.3.4.0-20170606 driver....

yuhhaurlin, 2017-08-03:
The code related to accounting has problem. I will try to fix it. Thanks.

aaron1959, 2017-08-04:
I'm finally connected with the serial cable and can reproduce the issue, so I just rebooted and and now logging until the web interface and SSH stop working then I should have some useful info I hope.

This is a WRT3200acm running DD-wrt r33006
.

aaron1959, 2017-08-06:
I Finaly was able to leave the router running and handling traffic as normal it was up for 29 hours when I lost access to the web ui so I checked the open putty window for a look at the hardware serial output when that happened and I'm hoping you find this information at least interesting.

[180834.005677] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[180859.552554] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180860.749391] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180863.003277] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180865.645190] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180866.463197] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180866.521133] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180872.776986] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180873.194873] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180874.467827] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180882.509471] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180883.106507] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180883.525501] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180884.009671] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180884.468401] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180885.153366] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180888.481252] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180889.047264] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180890.002251] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180894.004716] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[180901.637741] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180902.732650] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180903.508626] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180905.651564] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180907.003874] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180908.750428] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180914.506173] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180915.509287] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180917.640974] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180934.507732] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180935.507335] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180937.004250] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180938.688229] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180940.006211] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180941.007073] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180943.112008] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180943.523980] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180943.530641] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180954.003625] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[181015.002493] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[181075.001512] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[181134.000324] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[181193.999275] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[181241.186369] ieee80211 phy1: staid 4 deleted
[181541.591192] ieee80211 phy1: cmd 0x9143=GetSeqno timed out
[181541.596710] ieee80211 phy1: return code: 0x1143
[181541.601351] ieee80211 phy1: timeout: 0x1143
[181545.641033] ieee80211 phy1: cmd 0x9143=GetSeqno timed out
[181545.646550] ieee80211 phy1: return code: 0x1143
[181545.651186] ieee80211 phy1: timeout: 0x1143
[181545.655611] Unable to handle kernel NULL pointer dereference at virtual address 0000000c
[181545.663829] pgd = de024000
[181545.666632] [0000000c] *pgd=1e593831, *pte=00000000, *ppte=00000000
[181545.673050] Internal error: Oops: 17 [#1] SMP ARM
[181545.677858] Modules linked in: nf_nat_pptp nf_conntrack_pptp nf_nat_proto_gre 

nf_conntrack_proto_gre fuse ledtrig_usbport pwrseq_simple pwrseq_emmc ahci_mvebu ahci_platform ahci 

libahci_platform libahci ehci_orion sata_mv usb_storage sd_mod xhci_plat_hcd xhci_pci xhci_hcd 

ehci_pci ehci_platform ehci_hcd usbcore usb_common btmrvl_sdio btmrvl bluetooth mwifiex_sdio mvsdio 

sdhci_pxav3 sdhci_pltfm sdhci mmc_block mmc_core mwifiex mwlwifi mac80211 cfg80211 compat mii tmp421 

pwm_fan leds_pca963x leds_tlc591xx orion_wdt
[181545.724037] CPU: 1 PID: 1296 Comm: hostapd Not tainted 4.9.40 #35
[181545.730241] Hardware name: Marvell Armada 380/385 (Device Tree)
[181545.736270] task: de796b00 task.stack: de5fc000
[... 36 lines cut ...]
[181546.011353] dfa0: b6f49f30 c01076c0 00000000 00000000 00000005 bef16b2c 00000000 00000000
[181546.019650] dfc0: 00000000 00000000 b6f49f30 00000128 b6f49f30 00000000 b6eec9e0 00028790
[181546.027948] dfe0: bef16ab8 bef16aa4 b6f962d0 b6f957c4 60000010 00000005 00000000 00000000
[181546.036258] [<bf0bf6c4>] (_145 [mwlwifi]) from [<bf0bfc10>] (_144+0x27c/0x2e8 [mwlwifi])
[181546.044475] [<bf0bfc10>] (_144 [mwlwifi]) from [<bf0baad0>] (_218+0x38/0x1b8 [mwlwifi])
[181546.052615] [<bf0baad0>] (_218 [mwlwifi]) from [<bf05d054>] (_16+0x144/0x190 [mac80211])
[... cut ...]
[181546.166023] [<c03f02c0>] (__sys_sendmsg) from [<c01076c0>] (ret_fast_syscall+0x0/0x3c)
[181546.174061] Code: e5840010 e5d81004 e5c41014 e598c000 (e59c300c)
[181546.180296] ---[ end trace a91b4365f620c84a ]---

There it stopped giving data from the serial. 
I have Never witnessed this router ever REBOOT on it's own NEVER, it simply loses wifi connections along with no web ui access and loss of connection for serial data. the router still continues to work fine on a LAN and connections to the web through the wire.

s-pimenta, 2017-08-06:
@aaron1959 In my case both on DD-WRT and LEDE the router locks up completely! Wifi lost connection, and also via ethernet (fails to web browsing and fails to go web ui router config access). The router does not reboot immediately, but when I leave the router lockedup after 30min to 1 hour the automatic reboot happens.... 

Maybe your problem is different than mine.... 

However..... Since I flashed the new release of davidc502 (2017-07-30) with a kernel update (4.9.40) https://forum.openwrt.org/viewtopic.php?pid=362826#p362826 at the moment 4 days passed and using the tablet that causes problems and this router lockup NEVER happened again! (the build uses the same lastest commit of mwlwifi driver)

 I don't want to confirm the problem is gone, because I want to be sure on my claim, but right now It's very likely the problem is gone, because that lockup should already occurred to me if the problem still exists. 

But I was not reporting that more earlier because of reading here of other people having the same (maybe similar) problem. 

Davidc502 also talks about **amsdu** problem but I think this build I am using the patch if not included.... so the problem of the router lockup is provoked by the kernel? Reading the kernel change log is a bit out of my understanding so I leave to someone that understands better the kernel change log to analyse the differences between 4.4.37 to 4.4.40 (I don't remember the kernel version of the DD-WRT build)

s-pimenta, 2017-08-06:
The LEDE version of the davidc502 2017-07-30 build is Lede Reboot SNAPSHOT r4651-a6f6f8d (previous one that caused the problem  was Lede Reboot SNAPSHOT r4576-9fe9175) maybe reading the commit changes also helps to track what causes the lockup

aaron1959, 2017-08-06:
Edited my post no ui (wired or wireless).

s-pimenta, 2017-08-06:
Edited my post. Typed wrong kernel version

s-pimenta, 2017-08-06:
Talked too soon! The router just locked up with the davidc502 2017-07-30 build (kernel 4.9.40)...

 **So the problem stills persists!** 

Yes I was using that tablet that causes the problem, was just connected to the router 5ghz band, but not using the internet (just reading a magazine on the tablet) the network activity was only on android background tasks (maybe checking for new emails, etc) 

Immediately connected the computer via ethernet connection and also no Internet and no web ui router access (total lock-up) 

Leaving the router as is and waiting for the automatic reboot

s-pimenta, 2017-08-06:
After the router lock-up SSID is still broadcasting (both 2.4 and 5ghz)

Tryed other wifi devices to connect (because is still broadcasting the SSID) but no response (the wifi activity LEDs always ON but not even once they blink)

s-pimenta, 2017-08-06:
Router just rebooted by itself (passed about 1 hour after lockup), everything now works normal...

I will try to give information as much I can to track down this problem...

aaron1959, 2017-08-06:
I found that the router rebooted while I was away,  I have a serial log of the normal 24/7 warnings about mac:  and then the complete reboot and the following usual endless warnings about mac following the reboot. 
Serial hardware output....

[180884.009671] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180884.468401] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180885.153366] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180888.481252] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180889.047264] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180890.002251] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180894.004716] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[180901.637741] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180902.732650] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180903.508626] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180905.651564] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180907.003874] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180908.750428] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180914.506173] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180915.509287] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180917.640974] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180934.507732] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180935.507335] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180937.004250] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180938.688229] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180940.006211] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180941.007073] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180943.112008] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180943.523980] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180943.530641] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[180954.003625] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[181015.002493] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[181075.001512] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[181134.000324] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[181193.999275] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[181241.186369] ieee80211 phy1: staid 4 deleted
[181541.591192] ieee80211 phy1: cmd 0x9143=GetSeqno timed out
[181541.596710] ieee80211 phy1: return code: 0x1143
[181541.601351] ieee80211 phy1: timeout: 0x1143
[181545.641033] ieee80211 phy1: cmd 0x9143=GetSeqno timed out
[181545.646550] ieee80211 phy1: return code: 0x1143
[181545.651186] ieee80211 phy1: timeout: 0x1143
[181545.655611] Unable to handle kernel NULL pointer dereference at virtual address 0000000c
[181545.663829] pgd = de024000
[181545.666632] [0000000c] *pgd=1e593831, *pte=00000000, *ppte=00000000
[181545.673050] Internal error: Oops: 17 [#1] SMP ARM
[181545.677858] Modules linked in: nf_nat_pptp nf_conntrack_pptp nf_nat_proto_gre nf_conntrack_proto_gre fuse ledtrig_usbport pwrseq_simple pwrseq_emmc ahci_mvebu ahci_platform ahci libahci_platform libahci ehci_orion sata_mv usb_storage sd_mod xhci_plat_hcd xhci_pci xhci_hcd ehci_pci ehci_platform ehci_hcd usbcore usb_common btmrvl_sdio btmrvl bluetooth mwifiex_sdio mvsdio sdhci_pxav3 sdhci_pltfm sdhci mmc_block mmc_core mwifiex mwlwifi mac80211 cfg80211 compat mii tmp421 pwm_fan leds_pca963x leds_tlc591xx orion_wdt
[181545.724037] CPU: 1 PID: 1296 Comm: hostapd Not tainted 4.9.40 #35
[181545.730241] Hardware name: Marvell Armada 380/385 (Device Tree)
[181545.736270] task: de796b00 task.stack: de5fc000
[181545.740910] PC is at _145+0x98/0xf0 [mwlwifi]
[181545.745374] LR is at _145+0x34/0xf0 [mwlwifi]
[181545.749835] pc : [<bf0bf6c4>]    lr : [<bf0bf660>]    psr: 60000013
[181545.749835] sp : de5fdc88  ip : 00000000  fp : 00000000
[181545.761533] r10: d9c00b80  r9 : ca2e75c8  r8 : da3e8174
[181545.766865] r7 : d9c02844  r6 : d9c01540  r5 : 00000000  r4 : da3e4000
[181545.773504] r3 : 00000000  r2 : 00000002  r1 : 00000000  r0 : 00000000
[181545.780144] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[181545.787393] Control: 10c5387d  Table: 1e02404a  DAC: 00000051
[181545.793248] Process hostapd (pid: 1296, stack limit = 0xde5fc210)
[181545.799450] Stack: (0xde5fdc88 to 0xde5fe000)
[181545.803911] dc80:                   0000001f da3e8174 ca2e75c8 d9c045c0 da3e8168 bf0bfc10
[181545.812209] dca0: ca2e75c8 d9c01540 d9c00b80 df719c38 00000000 00000014 de5d6000 bf0baad0
[181545.820507] dcc0: ca2e7000 00000002 df7194c0 d86c6200 00000000 00000014 de5d6000 bf05d054
[181545.828804] dce0: 00000000 bf07b3c4 ca2e7000 ca2e7000 00000002 bf05fe34 00000002 00000000
[181545.837101] dd00: 00000014 ca2e7000 d9c00b80 bf0607bc cff46320 d86c6200 00000000 ca2e7000
[181545.845399] dd20: cff46320 bf060928 df7194c0 00000454 cff46320 bf0609a0 d9c00000 df719000
[181545.853697] dd40: de5fdd68 bf029078 cff46320 0002000c bf050710 bf046a4c cff46314 c04595d0
[181545.861996] dd60: bf046374 00000000 000017d6 0000050f cff46300 cff46310 cff46314 de5d6000
[181545.870293] dd80: d9c00000 df719000 cff46300 d86c6200 c04592b0 00000028 de5fddcc 00000000
[181545.878591] dda0: 00000000 c04588a8 c0927684 d86c6200 d5bda000 c04592a0 df5bdc00 d86c6200
[... cut ...]
[  141.615872] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[  166.642719] ieee80211 phy1: staid 2 deleted
[  197.965308] ip=9001a8c0 match mac 00:1b:c7:ff:3e:69= 00:1b:c7:ff:3e:69
[  201.614712] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[  204.288334] ieee80211 phy1: staid 2 deleted
[  261.615500] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[  321.615423] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e
[  382.609259] ip=8101a8c0 match mac 00:6b:9e:9f:17:0e= 00:6b:9e:9f:17:0e

yuhhaurlin, 2017-08-06:
I found problem with 10.3.4.0-20170713. Please help to check if 10.3.4.0-20170606 has problem or not. BTW, I need version information of mwlwifi. Please give version information of mwlwifi when you report problem.

aaron1959, 2017-08-06:
Sorry I can't remember the command lines for finding mwlwifi I will be happy to do that if you could be so kind as to post the code for that query.

aaron1959, 2017-08-06:
I have no Idea of how to integrate the driver into DD-WRT I am know nothing  about Linux , I'll have to wait to see if BrainSlayer will choose to put the "updated" driver into DD-WRT.
If he does that I will be happy to test I leave the serial cable connected at all times going right to the board.
I'm logging 24/7 both in the UI as well as from the serial.

yuhhaurlin, 2017-08-06:
1. cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info.
2. If you can't test 10.3.4.0-20170606, you can wait the fix for the problem I found and test the fix to see if the fix also fixes the problem you encountered.

aaron1959, 2017-08-06:
`So I can hope that BrainSlayer may try this one you are posting about. 

I don't understand the numbering as the "new" driver number seems like an older driver?

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170713
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:13:64:fa
2g: enable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0c00000
iobase1: e0e80000
tx limit: 1024
rx limit: 16384

s-pimenta, 2017-08-07:
@aaron1959 DD-WRT has the build with the lastest commit of mwlwifi driver (here: https://www.dd-wrt.com/site/support/other-downloads) Path: Downloads › betas › 2017 › 08-03-2017-r33006 › linksys-wrt3200acm

As I said before (https://github.com/kaloz/mwlwifi/issues/198#issuecomment-319357150) this problem also occurs on DD-WRT

aaron1959, 2017-08-07:
Sorry if you look that version uses the driver.

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: [color=#FF0000]10.3.4.0-20170713[/color]
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:13:64:fa
2g: enable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0c00000
iobase1: e0e80000
tx limit: 1024
rx limit: 16384

wongsyrone, 2017-08-08:
root@LEDE:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170606
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b3:35:6a
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000000
radio: disable
iobase0: e1080000
iobase1: e1300000
tx limit: 1024
rx limit: 16384

[308516.046631] ieee80211 phy1: staid 1 deleted
[312959.929158] ieee80211 phy1: staid 1 deleted
[317633.927247] ieee80211 phy1: staid 1 deleted
[331373.960488] ieee80211 phy1: staid 2 deleted
[331383.989185] ieee80211 phy1: cmd 0x9111=SetNewStation timed out
[331383.995146] ieee80211 phy1: return code: 0x1111
[331383.999781] ieee80211 phy1: timeout: 0x1111
[331402.464909] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[331402.470856] ieee80211 phy1: return code: 0x001d
[331402.475502] ieee80211 phy1: timeout: 0x001d
[331406.478854] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[331406.484809] ieee80211 phy1: return code: 0x001d
[331406.489444] ieee80211 phy1: timeout: 0x001d
[331410.492793] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[331410.498743] ieee80211 phy1: return code: 0x001d
[331410.503389] ieee80211 phy1: timeout: 0x001d
[331414.516733] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[331414.522687] ieee80211 phy1: return code: 0x001d
[331414.527340] ieee80211 phy1: timeout: 0x001d
[331418.530690] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[331418.536648] ieee80211 phy1: return code: 0x001d
[331418.541283] ieee80211 phy1: timeout: 0x001d
[331422.544609] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[331422.550557] ieee80211 phy1: return code: 0x001d
[331422.555204] ieee80211 phy1: timeout: 0x001d
[331432.464466] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[331432.470438] ieee80211 phy1: return code: 0x001d
[331432.475086] ieee80211 phy1: timeout: 0x001d
[331436.478422] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[331436.484378] ieee80211 phy1: return code: 0x001d
[331436.489014] ieee80211 phy1: timeout: 0x001d
[331440.491623] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[331440.497578] ieee80211 phy1: return code: 0x001d
[331440.502218] ieee80211 phy1: timeout: 0x001d
[331444.489928] ieee80211 phy1: cmd 0x8050=broadcast_ssid_enable timed out
[331444.496584] ieee80211 phy1: return code: 0x0050
[331444.501220] ieee80211 phy1: timeout: 0x0050
[331448.476232] ieee80211 phy1: cmd 0x8127=SetInformationElements timed out
[331448.482972] ieee80211 phy1: return code: 0x0127
[331448.487606] ieee80211 phy1: timeout: 0x0127

root@LEDE:~# uptime 
 17:47:06 up 3 days, 20:05,  load average: 1.78, 0.92, 0.40

yuhhaurlin, 2017-08-08:
@wongsyrone It looks like your problem is not related to the one I try to fix. I will create image for you to do test later. BTW, does it always need such a long time to reproduce your problem?

wongsyrone, 2017-08-08:
Yes.

wongsyrone, 2017-08-08:
You can create another branch in your fork, and I will compile it myself since I need some special software to be installed onto the router.

yuhhaurlin, 2017-08-08:
@wongsyrone I think I will create LEDE image for you to reproduce your problem after I fix the problem related to account processing. If it needs such a long time to reproduce the problem, maybe you can reboot the router every day to avoid this problem first. Thanks.

wongsyrone, 2017-08-08:
Rebooting is not a problem. My router won't lockup and reboot itself, only Wi-Fi channel becomes offline.

yuhhaurlin, 2017-08-08:
@wongsyrone The reason I hope you can reproduce the problem on my image is that your problem looks like firmware is gone. If you can reproduce problem on my image, I can try to hook ICE to debug the problem. Maybe we can check it later.

wongsyrone, 2017-08-08:
> firmware is gone

Do you mean other programs write firmware's memory?

yuhhaurlin, 2017-08-08:
@wongsyrone Command timeout means firmware won't response command any more. I will check with you later. If it needs such a long time to hit this problem, you can try to reboot router or restart wifi every day to avoid this problem first.

BrainSlayer, 2017-08-08:
i may do a build with full debug symbols so we see at least the exact function in mwlwifi and not just the stripped function name

s-pimenta, 2017-08-08:
Got new router lockup, but the tablet is not the only device that causes the problem... I just remove (forget) the wifi network of the WRT3200ACM on the tablet and its only connected to the ISP router, and still the lockup problem occured with my laptop connected to the WRT3200ACM...

aaron1959, 2017-08-08:
I'm willing to test anything on this wrt3200acm U.S. version Production version PCB so I have access to all interrnals and can go beyong the onboard hardware serial interface but since were only testing firmware this should be fine I can run 24/7 any firmware I can load and I can stress it a bit at 5 and 2.4 ghz as well as the usual internet things, these issues seem to be easy to replicate. I'm willing to test I'm back to the version of DD-wrt r32597 and still watching serial output.

yuhhaurlin, 2017-08-09:
I can reproduce RCU stall problem on 10.3.4.0-20170727 (it takes a period of time to reproduce it). The same test is all right for 10.3.4.0-20170606. The log shows problem related to the code for account processing. I try to fix it. I wonder if you encounter the same problem as I encountered. If you can't give me more useful information, you can wait for fix for the problem I found.

aaron1959, 2017-08-09:
@yuhhaurlin  what information would you like? would you like me to run the 10.3.4.0-20170727 again and log some more serial data when it locks up??

yuhhaurlin, 2017-08-10:
Please help to check 10.3.4.0-20170810. It fixes RCU stall problem.

aaron1959, 2017-08-10:
10.3.4.0-20170606 IS Crashing after about 20 hours I turned off a wireless client and tried to turn it back on and it would not authenticate then 5 ghz went down as well. Lost web UI, lost WIFI, still had wired connections no problem .  SSH and of course serial outputs of events preceding this event. please review attached text files containing direct serial output as well as one with SSH Log query.
[r32597 serial.txt](https://github.com/kaloz/mwlwifi/files/1215518/r32597.serial.txt)
[r32597 SSH.txt](https://github.com/kaloz/mwlwifi/files/1215519/r32597.SSH.txt)

BrainSlayer, 2017-08-10:
@yuhhaurlin the crashfunction is mwl_fwcmd_destroy_ba

yuhhaurlin, 2017-08-10:
Please enable kernel symbol.

yuhhaurlin, 2017-08-10:
BTW, how about 10.3.4.0-20170810?

yuhhaurlin, 2017-08-10:
Thanks. I will check it.

[133 more replies]

closed 2017-08-26 by yuhhaurlin