kaloz/mwlwifi #202: Linux global country code for WiFi gets messed up by this driver

howl, 2017-07-31:
When this driver is in use the global country code gets messed up till reboot. This doesn't happen when the module mwlwifi is not present and it's possible to manage the country code for other wifi cards, usb and even the sdio one also found inside the WRT3200ACM.

Despite the README of mwlwifi says not to change the country code, that is not an option at all, as the could be a little possible when the only wifi card in the system is a mwlwifi one, but if you use multiple ones you need to change the country code. For example the SDIO one let you change the country code and the default one is wrong even for the place the router has been bought.

The actual situation makes impossible to use other WiFi adapters with the WRT3200ACM without the global country code messed up to 98 and get restrictions that doesn't apply in the country where the router where bought for all the wifi adapters, even the mwlwifi ones.

howl, 2017-07-31:
It seems that this bug will be closed by @yuhhaurlin as he stated in #172 just because the country code can't be changed as stated in the README of this module, despite there is an explanation here about why that it's not an option. So it's not going to read this at all.

yuhhaurlin, 2017-07-31:
You are wrong. Only the firmware is programmed with device power table, then mwlwifi will disallow this kind of thing. In other word, it is the firmware to decide the behavior of mwlwifi. For WRT3200ACM, the device is programmed with device power table, so it won't allow user to change region code and tx power.

howl, 2017-07-31:
I will try to put it simplier not because you can't understand, just because not to talk about what is not the issue.

mwlwifi loaded, global country code messed up to 98, no mwlwifi loaded global country code works as expected with all other wifi adapters you want. Not changing country code not an option, that makes impossible to operate with other wifi adapters at the same time in the device.

Just make the driver not to mess up the global country code, leave the freaking behavior as you want but don't mess up with other software with the driver.

yuhhaurlin, 2017-07-31:
You don't understand, without this mechanism, there is no way for the device to be complied with FCC rules. In that case, the only way is that we can't allow the device to load open source firmware. If the firmware is not open source, user has no way to change region code and tx power of device. Do we need this kind of mechanism to forbidden user to hack the code and change anything to let the device can't be complied with FCC rules.

closed 2017-07-31 by yuhhaurlin

howl, 2017-07-31:
Leave the freaking mechanism, bug fix the global country code mess up for other adapters to work at the same time. It's not possible to explain simpler.

yuhhaurlin, 2017-07-31:
If you use WRT3200ACM and you don't change country code, you will not encounter problem. Unless you use 88W8964 module for other place, then you will encounter problem. We have another version of firmware to be used for this case, but for WRT3200ACM, this is the way to let the device to be complied with FCC rules.

yuhhaurlin, 2017-07-31:
Mwlwifi has no problem to change country code and tx power if the firmware is working with device without device power table. So I don't know what should I fix!

yuhhaurlin, 2017-07-31:
Even though we have another version of firmware which can ignore device power table. So the limitation is due to firmware and device power table, not mwlwifi.

howl, 2017-07-31:
Don't worry it's all OK don't fix anything, great driver, great foss support.

yuhhaurlin, 2017-07-31:
If you want to use 88W8964 module for other place, you need to check with Marvell. Thanks.

howl, 2017-07-31:
Don't worry, good work.

s-pimenta, 2017-07-31:
I have an WRT3200ACM-EU and the country code says FR (france) and I bought it from Amazon Spain and I live in Portugal... 

With this issue I cannot connect an USB Wifi card from Europe? or from USA?

howl, 2017-07-31:
You can, but then you can't use DFS channels at all for any of the wifi cards as the global country code will be messed up and Linux will refuse to do CAC for DFS. You can test it with the router itself as it has two wifi devices.

s-pimenta, 2017-07-31:
@howl Yeah I now understand... the main purpose of buying this router is to be able to use DFS channels, unless will be like any wifi Router on the market...

yuhhaurlin, 2017-07-31:
Why did you need another USB WiFi card? How many users will use this kind of application?

yuhhaurlin, 2017-07-31:
If you use client to connect to AP, it will run 802.11d and it will follow the country setting from AP.

yuhhaurlin, 2017-07-31:
I don't think user will connect another USB WiFi card to WRT3200ACM.

yuhhaurlin, 2017-07-31:
BTW, if the client had already set with country code, it will only scan channels allowed by the regulatory domain. I don't know why client will affect AP function of mwlwifi. If you mean you want to connect another USB WiFi card to WRT3200ACM via USB, I don't think it is a good idea.

s-pimenta, 2017-07-31:
I'm saying this because I want to connect the router to another wifi network if my Internet connection fails... And to be this possible I think this is only to add another USB wifi

yuhhaurlin, 2017-07-31:
If you use this USB as client, it will follows country setting by mwlwifi to scan channels.

yuhhaurlin, 2017-07-31:
Can you try what happened after connection with AP?

howl, 2017-07-31:
@yuhhaurlin how many users? For example all users that need two independent radios, but mwlwifi mess up this kind of setup. There are issues with the linux kernel stack to have one device acting as client and ap at the same time when the network the router it's connected it's intermittent and having two radios solves that.

Also you can see more practical use cases explained by the librerouter specs.

What I can see is that you aren't understanding the problem it's causing in the mac80211 stack of the linux kernel this module. With another wifi device you won't be able to use DFS channels cause the global country code will be set to 98 even you don't change it at all. You are one way with this is how to complain with FCC but you are wrong, there are more ways that plays fair and doesn't break other stuff.

Anyway I give up with this router, the pity is that I bought two time ago expecting them will work as they should.

yuhhaurlin, 2017-07-31:
I told you this is special design for WRT3200ACM. If you need to use 88W8964 module in other place, you can check with Marvell.

howl, 2017-08-01:
@yuhhaurlin I haven't told you nowhere I need to use 88W8964 in other place, I ignored you the first time you said that and now I don't know, because you understand what you want and talk about non related things when you can answer coherent things because that will make you give another one the reason that this modules is a real mess and a pain in the ass.

howl, 2017-08-01:
@s-pimenta I will answer you because the main developer won't do or if he do will be a non related answer.

You can't by no way use another wifi device with DFS channels while mwlwifi are installed. The only possible solutions I see is to use two wifi usb and remove mwlwifi from the system, or directly the more appropriate one find another router that doesn't have crap like WRT3200ACM. The most adequate solution will be that the developer understand the situation, but seems that will be impossible at least in this multiverse.

yuhhaurlin, 2017-08-01:
AP will run first, so mwlwifi will set channel first. USB WiFi will base on regulatory domain to scan channel. If it connect to remote AP, no matter it set country code or not, DFS channels already be used by mwlwifi. BTW, the regulatory domain should be the same for all APs.

yuhhaurlin, 2017-08-01:
In fact, mwlwifi already sets country code for you based on region code programmed in your device. In this way, you don't need to set country code again.

howl, 2017-08-02:
@yuhhaurlin indeed you are unable to understand the real mess you have done. There is anybody in the room aside yuhhaurlin that doesn't understand that when mwlwifi mess up by autosetting the global country code when there is another device the regulatory rules get messed up.

@yuhhaurlin what you say is not what happens, when there is another wifi device the regulatory domain gets dumbed by mwlwifi even if you NOT try to set another reg domain? As I said in this multiverse you are not going to understand that.

WRT3200ACM a good paperweight till you find someone to sell it that perhaps want to use it with the stock firmware.

yuhhaurlin, 2017-08-02:
Stock firmware uses the same firmware of WiFi module and same rules. It is not open source, you can't change anything.

BrainSlayer, 2017-08-03:
@howl : hint for you. a small change in mwlwifi will fix your issue and its easy to find.

BrainSlayer, 2017-08-03:
@howl: to be specific. remove the both regulatory_hint calls in mwl_reg_notifier. it wont change the behaviour of the mwldriver but it prevents reset of the countrycode back to the driver setting. once you want to change it

howl, 2017-08-03:
@yuhhaurlin go tell to whom I will find to sell the WRT3200ACMs.

@BrainSlayer thanks for the hint, but the even with it, there is no guarantee that in the future there is another important issue with the module and a way to make the main developer understand it (I really feel as I was talking with Eliza, or worst). I can only lament myself because I haven't heavy knowledge about reverse engineering and very good linux kernel module and mac80211 stack programming skills.

BrainSlayer, 2017-08-03:
@howl: if i study the way you talked to him its no wonder that he cannot follow you. you should switch off your feelings. if you would talk in that way to me, i wouldnt even reply. consider that he cannot change the way he implemented it, he is forced todo so. if you want to say thank you to someone, then visit the fcc. marvell is forced to preprogram all countrycodes and powertables. and consider also that he is no native speaker but chinese or taiwanese with the typical communication problems. i had already troubles with him in that way but i was always be able to resolv it by showing him practical examples why something wasnt done in the right way

yuhhaurlin, 2017-08-04:
The better way is that changing the mapping between region code and country and let mwlwifi set country code for you. However, you can only map country belongs to the region, otherwise, you will encounter problem.

seragh, 2017-08-04:
The current implementation is without a doubt sub par and there are a few alternatives which wouldn't conflict with FCC / Linksys / Marvell requirements / interests as far as I understand them and wouldn't cause this issue. <linux>/include/net/regulatory.h briefly discuss them. The most desirable would be to allow the user to further restrict what is used and only have the eeprom define the the superset of the allowable configuration.

howl, 2017-08-04:
@BrainSlayer the problem is that there is no way to make him understand, I have tried to explain different issues one way and another. The first one, to implement this using the linux kernel documented behaviour, as @seragh and others have said this doesn't conflict with FCC, and makes possible not to mess with the broadcasted country code. Here came up a supposed contract between Marvel and Belkin, but there si no one that knows about it to contact and discuss about it and how the linux kernel way doesn't go towards FCC and is more adequate. Finally I forgive.

Then derived from this issue, I see that when there is another radio that uses another module, the country code gets messed up even if you don't assign a country code and let mwlwifi do it. This is really a issue when there is a need of two radios, because you get limited depending the country you are, in the frequencies and power you can use. And if you have to use it with another devices already set to operate in the frequencies that has gone out with the country mess the router is useless.

I know that I get ruder from message to message and I sincerely expend my apologizes to @yuhhaurlin, but please try to understand the issues and be open mind to see that there are more solutions that doesn't generate problems and don't go towards FCC rules.

BrainSlayer, 2017-08-06:
i told you how to fix that issue. just remove 2 lines and the driver wont reset the countrycode anymore and you can do whatever you want, but the driver internally will still be fixed to his own countrycode

howl, 2017-08-06:
I know, but I won't be messing with the source code. I prefer to find a router with similar specs that works out of the box. At least the ones I have find there isn't support at all, so will wait and remain with a non MU-MIMO router with a supplemental usb wifi card.

Again thanks a lot @BrainSlayer

ValCher1961, 2017-08-14:
I've got a wrt3200acm destined for USA, I'm from Europe.
Using the American version of the router in Europe, I violated the legislative laws adopted in Europe.
I was able to change the WRT3200ACM code from FCC to ETSI, and thus unlock 12 and 13 channels in the G-band.
At the beginning, I posted the information and patch in the OpenWrt forum, but after removing the patch, I thought it might cause some harm.
I don't know how many scans were made and who could use it in the future.
Yuhhaurlin and the whole community, it's up to you to decide what to do.

howl, 2017-08-16:
@ValCher1961 you can read that @yuhhaurlin says that an AP from one country is not for use in another country. I didn't find that affirmation nowhere in the manual or conditions from linksys, but trust me that @yuhhaurlin it's just going to repeat he's previous affirmation that you can't do that.

Anyways if you follow the linux kernel guidelines what you did it's what it has to be done. And also refutes all the @yuhhaurlin arguments, as it's possible to do even with the original firmware.

ValCher1961, 2017-08-16:
@howl Read the topic of this driver discussion, and you'll get a lot of clarity.
Https://www.spinics.net/lists/linux-wireless/msg157416.html.
@yuhhaurlin I'm sorry for the aggravation of the subject. Everything I dug up would stay with me.

howl, 2017-08-16:
@ValCher1961 Aggravation? You arrived late.

AKTanara, 2018-08-01:
So long since the last post on this thread,
But I have a question that seems relevant hear...
I want to know that if I switch my Linksys WRT3200ACM-EU firmware to DD-WRT will it still be limited to regional TX Power limitations or I will be able to attain 30 dBm (1000mW) on both 2.4 & 5.0 GHz?

Regards

AKTanara, 2018-08-01:
@BrainSlayer
So long since the last post on this thread,
But I have a question that seems relevant hear...
I want to know that if I switch my Linksys WRT3200ACM-EU firmware to DD-WRT will it still be limited to regional TX Power limitations or I will be able to attain 30 dBm (1000mW) on both 2.4 & 5.0 GHz?

Regards

BrainSlayer, 2018-08-02:
@AKTanara : the chipset was never capable of doing 30 dbm, this would burn the device and no the mwlwifi driver doesnt support setting of the power limits on this chipset since all capabilities are preprogrammed on the chipset itself. there is no known way to override it. since the chipset is basicly controlled by a on chip firmware and not by the driver itself. if the device isnt preprogrammed you can set it in theory. but it seems that i'm the only person on this planed owning a wrt3200acm which is capable of doing that. but its a preproduction model which was given to me for development in the early days

AKTanara, 2018-08-02:
@BrainSlayer 
Thanks for reply, I just bought a Linksys WRT3200ACM-EU and haven't unpacked it. With info you provided I decided to return it...
Many Thanks

kristian, 2018-09-05:
@BrainSlayer just completely hypothetically... I mean... nobody *should & would* in theory consider to do someting horrible, horrible like this in the first place, but...: as far as I understood your explanation mwlwifi is by no means able to modify the region settings of any WRT3200ACM-* device, whichs is fine... but (again *only* hypothetically) couln't there be any build option which might could help you to "adjust" "some kind" of "table" ever so slightly, essentially reproducing very similar results? Just asking for a friend! ;)