kaloz/mwlwifi #203: version 20170727 still drops streaming audio

jacksonatl, 2017-07-31:
Streaming audio is dropped after about 30 minutes, every time. It's a simple mp3 stream from a radio station. I have no issues on an older Buffalo 2G router also running LEDE. 
All I get in the logs are a bunch of: 
ieee80211 phy0: staid 9 deleted
ieee80211 phy0: staid 3 deleted
 ieee80211 phy0: staid 5 deleted
 ieee80211 phy0: staid 2 deleted
 ieee80211 phy0: staid 2 deleted
 ieee80211 phy0: staid 5 deleted
 ieee80211 phy0: staid 9 deleted
 ieee80211 phy0: staid 7 deleted
 ieee80211 phy0: staid 6 deleted
 ieee80211 phy0: staid 9 deleted
 ieee80211 phy0: staid 7 deleted
 ieee80211 phy0: staid 5 deleted
 ieee80211 phy0: staid 10 deleted
 ieee80211 phy0: staid 5 deleted
 ieee80211 phy0: staid 10 deleted
 ieee80211 phy0: staid 7 deleted
 ieee80211 phy0: staid 10 deleted
 ieee80211 phy0: staid 9 deleted
 ieee80211 phy0: staid 9 deleted
 ieee80211 phy0: staid 5 deleted
 ieee80211 phy0: staid 5 deleted
 ieee80211 phy0: staid 9 deleted
 ieee80211 phy0: staid 9 deleted
 ieee80211 phy0: staid 9 deleted
 ieee80211 phy0: staid 9 deleted

Kernel log isn't much more interesting (10 minutes before drop and you can see me log in after drop)
Mon Jul 31 17:50:17 2017 daemon.warn odhcpd[1223]: DHCPV6 REQUEST IA_NA from 000100011e8f6afc847beb1c6653 on br-lan: ok fde1:a253:8add::146/128 
Mon Jul 31 17:52:47 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 30:75:12:89:31:1d
Mon Jul 31 17:55:18 2017 daemon.warn odhcpd[1223]: DHCPV6 SOLICIT IA_NA from 000100011e8f6afc847beb1c6653 on br-lan: ok fde1:a253:8add::146/128 
Mon Jul 31 17:55:18 2017 daemon.info dnsmasq[30128]: read /etc/hosts - 4 addresses
Mon Jul 31 17:55:18 2017 daemon.info dnsmasq[30128]: read /tmp/hosts/odhcpd - 0 addresses
Mon Jul 31 17:55:18 2017 daemon.info dnsmasq[30128]: read /tmp/hosts/dhcp.cfg02411c - 4 addresses
Mon Jul 31 17:55:18 2017 daemon.info dnsmasq-dhcp[30128]: read /etc/ethers - 0 addresses
Mon Jul 31 17:55:19 2017 daemon.warn odhcpd[1223]: DHCPV6 REQUEST IA_NA from 000100011e8f6afc847beb1c6653 on br-lan: ok fde1:a253:8add::146/128 
Mon Jul 31 17:55:19 2017 daemon.info dnsmasq[30128]: read /etc/hosts - 4 addresses
Mon Jul 31 17:55:19 2017 daemon.info dnsmasq[30128]: read /tmp/hosts/odhcpd - 1 addresses
Mon Jul 31 17:55:19 2017 daemon.info dnsmasq[30128]: read /tmp/hosts/dhcp.cfg02411c - 4 addresses
Mon Jul 31 17:55:19 2017 daemon.info dnsmasq-dhcp[30128]: read /etc/ethers - 0 addresses
Mon Jul 31 17:57:50 2017 daemon.notice hostapd: wlan0: AP-STA-POLL-OK 30:75:12:89:31:1d
Mon Jul 31 18:00:19 2017 daemon.warn odhcpd[1223]: DHCPV6 SOLICIT IA_NA from 000100011e8f6afc847beb1c6653 on br-lan: ok fde1:a253:8add::146/128 
Mon Jul 31 18:00:19 2017 daemon.info dnsmasq[30128]: read /etc/hosts - 4 addresses
Mon Jul 31 18:00:19 2017 daemon.info dnsmasq[30128]: read /tmp/hosts/odhcpd - 0 addresses
Mon Jul 31 18:00:19 2017 daemon.info dnsmasq[30128]: read /tmp/hosts/dhcp.cfg02411c - 4 addresses
Mon Jul 31 18:00:19 2017 daemon.info dnsmasq-dhcp[30128]: read /etc/ethers - 0 addresses
Mon Jul 31 18:00:20 2017 daemon.warn odhcpd[1223]: DHCPV6 REQUEST IA_NA from 000100011e8f6afc847beb1c6653 on br-lan: ok fde1:a253:8add::146/128 
Mon Jul 31 18:00:20 2017 daemon.info dnsmasq[30128]: read /etc/hosts - 4 addresses
Mon Jul 31 18:00:20 2017 daemon.info dnsmasq[30128]: read /tmp/hosts/odhcpd - 1 addresses
Mon Jul 31 18:00:20 2017 daemon.info dnsmasq[30128]: read /tmp/hosts/dhcp.cfg02411c - 4 addresses
Mon Jul 31 18:00:20 2017 daemon.info dnsmasq-dhcp[30128]: read /etc/ethers - 0 addresses
Mon Jul 31 18:02:18 2017 authpriv.info dropbear[32620]: Child connection from 192.168.1.100:45258

yuhhaurlin, 2017-08-01:
Did you have this problem on 10.3.4.0-20170606? Can you use following way to get more debug information:

# grep _level /var/run/hostapd-phy0.conf
logger_syslog_level=2
logger_stdout_level=2
# uci set wireless.radio0.log_level=1
# uci commit wireless
# wifi up
# grep _level /var/run/hostapd-phy0.conf
logger_syslog_level=1
logger_stdout_level=1

jacksonatl, 2017-08-02:
Ok, downgraded to 2017-06-06 driver and was still having the issue, after running tcpdump and looking that over I determined that the issue was actually with my ISP. Apologise for the false alarm, but I'm guessing a 10-80% packet drop rate has more to do with this issue than the wifi driver. Please close the ticket as Comcast is now working on their network to resolve this.

closed 2017-08-02 by jacksonatl