kaloz/mwlwifi #205: wrt1900acv1 txpower adjustment below regulatory limit does not work.

kchan345, 2017-08-13:
Hi,

1) The test is done with stationary phone placed on a table.
2) The txpower option in /etc/config/wireless is adjusted, after which the radio were re-enabled.
3) iw wlan0/1 info reports the txpower set in /etc/config/wireless

Test result:
channel, txpower setting, rssi
149, 10, -36 dbm
149, 20, -36 dbm
36, 10, -54 dbm
36, 20, -55 dbm

As seen, adjusting txpower has no measurable differences in the received signal strength by the phone. The only way to significantly change the metric is only by switching channel from 149 to 36, which has a different fcc regulatory limit 23 vs 30.

I suspect that the driver is ignoring mac80211 configuration and always uses a predefined maximum. For my project we need a really low txpower (~4-10 dBm) for high density wifi deployment.

I agree that one should not be able to set txpower higher than the regulatory limit, but one should be able to adjust the txpower within the regulatory limit.

yuhhaurlin, 2017-08-13:
Sorry. It is not supported.

closed 2017-08-13 by yuhhaurlin

kchan345, 2017-08-13:
Do you have some reference to previous discussion why turning down txpower below regulatory limit cannot be supported?