kaloz/mwlwifi #206: 5GHz band dropping clients sporadically under light sustained load

AjkayAlan, 2017-08-13:
I am encountering an issue where my 5GHz band is dropping clients when there is a steady load (but not maxing out my connection or anything). Here are some logs with various images:

Davidc502's LEDE image with 10.3.4.0-20170810:
[sysLog.txt](https://github.com/kaloz/mwlwifi/files/1220568/sysLog.txt)
[kernelLog.txt](https://github.com/kaloz/mwlwifi/files/1220569/kernelLog.txt)
I reconnected on 2.4GHz.

@yuhhaurlin 's image provided at https://github.com/kaloz/mwlwifi/issues/198#issuecomment-321714701:
[mwlwifiInfo.txt](https://github.com/kaloz/mwlwifi/files/1220573/mwlwifiInfo.txt)
[sysLog.txt](https://github.com/kaloz/mwlwifi/files/1220574/sysLog.txt)
[kernelLog.txt](https://github.com/kaloz/mwlwifi/files/1220575/kernelLog.txt)
I reconnected on 2.4GHz.

Another example from the above image:
[mwlwifiInfo.txt](https://github.com/kaloz/mwlwifi/files/1220576/mwlwifiInfo.txt)
[sysLog.txt](https://github.com/kaloz/mwlwifi/files/1220577/sysLog.txt)
[kernelLog.txt](https://github.com/kaloz/mwlwifi/files/1220578/kernelLog.txt)
I stayed on 5GHz and it reconnected after around 30 seconds.

@BrainSlayer's image provided in https://github.com/kaloz/mwlwifi/issues/198#issuecomment-321853842:
[mwlwifiInfo.txt](https://github.com/kaloz/mwlwifi/files/1220579/mwlwifiInfo.txt)
[ddwrtLog.txt](https://github.com/kaloz/mwlwifi/files/1220580/ddwrtLog.txt)
I reconnected on 2.4GHz.

Repetition steps for me:
1. Load up something that continuously streams (such as twitch)
2. Play a video game (such as Path Of Exile).

This issue only appears on the 5GHz band for me - the 2.4GHz doesn't drop for me at all.

In all above examples the device which was having issues was 68:1c:a2:14:02:d2. I can repeat the same issue on another device if needed. I will also try and repeat the same issue in stock firmware.

Chadster766, 2017-08-13:
To provide @yuhhaurlin with a different comparison could you test McDebian which is running mwlwifi 10.3.4.0-20170606?

https://github.com/Chadster766/McDebian

AjkayAlan, 2017-08-14:
> To provide @yuhhaurlin with a different comparison could you test McDebian which is running mwlwifi 10.3.4.0-20170606?

I am somewhat uncomfortable with something that doesn't have a UI unfortunately as I am relatively new to linux. Are you more focusing on a driver version? If so, would pulling something like LEDE and downgrading to 10.3.4.0-20170606 work instead?

> I will also try and repeat the same issue in stock firmware.
I actually am able to repeat this issue on stock firmware. It took a lot longer (over 4 hours of light load), but it did happen. Part of me wonders - could this be hardware failure instead of software issues?

yuhhaurlin, 2017-08-14:
If this only happens to one specific client, I suggest that you can test the client with other AP to see if client is all right.

AjkayAlan, 2017-08-14:
Unfortunately I don't have another 5GHz AP (lost my old one in a lightning strike). I will test and get logs for another client when I get home from work today.

BrushXue, 2017-08-14:
I have the exactly same issue. WRT1900ACSV2 LEDE 17.01.2

yuhhaurlin, 2017-08-15:
@BrushXue Previous code is not the same as the code for WRT3200ACM, I don't hear this problem for 88W8864.

yuhhaurlin, 2017-08-15:
@AjkayAlan 
1. When the problem happened, please ping the router to see if ping is all right.
2. Please help to check if you can reproduce this problem on other clients.

BrushXue, 2017-08-15:
From 10.3.0.17 to 10.3.4.0, I always have this problem. Usually it happens when downloading huge files. 5G connection will drop for every 10-20GB data transfer. After around 30 seconds it will be able to reconnect automatically.

yuhhaurlin, 2017-08-15:
@BrushXue Only happened for one specific client? Ping is all right when the problem happened.

AjkayAlan, 2017-08-15:
Will do. I have been trying to reproduce the issue on another client for about 6 hours and have been unable to. I made a few changes to try to eliminate more variables before testing:

1. I did a SSID impact analysis of my area and determined I needed to switch to some different channels (153 for 5GHz, 11 for 2.4ghz)
2. I moved the router a bit to get it farther away from anything which could impact a signal
3. I adjusted the antennas a bit to match the 45 degree angle picture at https://www.linksys.com/us/support-article?articleNum=138847.

After these changes and running both Davidc502's LEDE image with 10.3.4.0-20170810 and your image in the other thread I am not getting a replication. I will try and test my problem client more tomorrow along with continue load testing my second client. I won't be able to test for another 16 hours or so (sleep then work), so feel free to spread your attention elsewhere for your work day.

BrushXue, 2017-08-15:
@yuhhaurlin No. All of my roommates can trigger this issue when downloading.

yuhhaurlin, 2017-08-15:
@AjkayAlan Thanks.

BrushXue, 2017-08-15:
@yuhhaurlin One of my roommate told me he tried 2.4G. His connection no longer drops but had several 300ms pings when playing games.

yuhhaurlin, 2017-08-15:
@BrushXue Please help to test following LEDE images:

Factory: https://drive.google.com/open?id=0B3qLWtcWB9EdSUpCc3VrRDZMeVE
Upgrade: https://drive.google.com/open?id=0B3qLWtcWB9EdRlMwWXlLN1Z1Z0E

root@LEDE:/# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 00:50:43:22:88:8a
2g: disable
5g: enable
antenna: 4 4
irq number: 44
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000000
radio: disable
iobase0: f0e00000
iobase1: f1080000
tx limit: 1024
rx limit: 16384

yuhhaurlin, 2017-08-15:
@BrushXue Please use following ways to get more debug information for me:

# grep _level /var/run/hostapd-phy0.conf
logger_syslog_level=2
logger_stdout_level=2
# uci set wireless.radio0.log_level=1
# uci commit wireless
# wifi up
# grep _level /var/run/hostapd-phy0.conf
logger_syslog_level=1
logger_stdout_level=1

AjkayAlan, 2017-08-15:
@yuhhaurlin That image is tagged for WRT3200ACM, is it safe for his device (WRT1900ACSV2)?

yuhhaurlin, 2017-08-15:
@AjkayAlan You are right. I will give another links.

yuhhaurlin, 2017-08-15:
@BrushXue Sorry:
Factory: https://drive.google.com/open?id=0B3qLWtcWB9EdaERnbXU4NXcwelk
Upgrade: https://drive.google.com/open?id=0B3qLWtcWB9EdNnYweHpQczhINjQ

If you can reproduce your problem on above LEDE images, please help to check if open mode (without security) is all right.

BrushXue, 2017-08-15:
@yuhhaurlin Can I have the package of the wireless driver separately? So that I don't need to reinstall the system.

yuhhaurlin, 2017-08-15:
@BrushXue Can you get updated mwlwifi driver (10.3.4.0-20170810) for your LEDE image? I can't build the package for your LEDE image.

yuhhaurlin, 2017-08-15:
@BrushXue If you already use 10.3.4.0-20170810, you can use the way I described to get more debug information and help to check if open mode is all right.

BrushXue, 2017-08-15:
@yuhhaurlin The current driver version is 4.4.71+10.3.4.0.git-2017-06-06-1. There is a mwlwifi-bin but no update for over a year. Do I have to compile it by myself?
Alos, is there anything wrong with the security settings? Last week I changed to 802.1X so I removed wpad-mini and installed wpad & freeradius2.

yuhhaurlin, 2017-08-15:
@BrushXue Yes. You can build the updated mwlwifi by yourself. I just need more information. Basically, we did not find this kind of problem here.

BrushXue, 2017-08-15:
@yuhhaurlin I just found one repo
https://github.com/eduperez/mwlwifi_LEDE/releases
is that safe to install?

yuhhaurlin, 2017-08-15:
@BrushXue If it is built for the version of your LEDE image, you can use it.

BrushXue, 2017-08-15:
@yuhhaurlin OK I installed the latest driver and did a speedtest, looks good for now.
Once it drops again I'll upload the log file.

yuhhaurlin, 2017-08-15:
@BrushXue Make sure change log level to get more debug information. Thanks.

BrushXue, 2017-08-15:
@yuhhaurlin It seems like 99% of the log are from 802.1X authentication. How to suppress these messages?

yuhhaurlin, 2017-08-15:
Can you use PSK to do test?

BrushXue, 2017-08-15:
I just pursuaded my roommates to switch to 802.1X for security reason.(I created a guest wifi but they just ignore that and tell guests our main password) Now I'm telling them I'll switch back?

Can I just create another ssid on 5G using PSK?

yuhhaurlin, 2017-08-15:
Yes. You can just change log level for that VAP.

BrushXue, 2017-08-15:
@yuhhaurlin Could you please provide the code to change the log level for **wlan0-1** separately?

yuhhaurlin, 2017-08-15:
Sorry. It can't be changed separately.

AjkayAlan, 2017-08-16:
I unfortunately did not have as much time to load test today (had to work late). I load tested my problem client today while running a continuous ping to my router and did not have issues. Most pings are in the 1-3 ms range to the router, with a random 10-20ms ping every once in a while (5-10 minutes - not a big deal). 

I am going to load test my other client more while laying in bed for a while. I should have more time to test again tomorrow. Assuming that nothing goes wrong tomorrow, I would say that the issue could be closed for me as my issue was probably a combination of wifi congestion and bad positioning of the router and the antennas.

Here is an analysis of the wifi in my area, along with what my setup looks like now,
[image]
[image]

yuhhaurlin, 2017-08-16:
@AjkayAlan Thanks for your information. In fact, I had set group rekey as 60 seconds (one minute) and run iperf test with Macbook Air and Macbook Pro. The test works well. So if anyone reports problem related to security mode, I will check if it is specific client to make problem.

starcms, 2017-08-16:
@BrushXue, to fix high-pings in games, something called amsdu needs to be disabled.  On the 88W8864 chipset (1200/1900AC/ACS models), amsdu is broken and doesn't work properly, causing high-ping and packet loss that is especially noticeable in online gaming and VoIP.

You can download @david502's release of LEDE https://davidc502sis.dynamic-dns.net/releases/ and make sure you also download the latest driver that is at the top of the page.  Check the FAQ page for instructions on how to install it.  In the updated driver, @david502 has disabled amsdu.

Hopefully @yuhhaurlin will get a chance to fix it once he is done adding features to the 88W8964 (in the 3200ACM)

yuhhaurlin, 2017-08-16:
@starcms Replied you in the issue you created.

AjkayAlan, 2017-08-17:
I still have not been able to recreate my issue since adjusting my router's position, channels, and antennas. At this point, I believe my issue was due to random signal quality drops near the client (which would cause a disconnect seen in the logs when the signal got low enough).

From my perspective, this can be closed. Thanks for bearing with me!

closed 2017-08-17 by yuhhaurlin

BrushXue, 2017-08-17:
@yuhhaurlin With the 0810 driver I had one drop and my roommate had two. However, I checked with my roommate. Both of our motherboards use QCA6174 so I think this might be a motherboard issue. I know this chip has a lot of issue on Ubuntu so I'll try to fix my computer.

yuhhaurlin, 2017-08-17:
@BrushXue If this client has no problem with other AP, please let me know.Thanks.

s-pimenta, 2017-08-19:
Had again another lost internet connection (no ping to router), and here is a more verbose log from hostapd

Device in question (10:f0:05:08:ad:ea)

```
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea WPA: event 5 notification
Sat Aug 19 20:52:19 2017 daemon.notice hostapd: wlan0: CTRL-EVENT-EAP-STARTED 10:f0:05:08:ad:ea
Sat Aug 19 20:52:19 2017 daemon.notice hostapd: wlan0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: STA v IEEE 802.1X: Sending EAP Packet (identifier 227)
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: received EAP packet (code=2 id=227 len=12) from STA: EAP Response-Identity (1)
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: STA identity 'pimenta'
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: RADIUS Sending RADIUS message to authentication server
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: RADIUS Next RADIUS client retransmit in 3 seconds
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: RADIUS Received 94 bytes from RADIUS server
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: RADIUS Received RADIUS message
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea RADIUS: Received RADIUS packet matched with a pending request, round trip time 0.00 sec
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: decapsulated EAP packet (code=1 id=228 len=36) from RADIUS server: EAP-Request-unknown (52)
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: Sending EAP Packet (identifier 228)
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: received EAP packet (code=2 id=228 len=22) from STA: EAP Response-unknown (52)
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: RADIUS Sending RADIUS message to authentication server
Sat Aug 19 20:52:19 2017 daemon.debug hostapd: wlan0: RADIUS Next RADIUS client retransmit in 3 seconds
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: RADIUS Received 160 bytes from RADIUS server
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: RADIUS Received RADIUS message
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea RADIUS: Received RADIUS packet matched with a pending request, round trip time 0.05 sec
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: decapsulated EAP packet (code=1 id=229 len=102) from RADIUS server: EAP-Request-unknown (52)
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: Sending EAP Packet (identifier 229)
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: received EAP packet (code=2 id=229 len=102) from STA: EAP Response-unknown (52)
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: RADIUS Sending RADIUS message to authentication server
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: RADIUS Next RADIUS client retransmit in 3 seconds
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: RADIUS Received 96 bytes from RADIUS server
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: RADIUS Received RADIUS message
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea RADIUS: Received RADIUS packet matched with a pending request, round trip time 0.03 sec
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: decapsulated EAP packet (code=1 id=230 len=38) from RADIUS server: EAP-Request-unknown (52)
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: Sending EAP Packet (identifier 230)
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: received EAP packet (code=2 id=230 len=38) from STA: EAP Response-unknown (52)
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: RADIUS Sending RADIUS message to authentication server
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: RADIUS Next RADIUS client retransmit in 3 seconds
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: RADIUS Received 169 bytes from RADIUS server
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: RADIUS Received RADIUS message
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea RADIUS: Received RADIUS packet matched with a pending request, round trip time 0.01 sec
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.11: binding station to interface 'wlan0'
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: old identity 'pimenta' updated with User-Name from Access-Accept 'pimenta'
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: decapsulated EAP packet (code=3 id=230 len=4) from RADIUS server: EAP Success
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: Sending EAP Packet (identifier 230)
Sat Aug 19 20:52:20 2017 daemon.debug hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: authorizing port
Sat Aug 19 20:52:20 2017 daemon.info hostapd: wlan0: STA 10:f0:05:08:ad:ea RADIUS: starting accounting session 26CB8FFAB048E013
Sat Aug 19 20:52:20 2017 daemon.info hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.1X: authenticated - EAP type: 52 (unknown)
[... cut ...]
Sat Aug 19 20:53:47 2017 daemon.info hostapd: wlan0: STA 10:f0:05:08:ad:ea IEEE 802.11: authenticated
Sat Aug 19 20:53:49 2017 daemon.warn odhcpd[1153]: DHCPV6 CONFIRM IA_NA from 000423da6e23abb9a93645a8f6e25c3593ca on br-lan: not on-link fd62:e8dd:c69c::dfe/128 
Sat Aug 19 20:53:50 2017 daemon.warn odhcpd[1153]: DHCPV6 SOLICIT IA_NA from 000423da6e23abb9a93645a8f6e25c3593ca on br-lan: ok fd62:e8dd:c69c::dfe/128 
Sat Aug 19 20:53:51 2017 daemon.warn odhcpd[1153]: DHCPV6 REQUEST IA_NA from 000423da6e23abb9a93645a8f6e25c3593ca on br-lan: ok fd62:e8dd:c69c::dfe/128 

```

s-pimenta, 2017-08-19:
Edited previous post (https://github.com/kaloz/mwlwifi/issues/206#issuecomment-323544676) to add more information

AjkayAlan, 2017-08-19:
I will say that I am still getting this issue - I must have been getting lucky during the week or something. Here are the logs from the stock firmware (using http://192.168.1.1/sysinfo.cgi/):

[stockSysInfo.txt](https://github.com/kaloz/mwlwifi/files/1236638/stockSysInfo.txt)

I still think I am going to have troubles reproducing on another client since the only other decent 5GHz client I have is a low powered laptop that won't be able to perform the same exact steps I do on my gaming computer.

AjkayAlan, 2017-08-19:
Another interesting thing I noticed, pings to the router on 2.4GHz are way different than on 5GHz when there is light load. I am sitting pretty close to the router with my laptop (the client I could not reproduce the issue on), and here is pinging of the router while watching https://www.twitch.tv/nalcs1:

```
Pinging 192.168.1.1 with 32 bytes of data:
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=3ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=3ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=3ms TTL=64
Reply from 192.168.1.1: bytes=32 time=3ms TTL=64
Reply from 192.168.1.1: bytes=32 time=3ms TTL=64
Reply from 192.168.1.1: bytes=32 time=6ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=4ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=33ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=3ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=4ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
Reply from 192.168.1.1: bytes=32 time=7ms TTL=64
Reply from 192.168.1.1: bytes=32 time=3ms TTL=64
Reply from 192.168.1.1: bytes=32 time=4ms TTL=64
Reply from 192.168.1.1: bytes=32 time=1ms TTL=64
--Switched to 2.4GHz
Reply from 192.168.1.1: bytes=32 time=4ms TTL=64
Reply from 192.168.1.1: bytes=32 time=6ms TTL=64
Reply from 192.168.1.1: bytes=32 time=3ms TTL=64
Reply from 192.168.1.1: bytes=32 time=40ms TTL=64
Reply from 192.168.1.1: bytes=32 time=5ms TTL=64
Reply from 192.168.1.1: bytes=32 time=116ms TTL=64
Reply from 192.168.1.1: bytes=32 time=50ms TTL=64
Reply from 192.168.1.1: bytes=32 time=232ms TTL=64
Reply from 192.168.1.1: bytes=32 time=104ms TTL=64
Reply from 192.168.1.1: bytes=32 time=105ms TTL=64
Reply from 192.168.1.1: bytes=32 time=182ms TTL=64
Reply from 192.168.1.1: bytes=32 time=18ms TTL=64
Reply from 192.168.1.1: bytes=32 time=217ms TTL=64
Reply from 192.168.1.1: bytes=32 time=14ms TTL=64
Reply from 192.168.1.1: bytes=32 time=26ms TTL=64
Reply from 192.168.1.1: bytes=32 time=12ms TTL=64
Reply from 192.168.1.1: bytes=32 time=12ms TTL=64
Reply from 192.168.1.1: bytes=32 time=125ms TTL=64
Reply from 192.168.1.1: bytes=32 time=9ms TTL=64
Reply from 192.168.1.1: bytes=32 time=94ms TTL=64
Reply from 192.168.1.1: bytes=32 time=2ms TTL=64
Reply from 192.168.1.1: bytes=32 time=65ms TTL=64
Reply from 192.168.1.1: bytes=32 time=10ms TTL=64
Reply from 192.168.1.1: bytes=32 time=162ms TTL=64
Reply from 192.168.1.1: bytes=32 time=25ms TTL=64
Reply from 192.168.1.1: bytes=32 time=156ms TTL=64
Reply from 192.168.1.1: bytes=32 time=8ms TTL=64
Reply from 192.168.1.1: bytes=32 time=313ms TTL=64
Reply from 192.168.1.1: bytes=32 time=10ms TTL=64
Reply from 192.168.1.1: bytes=32 time=5ms TTL=64
Reply from 192.168.1.1: bytes=32 time=17ms TTL=64
Reply from 192.168.1.1: bytes=32 time=5ms TTL=64
Reply from 192.168.1.1: bytes=32 time=6ms TTL=64
Reply from 192.168.1.1: bytes=32 time=43ms TTL=64
Reply from 192.168.1.1: bytes=32 time=306ms TTL=64
Reply from 192.168.1.1: bytes=32 time=3ms TTL=64
Reply from 192.168.1.1: bytes=32 time=7ms TTL=64
```

yuhhaurlin, 2017-08-19:
1. If you can reproduce this problem on stock firmware, please report to where you should report.
2. 2g band may have inference, you can switch channel 1, 6, and 11 to check it.

AjkayAlan, 2017-08-19:
Running Davicc502's lede with the 0810 wifi driver, here is another logged event where my desktop dropped, but my laptop stayed connected. I was pinging my router on both machines:

[syslog.txt](https://github.com/kaloz/mwlwifi/files/1236697/syslog.txt)
[kernellog.txt](https://github.com/kaloz/mwlwifi/files/1236698/kernellog.txt)
[image]
[pinglog.txt](https://github.com/kaloz/mwlwifi/files/1236699/pinglog.txt)

Interestingly enough, I didn't switch the network at all, and windows ended up killing the adapter and restarting it, along with throwing an error in the event log:

WLAN Extensibility Module has failed to start.

Module Path: C:\WINDOWS\System32\bcmihvsrv64.dll
Error Code: 21

[event.txt](https://github.com/kaloz/mwlwifi/files/1236701/event.txt)

I went ahead and reported the issue on stock firmware through their channels as well.

yuhhaurlin, 2017-08-19:
If you can reproduce this problem on stock firmware, please report to where you should report.

AjkayAlan, 2017-08-21:
Alright, I replaced my wireless adapter and am no longer having issues. My problem adapter was a Rosewill rnx-ac1900pce. I bought a Netgear AC1200 (A6210) and cant cause any drops. Sorry again for the extra noise.