kaloz/mwlwifi #207: Amsdu broken on 88W8864

starcms, 2017-08-16:
Please see issue https://github.com/kaloz/mwlwifi/issues/74 for plenty of details.

Hopefiully once work is done on 88W8964, this can be revisited and fixed.  Not sure why issue #74 was closed.

yuhhaurlin, 2017-08-16:
AMSDU works well. For VoIP, the packet will get high priority. Due to AMSDU will flush packet within 10ms if no following packets come to the same STA.

starcms, 2017-08-16:
Multiplayer games are made practically impossible to play.  Have you conducted the simple ping test I described above?

yuhhaurlin, 2017-08-16:
VoIP should not have problem. It goes to another AC and it will get high priority. About multiplayer games, can you let me know what the problem is?

starcms, 2017-08-16:
In any multiplayer gaming, ping is 30-100ms higher than compared to another router, with the driver compiled with amsdu disabled, or over cat5.  It really is a significant issue.  Not present on the 3200ACM though (88W8964).

yuhhaurlin, 2017-08-16:
All right. I will add debugfs to allow user to disable AMSDU later.

starcms, 2017-08-16:
Thank you so much for reconsidering and future work on this issue.  Can you re-open issue until then?

jbsky, 2017-09-04:
I confirm with devices such as iphone and ipad, I have much less latency on games with the option wmm_enabled=0 in the hostapd configuration file.

I've even got the ipad that goes below 10ms.

To check the problem, just ping from router to device. The problem does not necessarily appear when pinking in the opposite direction.

starcms, 2017-09-05:
@jbsky, I actually have a WRT1200AC, but this issue affects all devices with the 88W8864 wifi chip.  So that means all WRT1200/1900/ac/acs models.  Only device not affected is the WRT3200ACM (with its 88W8964 chip).

@davidc502 compiles his LEDE builds with amsdu disabled in mwlwifi (to fix this issue until @yuhhaurlin 
 can make a fix or easier work-around) and I highly recommend them.  https://davidc502sis.dynamic-dns.net/

kylejvrsa, 2017-09-05:
I can confirm what @starcms is saying. We have 3 WRT1900ACS routers and all 3 have the same behaviour:

Lede 17.01.2

cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8864
hw version: 7
driver version: 10.3.4.0-20170606
firmware version: 0x0702091a
power table loaded from dts: no
firmware region code: 0x10
mac address: 00:25:9c:14:12:31
2g: disable
5g: enable
antenna: 4 4
irq number: 104
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000003
radio: enable
iobase0: e0e80000
iobase1: e1000000
tx limit: 768
rx limit: 64
qe trigger number: 3722122

From Client to Router:
```
 icmp_seq=2 ttl=64 time=1.26 ms
 icmp_seq=3 ttl=64 time=4.86 ms
 icmp_seq=4 ttl=64 time=3.85 ms
 icmp_seq=5 ttl=64 time=1.40 ms
 icmp_seq=6 ttl=64 time=1.26 ms
 icmp_seq=7 ttl=64 time=1.65 ms
 icmp_seq=8 ttl=64 time=1.45 ms
 icmp_seq=9 ttl=64 time=1.65 ms
 icmp_seq=10 ttl=64 time=2.79 ms
 icmp_seq=11 ttl=64 time=1.22 ms
 icmp_seq=12 ttl=64 time=7.82 ms
 icmp_seq=13 ttl=64 time=2.07 ms
 icmp_seq=14 ttl=64 time=8.66 ms
 icmp_seq=15 ttl=64 time=1.66 ms
 icmp_seq=16 ttl=64 time=2.19 ms
 icmp_seq=17 ttl=64 time=1.75 ms
 icmp_seq=18 ttl=64 time=1.95 ms
 icmp_seq=19 ttl=64 time=1.53 ms
 icmp_seq=20 ttl=64 time=2.59 ms
 icmp_seq=21 ttl=64 time=8.84 ms
 icmp_seq=22 ttl=64 time=2.11 ms
 icmp_seq=23 ttl=64 time=2.80 ms
 icmp_seq=24 ttl=64 time=1.62 ms
 icmp_seq=25 ttl=64 time=4.52 ms
 icmp_seq=26 ttl=64 time=16.3 ms
 icmp_seq=27 ttl=64 time=1.99 ms
 icmp_seq=28 ttl=64 time=1.48 ms
 icmp_seq=29 ttl=64 time=6.03 ms
 icmp_seq=30 ttl=64 time=3.18 ms
 icmp_seq=31 ttl=64 time=3.46 ms
 icmp_seq=32 ttl=64 time=1.65 ms
 icmp_seq=33 ttl=64 time=1.42 ms
 icmp_seq=34 ttl=64 time=1.26 ms
```
From Router to Client:
```
 seq=29 ttl=64 time=16.021 ms                                                                                                                                      
 seq=30 ttl=64 time=287.967 ms
 seq=31 ttl=64 time=211.757 ms
 seq=32 ttl=64 time=1.698 ms
 seq=33 ttl=64 time=265.176 ms
 seq=34 ttl=64 time=181.003 ms
 seq=35 ttl=64 time=275.410 ms
 seq=36 ttl=64 time=125.292 ms
 seq=37 ttl=64 time=148.873 ms
 seq=38 ttl=64 time=172.421 ms
 seq=39 ttl=64 time=16.658 ms
 seq=40 ttl=64 time=117.598 ms
 seq=41 ttl=64 time=40.152 ms
 seq=42 ttl=64 time=2.030 ms
 seq=43 ttl=64 time=87.389 ms
 seq=44 ttl=64 time=112.223 ms
 seq=45 ttl=64 time=139.988 ms
 seq=46 ttl=64 time=159.508 ms
 seq=47 ttl=64 time=1.379 ms
 seq=48 ttl=64 time=105.332 ms
 seq=49 ttl=64 time=17.608 ms
 seq=50 ttl=64 time=153.549 ms
 seq=51 ttl=64 time=285.671 ms
 seq=52 ttl=64 time=6.550 ms
 seq=53 ttl=64 time=2.652 ms
 seq=54 ttl=64 time=147.567 ms
 seq=55 ttl=64 time=178.349 ms
 seq=56 ttl=64 time=93.623 ms
 seq=57 ttl=64 time=21.106 ms
 seq=58 ttl=64 time=36.999 ms
 seq=59 ttl=64 time=14.978 ms
 seq=60 ttl=64 time=289.808 ms
 seq=61 ttl=64 time=108.879 ms
 seq=62 ttl=64 time=2.748 ms
 seq=63 ttl=64 time=3.296 ms
 seq=64 ttl=64 time=180.061 ms
 seq=65 ttl=64 time=209.622 ms
 seq=66 ttl=64 time=228.060 ms
 seq=67 ttl=64 time=256.919 ms
 seq=68 ttl=64 time=275.787 ms
 seq=69 ttl=64 time=13.851 ms
```

I know the driver is outdated, I just wanted to confirm that with even the older driver the issue was already there.

yuhhaurlin, 2017-09-05:
From router to client, it could be power save mode of your client.

yuhhaurlin, 2017-09-05:
You can disable power save mode of client and redo the test.

kylejvrsa, 2017-09-05:
Power management is disabled for that client. It's a Laptop with Debian and KDE plugged in with power management disabled.

Here is the output from another client:
Router to Client:
```
seq=0 ttl=64 time=1.217 ms
 seq=1 ttl=64 time=1.053 ms
 seq=2 ttl=64 time=1.311 ms
 seq=3 ttl=64 time=0.911 ms
 seq=4 ttl=64 time=0.973 ms
 seq=5 ttl=64 time=0.923 ms
 seq=6 ttl=64 time=0.943 ms
 seq=7 ttl=64 time=0.817 ms
 seq=8 ttl=64 time=0.969 ms
 seq=9 ttl=64 time=0.937 ms
 seq=10 ttl=64 time=0.995 ms
 seq=11 ttl=64 time=1.764 ms
 seq=12 ttl=64 time=1.450 ms
 seq=13 ttl=64 time=1.113 ms
 seq=14 ttl=64 time=0.997 ms
 seq=15 ttl=64 time=1.275 ms
 seq=16 ttl=64 time=1.493 ms
 seq=17 ttl=64 time=0.927 ms
 seq=18 ttl=64 time=0.987 ms
 seq=19 ttl=64 time=0.925 ms
 seq=20 ttl=64 time=0.876 ms
 seq=21 ttl=64 time=1.395 ms
 seq=22 ttl=64 time=0.965 ms
 seq=23 ttl=64 time=1.072 ms
 seq=24 ttl=64 time=0.951 ms
 seq=25 ttl=64 time=0.981 ms
 seq=26 ttl=64 time=1.234 ms
 seq=27 ttl=64 time=0.820 ms
 seq=28 ttl=64 time=1.507 ms
 seq=29 ttl=64 time=1.443 ms
 seq=30 ttl=64 time=1.711 ms
 seq=31 ttl=64 time=0.879 ms
 seq=32 ttl=64 time=0.864 ms
 seq=33 ttl=64 time=1.547 ms
 seq=34 ttl=64 time=0.748 ms
 seq=35 ttl=64 time=0.768 ms
 seq=36 ttl=64 time=0.955 ms
 seq=37 ttl=64 time=0.999 ms
 seq=38 ttl=64 time=1.133 ms
 seq=39 ttl=64 time=1.451 ms
 seq=40 ttl=64 time=0.996 ms
 seq=41 ttl=64 time=0.736 ms
 seq=42 ttl=64 time=0.891 ms
 seq=43 ttl=64 time=1.112 ms
 seq=44 ttl=64 time=1.196 ms
 seq=45 ttl=64 time=0.861 ms
 seq=46 ttl=64 time=2.069 ms
 seq=47 ttl=64 time=1.930 ms
 seq=48 ttl=64 time=1.181 ms
 seq=49 ttl=64 time=1.448 ms
 seq=50 ttl=64 time=174.146 ms
 seq=51 ttl=64 time=1.910 ms
 seq=52 ttl=64 time=221.782 ms
 seq=53 ttl=64 time=245.565 ms
 seq=54 ttl=64 time=21.374 ms
 seq=55 ttl=64 time=89.200 ms
 seq=56 ttl=64 time=1.367 ms
 seq=57 ttl=64 time=134.291 ms
 seq=58 ttl=64 time=187.147 ms
 seq=59 ttl=64 time=1.480 ms
 seq=60 ttl=64 time=2.670 ms
 seq=61 ttl=64 time=247.285 ms
 seq=62 ttl=64 time=257.378 ms
 seq=63 ttl=64 time=31.275 ms
 seq=64 ttl=64 time=29.108 ms
 seq=65 ttl=64 time=120.618 ms
 seq=66 ttl=64 time=144.383 ms
 seq=67 ttl=64 time=170.743 ms
 seq=68 ttl=64 time=194.809 ms
 seq=69 ttl=64 time=34.461 ms
 seq=70 ttl=64 time=0.878 ms
 seq=71 ttl=64 time=30.269 ms
 seq=72 ttl=64 time=290.440 ms
 seq=73 ttl=64 time=1.381 ms
 seq=74 ttl=64 time=1.459 ms
 seq=75 ttl=64 time=1.183 ms
 seq=76 ttl=64 time=0.883 ms
 seq=77 ttl=64 time=1.122 ms
 seq=78 ttl=64 time=1.292 ms
 seq=79 ttl=64 time=1.349 ms
 seq=80 ttl=64 time=1.279 ms
 seq=81 ttl=64 time=1.276 ms
 seq=82 ttl=64 time=0.885 ms
 seq=83 ttl=64 time=1.100 ms
 seq=84 ttl=64 time=0.995 ms
 seq=85 ttl=64 time=0.812 ms
 seq=86 ttl=64 time=0.918 ms
 seq=87 ttl=64 time=2.383 ms
 seq=88 ttl=64 time=263.737 ms
 seq=89 ttl=64 time=50.792 ms
 seq=90 ttl=64 time=106.938 ms
 seq=91 ttl=64 time=128.441 ms
 seq=92 ttl=64 time=1.316 ms
 seq=93 ttl=64 time=31.274 ms
 seq=94 ttl=64 time=1.038 ms
 seq=95 ttl=64 time=226.429 ms
 seq=96 ttl=64 time=250.652 ms
 seq=97 ttl=64 time=274.277 ms
 seq=98 ttl=64 time=298.246 ms
 seq=99 ttl=64 time=59.700 ms
 seq=100 ttl=64 time=1.788 ms
 seq=101 ttl=64 time=166.251 ms
 seq=102 ttl=64 time=189.368 ms
 seq=103 ttl=64 time=212.968 ms
 seq=104 ttl=64 time=69.469 ms
 seq=105 ttl=64 time=55.687 ms
 seq=106 ttl=64 time=41.134 ms
 seq=107 ttl=64 time=310.335 ms
 seq=108 ttl=64 time=29.308 ms
 seq=109 ttl=64 time=75.951 ms
 seq=110 ttl=64 time=2.511 ms
 seq=111 ttl=64 time=1.070 ms
 seq=112 ttl=64 time=0.910 ms
 seq=113 ttl=64 time=14.174 ms
 seq=114 ttl=64 time=1.469 ms
 seq=115 ttl=64 time=19.454 ms
 seq=116 ttl=64 time=22.207 ms
 seq=117 ttl=64 time=23.781 ms
 seq=118 ttl=64 time=24.311 ms
 seq=119 ttl=64 time=1.206 ms
 seq=120 ttl=64 time=0.978 ms
 seq=121 ttl=64 time=1.575 ms
 seq=122 ttl=64 time=1.504 ms
 seq=123 ttl=64 time=0.916 ms
 seq=124 ttl=64 time=1.003 ms
 seq=125 ttl=64 time=1.226 ms
 seq=126 ttl=64 time=1.167 ms
 seq=127 ttl=64 time=1.061 ms
 seq=128 ttl=64 time=1.010 ms
 seq=129 ttl=64 time=0.921 ms
 seq=130 ttl=64 time=0.914 ms
 seq=131 ttl=64 time=1.513 ms
 seq=132 ttl=64 time=63.313 ms
 seq=133 ttl=64 time=25.122 ms
 seq=134 ttl=64 time=69.917 ms
 seq=135 ttl=64 time=73.055 ms
 seq=136 ttl=64 time=1.275 ms
 seq=137 ttl=64 time=0.827 ms
 seq=138 ttl=64 time=84.193 ms
 seq=139 ttl=64 time=88.727 ms
 seq=140 ttl=64 time=91.934 ms
 seq=141 ttl=64 time=93.933 ms
 seq=142 ttl=64 time=79.707 ms
 seq=143 ttl=64 time=100.891 ms
 seq=144 ttl=64 time=103.208 ms
 seq=145 ttl=64 time=107.363 ms
 seq=146 ttl=64 time=110.987 ms
 seq=147 ttl=64 time=0.884 ms
 seq=148 ttl=64 time=23.959 ms
 seq=149 ttl=64 time=1.631 ms
 seq=150 ttl=64 time=0.908 ms
 seq=151 ttl=64 time=21.435 ms
 seq=152 ttl=64 time=1.072 ms
 seq=153 ttl=64 time=0.769 ms
 seq=154 ttl=64 time=1.086 ms
 seq=155 ttl=64 time=2.374 ms
 seq=156 ttl=64 time=145.457 ms
 seq=157 ttl=64 time=1.098 ms
 seq=158 ttl=64 time=153.038 ms
 seq=159 ttl=64 time=1.062 ms
 seq=160 ttl=64 time=143.205 ms
 seq=161 ttl=64 time=158.647 ms
 seq=162 ttl=64 time=77.030 ms
```

The spikes seem to happen when there's activity from other clients on the WiFi, which is to be expected but not to this extent I would imagine?

yuhhaurlin, 2017-09-05:
2g band?

kylejvrsa, 2017-09-05:
Both of them running on 5G

jbsky, 2017-09-05:
I would like to point out that according to my research, the ping latency for an iphone improves punctually only if there is significant bit rate like loading a video on utube. In fact, for games that require very little throughput, the latency is blatant. I also want to point out that I don't have this problem with ISP wifi boxes.
Applied sta_info->is_amsdu_allowed = false doesn't change anything at all to the problem as it could be suggested at the end "High ping on WRT1900ACS #74 ".
I only use the 5Ghz bandwidth.

yuhhaurlin, 2017-09-05:
Without AMSDU, driver won't buffer packets, the ping should be all right. The only possibility is power save mode of client and noise in the air.

starcms, 2017-09-08:
@jbsky, compiling the driver with sta_info->is_amsdu_allowed = false (which disables AMSDU) makes a huge difference in the latency of multiplayer games.  HUGE.  Pings go from 200ms+ (unplayable) to 30ms with it disabled.

jbsky, 2017-09-08:
I don't have the problem with my laptops. I have a ping around 1ms. which increases slightly with a spike of 5ms so big activity like downloading a huge file.

Let me rephrase my statement. I have a similar problem with mobile phones but unlike you, amsdu doesn't change anything.

I'll sum it up at home:

* I have an unstable ping from the router to the device.
`32 packets transmitted, 31 received, 3% packet loss, time 31149ms`
`rtt min/avg/max/mdev = 1.411/654.737/2160.414/657.971 ms, pipe 3`

* The ping is punctually better (but it's not the best either) when I have bit rate to the device, i. e. downloading a video for example or a bit rate test.
`45 packets transmitted, 45 received, 0% packet loss, time 44060ms`
`rtt min/avg/max/mdev = 1.294/166.566/1787.245/350.909 ms, pipe 2`

* I don't have a problem with the ISP box wifi.

Should I open a new ticket?

yuhhaurlin, 2017-09-08:
Can you ping to router as same time? It looks like it is power save of client.

jbsky, 2017-09-08:
I did what you asked and the ping is stable but very high.
`31 packets transmitted, 31 received, 0% packet loss, time 30035ms`
`rtt min/avg/max/mdev = 133.288/180.480/226.579/29.468 ms`

How do I disable the power saving of an iphone? I'm not sure it's possible because this device is battery powered and Apple won't change anything.

Strangely, I don't have this problem with the ISP box wifi, whereas this problem exists with the standard firmware (already tested).

Can the router's wifi driver solve this problem?

yuhhaurlin, 2017-09-08:
This is correct behavior of client in order to save power. In fact, if there is no problem when you run your application, you should not change it.

yuhhaurlin, 2017-09-08:
In fact, DTIM will affect the time for client to wake up. However, I don't think you need to change it. As mobile device, you should not wake it up too frequently if there is no data for the device.

tapper82, 2017-09-21:
Hi yuhhaurlin  are you still working on this driver? There has bin no updated code since aug 10

nidstigator, 2017-10-17:
@starcms I just want to chime in here and say thank you for finding the root cause for this. Been running LEDE on my WRT1900ACS since Oct 2016 and this issue has been puzzling me.

I just built a custom image with AMSDU disabled and the ping issue is completely gone!

Thanks again!

bbogey, 2017-10-18:
Will it cause any other issues when disable AMSDU, less BW or something?
If in @davidc502 build AMSDU is disabled, which is used also as a reference build for testing,  so should we all disable AMSDU?

nidstigator, 2017-10-18:
@bbogey Nothing i've noticed anyway. I build my own image because I enable cryptodev for hardware-accelerated vpn decryption. Otherwise I'd be using @davidc502's build.

yuhhaurlin, 2017-11-06:
Please check https://github.com/kaloz/mwlwifi/commit/574e24e7dd661676e585dacd54e4f93b24e78ba0. I have added debugfs file "tx_amsdu" to enable or disable tx amsdu. Using this way, you can disable tx amsdu if this feature bothers you. The setting only affects stations associated later. You can use "cat sta" to check if tx amsdu is enable or not for associated stations.

Note: This commit only affects 88W8864. 

root@LEDE:/sys/kernel/debug/ieee80211/phy0/mwlwifi# cat sta

mac address: 5c:f9:38:96:62:e0
aid: 1
ampdu: true
amsdu: false
ht_cap: 0x086f, ampdu: 03, 05
rx_mask: 0xff, ff, 00, 00
vht_cap: 0x0f8158b2, mcs: fffa, fffa
rx_bw: 2, rx_nss: 2
tdls: 0, tdls_init: 0
wme: 1, mfp: 0
IV: 000000000001

closed 2017-11-06 by yuhhaurlin

thagabe, 2017-11-07:
@yuhhaurlin so it's just a work around....until you can fix this right?

yuhhaurlin, 2017-11-07:
If I have time, I will check it later.

starcms, 2017-11-20:
@yuhhaurlin, I see tx_amsdu present in /sys/kernel/debug/ieee80211/phy0/mwlwifi.  I can't delete it.  Is this the right file to delete to disable amsdu?  Thanks!

starcms, 2017-11-20:
@yuhhaurlin, so simple, thanks!!

starcms, 2017-11-20:
Just in-case anyone else has the same question, the commands to disable AMSDU on 5ghz and 2.4ghz are as follows:

echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu
echo 0 > /sys/kernel/debug/ieee80211/phy1/mwlwifi/tx_amsdu

They have to be entered on every boot and before any devices connect to the AP.

thagabe, 2018-03-05:
@yuhhaurlin 

Was this issue ever fixed without a workaround?

starcms, 2018-04-29:
@thagabe, nope.  I don't think @yuhhaurlin has any plans to either.

yunesj, 2021-08-09:
> Just in-case anyone else has the same question, the commands to disable AMSDU on 5ghz and 2.4ghz are as follows:
>
> echo 0 > /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu
> echo 0 > [...]

Will it work to put it in /etc/rc.local via LuCi?

Is this the best place to watch the issue?

gileri, 2021-08-10:
>Will it work to put it in /etc/rc.local via LuCi?

yes