kaloz/mwlwifi #208: 3200ACM MU-MIMO

p3x-robot, 2017-08-21:
What is the status of the  MU-MIMO implementation?

p3x-robot, 2017-08-21:
is there anything I can help to make it work?

p3x-robot, 2017-08-22:
so are you saying for ```3200ACM```, just disable ```kmod-mwifiex-sdio``` and  ```mwifiex-sdio-firmware``` and ```mwlwifi``` will automatically do it?

BrainSlayer, 2017-08-22:
from my knowledge mu mimo is supported. you just need to configure the vht flags correct in hostapd

p3x-robot, 2017-08-22:
ciao!
how are you? 
thanks so much for replying!
so i delete kmod-mwifiex-sdio and mwifiex-sdio-firmware and configure the vht flags correct in hostapd?
do you know where the vht flags correct in hostapd?

see ya!

p3x-robot, 2017-08-22:
ahh and whichi version that supports? lede 17.0.1.2 is enough? or newer?
thanks so much!

yuhhaurlin, 2017-08-22:
This feature will be added later.

p3x-robot, 2017-08-25:
I found this:
```
docker@34e9a52707ae:/build/source$ cat /build/source/package/network/services/hostapd/files/hostapd.sh | grep vht
		ac) append base_cfg "require_vht=1" "$N";;
```

@BrainSlayer  Is it where I could set it up the VHT somehow, there is no VHT is hostapd anywhere also.

I found these:
```
/build/source/package/network/services/hostapd/patches/004-hostapd-Add-possibility-to-send-debug-messages-to-sy.patch
/build/source/package/network/services/hostapd/patches/430-hostapd_cli_ifdef.patch
/build/source/package/network/services/hostapd/files/hostapd-full.config
/build/source/package/network/services/hostapd/files/hostapd-mini.config
/build/source/package/network/services/hostapd/files/hostapd.sh
/build/source/dl/hostapd-2016-12-19-ad02e79d.tar.xz
```
But all empty with VHT with grep.

p3x-robot, 2017-08-25:
@BrainSlayer  in the router I found this files:
```
/lib/netifd/hostapd.sh # has vht
/tmp/run/hostapd-phy0.conf # has vht
```
```
root@digi:~# cat /rom/lib/netifd/hostapd.sh | grep vht
		ac) append base_cfg "require_vht=1" "$N";;
```
```
root@digi:~# cat /tmp/run/hostapd-phy0.conf | grep vht
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=-6
vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][SU-BEAMFORMER][SU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]
```

Do you know where this could turn it on???
:)

**This is it, isn'it ?**
```
root@digi:~# cat /tmp/run/hostapd-phy0.conf
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=FR
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
channel=acs_survey

ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=-6
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][SU-BEAMFORMER][SU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=password
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=patrikx3-digi
bridge=br-lan
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=60:38:e0:bd:7a:6a
```

BrainSlayer, 2017-08-25:
you need to fix the vht_capab info MU-BEAMFORMER etc is missing. but i need to check on my device first which supported flags are returned by the chipset itself. would take some hours since i just woke up

BrainSlayer, 2017-08-25:
okay checked it. the chipset does return mu-beamformee etc. as unsupported so the chipset isnt capable of doing mu-mimo or the mwlwifi drivers firmware isnt capable of doing it. @yuhhaurlin  may habe a answer to it

yuhhaurlin, 2017-08-25:
I already told you that MU-MIMO will be added later. 88W8964 is capable to do that, but it needs extra code to let it work. I have no schedule now, but it should be done ASAP.

yuhhaurlin, 2017-08-25:
BTW, mac80211.sh will add capability of chip to hostapd configuration file. If you use LEDE/OpenWrt, there is no need for you to worry about it.

gufus, 2017-08-25:
How about WDS?  Maybe next?

Please :-)

yuhhaurlin, 2017-08-25:
I think WDS will be done first. I have no schedule for these two features, but they will be done ASAP.

gufus, 2017-08-25:
Ah thanks!

There will be something extra under the Christmas tree 4 ya this year :-)

p3x-robot, 2017-08-25:
ok, then I cannot patch it myself,thanks for the info, i close it, ciao!

closed 2017-08-25 by p3x-robot