kaloz/mwlwifi #209: Loss of Authentication on ath0/5Ghz

aaron1959, 2017-08-26:
After some time apx 20 to 24 hours I lose my 5 Ghz connections and attempts to reconnect will give the message "cannot authenticate"
I will post Syslog and serial log when this next occurs.

aaron1959, 2017-08-27:
5 Ghz lost authentication? will not authenticate!
Here are the syslog and serial output from before the event to the event.
[latest syslog for 5Ghz deauthentication.txt](https://github.com/kaloz/mwlwifi/files/1254069/latest.syslog.for.5Ghz.deauthentication.txt)
[Latest serial log of 5Ghz failure to authenticate..txt](https://github.com/kaloz/mwlwifi/files/1254070/Latest.serial.log.of.5Ghz.failure.to.authenticate.txt)

yuhhaurlin, 2017-08-27:
Yes. This is host command timeout. I need to reproduce it first before debugging it. If you can reproduce the problem on the LEDE image and let me know how to reproduce it, it will help me to debug it.

yuhhaurlin, 2017-08-27:
Another strange is that even though system reboot and redownload firmware, host command is still timeout. Can you try another WRT3200ACM?

aaron1959, 2017-08-27:
I  tried already with LEDE and let run for days.

yuhhaurlin, 2017-08-27:
@aaron1959 You run exactly the same test on the LEDE image I sent out, and this problem won't happen?

aaron1959, 2017-08-27:
Yes I ran the LEDE you gave me last time with the 810 driver and I let it run for at least 48 hours and it never had any issue with 5 Ghz losing authentication/host command timeout.
also there were far less warnings, the DD-WRT has hundreds of warnings and eventually a lot of kernel errors before the "host command timeout"

andyVSR, 2017-08-27:
dd-wrt is using a different version of hostapd and wpa_supplicant. it is 2.6 devel vs 2.7 devel on LEDE.i also encounter some issues with some (not all) 5gz devices on dd-wrt, which doesnt happen with LEDE. one issue is that if i turn off wifi on some 5gz device , it will not be deauth from router and connections shows stii active on dd-wrt. other issue is that randomly on reboots the wifi devices does not reconnect.
 maybe @BS can confirm that is a different version of hostapd on dd-wrt.

aaron1959, 2017-08-27:
@andyVSR  can you post logs? If hostpd is not the same then ut seems that this is likely the issue with not seeing this behavior in LEDE, IE Host Command Timeout.??

andyVSR, 2017-08-27:
for my issues there is no records in logs (the connection to the wifi device is just not deauth and not a single line in dmesg output).
 you can use :
 hostaph -v
 wpa_supplicant -v
to check versions. also from dd-wrt svn, the most recent hostapd is 2016-09-05, and for LEDE is 2016-12-19. i dont know what hostapd used @yuhhaurlin in his latest version providet to you for testing.
 better @BS and @yuhhaurlin can clarify this.

aaron1959, 2017-08-27:
Interesting I'm obviously testing a older version of DD-WRT 33180 so My info may be useless as there may have been significant changes. And the issue you report is totally different then this one.  
I will load r33215 and resume testing. 
For now this issue can be closed I think. I will reopen later if same issue with hostpd.

BrainSlayer, 2017-08-27:
its very unlikelly that this is related to hostapd etc. the error shows in the log is related to the devices own firmware which seem to have crashed or is hanging

aaron1959, 2017-08-27:
Would you recommend clearing Nvram and then trying again?
I have not been clearing NVRAM as per earlier posts about it not being needed?

aaron1959, 2017-08-27:
I'm just going to try to nvram erase on this latest version and look some more. The device doesn't do any of this with the version of LEDE that I tested also the Linksys firmware loaded on the other partition seems to work 24/7 when I tested it last.

aaron1959, 2017-08-28:
Since this latest DD-WRT and Nvram erase it seems to be working, but only 20 hours. 
Before Nvram erase it lasted 23 hours then with only rebooting it would only last a couple of hours, so I'm keeping my fingers crossed that the firmware crash was due to corruption in the Nvram.
If this same issue doesn't happen in the next 20 hours I will close this. 
Thanks for the help.

BrainSlayer, 2017-08-28:
i dont expect that. if the nvram really would be corrupted which is very unlikelly, you will see much bigger effects

aaron1959, 2017-08-28:
Well I was hoping it's @ up 22:25 and still OK so I'm waiting, if it makes it three days I'm convinced it's OK for me.

aaron1959, 2017-08-28:
All WiFi failed.  
Here is the syslog of all of the errors and some log before that.
[all wifi fail 8_28.txt](https://github.com/kaloz/mwlwifi/files/1258701/all.wifi.fail.8_28.txt)
If you want serial log for some reason I can post.

yuhhaurlin, 2017-08-29:
Same as before. If you can't reproduce this problem on LEDE image, I don't know how to debug it. In fact, there is no need for you to send out more log If it happened on dd-WRT.

aaron1959, 2017-08-29:
OK I won't bother it does seem like a issue unique to DD-WRT. Please close this.
I give up for now. 

I will be trying any router without a marvel chip.

rs-se, 2017-08-29:
@aaron1959 

What kind of WiFi clients are you using for 5Ghz, once in awhile my Amazon Echo Dots loose connectivity but not since latest version.

rs-se, 2017-08-29:
@aaron1959 

Your best bet is not to buy residential quality networking equipment, ubiquiti wifi products are a definite alternative.

aaron1959, 2017-08-29:
This time ALL WiFi clients lost connectivity, clients are Samsung s7 android @866 Mhz and a Dell laptop with intel wireless @ 300 Mhz  on 5 Ghz and a bunch of clients on 2.4 @ 72 Mhz or so. Wired clients are just fine. Unless BrainSlayer has something, I'm going to quietly leave and figure out some other option. I am sad but that's the way it goes sometimes.

aaron1959, 2017-08-29:
Well this thing is a Frankenstein 
Linksysbelkinmarvel   wrt3200acm 
POS. 
NO MORE MARVEL!

yuhhaurlin, 2017-08-29:
@aaron1959 Yes, it is host command time out, so all clients will get problem. In order to debug this kind of problem:
1. Reproduce the problem on the LEDE image I sent out or LEDE image from https://davidc502sis.dynamic-dns.net/.
2. Once if the problem happened, please give me the log to make sure it is host command timeout.
3. Let me know your setting and test environment, so I can reproduce the problem.
4. Once if the problem can be reproduced, I can hook ICE to debug the problem.

Thanks.

yuhhaurlin, 2017-08-29:
@aaron1959 If the problem always happened after 20 hours, maybe you can check if calling "wifi up" to reinitialize AP every 12 hours can work around your problem.

aaron1959, 2017-08-29:
That is an idea for now I may try that and If I get some time I will try LEDE again I probably didn't let it run long enough. I do know once it fails it will fail again after reboot much faster if I erase Nvram then it always stayed up about 20 hours.
I have no idea about how to call "wifi up" but I do know how to use google and will attempt to do this.

yuhhaurlin, 2017-08-29:
BTW, does any other users encounter this kind of problem on dd-WRT or LEDE? If you have this kind of problem on LEDE, please let me know your setting and test environment. Thanks.

yuhhaurlin, 2017-08-29:
@aaron1959 Basically mwlwifi won't touch NVRAM. I already query if other users have the same issue on dd-WRT. If other users won't have this kind of problem, maybe you can try to use another WRT3200ACM. Thanks.

yuhhaurlin, 2017-08-29:
@aaron1959 Another way is that when time is allowable, I will do more tests on LEDE image. However, if you can reproduce the problem on it and let me know how to do it, it will be helpful. Thanks.

davidc502, 2017-08-29:
Not to say others have not, but I've never run across this problem.. Knock on wood :)

yuhhaurlin, 2017-08-29:
@davidc502 Thanks for your information.

aaron1959, 2017-08-29:
@ davidc502 are you talking about LEDE? If so I nowthink this issue is unique to DD-WRT so I don't think there is any point in going further here. 
I can try to give this information to BrainSlayer (at risk of having the fan club tell me I'm wrong) but I think he basically said my router may be defective,  yet it works in that respect with the LEDE you posted so I'm at a loss. 
but I really think we have beat the crap out of this horse, and It died for me. 
Bye for now.

s-pimenta, 2017-08-29:
The 2017.06.06 Driver is just fine for me both on LEDE/OpenWrt and DD-WRT... 

I also just give up... almost 1 month reporting a bug and seems to be ignored... To me doesn't matter if the Wi-Fi signal is reported in the router... if it the router works it's fine for me... 

Even with the lastest OEM Linksys Firmware (i think is 1.0.6) have also wifi disconnects and lockups that needs manual reboot... 

Paid a lot of money for this router... 5 months latter the router is not stable...  (for the same price could buy a enterprise grade router) 

Linksys recently released a "new" router the Linksys WRT32X that's only a rebranded black router with some broken firmware with optimization for gaming... The specs ARE EXACLY THE SAME!!! I bet this new router firmware can be installed on the WRT3200ACM... 

@aaron1959 if you want the old 2017.06.06 wifi driver I can send to you, both LEDE/OpenWrt or DD-WRT

s-pimenta, 2017-08-29:
At this pace a new Wi-Fi standard is out and this router is outdated... 

The development is slow because only @yuhhaurlin can really develop the driver... BINARY BLOBS is just stupid... Marvell / Linksys could be selling this router like hot cakes because of the Open Source community... 

The WRT3200ACM is one of the best routers in terms of specs right now... but great specs mean NOTHING without a GREAT software... so it worthless!

yuhhaurlin, 2017-08-29:
Please just report technical issue here. Without useful log information and steps to reproduce problem, I can't help you. One more thing, this problem won't let router reboot.

yuhhaurlin, 2017-08-29:
BTW, I work for this project for a period of time. I have checked discussion in some communities. It looks like current driver is stable for most of users.

yuhhaurlin, 2017-08-29:
If you think 2017.06.06 is suitable for you, you can just use it. This is open source, you can use any versions you like.

BrainSlayer, 2017-08-29:
@aaron1959 you dont need to inform me. i'm here and following this thread. for @yuhhaurlin is right, for reproducing he needs that you run this on LEDE since he doesnt work with dd-wrt. i just can guarantee that dd-wrt uses the same driver as lede (or at least whats here in github) including the same mac80211 drivers. and beside this i believe this problem is client device specific. my own wrt3200acm runs since a long time always using the latest fw with dd-wrt and does not show up with such problems. for me it looks like the marvell firmware blob is crashing. thats the only thing i know which explains this behaviour. but the firmware doesnt offer a crashlog or other diagnostics i know to track this down.

BrainSlayer, 2017-08-29:
@s-pimenta i know one better router :-) (netgear r9000 with quadcore 1.8 ghz cpu, fibre port and 8 port switch and 2 qca wifi chipsets + 1x 60 ghz wil6210 chipset, but thats another price level and no sata port)
but personally i really want to have the wrt3200acm stable at the end with features like wds sta / ap and all the other goods which are missing right now, but technically possible

aaron1959, 2017-08-29:
@s-pimenta Can you send me that driver in DD-wrt ? aaronramsdell@yahoo.com .
OK I will once again run that last version that yuhhaurlin posted if I can find it? Please yuhhaurlin can you post a link again just so I know for sure I'm testing the exact same version you want me to test. If the Marvel "blob" is failing only in this router it seems that it would fail for all firmware loaded onto the device. so I think I did not run LEDE long enough.
@BrainSlayer  I understand you are watching many threads, I am wondering If there will be DD-wrt for the r9000?

BrainSlayer, 2017-08-29:
@aaron1959 : stupid question :-) since about 6 months (lede doesnt support it by way) ftp://ftp.dd-wrt.com/betas/2017/08-25-2017-r33215/netgear-r9000/

aaron1959, 2017-08-29:
Yea I thought it was stupid after I asked it. :-) I'm looking for one r9000 now. 
Maybe I can get rid of this one with stock linksys on it for $5 USD lol.
Or trade up to a wrt54GS. :-)

BrainSlayer, 2017-08-29:
feeling bad now since belkin supported us financially to support this device. never thought that it would take that long to get a useable driver which is still feature incomplete. hey i still have 2 WRT54Gv5 with 8 mb ram and 2 mb flash. just flashed them with dd-wrt some days so see if its still working and yes it did. but painfull low left memory for sure

s-pimenta, 2017-08-29:
@BrainSlayer What?! Belkin did pay to have support of WRT3200ACM?! Sorry... but that's unacceptable! Taking so long to have an driver working... But the problem is only @yuhhaurlin directly developing the driver... Marvell needs to add more people to speed things up... But everything I say seems to mean nothing to Marvell...

To me Belkin did not pay me anything... got $300 off my pocket...

s-pimenta, 2017-08-29:
@aaron1959 I've just send the firmware images with 2017.06.06 wifi driver via WeTransfer... check your e-mail

BrainSlayer, 2017-08-29:
@s-pimenta well its not unacceptable, because they payed already and its good for supporting our project its also not the fault of belkin that marvell doesnt support opensource that well. for the previous models i got closed sources drivers but the quality was bad. so the mwlwifi approach has much better quality and both drivers are using the same binary blob. i just hope that @yuhhaurlin is able to fix the outstanding problems in near future. i mean its now more than a year that i received the wrt3200acm sample device (its was long before market start) and we are still not at the same level as the ath10k driver which is problematic in the same way. but the ath10k binary blob is better documented and qca gave me access to the sourcecodes of it, so i was able to help the ath10k project in some ways to get all working

BrainSlayer, 2017-08-29:
@s-pimenta the driver alone will not help. consider the the 06.06 version uses kernel 3.18. but the current dd-wrt version is based on 4.9

Chadster766, 2017-08-29:
@BrainSlayer 06.06 builds fine for kernel 4.9.

yuhhaurlin, 2017-08-29:
Reproduce this problem on LEDE image first. Please stop discussing thing like this. Otherwise, I don't think I will support on this issue anymore.
I don't know what do you mean for major problems. I wait for reproducing this problem for a long time, but what I got is a lot of complains.

yuhhaurlin, 2017-08-29:
If you want to promote other products, please go somewheree else.
Here is for issues of WRT series problems.
BTW, I will only reply problems when I come to office now. Thanks.

BrainSlayer, 2017-08-29:
@yuhhaurlin i dont think that you understand the situation. nobody blamed you for your work. people are just frustrated that there is no progress in alot of points. no matter if its wds or client chipset specific connection problems (see ESP chipset) and we all think that its the lack of support from marvell for the opensource community and i say again, we dont think that this is your fault. you're simply just alone here and this might be not enough from the efforts marvell is doing.

yuhhaurlin, 2017-08-29:
1. WDS will be done later due to other high priority jobs.
2. ESP: someone said it is working, so I close it. For previous debugging, I can make sure connection is established and I want user to make sure data is working, but he can make sure that.
3. For this issue, I need the problem to be reproduced on LEDE image, but what I got is same messages from dd-WRT.

If things go like this, I really don't know what should I do.

yuhhaurlin, 2017-08-29:
Marvell fully supports this project.

yuhhaurlin, 2017-08-29:
Currently the driver is tested with LEDE image. I don't think I have time to get familiar with dd-WRT.  This issue must be reproduced on LEDE first before debugging it.

yuhhaurlin, 2017-08-29:
Especially this issue is almost not to happen for other users.

eduperez, 2017-08-29:
I can confirm this works on LEDE: I have a WRT3200ACM as a AP for my network, plus a WRT4300 as a STA to extend the WIFI range, both running LEDE,, and the connection simply does not drop.

BrainSlayer, 2017-08-29:
in addition something may compare the hostapd configurations in lede and dd-wrt. both might differ. especially with vht_caps

aaron1959, 2017-08-29:
> in addition something may compare the hostapd configurations in lede and dd-wrt. both might differ. especially with vht_caps
So I don't know what I can do here?

BrainSlayer, 2017-08-29:
just compare the hostapd configs in both systems. especially the line "vht_capab" as i said
on the qca system in dd-wrt it looks like this
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][DSSS_CCK-40][MAX-AMSDU-7935]
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMER][MU-BEAMFORMER][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7][BF-ANTENNA-2][SOUNDING-DIMENSION-2][BF-ANTENNA-3][SOUNDING-DIMENSION-3]

(my wrt3200acm isnt on right now)
in dd-wrt you will find it in the /tmp directory. ath0_hostapd.conf for example

gufus, 2017-08-29:
YES!

I agree.

[15 more replies]

closed 2017-09-04 by yuhhaurlin