kaloz/mwlwifi #211: The kernel does not display the power table

jbsky, 2017-08-30:
Hello, @yuhhaurlin 

Don't worry, the pilot seems to be working well. However, I would like to be able to adjust the signal strength. First of all, I would like to point out that:
* The router is the WRT1900AC v1, 88W8864
* The dts sheet is filled in with the original power table.
* I'm using the latest driver.

With the following command: 
`cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/device_pwrtbl`

 I have:
`power table loaded from dts: yes`
`firmware region code: 0x0`
`number of channel: 0`

Yet in the code, normally, the table is displayed but not here.

Do you have any leads? Is that normal?

Thank you for considering my request.

yuhhaurlin, 2017-08-31:
This is correct behavior. This debugfs file is used to dump power table programmed with your device. Because your device does not programmed with device power table, so it won't be displayed.
For your device, power table is loaded from DTS.

yuhhaurlin, 2017-08-31:
I close this issue.

closed 2017-08-31 by yuhhaurlin

jbsky, 2017-08-31:
Okay, now I wish I could change the signal strength. I wish, if it is possible that you could explain the composition of the table to us. At the limit, I may have understood the first column that represents the channel number.

`[...]`
`			mwlwifi {`
`			marvell,5ghz = <0>;`
`			marvell,chainmask = <4 4>;`
`			marvell,powertable {`
`				FCC =`
`					<1 0 0x16 0x15 0x14 0x13 0x12 0x11 0x10 0x9 0x8 0x7 0x6 0x5 0x4 0x3 0x2 0x0 0 0xf>,`
`					<2 0 0x16 0x15 0x14 0x13 0x12 0x11 0x10 0x9 0x8 0x7 0x6 0x5 0x4 0x3 0x2 0x0 0 0xf>,`
`					<3 0 0x16 0x15 0x14 0x13 0x12 0x11 0x10 0x9 0x8 0x7 0x6 0x5 0x4 0x3 0x2 0x0 0 0xf>,`
`[...]`

Being in France, I use the ETSI part, not the FCC part. The only parameter in the influential dts file is  marvell,5ghz = <0>;

Then to change the signal, I use the following commands to test:
iw wlan1 set txpower fixed 100
iw wlan1 set txpower fixed 1600

But it has no effect. Is it possible to change the signal strength? Do I have to switch off and restart the wifi card or access point to take the setting into account?

* To see if the signal strength changes, I use Acrylic.
* Personally, I'm only interested in the 5Ghz part.

Thank you @yuhhaurlin

yuhhaurlin, 2017-08-31:
Sorry. I am not the right person to ask this question.

jbsky, 2017-08-31:
I think you're the right person because it concerns the driver of the wifi card.

yuhhaurlin, 2017-08-31:
The driver doesn't not support user to change power level.