kaloz/mwlwifi #212: 2.4 GHz radio crash (stopped accepting new clients)

s-pimenta, 2017-08-30:
Another client (tablet MAC: f8:32:e4:0f:6c:C9) connection has dropped and 2.4 GHz radio router stopped accepting new clients connections. 2.4Ghz SSID still broadcasting but no one can connect to it... Ethernet and 5Ghz radio is fine and still working... Manual router reboot solves the problem. 

Running: WRT-3200ACM-EU
5GHz (channel 100), WPA2 802.1X (CCMP) EAP-PWD
2.4GHz (channel 11) WPA2 802.1X (CCMP) EAP-PWD
Lede Reboot SNAPSHOT r4707-3e6d303 / LuCI Master (git-17.231.35495-ad81dc1)
Kernel Version: 4.9.44
mwlwifi: 2017-08-10

```
root@LEDE:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:b9:0b:29
2g: enable
5g: disable
antenna: 4 4
irq number: 46
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1480000
iobase1: e1700000
tx limit: 1024
rx limit: 16384
```
Before the problem happened the syslog says this about the client at **21:43**

```
Wed Aug 30 21:43:34 2017 daemon.debug hostapd: wlan1: WPA rekeying GTK
Wed Aug 30 21:43:34 2017 daemon.debug hostapd: wlan1: STA f8:32:e4:0f:6c:c9 WPA: sending 1/2 msg of Group Key Handshake
Wed Aug 30 21:43:35 2017 daemon.debug hostapd: wlan1: STA f8:32:e4:0f:6c:c9 WPA: EAPOL-Key timeout
Wed Aug 30 21:43:35 2017 daemon.debug hostapd: wlan1: STA f8:32:e4:0f:6c:c9 WPA: sending 1/2 msg of Group Key Handshake
Wed Aug 30 21:43:35 2017 daemon.debug hostapd: wlan1: STA f8:32:e4:0f:6c:c9 WPA: received EAPOL-Key frame (2/2 Group)
Wed Aug 30 21:43:35 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 WPA: group key handshake completed (RSN)
Wed Aug 30 21:43:35 2017 daemon.debug hostapd: wlan1: STA f8:32:e4:0f:6c:c9 WPA: received EAPOL-Key 2/2 Group with unexpected replay counter
```

A few minutes (at 21:50) the syslog is full of this: (syslog is so full of this that I can't see logs before that hour)
```
Wed Aug 30 22:28:36 2017 kern.err kernel: [15918.613620] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:36 2017 kern.err kernel: [15918.793616] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:36 2017 kern.err kernel: [15918.853615] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:36 2017 kern.err kernel: [15919.003613] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:36 2017 kern.err kernel: [15919.063612] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.229806] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.234629] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.239440] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.243610] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.249846] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.254661] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.259479] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.264437] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.269247] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.274067] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.278877] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.283694] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.288504] ieee80211 phy1: adapter does not exist
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.293613] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.453605] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.523603] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.683601] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.763598] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.923596] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.993595] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:38 2017 kern.err kernel: [15920.153594] ieee80211 phy1: buffer is NULL for tx done ring
Wed Aug 30 22:28:38 2017 kern.err kernel: [15920.233591] ieee80211 phy1: buffer is NULL for tx done ring
```
Seems this was already reported https://github.com/kaloz/mwlwifi/issues/196

yuhhaurlin, 2017-08-30:
Adapter is not existed, it means your HW has problem. Please check if it will happen again. I close this one.

s-pimenta, 2017-08-31:
This problem has happened several times!  with this particular device (f8:32:e4:0f:6c:C9)

Why you close so fast?

yuhhaurlin, 2017-08-31:
It shows your PCI WiFi adapter is done.

yuhhaurlin, 2017-08-31:
Wed Aug 30 22:28:37 2017 kern.err kernel: [15919.249846] ieee80211 phy1: adapter does not exist => Driver can't access register of PCI adapter.

s-pimenta, 2017-08-31:
What do you mean by "your PCI WiFi adapter is done"? Has a hardware defect? 

How this can be possible? This problem only happens with this specific client, otherwise the 2.4Ghz is fine! 

This problem has been reported previously and since this specific client is used by a person that can't tolerates Internet connection interruptions, stopped this client to connecting to the Linksys router by several weeks... 

Since then the 2.4GHz adapter never happened this kind of problem! 

So... the software is faulty not the hardware... 

I will try to give more logs, because the syslog get flooded with this error and I can't see any further back...

yuhhaurlin, 2017-08-31:
I reopen it. But the problem is pretty strange. I will put it as low priority.

yuhhaurlin, 2017-08-31:
I just feels confused. The log shows driver can't access registers of WiFi adapter. If this only happens for this client and you can't give other kind of log messages, I don't think I can fix it.

yuhhaurlin, 2017-08-31:
But  how can client makes this kind of problem happed? I have no idea.

BrainSlayer, 2017-08-31:
@s-pimenta can you post a full dmesg? if the hardware is fault we should see pci errors in the kernel log. or a pci bus reset. @yuhhaurlin things may happen if the hardware shuts down for some reason. hard to track for sure. but lets see what the kernel log says. if the hardware is fault the kernel should print out pci errors in a particular way

yuhhaurlin, 2017-08-31:
Normally I don't think I can fix it. No kernel panic, no other kind of error log, but driver can't access this adapter. Maybe he can give more log messages.

BrainSlayer, 2017-08-31:
if pci faults the hw itself must be crashed. which is more than just a firmware bug. this is something for deeper research if reproduceable with a specifific client

yuhhaurlin, 2017-08-31:
It is hard to believe client will make this kind of problem happen!

BrainSlayer, 2017-08-31:
hard to believe, yes. but interesting issue anyway if he is true and i happens only with a specific client

BrainSlayer, 2017-08-31:
@s-pimenta can you tell me which client hardware you're using?

yuhhaurlin, 2017-08-31:
If he can try another WRT3200ACM, it would be better.

BrainSlayer, 2017-08-31:
@yuhhaurlin yes but not many people have 2 of this expensive device. i also have just one and this is a preproduction model

BrainSlayer, 2017-08-31:
i'm doing a private research right now and we have seen a indicator for a heat issue. in dd-wrt we can monitor the cpu and wifi sensors and it seems that some problems do not occur with a additiona cooling solution. but still need time to proof this

yuhhaurlin, 2017-08-31:
All right. If you can get this client and prove this problem can happen, I will try to reproduce it here.

yuhhaurlin, 2017-08-31:
If what you said is true, the problem should not just happen for a specific client. Whatever, I will wait for you to reproduce the problem with this client. Thanks.

BrainSlayer, 2017-08-31:
depends on the phy operation mode and modulation scheme. the biggest heat source is the phy unit.
wrong antennas may also create heat issues (standing wave issue)

s-pimenta, 2017-08-31:
@yuhhaurlin 
> It is hard to believe client will make this kind of problem happened!

2 days ago I was trying to connect the problematic client for the first time _since 17 July_ (https://github.com/kaloz/mwlwifi/issues/196), to the 2.4GHz (_the client doesn't have 5GHz radio_), and I simply cannot connect, no IP nothing, looking like the password credentials was wrong... tryed several 2-3 times to check if I misstyped the password and no connect... checked the router syslog.... the log flooded of:
```
phy1: buffer is NULL for tx done ring
kern.err kernel: [15919.229806] ieee80211 phy1: adapter does not exist
```
So the **client causes the crash**...

Last time happened (yesterday) after connected by about 30min (light use, some facebook and youtube), 2.4GHz crashed again... 

I noticed also at the same time also caused my tablet to loss connection on 5GHz... reconnect solved the issue (_maybe is related to **wpa_group_rekey**_ https://github.com/kaloz/mwlwifi/issues/210)

@BrainSlayer 
> @s-pimenta can you tell me which client hardware you're using?

**Asus ZenPad S 8.0 Z580C**
This models varies a lot... the USA version seems to have Wifi AC adapter, but this one doesn't have, BGN only...

https://www.asus.com/pt/Tablets/ASUS_ZenPad_S_80_Z580C/specifications/

The box says 
[image]

@yuhhaurlin 

> If he can try another WRT3200ACM, it would be better.

No I don't have other Linksys WRT3200ACM router...

@BrainSlayer 
> i'm doing a private research right now and we have seen a indicator for a heat issue. in dd-wrt we can monitor the cpu and wifi sensors and it seems that some problems do not occur with a additiona cooling solution. but still need time to proof this

Maybe I suspect that, the routers seems too hot even when idle (nothing connected to the router), having a CPU always at about 80 ºC is not very healthy... they say is a SoC up to 105 ºC ... Missing a proper CPU scaling... (https://forum.lede-project.org/t/cpu-frequency-scaling-driver-for-mvebu-wrt3200acm-etc/2808)

I can try put a fan to see if the problem happens again... but this only happens to an specific client...

@BrainSlayer 

> depends on the phy operation mode and modulation scheme. the biggest heat source is the phy unit.
> wrong antennas may also create heat issues (standing wave issue)

Using stock antennas, with proper ventilation (on top of an furniture, never with direct sun light, room at about 26ºC)

[image]

I will try to reproduce the problem to give more detailed logs...

Do I need to change anything to give full **dmesg**?

BrainSlayer, 2017-08-31:
mine is also at 80 degrees. thats normal. back do your client. can you check the wifi chipset? simple look at the driver which issed for that chipset. that will help,

ValCher1961, 2017-09-01:
The sensors utility show that the temperature in the WIFI modules is approximately 50 C, but I feel it is somewhat higher. Tonight, I'll check the real temperature of the Wifi modules with a contact sensor.

ValCher1961, 2017-09-01:
Measured the temperature on the radiator wlan0 in an inactive and active state.
The temperature of the inactive wlan0 measured by the external thermocouple was consistent with the utility "sensor" evidence and was 47 °C.
With the active use of Wlan0 (iperf3), the temperature at the external thermocouple rose to 60 °C, while the utility "sensors" showed a temperature of 50 °C, i.e. the difference was approximately 10 °C.

Amazing temperature behavior.

gufus, 2017-09-01:
I'm sure you know wlan0 is the 2g radio.

ValCher1961, 2017-09-01:
It's equivalent to the WLAN1-2.4 GHz!

gufus, 2017-09-01:
@ ValCher1961 

FYI
https://www.smallnetbuilder.com/wireless/wireless-reviews/33030-linksys-wrt3200acm-ac3200-mu-mimo-gigabit-wi-fi-router-reviewed

ValCher1961, 2017-09-01:
And what?
https://github.com/ValCher1961/McDebian_WRT3200ACM/tree/master/hardware-mods

s-pimenta, 2017-09-03:
Happened again, but this time after 2min the crash happened the router has rebooted by itself

Logs from USB pendrive:

**Syslog:**
```
Sun Sep  3 20:42:07 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 WPA: group key handshake completed (RSN)
Sun Sep  3 20:42:07 2017 daemon.info hostapd: wlan1: STA c4:b3:01:03:9d:66 WPA: group key handshake completed (RSN)

Sun Sep  3 20:43:10 2017 daemon.info hostapd: wlan0: STA 74:23:44:5a:a1:f6 WPA: group key handshake completed (RSN)

Sun Sep  3 20:44:11 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 IEEE 802.11: associated (aid 1)
Sun Sep  3 20:44:11 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 8c:3a:e3:56:71:77
Sun Sep  3 20:44:11 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 WPA: pairwise key handshake completed (RSN)
Sun Sep  3 20:44:11 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 IEEE 802.11: authenticated

Sun Sep  3 20:44:15 2017 daemon.info dnsmasq-dhcp[3524]: DHCPDISCOVER(br-lan) 8c:3a:e3:56:71:77 
Sun Sep  3 20:44:15 2017 daemon.info dnsmasq-dhcp[3524]: DHCPOFFER(br-lan) 192.168.1.233 8c:3a:e3:56:71:77 
Sun Sep  3 20:44:15 2017 daemon.info dnsmasq-dhcp[3524]: DHCPDISCOVER(br-lan) 8c:3a:e3:56:71:77 
Sun Sep  3 20:44:15 2017 daemon.info dnsmasq-dhcp[3524]: DHCPOFFER(br-lan) 192.168.1.233 8c:3a:e3:56:71:77 
Sun Sep  3 20:44:15 2017 daemon.info dnsmasq-dhcp[3524]: DHCPREQUEST(br-lan) 192.168.1.233 8c:3a:e3:56:71:77 
Sun Sep  3 20:44:15 2017 daemon.info dnsmasq-dhcp[3524]: DHCPACK(br-lan) 192.168.1.233 8c:3a:e3:56:71:77 android-b5be18f4f72a8eaf

Sun Sep  3 20:48:22 2017 user.warn igmpproxy[3083]: MRT_DEL_MFC; Errno(2): No such file or directory

Sun Sep  3 20:49:48 2017 daemon.notice hostapd: wlan1: AP-STA-POLL-OK f8:32:e4:0f:6c:c9

Sun Sep  3 20:52:06 2017 daemon.info hostapd: wlan1: STA c4:b3:01:03:9d:66 WPA: group key handshake completed (RSN)
Sun Sep  3 20:52:07 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 WPA: group key handshake completed (RSN)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
This MAC is the same tablet (Asus ZenPad S 8.0 Z580C) causing the problem?! f8:32:e4:0f:6c:c9

---------------(seconds latter)-----------------

Sun Sep  3 20:52:28 2017 kern.err kernel: [10238.149979] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:52:28 2017 kern.err kernel: [10238.299976] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:52:28 2017 kern.err kernel: [10238.359989] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:52:29 2017 kern.err kernel: [10238.549973] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:52:29 2017 kern.err kernel: [10238.609971] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:52:29 2017 kern.err kernel: [10238.749973] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:52:29 2017 kern.err kernel: [10238.819967] ieee80211 phy1: buffer is NULL for tx done ring

---------------------(Removed some repeated lines)---------------------

Sun Sep  3 20:52:32 2017 kern.err kernel: [10241.979916] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.039914] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.179913] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.239912] ieee80211 phy1: buffer is NULL for tx done ring

-----------------------(PCI ERROR?!)-------------------

Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.265546] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.271409] ieee80211 phy1: return code: 0x001d
Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.275956] ieee80211 phy1: timeout: 0x001d

Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.280168] ieee80211 phy1: adapter does not exist
Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.284977] ieee80211 phy1: adapter does not exist
Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.289786] ieee80211 phy1: adapter does not exist
Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.294600] ieee80211 phy1: adapter does not exist
Sun Sep  3 20:52:32 2017 user.warn igmpproxy[3083]: MRT_DEL_MFC; Errno(2): No such file or directory
Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.389912] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:52:32 2017 kern.err kernel: [10242.459908] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:52:33 2017 kern.err kernel: [10242.609908] ieee80211 phy1: buffer is NULL for tx done ring

-------------------------(Removed some repeated lines)-----------------------------

Sun Sep  3 20:53:10 2017 kern.err kernel: [10280.289369] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:53:10 2017 kern.err kernel: [10280.349369] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 20:53:10 2017 kern.err kernel: [10280.519366] ieee80211 phy1: buffer is NULL for tx done ring
[... 40 lines cut ...]
[10242.265546] ieee80211 phy1: cmd 0x801d=MEMAddrAccess timed out
[10242.271409] ieee80211 phy1: return code: 0x001d
[10242.275956] ieee80211 phy1: timeout: 0x001d
[10242.280168] ieee80211 phy1: adapter does not exist
[10242.284977] ieee80211 phy1: adapter does not exist
[10242.289786] ieee80211 phy1: adapter does not exist
[10242.294600] ieee80211 phy1: adapter does not exist
[10242.389912] ieee80211 phy1: buffer is NULL for tx done ring
[10242.459908] ieee80211 phy1: buffer is NULL for tx done ring
```

[... cut ...]
Sun Sep  3 19:02:07 2017 daemon.info hostapd: wlan1: STA c4:b3:01:03:9d:66 WPA: group key handshake completed (RSN)

Sun Sep  3 19:02:15 2017 kern.err kernel: [ 3624.683300] ieee80211 phy0: create ba result error 1
```

**Don't know what to say more to help debug this...** Maybe test with a fan to help cool it down?

@yuhhaurlin This problem needs to be addressed (_this is serious problem_), maybe is related to what @BrainSlayer said of Wifi driver crashing... https://github.com/kaloz/mwlwifi/issues/210#issuecomment-326803367

gufus, 2017-09-03:
IMHO

It's that "ba" error again

s-pimenta, 2017-09-03:
Just after the previous comment posted, put the USB pen drive again and SSH to DMESG and LOGREAD to a file, and the problem **appeared again!!**

But this time is a _bit different_... **No one was connected to 2.4GHz** and only 1 client connected to 5GHz, and then... **1-2min later get kicked off 5GHz**... the router did **NOT rebooted**, checked what is happening and saw this:

Syslog:
```
Sun Sep  3 21:37:07 2017 daemon.info hostapd: wlan0: STA 74:23:44:5a:a1:f6 WPA: group key handshake completed (RSN)
Sun Sep  3 21:37:37 2017 authpriv.info dropbear[4525]: Exit (root): Disconnect received
Sun Sep  3 21:38:34 2017 daemon.info dnsmasq-dhcp[3457]: DHCPREQUEST(br-lan) 192.168.1.157 40:b0:34:c1:ce:05 
Sun Sep  3 21:38:34 2017 daemon.info dnsmasq-dhcp[3457]: DHCPACK(br-lan) 192.168.1.157 40:b0:34:c1:ce:05 spimenta-laptop
Sun Sep  3 21:38:36 2017 daemon.warn odhcpd[1487]: DHCPV6 CONFIRM IA_NA from 000423da6e23abb9a93645a8f6e25c3593ca on br-lan: not on-link 2001:8a0:e475:8f00::dfe/128 fdf8:3f9d:8154::dfe/128 
Sun Sep  3 21:38:36 2017 daemon.warn odhcpd[1487]: DHCPV6 SOLICIT IA_NA from 000423da6e23abb9a93645a8f6e25c3593ca on br-lan: ok 2001:8a0:e475:8f00::dfe/128 fdf8:3f9d:8154::dfe/128 
Sun Sep  3 21:38:37 2017 daemon.warn odhcpd[1487]: DHCPV6 REQUEST IA_NA from 000423da6e23abb9a93645a8f6e25c3593ca on br-lan: ok 2001:8a0:e475:8f00::dfe/128 fdf8:3f9d:8154::dfe/128 
Sun Sep  3 21:40:50 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 IEEE 802.11: associated (aid 1)
Sun Sep  3 21:40:50 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 8c:3a:e3:56:71:77
Sun Sep  3 21:40:50 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 WPA: pairwise key handshake completed (RSN)
Sun Sep  3 21:40:51 2017 daemon.info hostapd: wlan0: STA 30:c7:ae:02:a8:32 IEEE 802.11: authenticated
Sun Sep  3 21:40:51 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 IEEE 802.11: authenticated
Sun Sep  3 21:40:56 2017 daemon.info dnsmasq-dhcp[3457]: DHCPDISCOVER(br-lan) 8c:3a:e3:56:71:77 
Sun Sep  3 21:40:56 2017 daemon.info dnsmasq-dhcp[3457]: DHCPOFFER(br-lan) 192.168.1.233 8c:3a:e3:56:71:77 
Sun Sep  3 21:40:56 2017 daemon.info dnsmasq-dhcp[3457]: DHCPDISCOVER(br-lan) 8c:3a:e3:56:71:77 
Sun Sep  3 21:40:56 2017 daemon.info dnsmasq-dhcp[3457]: DHCPOFFER(br-lan) 192.168.1.233 8c:3a:e3:56:71:77 
Sun Sep  3 21:40:56 2017 daemon.info dnsmasq-dhcp[3457]: DHCPREQUEST(br-lan) 192.168.1.233 8c:3a:e3:56:71:77 
Sun Sep  3 21:40:56 2017 daemon.info dnsmasq-dhcp[3457]: DHCPACK(br-lan) 192.168.1.233 8c:3a:e3:56:71:77 android-b5be18f4f72a8eaf

Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.451235] ieee80211 phy1: unknown control case: 31

----------------(unknown control case: 31)-----------------

Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.456292] ieee80211 phy1: slow: space 3774(8501) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.462335] ieee80211 phy1: slow: space 3774(46022) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.468711] ieee80211 phy1: slow: space 3774(8352) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.474756] ieee80211 phy1: slow: space 3774(41259) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.480879] ieee80211 phy1: slow: space 3774(52749) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.486997] ieee80211 phy1: slow: space 3774(12965) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.493239] ieee80211 phy1: slow: space 3774(14003) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.499480] ieee80211 phy1: slow: space 3774(58706) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.505600] ieee80211 phy1: slow: space 3774(54075) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.511726] ieee80211 phy1: slow: space 3774(35063) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.517846] ieee80211 phy1: slow: space 3774(9368) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.523888] ieee80211 phy1: slow: space 3774(27465) is not enough

-------------------------(Removed some repeated lines)-----------------------------

Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.324118] ieee80211 phy1: slow: space 3774(17394) is not enough
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.330244] ieee80211 phy1: slow: space 3774(12688) is not enough
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.336377] ieee80211 phy1: slow: space 3774(6672) is not enough
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.342429] ieee80211 phy1: slow: space 3774(12912) is not enough
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.348556] ieee80211 phy1: slow: space 3774(60530) is not enough
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.354686] ieee80211 phy1: slow: space 3774(58668) is not enough
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.358273] ieee80211 phy1: adapter does not exist
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.358275] ieee80211 phy1: adapter does not exist
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.358278] ieee80211 phy1: adapter does not exist
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.358280] ieee80211 phy1: adapter does not exist
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.358283] ieee80211 phy1: adapter does not exist
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.358285] ieee80211 phy1: adapter does not exist
Sun Sep  3 21:43:06 2017 kern.err kernel: [ 2839.358413] ieee80211 phy1: adapter does not exist
[... cut ...]
Sun Sep  3 21:53:25 2017 kern.err kernel: [ 3458.678991] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 21:53:25 2017 kern.err kernel: [ 3458.818987] ieee80211 phy1: buffer is NULL for tx done ring
Sun Sep  3 21:53:25 2017 kern.err kernel: [ 3458.878987] ieee80211 phy1: buffer is NULL for tx done ring
```
And then manualy rebooted the router...

@yuhhaurlin **This need to addressed ASAP!** To try something different (and to stop this problem of happening) I will **DISABLE the 2.4GHz radio**, and see what happens... 

This problem only happens on the 2.4GHz radio...

s-pimenta, 2017-09-03:
The **ieee80211** seems to report most of the problems... There is a way to get a debug or more verbose?

yuhhaurlin, 2017-09-03:
This is the same router as previous one you report connection lost?
The problem happens too quickly and it shows driver can't access the 2g adapter.
If disable 2g and this problem, I suspect it is problem of 2g adapter. Especially you said that no client connected to 2g and the problem still happen.

s-pimenta, 2017-09-04:
> This is the same router as previous one you report connection lost?

Yes

> The problem happens too quickly and it shows driver can't access the 2g adapter.
> If disable 2g and this problem is gone, I suspect it is problem of 2g adapter. Especially you said that no client connected to 2g and the problem still happen.

Yes, it seems that the 2g adapter is broken, but @BrainSlayer said he have seen this problem is several logs... so this is a common problem? I could RMA this router, but this is common seems not solving the issue, the 2.4GHz has a defect...

This can be an overheat problem, I will try to put a fan on top of the router to see if this problem occurs or not...

s-pimenta, 2017-09-04:
```
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.451235] ieee80211 phy1: unknown control case: 31
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.456292] ieee80211 phy1: slow: space 3774(8501) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.462335] ieee80211 phy1: slow: space 3774(46022) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.468711] ieee80211 phy1: slow: space 3774(8352) is not enough
Sun Sep  3 21:43:04 2017 kern.err kernel: [ 2837.474756] ieee80211 phy1: slow: space 3774(41259) is not enough
```

Seems @seelrat2000 also had the same log I had, and you said to ignore?!

https://github.com/kaloz/mwlwifi/issues/181

BrainSlayer, 2017-09-04:
the space message might also be a out of memory problem. do you have informations about the available ram at this time?

yuhhaurlin, 2017-09-04:
He means host command timeout. Your problem is that adapter is gone. Under this case, unexpected problem could happen.

s-pimenta, 2017-09-04:
@BrainSlayer 

> the space message might also be a out of memory problem. do you have informations about the available ram at this time?

I checked the RAM and seems to be normal usage (not full), but I checked after the message (ieee80211 phy1: slow: space 3774(18938) is not enough) disappeared, not at the moment that the message is appearing...

-----------------------------

@yuhhaurlin seems @aaron1959 also have the same **cmd 0x801d=MEMAddrAccess timed out** https://github.com/kaloz/mwlwifi/issues/209#issuecomment-325515084

Did you noticed that this problem **always happens on phy1 = 2g adapter**?

@yuhhaurlin And you also saying the **router is defective**.... https://github.com/kaloz/mwlwifi/issues/209#issuecomment-325539295

**Seems Marvel did make a defective 2g adapter... or.... _most likely an defective driver_....** :angry: 

----------------------

I will try to do a **iperf** test to stress test the 2g adapter, and see if the problem happens again, and then put a fan on top with **iperf** running if does happen again...

**Maybe I need to return this router and choose another one...** but this is the only router that has DFS...

Asus just launched an _802.1AX_ Wifi router.... (https://www.snbforums.com/threads/and-so-it-begins-asus-announces-first-802-11ax-router.41020/)  just kidding :smile: I don't have devices that supports it... Maybe just buy Ubiquiti AP's, and have an PfSense server...

yuhhaurlin, 2017-09-04:
You always mixed a lot of things. I don't know what to say. I don't think your problem is the normal one. I can't help you. Sorry.

yuhhaurlin, 2017-09-04:
You can take a look at the code to know under what condition the message "Adapter is not existed". When I work for you about problem related list connection, the final result is just "setting error". If you let me know exactly your setting first, maybe it won't waste a lot of my times.

yuhhaurlin, 2017-09-04:
I am pretty busy. The reason why I check these issues is that I think they could be severe if they are real problems and everyone will encounter the same issues.
For host command timeout, I need the problems to be reproduced on LEDE image, but no one can reproduce it.
About this issue, I never heard anyone encounter it and from the log message it shows adapter is gone even though there is no client connected to it. Obviously the router won't encounter problem if 2g is disabled. If it is really problem of driver, I think everyone will encounter the same problem.

s-pimenta, 2017-09-04:
When this problem happens the 2g was always enabled...
Since the 2g is the problem, yesterday I disabled it, never happened again...

I'm using LEDE (davidc502) image, with latest wifi driver 
The only setting I put was 5GHz at channel 100 with WPA2 CCMP
and the 2.4GHz at channel 11 with also WPA2 CCMP
**I don't know what I doing wrong now...**

```
root@LEDE:~# cat /sys/kernel/debug/ieee80211/phy1/mwlwifi/info 

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x30
mac address: 60:38:e0:b9:0b:29
2g: enable
5g: disable
antenna: 4 4
irq number: 46
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1480000
iobase1: e1700000
tx limit: 1024
rx limit: 16384
```

```
root@LEDE:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11a'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0'
	option htmode 'VHT80'
	option channel '100'
	option country 'FR'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'dd-wrt 5GHz'
	option encryption 'psk2+ccmp'
	option key 'password'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '11'
	option hwmode '11g'
	option path 'soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
	option htmode 'HT20'
	option country 'FR'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'dd-wrt 2.4GHz'
	option encryption 'psk2+ccmp'
	option key 'password'

```

I cannot use 802.1X, now I cannot use 2g....

BrainSlayer, 2017-09-04:
@yuhhaurlin : i know you concern that people can't reproduce it with lede, but as it seems just a few people are around here and so far just a few are using lede. i always tell them to try the  test scenario with it to get it reproduced, but so far people still running my firmware with it which isnt really helpfull for you. could you add some diagnostic code i might include in my testversions?

yuhhaurlin, 2017-09-04:
I need to check later.

yuhhaurlin, 2017-09-05:
@s-pimenta 

static bool pcie_chk_adapter(struct pcie_priv *pcie_priv)
{
	struct mwl_priv *priv = pcie_priv->mwl_priv;
	u32 regval;

	regval = readl(pcie_priv->iobase1 + MACREG_REG_INT_CODE);

	if (regval == 0xffffffff) {
		wiphy_err(priv->hw->wiphy, "adapter does not exist\n");
		return false;
	}

	return true;
}

yuhhaurlin, 2017-09-06:
I close this one.

closed 2017-09-06 by yuhhaurlin

s-pimenta, 2017-09-16:
After 2 weeks of testing, finaly **I FOUND WHAT CAUSES THE PROBLEM**!!!

After some tests, and 11 days of router uptime, the problem is by using: **Channel 100 on 5GHz at 80MHz** bandwidth!! Believe or not this **IS THE CAUSE OF THE PROBLEM!** Crashes reboots everything...

during the 11 days of uptime started with the 2g disabled, and 5g at 20mhz at channel 36... router just fine
then enabled the 2g ch 11 at 20mhz and 5g at ch 36 at 20mhz... router fine
2g ch 11 at 20mhz and put 5g CH:36 at 80mhz.... router just fine
2g ch 11 at 20mhz and put 5g CH:52 (a DFS channel) at 20mhz.... router just fine
2g ch 11 at 20mhz and put 5g CH:52 at 80mhz.... router just fine....

after 10days of uptime...
2g ch 11 at 20mhz and put 5g CH:100 at 80mhz... after 2 days.... REBOOT!!!!

If you want I can create a new issue to address this problem...

[image]

yuhhaurlin, 2017-09-16:
2g and 5g are two different 88W8964 modules.
I don't think channel setting as 100 will make the router reboot. Did you get any log messages before the reboot?

pmcduff, 2017-09-17:
This happened to me after I configured wifi LEDs on LEDE Reboot 17.01.2 r3435-65eec8bd5f.

s-pimenta, 2017-09-19:
The log of the last time has happened, has nothing on pen drive related to the crash...

```
Sat Sep 16 18:52:41 2017 daemon.info hostapd: wlan1: STA c0:ee:fb:20:60:c0 IEEE 802.11: authenticated
Sat Sep 16 18:55:05 2017 user.warn igmpproxy[2114]: MRT_DEL_MFC; Errno(2): No such file or directory
Sat Sep 16 18:56:22 2017 daemon.notice netifd: wan (1644): udhcpc: sending renew
Sat Sep 16 18:56:22 2017 daemon.notice netifd: wan (1644): udhcpc: lease of 188.83.231.133 obtained, lease time 3600
Sat Sep 16 18:59:15 2017 user.warn igmpproxy[2114]: MRT_DEL_MFC; Errno(2): No such file or directory
Sat Sep 16 19:00:48 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 IEEE 802.11: associated (aid 2)
Sat Sep 16 19:00:48 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 8c:3a:e3:56:71:77
Sat Sep 16 19:00:48 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 WPA: pairwise key handshake completed (RSN)
Sat Sep 16 19:00:48 2017 daemon.info dnsmasq-dhcp[2544]: DHCPDISCOVER(br-lan) 8c:3a:e3:56:71:77 
Sat Sep 16 19:00:48 2017 daemon.info dnsmasq-dhcp[2544]: DHCPOFFER(br-lan) 192.168.1.233 8c:3a:e3:56:71:77 
Sat Sep 16 19:00:48 2017 daemon.info dnsmasq-dhcp[2544]: DHCPREQUEST(br-lan) 192.168.1.233 8c:3a:e3:56:71:77 
Sat Sep 16 19:00:48 2017 daemon.info dnsmasq-dhcp[2544]: DHCPACK(br-lan) 192.168.1.233 8c:3a:e3:56:71:77 android-b5be18f4f72a8eaf
Sat Sep 16 19:00:51 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 IEEE 802.11: authenticated
Sat Sep 16 19:01:20 2017 daemon.info hostapd: wlan0: STA 74:23:44:5a:a1:f6 WPA: group key handshake completed (RSN)
Sat Sep 16 19:01:21 2017 daemon.info hostapd: wlan0: STA 8c:3a:e3:56:71:77 WPA: group key handshake completed (RSN)
Sat Sep 16 19:02:36 2017 daemon.info hostapd: wlan1: STA 68:48:98:b5:94:3f WPA: group key handshake completed (RSN)
Sat Sep 16 19:02:36 2017 daemon.info hostapd: wlan1: STA c0:ee:fb:20:60:c0 WPA: group key handshake completed (RSN)
Sat Sep 16 19:02:37 2017 daemon.info hostapd: wlan1: STA f8:32:e4:0f:6c:c9 WPA: group key handshake completed (RSN)
                                                                                                     
```
the STA f8:32:e4:0f:6c:c9 is the last message... and guess what? it the same Asus tablet causing the problem on 2g...

But this tablet was connected several times when the router has 11 days of uptime, because I was **not using** channel 100 at 80MHz bandwidth on 5g...

So the channel 100 is the problem... I'm now using on channel 52 at 80MHz bandwidth and never got this problem of random reboots and lockups, and 2g adapter not found, etc...

s-pimenta, 2017-09-19:
Just to clarify the STA f8:32:e4:0f:6c:c9 donesnt have a 5g radio... I know it's weird the 2g STA causing issue with the 5g on channel 100, but that's what's happening... when I change to channel 36 or 52 the reboot/crashes are gone!

yuhhaurlin, 2017-09-19:
The log is normal. I can't understand why router reboot? No Oops and if driver has problem to access adapter or host command timeout, log should show error messages. Router won't reboot if driver can't access adapter or host command timeout.

BrainSlayer, 2017-09-19:
this is syslog. in syslog you see no oops. most people here dont have serial access. i do have, but i just have a view clients for testing which do not suffer from that problem.

s-pimenta, 2017-09-19:
@yuhhaurlin But it rebooted... Sometimes has an total lockup (saying the 2g adapter does not exist), other times simply reboots... I know it is weird, especialy to reproduce the problem, but it what happens... I think I narrow sufficiently to reproduce the problem... This only happens when I set the 5g to channel 100...

Maybe I will void the warranty and attach an serial port with an Arduino to see the logs...

aaron1959, 2017-09-19:
I don't think you need to use an arduino I just left an old laptop hooked up for logging, If you do use an arduino can you send me the code to get it to log serial data? that might be better for me also.
I have an Arduino Uno. 

aaronramsdell@yahoo.com 

I have hours and hours of serial logs but none of the ones I've posted seem to be of use. 
I'm just using Putty I can log the serial as well as ssh at the same time, running two instances of putty.

s-pimenta, 2017-09-19:
@aaron1959 https://wiki.openwrt.org/doc/hardware/port.serial.cables#arduino_uno_rev3_as_usb_to_serial_cable

Since arduino runs on 5v, maybe is not an good idea... Serial to USB are very cheap on Ebay...

In Europe Warranty does not work like that... if the router has some sticker or if the screws has some markings to proof that has been already opened, warranty is gone... But I will investigate if the router has that...

Chadster766, 2017-09-19:
Just pipe (nohup) syslog to a file.

aaron1959, 2017-09-19:
@Chadster766 
Please see this above post by Brain Slayer
"this is syslog. in syslog you see no oops. most people here dont have serial access. i do have, but i just have a view clients for testing which do not suffer from that problem."

s-pimenta, 2017-09-19:
@Chadster766 

Like that?
`nohup | logread -f >> /mnt/usb/wrt3200acm-log &`

aaron1959, 2017-09-19:
@s-pimenta 
Yes I already have a serial to usb cable and that's normally what I'm using, there is no problem that ardurino runs on 5 v uno runs on 3.3 v actually. On the U.S. routers they only have a very small sticker that can be fixed if you are careful. Many are already out of warranty now. arduino will accept the serial data from a wrt3200acm no problem.

[25 more replies]