kaloz/mwlwifi #215: Please open the development process

kubrickfr, 2017-09-18:
Hi,

The driver is open-source, great, but the development happens behind closed doors and has stalled again.
Can you please release the firmware documentation? You can keep your private functions private, but it would be great if you could adopt a open-source development model and create a community?
I'm not criticizing @yuhhaurlin work here, but Marvell has obviously given him just enough time to do the bare minimum and then moved on. So please help us help you.

F.

yuhhaurlin, 2017-09-18:
Sorry. I don't know what you mean. Mwlwifi driver is open source, you can modify it as you want.

closed 2017-09-18 by yuhhaurlin

kubrickfr, 2017-09-18:
haha

BrainSlayer, 2017-09-18:
@yuhhaurlin : he is talking about the private firmware api which is undocumented

yuhhaurlin, 2017-09-18:
@BrainSlayer Private firmware is not open source.

kubrickfr, 2017-09-18:
@yuhhaurlin my point exactly.
and if on top of being closed source the firmware is undocumented, then the _code_ may be open source but it will never be an open source _project_ as only you can develop it further and users will remain at at Marvell's mercy for improvements.
So at the moment we have a driver that is open source but that we can't really understand or maintain. Call that open source if you want...

yuhhaurlin, 2017-09-18:
@kubrickfr If you have project which needs source code of firmware, you need to check with Marvell. This is the same way for most open source wifi drivers. Only driver is open source, firmware is private.

kubrickfr, 2017-09-18:
@yuhhaurlin You misread me, I have not asked for the firmware to be open-sourced. I understand, reluctantly, that the firmware being closed-source is even required for FCC compliance with 5GHz regulations. Fine.
All I'm asking for is for the firmware API **documentation** to be released.

yuhhaurlin, 2017-09-18:
@kubrickfr I don't have the right to own or release the document. Sorry.

kubrickfr, 2017-09-18:
@yuhhaurlin I imagined this was the case, this is why I opened this issue so that you can officially ask for this permission to whomever at Marvell to do so.
The fact that it is a rather common industry practice doesn't make it less counterproductive, I mean, what are people going to do with it that could harm Marvell? Especially since you don't have to document all your secret functions that are such valuable intellectual property... Atheros went as far as opensourcing their firmware https://github.com/qca/open-ath9k-htc-firmware and they didn't go bankrupt as a result...

yuhhaurlin, 2017-09-18:
I work for Marvell. To understand and use firmware APIs is not a trivial job. Most of the time you need to trace source code of firmware to know how to use them correctly. According to current policy, we can't release documents related to firmware to open source.

rs-se, 2017-09-18:
I understand and appreciate all your work here to get this driver open
source, at this time in time we are one to two years away from this open
source driving being fully functional and/or stable.

In the last 13 months, the driver commercial and open source has reached
many great milestones in the stride towards stability and functionality.

It is very sad that it has been 13 months and this driver doesn't fully
function on DFS channels where there is no Government or weather radio
present on selected channels however outside RF interference causes the
chip to skip channels until it drops down to channel 36 rendering use of
DFS channels unusable and MIMO functions nonexistent.

I truly enjoy the WRT3200ACM, however I feel at this point, buying a
router/Wi-Fi bundled device is currently out of the question and all future
devices  instead will be an external Access Point solution with two
standalone routers.

I truly hope Marvell understands the seriousness of this situation of not
allowing the driver to be completely open source with documation of the
Firmware API being available is absolute key to support development of this
driver. The experiences the community has obersved from this driver could
result in people boycottimg Marvell.

I don't mean to complain and you have done lots for us, however
frustrations are shared among others that we could be at a much better
place for stability and functionality of the Wi-Fi chips if Marvell had
treated this project in higher regard.

aaron1959, 2017-09-18:
Maybe Marvell feel that their product is selling very good so they don't care about small details like thousands of users who are not happy. 
I have [u]no choice[/u] I went back to broadcom device where I can find some help.

yuhhaurlin, 2017-09-18:
1. DFS is working for WRT3200ACM.
2. The driver is pretty stable now.
3. MU-MIMO will be done in the future.
4. More features and chips will be supported by mwlwifi.

yuhhaurlin, 2017-09-18:
@aaron1959 Please only report issues here. I am a technical engineer and I only take care of technical issues.

gufus, 2017-09-18:
And when we see WDS working?

rs-se, 2017-09-18:
DFS is working but is not entirely stable when there nearby RF
interference, the chip incorrectly flags it as Government or weather radio
and skips channel(s) unnecessarily.

I would love to help with making this driver more stable and DFS more
functional, at this time I have no means to debug it further. I work for a
telecommunications company that takes up most of my time.

yuhhaurlin, 2017-09-18:
@gufus WDS for WRT3200ACM will be done in the future too.

yuhhaurlin, 2017-09-18:
@rs-se Mwlwifi uses the same firmware of 88W8964 as the stock firmware. Please reproduce the problem on stock firmware and report it to Belkin.

jacksonatl, 2017-09-18:
I must object again to this driver being described as "pretty stable". I
can reliably crash the driver simply by streaming an online radio station.
It happens every time after less than 1 hour of streaming. A 7 year old
Buffalo using the ar71xx chipset has no trouble under the identical load. I
do not have the resources to get serial output, but considering how
reliable the crash is, it cannot be an issue to replicate the load on your
side.

yuhhaurlin, 2017-09-18:
You can create an issue and give detail information for me.

gufus, 2017-09-18:
+1

I need/use wds in my office.

s-pimenta, 2017-09-19:
Also have random lockups and reboots when I use the 5g on channel 100... definitely not "pretty stable", the router is still very finicky... you are very constrained to some specific settings, otherwise you can get problems...

yuhhaurlin, 2017-09-19:
Please give log message. You have gave a lot of different information. I am tired of that. If this driver is not pretty stable, there must be a lot of users open issues here.

s-pimenta, 2017-09-19:
@yuhhaurlin Maybe I give a lot of different information... but I was trying to give the best tips to try to find the source of the problem (since you know better the driver than me...)

haribert, 2017-09-19:
@yuhhaurlin I know, you are only the technical person. But you are our only contact for the opensource statement of the WRT device. So you are probably the person here who could convince Marvell the most, to publish the driver API. It is a huge opportunity for this product if there are people who could support your work and improve the driver. Please make a request at Marvell to publish the necessary documentation.

At the moment I am a little bit disappointed with the progress of the driver. The last commit was at beginning of August and almost all new issues where closed almost immediately. So I would never say that the product is stable if I look at the issues since the last commit and their reported solutions:
[image]

I know you are not concerned about it, but I would not buy the router again. I am even thinking of buying an other one.

Can you give a coarse roadmap? When will the features from the stock firmware be implemented (WDS, MU-MIMO, ...), will these happen in the next week, until end of the year or sometime in 2018? Are you working for mwlwifi only or do you have other projects in your backlog?

yuhhaurlin, 2017-09-20:
Mac80211 only supports MU-MIMO for client, I am curious what open source product will support MU-MIMO for AP mode? 
WDS for 88W8964 is not a trivial job, however, it will be done later.
The remaining issues are not critical and they will be fixed later.
In fact, if any issues I think it is critical, I will check and fix it ASAP.

yuhhaurlin, 2017-09-20:
I am not idle. I am busy on mwlwifi for some features. Once if it is done, code will be merged back to here and pending features and issues for WRT will be implemented and checked later.

aaron1959, 2017-09-20:
Believe me nobody thinks you're idle. Most of us think you can use some help, it's not shameful to ask for or to need help. 
With the exception of myself there are a lot of dedicated and talented people who would be happy to help you,.....  but if Marvell won't let you,  I guess, as we say 'your hands are tied'.

aaron1959, 2017-09-21:
Here is serial of router crash/lockup/malfunction I found that I can run this router foe over one week with absolutely no issues, I tried everything I could think of to crash it but it still kept working. AS SOON as I reintroduced the android running @ 866 @ VHT80 Samsung Galaxy S7 it crashed/lockedup/lost web access, both wired and wireless.
[9_21_17 WRT3200acm CRASH.txt](https://github.com/kaloz/mwlwifi/files/1322239/9_21_17.WRT3200acm.CRASH.txt)

BrainSlayer, 2017-09-21:
@aaron1959 yuhhaurlin refuses todo anything if its not reproduceable with lede. this log is from dd-wrt.  but at the end this again looks like a firmware bug in the mwlwifi firmware since you see a failing message in firmware communication just before the crash

aaron1959, 2017-09-21:
Yes I agree, I just don't understand why he can't use this data. I just guess I'll have to reload LEDE and let it run, but if this data isn't a big clue to the problem I am in great doubt that it will make a difference in LEDE but I can only hope. When I get time I'll mess with it more. It looks to me that there is a firmware problem and it's saying so.

rs-se, 2017-09-22:
I have found a solid way on my WRT3200ACM to replicate a firmware crash of the mwlwifi driver.

iw dev ath0 interface add mon0 type monitor
ifconfig mon0 up
tcpdump -i mon0

Within 5 minutes, the router had a kernal panic and restarted.

yuhhaurlin, 2017-09-22:
The same as issue #60.

haribert, 2017-09-22:
In issue #60 you [write the following](https://github.com/kaloz/mwlwifi/issues/60#issuecomment-324628462):
> If you use WRT3200ACM, you will not encounter this problem. For 88W8864, this problem will be fixed later.

But as far I understand it is happening on a WRT3200ACM device?

And on January 11 you wrote:

> Root cause is found. I will fix this problem later.

We all know that you are working on it, but still, it would be very great if you could publish a coarse roadmap with months or even quarters. What does "later" mean. In the case of issue #60 it is at the moment more than 8 months...

yuhhaurlin, 2017-09-22:
Sorry. For 88W8864, root cause is found. For 88W8964, if user encounters problem, we will check it. But priority will be lower. Thanks.