kaloz/mwlwifi #216: wrt3200acm Wlan crashes then reboots using latest driver.

pmcduff, 2017-09-22:
```
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Executing command: ipta                                                                                        bles -t mangle -A ndsOUT -s 10.0.14.0 -m mac --mac-source 40:0e:85:5b:58:71 -j M                                                                                        ARK --or-mark 0xd400
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Setting default SIGCHLD                                                                                         handler SIG_DFL
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Waiting for PID 2073 to                                                                                         exit
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Process PID 2073 exited                                                                                         normally, status 0
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Restoring previous SIGC                                                                                        HLD handler
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Executing command: ipta                                                                                        bles -t mangle -A ndsINC -d 10.0.14.0 -j MARK --or-mark 0xd400
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Setting default SIGCHLD                                                                                         handler SIG_DFL
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Waiting for PID 2075 to                                                                                         exit
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Process PID 2075 exited                                                                                         normally, status 0
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Restoring previous SIGC                                                                                        HLD handler
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Executing command: ipta                                                                                        bles -t mangle -A ndsINC -d 10.0.14.0 -j ACCEPT
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Setting default SIGCHLD                                                                                         handler SIG_DFL
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Waiting for PID 2077 to                                                                                         exit
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Process PID 2077 exited                                                                                         normally, status 0
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Restoring previous SIGC                                                                                        HLD handler
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Unlocking client list
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Client list unlocked
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Thread 3 returned from                                                                                         httpdProcessRequest() for 10.0.14.0
Fri Sep 22 05:08:57 2017 daemon.debug nodogsplash[1394]: Thread 3 ended request                                                                                         from 10.0.14.0
Fri Sep 22 05:08:59 2017 daemon.info hostapd: wlan0: STA a8:be:27:54:79:b3 IEEE 802.11: associated (aid 3)
Fri Sep 22 05:08:59 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED a8:be:27:54:79:b3
Fri Sep 22 05:08:59 2017 daemon.debug nodogsplash[1394]: 0 current httpd threads.
Fri Sep 22 05:08:59 2017 daemon.info nodogsplash[1394]: Creating httpd request thread 4 for 10.0.35.79
Fri Sep 22 05:08:59 2017 daemon.debug nodogsplash[1394]: Thread 4 calling httpdProcessRequest() for 10.0.35.79
Fri Sep 22 05:08:59 2017 daemon.info nodogsplash[1394]: Capturing as 404 request from 10.0.35.79 for [c.whatsapp.net/chat]
Fri Sep 22 05:08:59 2017 daemon.debug nodogsplash[1394]: Thread 4 returned from httpdProcessRequest() for 10.0.35.79
Fri Sep 22 05:08:59 2017 daemon.debug nodogsplash[1394]: Thread 4 ended request from 10.0.35.79
Fri Sep 22 05:08:59 2017 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED a8:be:27:54:79:b3
Fri Sep 22 05:08:59 2017 daemon.info hostapd: wlan0: STA a8:be:27:54:79:b3 IEEE 802.11: disassociated
Fri Sep 22 05:08:59 2017 daemon.info hostapd: wlan0: STA a8:be:27:54:79:b3 IEEE 802.11: disassociated
Fri Sep 22 05:08:59 2017 kern.info kernel: [   35.847262] ieee80211 phy0: staid 3 deleted
Fri Sep 22 05:09:00 2017 daemon.info hostapd: wlan0: STA a8:be:27:54:79:b3 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Sep 22 05:09:02 2017 daemon.notice hostapd: handle_auth_cb: STA a8:be:27:54:79:b3 not found
Fri Sep 22 05:09:02 2017 daemon.notice hostapd: handle_auth_cb: STA a8:be:27:54:79:b3 not found
Fri Sep 22 05:09:02 2017 daemon.warn dnsmasq[2006]: possible DNS-rebind attack detected: proxy.mobile.att.net
Fri Sep 22 05:09:02 2017 daemon.warn dnsmasq[2006]: possible DNS-rebind attack detected: proxy.mobile.att.net
Fri Sep 22 05:09:18 2017 daemon.info dnsmasq[2006]: read /etc/hosts - 4 addresses
Fri Sep 22 05:09:18 2017 daemon.info dnsmasq[2006]: read /tmp/hosts/dhcp.cfg02411c - 2 addresses
Fri Sep 22 05:09:18 2017 daemon.info dnsmasq-dhcp[2006]: read /etc/ethers - 0 addresses
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.081681] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.151681] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.181680] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.251682] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.281683] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.351684] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.381685] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.451687] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.481687] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.551690] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.581690] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.651693] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.681693] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.751695] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.781696] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.851697] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.881698] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.951700] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:34 2017 kern.err kernel: [   70.981701] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:35 2017 kern.err kernel: [   71.051703] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:35 2017 kern.err kernel: [   71.081704] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:35 2017 kern.err kernel: [   71.151706] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:35 2017 kern.err kernel: [   71.181706] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:35 2017 kern.err kernel: [   71.251708] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:35 2017 kern.err kernel: [   71.281709] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:35 2017 kern.err kernel: [   71.351711] ieee80211 phy1: buffer is NULL for tx done ring
Fri Sep 22 05:09:35 2017 kern.err kernel: [   71.381712] ieee80211 phy1: buffer is NULL for tx done ring
[... cut ...]
Fri Sep 22 05:09:38 2017 kern.warn kernel: [   72.642592] ieee80211 phy1: card unplugged?
Fri Sep 22 05:09:38 2017 kern.warn kernel: [   72.646793] ieee80211 phy1: card unplugged?
Fri Sep 22 05:09:38 2017 kern.warn kernel: [   72.650995] ieee80211 phy1: card unplugged?
Fri Sep 22 05:09:38 2017 kern.warn kernel: [   72.655197] ieee80211 phy1: card unplugged?
Fri Sep 22 05:09:38 2017 kern.warn kernel: [   72.659398] ieee80211 phy1: card unplugged?
```

Some help regarding the improvement of stability would be appreciated.

yuhhaurlin, 2017-09-22:
Adapter can't be accessed by driver.

yuhhaurlin, 2017-09-22:
How to make this problem happened?

pmcduff, 2017-09-22:
Normal operation on LEDE 17.01.2. Wlan 5ghz channel 36 and 42. 2.4ghz on channel 5.

pmcduff, 2017-09-22:
I can get a serial cable tomorrow, but this is what logread -f outputs before crash:
```
root@LEDE:~# logread -f
Fri Sep 22 05:33:30 2017 daemon.info hostapd: wlan0: STA 40:0e:85:5b:58:71 IEEE 802.11: associated (aid 2)
Fri Sep 22 05:33:30 2017 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 40:0e:85:5b:58:71

```

yuhhaurlin, 2017-09-22:
phy1 shows "card unplugged?".

yuhhaurlin, 2017-09-22:
What is the output for "iw dev"?

pmcduff, 2017-09-22:
iw dev:
```
root@LEDE:~# iw dev
phy#2
        Interface wlan2
                ifindex 11
                wdev 0x200000002
                addr 60:38:e0:b9:f3:cb
                ssid Here is your fast Wi-Fi
                type AP
                channel 44 (5220 MHz), width: 40 MHz, center1: 5230 MHz
                txpower 0.00 dBm
phy#1
        Interface wlan1
                ifindex 9
                wdev 0x100000002
                addr 60:38:e0:b9:f3:c9
                ssid Here is your Wi-Fi
                type AP
                channel 5 (2432 MHz), width: 20 MHz, center1: 2432 MHz
                txpower 30.00 dBm
phy#0
        Interface wlan0
                ifindex 10
                wdev 0x2
                addr 60:38:e0:b9:f3:ca
                ssid Here is your fastest Wi-Fi
                type AP
                channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 23.00 dBm
```
iw list:
```
root@LEDE:~# iw list
Wiphy phy2
        max # scan SSIDs: 10
        max scan IEs length: 256 bytes
        max # sched scan SSIDs: 10
        max # match sets: 10
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Fragmentation threshold: 2346
        RTS threshold: 2347
        Retry short limit: 9
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Available Antennas: TX 0x1 RX 0x1
        Configured Antennas: TX 0x1 RX 0x1
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x17f
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
[... 289 lines cut ...]
                        * 5560 MHz [112] (23.0 dBm) (radar detection)
                        * 5580 MHz [116] (23.0 dBm) (radar detection)
                        * 5600 MHz [120] (23.0 dBm) (radar detection)
                        * 5620 MHz [124] (23.0 dBm) (radar detection)
                        * 5640 MHz [128] (23.0 dBm) (radar detection)
                        * 5660 MHz [132] (23.0 dBm) (radar detection)
                        * 5680 MHz [136] (23.0 dBm) (radar detection)
                        * 5700 MHz [140] (23.0 dBm) (radar detection)
                        * 5720 MHz [144] (23.0 dBm) (radar detection)
                        * 5745 MHz [149] (30.0 dBm)
                        * 5765 MHz [153] (30.0 dBm)
                        * 5785 MHz [157] (30.0 dBm)
                        * 5805 MHz [161] (30.0 dBm)
        valid interface combinations:
                 * #{ AP } <= 16, #{ managed } <= 1,
                   total <= 16, #channels <= 1, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 160 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
```

pmcduff, 2017-09-22:
I can get a serial log later today.

pmcduff, 2017-09-22:
```
 BootROM - 1.73
Booting from NAND flash

General initialization - Version: 1.0.0
AVS selection from EFUSE disabled (Skip reading EFUSE values)
mvSysEnvIsFlavourReduced: TWSI Read of 'flavor' failed
Detected Device ID 6820
High speed PHY - Version: 2.0

Init RD NAS topology Serdes Lane 3 is USB3
Serdes Lane 4 is SGMII
board SerDes lanes topology details:
 | Lane # | Speed|    Type     |
 ------------------------------|
 |   0    |  6   |  SATA0      |
 |   1    |  5   |  PCIe0      |
 |   2    |  6   |  SATA1      |
 |   3    |  5   |  USB3 HOST1 |
 |   4    |  5   |  PCIe1      |
 |   5    |  0   |  SGMII2     |
 -------------------------------
:** Link is Gen1, check the EP capability
PCIe, Idx 0: Link upgraded to Gen2 based on client cpabilities
:** Link is Gen1, check the EP capability
PCIe, Idx 1: remains Gen1
High speed PHY - Ended Successfully
DDR3 Training Sequence - Ver TIP-1.46.0
mvSysEnvGetTopologyUpdateInfo: TWSI Read failed
DDR3 Training Sequence - Switching XBAR Window to FastPath Window
DDR3 Training Sequence - Ended Successfully
Not detected suspend to RAM indication
BootROM: Image checksum verification PASSED

 __   __                      _ _
|  \/  | __ _ _ ____   _____| | |
| |\/| |/ _` | '__\ \ / / _ \ | |
| |  | | (_| | |   \ V /  __/ | |
|_|  |_|\__,_|_|    \_/ \___|_|_|
         _   _     ____              _
        | | | |   | __ )  ___   ___ | |_
        | | | |___|  _ \ / _ \ / _ \| __|
        | |_| |___| |_) | (_) | (_) | |_
         \___/    |____/ \___/ \___/ \__|
 ** LOADER **

U-Boot 2013.01 (Apr 08 2016 - 15:47:50) Marvell version: 2015_T1.QA.0p16

Boot version : v0.0.4

Board: RD-NAS-88F6820-DDR3
SoC:   MV88F6820 Rev A0
       running 2 CPUs
CPU:   ARM Cortex A9 MPCore (Rev 1) LE
       CPU 0
       CPU    @ 1866 [MHz]
       L2     @ 933 [MHz]
       TClock @ 200 [MHz]
       DDR3    @ 933 [MHz]
       DDR3 32 Bit Width,FastPath Memory Access, DLB Enabled, ECC Disabled
DRAM:  512 MiB
NAND:  256 MiB
MMC:   mv_sdh: 0
DEVINFO offset == 0x7e0000
U-ENV offset == 0x200000
U-ENV offset == 0x200000
S-ENV offset == 0x220000

#### auto_recovery ####
[u_env] get auto_recovery == yes
[... 356 lines cut ...]
[   14.469925] br-lan: port 3(wlan1) entered disabled state
[   14.475547] device wlan0 entered promiscuous mode
[   14.481395] ieee80211 phy1: change: 0x100
[   14.494443] ieee80211 phy1: change: 0x42
[   14.646297] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   14.652731] br-lan: port 3(wlan1) entered forwarding state
[   14.658249] br-lan: port 3(wlan1) entered forwarding state
[   14.716679] ieee80211 phy0: change: 0x100
[   14.729791] ieee80211 phy0: change: 0x40
[   14.931289] ieee80211 phy0: change: 0x40
[   14.951211] IPv6: ADDRCONF(NETDEV_CHANGE): wlan2: link becomes ready
[   14.957658] br-lan: port 2(wlan2) entered forwarding state
[   14.963187] br-lan: port 2(wlan2) entered forwarding state
[   15.151298] ieee80211 phy0: change: 0x40
[   15.211281] br-lan: port 1(eth0) entered forwarding state
[   15.251298] ieee80211 phy0: change: 0x100
[   15.264516] ieee80211 phy0: change: 0x100
[   15.277585] ieee80211 phy0: change: 0x42
[   15.470415] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   15.476868] br-lan: port 4(wlan0) entered forwarding state
[   15.482398] br-lan: port 4(wlan0) entered forwarding state
[   16.651281] br-lan: port 3(wlan1) entered forwarding state
[   16.961275] br-lan: port 2(wlan2) entered forwarding state
[   17.481253] br-lan: port 4(wlan0) entered forwarding state
```

yuhhaurlin, 2017-09-22:
？？？

pmcduff, 2017-09-22:
It still crashes after about a minute with phy2 disabled

pmcduff, 2017-09-22:
I get this from serial.
`[   13.277195] mwifiex_sdio mmc0:0001:1: mwifiex_process_cmdresp: cmd 0x242 failed during       initialization`

BrainSlayer, 2017-09-22:
this does not belong to the mwlwifi driver. this is related to phy2 shich is a totally different chipset

BrainSlayer, 2017-09-22:
phy2 works in dd-wrt, but havent seen it working in lede right now. but it should be off topic here. its another chipset

pmcduff, 2017-09-22:
I've disabled from LEDE web gui, but it still crashes anyways.

BrainSlayer, 2017-09-22:
would be good to see the crashlog which can only be seen on a serial log. this is what @yuhhaurlin  is waiting for. he may be already aware of this issue, but right now he is unable to reproduce it for himself. but consider that you're not the only one right now. so if you're able to get a serial log, it would be helpfull for him

aaron1959, 2017-09-22:
@pmcduff  Please keep serial logging LEDE for crash data PLEASE!

yuhhaurlin, 2017-09-22:
Please try:
opkg remove kmod-mwifiex-sdio
opkg remove mwifiex-sdio-firmware
reboot
In order to remove mwifiex.

pmcduff, 2017-09-22:
How do I get a serial log? I already have a cable.

yuhhaurlin, 2017-09-22:
You still encounter problem after removing mwifiex?

gufus, 2017-09-22:
@ pmcduff

1 laptop 1 serial cable
 https://github.com/Chadster766/McWRT/wiki/Linksys-WRT1900AC-Serial-Port 

setup "putty" on the laptop attach the cable to the unit and log the lockup/crash

pmcduff, 2017-09-22:
yes

pmcduff, 2017-09-22:
```
BootROM - 1.73
Booting from NAND flash

General initialization - Version: 1.0.0
AVS selection from EFUSE disabled (Skip reading EFUSE values)
mvSysEnvIsFlavourReduced: TWSI Read of 'flavor' failed
Detected Device ID 6820
High speed PHY - Version: 2.0

Init RD NAS topology Serdes Lane 3 is USB3
Serdes Lane 4 is SGMII
board SerDes lanes topology details:
 | Lane # | Speed|    Type     |
 ------------------------------|
 |   0    |  6   |  SATA0      |
 |   1    |  5   |  PCIe0      |
 |   2    |  6   |  SATA1      |
 |   3    |  5   |  USB3 HOST1 |
 |   4    |  5   |  PCIe1      |
 |   5    |  0   |  SGMII2     |
 -------------------------------
:** Link is Gen1, check the EP capability
PCIe, Idx 0: Link upgraded to Gen2 based on client cpabilities
:** Link is Gen1, check the EP capability
PCIe, Idx 1: remains Gen1
High speed PHY - Ended Successfully
DDR3 Training Sequence - Ver TIP-1.46.0
mvSysEnvGetTopologyUpdateInfo: TWSI Read failed
DDR3 Training Sequence - Switching XBAR Window to FastPath Window
DDR3 Training Sequence - Ended Successfully
Not detected suspend to RAM indication
BootROM: Image checksum verification PASSED

 __   __                      _ _
|  \/  | __ _ _ ____   _____| | |
| |\/| |/ _` | '__\ \ / / _ \ | |
| |  | | (_| | |   \ V /  __/ | |
|_|  |_|\__,_|_|    \_/ \___|_|_|
         _   _     ____              _
        | | | |   | __ )  ___   ___ | |_
        | | | |___|  _ \ / _ \ / _ \| __|
        | |_| |___| |_) | (_) | (_) | |_
         \___/    |____/ \___/ \___/ \__|
 ** LOADER **

U-Boot 2013.01 (Apr 08 2016 - 15:47:50) Marvell version: 2015_T1.QA.0p16

Boot version : v0.0.4

Board: RD-NAS-88F6820-DDR3
SoC:   MV88F6820 Rev A0
       running 2 CPUs
CPU:   ARM Cortex A9 MPCore (Rev 1) LE
       CPU 0
       CPU    @ 1866 [MHz]
       L2     @ 933 [MHz]
       TClock @ 200 [MHz]
       DDR3    @ 933 [MHz]
       DDR3 32 Bit Width,FastPath Memory Access, DLB Enabled, ECC Disabled
DRAM:  512 MiB
NAND:  256 MiB
MMC:   mv_sdh: 0
DEVINFO offset == 0x7e0000
U-ENV offset == 0x200000
U-ENV offset == 0x200000
S-ENV offset == 0x220000

#### auto_recovery ####
[u_env] get auto_recovery == yes
[... 344 lines cut ...]
[   15.348297] br-lan: port 2(wlan1) entered forwarding state
[   15.353822] br-lan: port 2(wlan1) entered forwarding state
[   15.359365] br-lan: port 2(wlan1) entered disabled state
[   15.364971] device wlan0 entered promiscuous mode
[   15.370360] ieee80211 phy1: change: 0x100
[   15.383417] ieee80211 phy1: change: 0x42
[   15.534318] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   15.540741] br-lan: port 2(wlan1) entered forwarding state
[   15.546263] br-lan: port 2(wlan1) entered forwarding state
[   15.606242] ieee80211 phy0: change: 0x100
[   15.618483] ieee80211 phy0: change: 0x40
[   15.821295] ieee80211 phy0: change: 0x40
[   16.031290] ieee80211 phy0: change: 0x40
[   16.141302] ieee80211 phy0: change: 0x100
[   16.154549] ieee80211 phy0: change: 0x100
[   16.167594] ieee80211 phy0: change: 0x42
[   16.301286] br-lan: port 1(eth0) entered forwarding state
[   16.341372] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   16.347793] br-lan: port 3(wlan0) entered forwarding state
[   16.353333] br-lan: port 3(wlan0) entered forwarding state
[   17.541279] br-lan: port 2(wlan1) entered forwarding state
[   18.350557] br-lan: port 3(wlan0) entered forwarding state
[   60.068713] ieee80211 phy1: staid 1 deleted
```
There's another crashlog for you.

yuhhaurlin, 2017-09-22:
？？？ I can't find problem from the log!

pmcduff, 2017-09-22:
I've just reinstalled LEDE.

yuhhaurlin, 2017-09-22:
So this is not log for crash?

yuhhaurlin, 2017-09-22:
At least, mwifiex is removed. Wait for your crash log.

pmcduff, 2017-09-22:
The router would just reboot on its own after a few minutes. No logs.

pmcduff, 2017-09-22:
I'm sorry that i've given up on using the latest firmware. I have 30 people who have paid for wifi on my back so I had to revert. Best of luck. Will install the next firmware iteration.

gufus, 2017-09-22:
@pmcduff  
Are you loging to a file in puddy?

Just right as it crashes close puddy, it will log the crash

yuhhaurlin, 2017-09-22:
It looks like this is not problem of mwlwifi. I close this issue. You can create a new issue if you get log related to mwlwifi. Thanks.

closed 2017-09-22 by yuhhaurlin

pmcduff, 2017-09-24:
This might be helpful
```
[  237.961237] INFO: rcu_sched self-detected stall on CPU
[  237.966411]  1-...: (1 GPs behind) idle=173/140000000000001/0 softirq=6360/6361 fqs=3000
[  237.971238] INFO: rcu_sched detected stalls on CPUs/tasks:
[  237.971242]  1-...: (1 GPs behind) idle=173/140000000000001/0 softirq=6360/6361 fqs=3000
[  237.971245]  (detected by 0, t=6002 jiffies, g=2501, c=2500, q=891)
[  237.971246] Task dump for CPU 1:
[  237.971250] kworker/u4:1    R  running task        0    17      2 0x00000002
[  237.971263] Workqueue: phy1 _113 [mwlwifi]
[  237.971272] [<c05272b8>] (__schedule) from [<df41d018>] (0xdf41d018)
[  238.015414]   (t=6006 jiffies g=2501 c=2500 q=891)
[  238.020234] Task dump for CPU 1:
[  238.023472] kworker/u4:1    R  running task        0    17      2 0x00000002
[  238.030574] Workqueue: phy1 _113 [mwlwifi]
[  238.034703] [<c010ee38>] (unwind_backtrace) from [<c010ad00>] (show_stack+0x10/0x14)
[  238.042480] [<c010ad00>] (show_stack) from [<c0196290>] (rcu_dump_cpu_stacks+0x94/0xcc)
[  238.050519] [<c0196290>] (rcu_dump_cpu_stacks) from [<c016d890>] (rcu_check_callbacks+0x2bc/0x7b4)
[  238.059516] [<c016d890>] (rcu_check_callbacks) from [<c016fb78>] (update_process_times+0x34/0x5c)
[  238.068426] [<c016fb78>] (update_process_times) from [<c017e644>] (tick_sched_timer+0x214/0x258)
[  238.077247] [<c017e644>] (tick_sched_timer) from [<c0170750>] (__hrtimer_run_queues+0xd0/0x18c)
[  238.085981] [<c0170750>] (__hrtimer_run_queues) from [<c017096c>] (hrtimer_interrupt+0xb8/0x20c)
[  238.094802] [<c017096c>] (hrtimer_interrupt) from [<c010e328>] (twd_handler+0x30/0x40)
[  238.102753] [<c010e328>] (twd_handler) from [<c0164e28>] (handle_percpu_devid_irq+0x78/0x144)
[  238.111314] [<c0164e28>] (handle_percpu_devid_irq) from [<c0160414>] (generic_handle_irq+0x24/0x34)
[  238.120397] [<c0160414>] (generic_handle_irq) from [<c0160978>] (__handle_domain_irq+0x90/0xa4)
[  238.129131] [<c0160978>] (__handle_domain_irq) from [<c0101480>] (gic_handle_irq+0x48/0x74)
[  238.137518] [<c0101480>] (gic_handle_irq) from [<c010b7cc>] (__irq_svc+0x6c/0x90)
[  238.145029] Exception stack(0xdf4d3e90 to 0xdf4d3ed8)
[  238.150100] 3e80:                                     00000000 da3fc740 bf0c38a4 e1280000
[  238.158310] 3ea0: d9c96448 d9c9688c da3e1540 da3e4740 000061bc 0000688c da3e4540 df41d000
[  238.166520] 3ec0: 00000000 df4d3ee0 00000001 bf0c394c 80000013 ffffffff
[  238.173166] [<c010b7cc>] (__irq_svc) from [<bf0c394c>] (_256+0xa8/0x31c [mwlwifi])
[  238.180774] [<bf0c394c>] (_256 [mwlwifi]) from [<c01376b4>] (process_one_work+0x240/0x3d0)
[  238.189074] [<c01376b4>] (process_one_work) from [<c013878c>] (worker_thread+0x320/0x550)
[  238.197285] [<c013878c>] (worker_thread) from [<c013ccf0>] (kthread+0xe8/0xf4)
[  238.204538] [<c013ccf0>] (kthread) from [<c0107778>] (ret_from_fork+0x14/0x3c)
```

gufus, 2017-09-24:
Ouch