kaloz/mwlwifi #217: Wrt3200acm freezes up.

pmcduff, 2017-09-24:
I reverted back to dd-wrt as the stock firmware was too unstable. I got these crash logs after about 4 mins of uptime. It causes all clients to lose internet connectivity, and the router becomes unpingable.
```
[  237.961237] INFO: rcu_sched self-detected stall on CPU
[  237.966411]  1-...: (1 GPs behind) idle=173/140000000000001/0 softirq=6360/6361 fqs=3000
[  237.971238] INFO: rcu_sched detected stalls on CPUs/tasks:
[  237.971242]  1-...: (1 GPs behind) idle=173/140000000000001/0 softirq=6360/6361 fqs=3000
[  237.971245]  (detected by 0, t=6002 jiffies, g=2501, c=2500, q=891)
[  237.971246] Task dump for CPU 1:
[  237.971250] kworker/u4:1    R  running task        0    17      2 0x00000002
[  237.971263] Workqueue: phy1 _113 [mwlwifi]
[  237.971272] [<c05272b8>] (__schedule) from [<df41d018>] (0xdf41d018)
[  238.015414]   (t=6006 jiffies g=2501 c=2500 q=891)
[  238.020234] Task dump for CPU 1:
[  238.023472] kworker/u4:1    R  running task        0    17      2 0x00000002
[  238.030574] Workqueue: phy1 _113 [mwlwifi]
[  238.034703] [<c010ee38>] (unwind_backtrace) from [<c010ad00>] (show_stack+0x10/0x14)
[  238.042480] [<c010ad00>] (show_stack) from [<c0196290>] (rcu_dump_cpu_stacks+0x94/0xcc)
[  238.050519] [<c0196290>] (rcu_dump_cpu_stacks) from [<c016d890>] (rcu_check_callbacks+0x2bc/0x7b4)
[  238.059516] [<c016d890>] (rcu_check_callbacks) from [<c016fb78>] (update_process_times+0x34/0x5c)
[  238.068426] [<c016fb78>] (update_process_times) from [<c017e644>] (tick_sched_timer+0x214/0x258)
[  238.077247] [<c017e644>] (tick_sched_timer) from [<c0170750>] (__hrtimer_run_queues+0xd0/0x18c)
[  238.085981] [<c0170750>] (__hrtimer_run_queues) from [<c017096c>] (hrtimer_interrupt+0xb8/0x20c)
[  238.094802] [<c017096c>] (hrtimer_interrupt) from [<c010e328>] (twd_handler+0x30/0x40)
[  238.102753] [<c010e328>] (twd_handler) from [<c0164e28>] (handle_percpu_devid_irq+0x78/0x144)
[  238.111314] [<c0164e28>] (handle_percpu_devid_irq) from [<c0160414>] (generic_handle_irq+0x24/0x34)
[  238.120397] [<c0160414>] (generic_handle_irq) from [<c0160978>] (__handle_domain_irq+0x90/0xa4)
[  238.129131] [<c0160978>] (__handle_domain_irq) from [<c0101480>] (gic_handle_irq+0x48/0x74)
[  238.137518] [<c0101480>] (gic_handle_irq) from [<c010b7cc>] (__irq_svc+0x6c/0x90)
[  238.145029] Exception stack(0xdf4d3e90 to 0xdf4d3ed8)
[  238.150100] 3e80:                                     00000000 da3fc740 bf0c38a4 e1280000
[  238.158310] 3ea0: d9c96448 d9c9688c da3e1540 da3e4740 000061bc 0000688c da3e4540 df41d000
[  238.166520] 3ec0: 00000000 df4d3ee0 00000001 bf0c394c 80000013 ffffffff
[  238.173166] [<c010b7cc>] (__irq_svc) from [<bf0c394c>] (_256+0xa8/0x31c [mwlwifi])
[  238.180774] [<bf0c394c>] (_256 [mwlwifi]) from [<c01376b4>] (process_one_work+0x240/0x3d0)
[  238.189074] [<c01376b4>] (process_one_work) from [<c013878c>] (worker_thread+0x320/0x550)
[  238.197285] [<c013878c>] (worker_thread) from [<c013ccf0>] (kthread+0xe8/0xf4)
[  238.204538] [<c013ccf0>] (kthread) from [<c0107778>] (ret_from_fork+0x14/0x3c)
```

gufus, 2017-09-24:
Again

yuhhaurlin, 2017-09-25:
CU stall is fixed by 0810.

closed 2017-09-25 by yuhhaurlin

aaron1959, 2017-09-25:
This is the driver that has problem with 'Stall On CPU'
how is 0810 fixed????

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:13:64:f9
2g: enable
5g: disable
antenna: 4 4
irq number: 47
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e1000000
iobase1: e1280000
tx limit: 1024
rx limit: 16384

aaron1959, 2017-09-25:
0810 is not fixed for CPU STALL!

yuhhaurlin, 2017-09-25:
You got this problem?

yuhhaurlin, 2017-09-25:
0913 LEDE main trunk still uses 0713.

rs-se, 2017-09-25:
Why are we biased against dd-wrt, OpenWRT/LEDE is not a perfect project nor
is it easily configurable.

Is LEDE sponsoring this driver or is Belkin pushing for support mainly for
LEDE?

aaron1959, 2017-09-25:
047] ieee80211 phy0: staid 1 deleted
[ 8535.148588] ieee80211 phy0: staid 1 deleted
[ 8567.554564] ieee80211 phy1: staid 2 deleted
[ 8618.614571] ieee80211 phy1: staid 3 deleted
[ 8629.416566] ieee80211 phy1: staid 3 deleted
[ 8638.378584] ieee80211 phy1: staid 3 deleted
[ 8700.748423] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[ 8700.754294] ieee80211 phy0: return code: 0x051d
[ 8700.758844] ieee80211 phy0: timeout: 0x001d
[ 8704.762414] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[ 8704.768275] ieee80211 phy0: return code: 0x001d
[ 8704.772833] ieee80211 phy0: timeout: 0x001d
[ 8708.776421] ieee80211 phy0: cmd 0x801d=MEMAddrAccess timed out
[ 8708.782289] ieee80211 phy0: return code: 0x001d
[ 8708.786838] ieee80211 phy0: timeout: 0x001d
[ 8839.210401] ieee80211 phy0: cmd 0x9143=GetSeqno timed out
[ 8839.215836] ieee80211 phy0: return code: 0x1143
[ 8839.220409] ieee80211 phy0: timeout: 0x1143
[ 9138.180444] ieee80211 phy0: can't send packet during ADDBA
[ 9142.190238] Unable to handle kernel NULL pointer dereference at virtual address 0000000c
[ 9142.198380] pgd = d4cc0000
[ 9142.201105] [0000000c] *pgd=12f1b831, *pte=00000000, *ppte=00000000
[ 9142.207431] Internal error: Oops: 17 [#1] SMP ARM
[ 9142.212153] Modules linked in: nf_nat_pptp nf_conntrack_pptp nf_nat_proto_gre nf_conntrack_proto_gre mip6 ip6_tunnel xfrm6_mode_beet xfrm6_mode_transport xfrm6_mode_tunnel xfrm6_tunnel ipv6 tunnel6 fuse ledtrig_usbport pwrseq_simple pwrseq_emmc ahci_mvebu ahci_platform ahci libahci_platform libahci ehci_orion sata_mv usb_storage sd_mod xhci_plat_hcd xhci_pci xhci_hcd ehci_pci ehci_platform ehci_hcd usbcore usb_common btmrvl_sdio btmrvl bluetooth mwifiex_sdio mvsdio sdhci_pxav3 sdhci_pltfm sdhci mmc_block mmc_core mwifiex mwlwifi mac80211 cfg80211 compat mii tmp421 pwm_fan leds_pca963x leds_tlc591xx orion_wdt [last unloaded: nf_conntrack_proto_gre]
[ 9142.270300] CPU: 0 PID: 2429 Comm: hostapd Not tainted 4.9.50 #51
[ 9142.276418] Hardware name: Marvell Armada 380/385 (Device Tree)
[ 9142.282362] task: df4e0540 task.stack: d59b6000
[ 9142.286918] PC is at _145+0x98/0xf0 [mwlwifi]
[ 9142.291295] LR is at _145+0x34/0xf0 [mwlwifi]
[ 9142.295669] pc : [<bf0c36c0>]    lr : [<bf0c365c>]    psr: 60000013
[ 9142.295669] sp : d59b7c88  ip : 00000000  fp : 00000000
[ 9142.307196] r10: de0a0bc0  r9 : daa88de0  r8 : de6a0054
[ 9142.312442] r7 : de0a28a4  r6 : de0a15a0  r5 : 00000000  r4 : de7ac000
[ 9142.318997] r3 : 00000000  r2 : 00000002  r1 : 00000000  r0 : 00000000
[ 9142.325551] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[ 9142.332717] Control: 10c5387d  Table: 14cc004a  DAC: 00000051
[ 9142.338486] Process hostapd (pid: 2429, stack limit = 0xd59b6210)
[ 9142.344604] Stack: (0xd59b7c88 to 0xd59b8000)
[ 9142.348979] 7c80:                   00000007 de6a0054 daa88de0 de0a4560 de6a0048 bf0c3c0c
[ 9142.357194] 7ca0: daa88de0 de0a0bc0 de0a15a0 d3676700 da4efc68 00000014 dec30000 bf0bead0
[ 9142.365408] 7cc0: daa88800 00000002 da4ef4c0 d3676700 00000000 00000014 dec30000 bf05ff9c
[ 9142.373621] 7ce0: 00000000 bf07eafc daa88800 daa88800 00000002 bf062d78 00000002 00000000
[ 9142.381835] 7d00: 00000014 daa88800 de0a0bc0 bf063714 d3676620 d3676700 00000000 daa88800
[ 9142.390049] 7d20: d3676620 bf063880 da4ef4c0 00000454 d3676620 bf0638f8 de0a0000 da4ef000
[ 9142.398263] 7d40: d59b7d68 bf0299f0 d3676620 0002000c bf052040 bf048298 d3676614 c045a348
[ 9142.406477] 7d60: bf047b68 00000000 59c851c5 00000972 d3676600 d3676610 d3676614 dec30000
[ 9142.414691] 7d80: de0a0000 da4ef000 d3676600 d3676700 c045a028 00000028 d59b7dcc 00000000
[ 9142.422905] 7da0: 00000000 c0459620 c09276c4 d3676700 d2f1e000 c045a018 df5c5c00 d3676700
[ 9142.431119] 7dc0: d2f1e000 c0458fc4 00000008 7fffffff 00000028 d59b7f68 00000028 d2f1e000
[ 9142.439333] 7de0: 00000008 d3676700 024000c0 c0459460 00000000 c02c444c d59b7e70 d59b7e04
[ 9142.447546] 7e00: 0000000c d8569f40 00000000 0000097d 00000000 00000000 00000000 d59b7f68
[ 9142.455760] 7e20: 00000000 00000000 df04d300 d59b7e50 00000000 d59b7e50 00028778 c03ef8a8
[ 9142.463974] 7e40: d59b7f68 c03f02e4 00000001 00000000 00000000 00000000 dfbd2f70 00000850
[ 9142.472187] 7e60: d3676700 00000000 00000400 00000002 017d4010 00000028 d59b7ed0 00000002
[ 9142.480401] 7e80: df781b30 00000000 d435c848 00000000 00000000 daf989c0 dfbd2f70 daf98a40
[ 9142.488615] 7ea0: c0843f40 c0904464 dfbd2f40 df4e0868 00000010 00000000 00000000 00000000
[ 9142.496829] 7ec0: 00000002 c03ef8a8 00000000 c03f0c50 c0904464 c0902d00 0000007e 00000000
[ 9142.505042] 7ee0: beae36e0 0000007e 00000000 00000000 00000001 df4e0540 afb50401 d59b7ee8
[ 9142.513256] 7f00: 00000000 00000000 00000000 00000000 00000000 00000001 00000000 fffffffe
[ 9142.521470] 7f20: 00000000 c013df00 d3412018 dfbd2f40 df4e0540 beae3b2c 00000000 df04d300
[ 9142.529684] 7f40: 00000128 c0107884 d59b6000 00000000 00028778 c03f0f90 00000000 00000000
[ 9142.537897] 7f60: 00000000 fffffff7 d59b7eb0 0000000c 00000001 00000000 00000000 d59b7e78
[ 9142.546111] 7f80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 9142.554324] 7fa0: b6eceee0 c01076c0 00000000 00000000 00000005 beae3b2c 00000000 00000000
[ 9142.562538] 7fc0: 00000000 00000000 b6eceee0 00000128 b6eceee0 00000000 b6ece470 00028778
[ 9142.570752] 7fe0: beae3ab8 beae3aa4 b6f972d0 b6f967c4 60000010 00000005 92000fca 0fc30740
[ 9142.578979] [<bf0c36c0>] (_145 [mwlwifi]) from [<bf0c3c0c>] (_144+0x27c/0x2e8 [mwlwifi])
[... cut ...]
[ 9142.683710] [<c0459460>] (netlink_sendmsg) from [<c03ef8a8>] (sock_sendmsg+0x14/0x3c)
[ 9142.691577] [<c03ef8a8>] (sock_sendmsg) from [<c03f02e4>] (___sys_sendmsg+0x208/0x21c)
[ 9142.699529] [<c03f02e4>] (___sys_sendmsg) from [<c03f0f90>] (__sys_sendmsg+0x40/0x64)
[ 9142.707396] [<c03f0f90>] (__sys_sendmsg) from [<c01076c0>] (ret_fast_syscall+0x0/0x3c)
[ 9142.715349] Code: e5840010 e5d81004 e5c41014 e598c000 (e59c300c)
[ 9142.721493] ---[ end trace 29675d19bbabbd83 ]---
  The more you ignore the truth the less credible you are.

yuhhaurlin, 2017-09-25:
@rs-se For LEDE with updated mwlwifi driver, please check https://davidc502sis.dynamic-dns.net/,

yuhhaurlin, 2017-09-25:
@aaron1959 This is still happened on dd-WRT, right? Please reproduce it on LEDE via https://davidc502sis.dynamic-dns.net/. Let me know the LEDE image and the way to reproduce the problem. I will hook ICE to check this problem.

yuhhaurlin, 2017-09-25:
@rs-se I don't know who takes care of updating mwlwifi with LEDE. But link here https://davidc502sis.dynamic-dns.net/ will keep updated mwlwifi driver with LEDE image for WRT series devices.

aaron1959, 2017-09-25:
No need for LEDE the needed info is already posted above. No need to hook ICE to check.

yuhhaurlin, 2017-09-25:
@aaron1959 This is host command time out. Please follow the way I told you to reproduce the problem. So I can check the problem. Otherwise, I can't help you.

BrainSlayer, 2017-09-25:
@yuhhaurlin i already use this newer firmware in dd-wrt and posted it in our forum for testing. i extracted it from the linksys firmware. so far no new issues. some people say its somewhat faster. but the reboot / crash problems remain with certain clients

yuhhaurlin, 2017-09-25:
@BrainSlayer Can  you try to load the image I posted (9.3.0.8) to make sure it is all right? If yes, I will post the firmware to this GitHub. Thanks.

yuhhaurlin, 2017-09-25:
@BrainSlayer Another strange thing is that why they can't reproduce this problem on LEDE image?

BrainSlayer, 2017-09-25:
@yuhhaurlin the problem is see is that right now all people here avoid to use lede for some reason. all people who are posting about this issue are using mainly dd-wrt. i just can guarantee that i use the plain unmodified mwlwifi driver from github here using a 4.9 kernel. the kernel patches required for the wrt3200acm is based on lede sources. but however i may use different hostapd configs. so i may use hostapd vht flags which arent enabled in lede

yuhhaurlin, 2017-09-25:
@BrainSlayer Capability will be added to hostapd configuration file via script file. I wonder what kind of VHT capability won't be added to configuration file of hostapd?

BrainSlayer, 2017-09-25:
these are my capabilities with vht160

ht_capab=[HT40-][LDPC][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40][MAX-AMSDU-7935]
vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][RX-STBC-1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][MAX-MPDU-11454][VHT160][MAX-A-MPDU-LEN-EXP7]

BrainSlayer, 2017-09-25:
i read out the supported caps from the mwlwifi driver and generated it in that way.

yuhhaurlin, 2017-09-25:
I think LEDE do the same thing, right?

yuhhaurlin, 2017-09-25:
Script files will do this kind of thing. In order to support some features, I had patched mac80211.sh and hostpad.sh. And I think these script files will add capability of driver to hostapd configuration file.

BrainSlayer, 2017-09-25:
if i remember correct lede does not support all of this flags. can you check your own flags and compare it with mine? or simply post it here and i compare it

BrainSlayer, 2017-09-25:
so the script  in lede supports only features known by the script itself.

yuhhaurlin, 2017-09-25:
I don't have 88W8964 module now. Can anyone help to post hostapd configuration file of WRT3200ACM here (/var/run/hostapd-phy0.conf and /var/run/hostapd-phy1.conf, one is 2g and another one is 5g).

rmandrad, 2017-09-25:
LEDE is fine - I weekly build my own latest version - no crashes - no issues with WRT3200ACM

DD-WRT is basically behind support - hostpad, linux kernel etc

this is a red herring basically

yuhhaurlin, 2017-09-25:
@rmandrad Can you help to dump hostapd configuration file? Thanks.

rmandrad, 2017-09-25:
let me know if this is what you want ....

hostapd-phy0.conf

driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=FR
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
channel=124

ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]
vht_oper_chwidth=2
vht_oper_centr_freq_seg0_idx=114
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][SU-BEAMFORMER][SU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][VHT160][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

interface=wlan0
ctrl_interface=/var/run/hostapd
ap_isolate=1
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=xxx
wpa=2
wpa_pairwise=CCMP
ssid=xx
bridge=br-lan
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=xxxx

rmandrad, 2017-09-25:
note that I removed phy2 ... the phy1 runs on 2g

yuhhaurlin, 2017-09-25:
@rmandrad Yes. Thanks.

yuhhaurlin, 2017-09-25:
@BrainSlayer I think script files of LEDE completely add capability of mwlwifi driver to hostapd configuration file. And this is what I think.

BrainSlayer, 2017-09-25:
its identical. except that in dd-wrt Single user beamforming is disabled by default and just enabled if user enables it in gui

BrainSlayer, 2017-09-25:
@rmandrad : dd-wrt uses kernel 4.9. so what you say is not correct. its always latest. for hostapd i use a little bit older versions since there was no issue right now which forces my to use a newer version

BrainSlayer, 2017-09-25:
but phy2 is enabled by default in dd-wrt.  but can be disabled

rmandrad, 2017-09-25:
which version of hostapad does DD-WRT run ? I remember that LEDE had an issue with hostapad on versions earlier than 2017... suggest you upgrade to latest hostapad

BrainSlayer, 2017-09-25:
i dont have issues. its october 2016 i use. it was originally takes from openwrt. but i avoid updates without a reason since hostapd growes by codesize alot each time which causes troubles for smaller routers

BrainSlayer, 2017-09-25:
but hostapd is unlikelly to cause crashes in the kernel. if so, its still a driver bug

rmandrad, 2017-09-25:
well ... I don't know DD-WRT ...... compiling LEDE I found some issue with GCC vs MUSL... so I use MUSL

BrainSlayer, 2017-09-25:
you mean uclibc or glibc. dd-wrt uses musl too

rmandrad, 2017-09-25:
Yes musl works fine not gcc
.. don't know uclibc

BrainSlayer, 2017-09-25:
gcc is no library. gcc is a compiler which uses uclibc, glibc or musl as backend c library

rmandrad, 2017-09-25:
sorry .. what I am saying it may be worth compiling dd-wrt with musl ...

pmcduff, 2017-09-25:
Here's the hostapd conf from dd-wrt:
ath0:
```
ieee80211n=1
dynamic_ht40=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40][MAX-AMSDU-7935]
vht_capab=[RXLDPC][SHORT-GI-80][RX-STBC-1][SU-BEAMFORMER][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7][BF-ANTENNA-2][SOUNDING-DIMENSION-2][BF-ANTENNA-3][SOUNDING-DIMENSION-3]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
hw_mode=a
channel=36
frequency=5180
beacon_int=100
```
ath1:
```
ieee80211n=1
dynamic_ht40=0
ht_capab=[HT20][LDPC][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40][MAX-AMSDU-7935]
vht_capab=[RXLDPC][RX-STBC-1][SU-BEAMFORMER][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7][BF-ANTENNA-2][SOUNDING-DIMENSION-2][BF-ANTENNA-3][SOUNDING-DIMENSION-3]
vendor_vht=1
hw_mode=g
channel=9
frequency=2452
beacon_int=100
```

BrainSlayer, 2017-09-25:
@rmandrad  and all what i say is that dd-wrt is using musl since more than 2 years if not longer

BrainSlayer, 2017-09-25:
@yuhhaurlin the only main difference is that dd-wrt makes use of single user beamforming if selected. it can be disabled in wireless settings. but if this is causing the big fail i may exlude this feature from mwlwifi until its fixes. 
@pmcduff you can disable single user beamforming at wireless basic advanced settings. try if its more stable then

aaron1959, 2017-09-25:
DD-WRT single user beam forming is disabled by default.

pmcduff, 2017-09-25:
What really helped was to change txpower from 30dBm to 27, it made a huge difference. I'm getting about an hour of stable operation as opposed to a few minutes even with single-user beamforming enabled. I will try disabling it to see if that helps even more, but it is a nice feature that I would not like to disable.
DD-WRT v3.0-r33006 std(08/03/17)

yuhhaurlin, 2017-09-25:
Tx power is not supported.

pmcduff, 2017-09-25:
It crashed after 20 minutes with single-user beamforming disabled.

yuhhaurlin, 2017-09-25:
dd-WRT? What log message?

yuhhaurlin, 2017-09-25:
Can you try LEDE? I have posted the link.

aaron1959, 2017-09-25:
In DD-WRT power setting is not functional, changes are not actually sent to the firmware as it's locked in that sense so probably a coincidence.

aaron1959, 2017-09-25:
DD-WRT r33006 is VERY very OLD!

BrainSlayer, 2017-09-25:
@aaron1959 power setting is not functional for lede too. the chipset does not support power setting

yuhhaurlin, 2017-09-25:
1. RCU stall is fixed by 0810, no matter for LEDE or dd-WRT.
2. Host command timeout only reported by one user on dd-WRT. I hope this problem can be reproduced on LEDE, so I can try to reproduce and debug it.
3. Please don't report problems on previous version of driver.
Thanks.

yuhhaurlin, 2017-09-25:
Before you report problem, please check what version of mwlwifi tested by you.

aaron1959, 2017-09-25:
@BrainSlayer  "@aaron1959 power setting is not functional for lede too. the chipset does not support power setting" yes I understand that it i now locked for everyone per F.C.C. rules?

aaron1959, 2017-09-25:
@yuhhaurlin  if you can borrow a Galaxy S7 running Nougat and test your LEDE @ 5 Ghz with no security you can find the answer yourself if you would choose.

yuhhaurlin, 2017-09-25:
@aaron1959 Please reproduce this problem on LEDE first. Your test environment can hit this problem on dd-WRT, it should be easier to reproduce the same problem on LED by you.

aaron1959, 2017-09-25:
I don't work for Marvell.

[54 more replies]