kaloz/mwlwifi #218: Interpret pwrtbl

lantis1008, 2017-09-26:
Hi @yuhhaurlin ,

Are you able to provide information on how these tables translate into transmission power values?
e.g.
`AU =	<36 0 0xd 0xd 0xd 0xd 0xd 0xd 0xd 0xd 0xe 0xe 0xe 0xe 0xe 0xe 0xe 0xe 0 0xf>,`
How does this line from the DTS for mamba for channel 36 in Australia relate to
`(5150 - 5250 @ 80), (23), NO-OUTDOOR, AUTO-BW`
this regulatory limit?

Do these numbers somehow add to 23dBm?

yuhhaurlin, 2017-09-26:
Sorry. I have no idea about that. These numbers are provided by Belkin.

closed 2017-09-26 by yuhhaurlin

lantis1008, 2017-09-26:
Thanks anyway

jbsky, 2017-09-26:
Allow me, it is possible to overwrite the power table by modifying the driver. Attention, it works with the WRT1900AC router. It seems to me that it can work when the power table is loaded from the dts file. 
I attach here the link to my patch :
https://github.com/jbsky/Debian-On-WRT1900AC-V1/blob/master/kernel-4.12.9/ability_to_change_the_signal_strength.patch
try the commands:
iw wlan1 set txpower fixed 0
iw wlan1 set txpower fixed 2500
Verifiable with the Acrylic application.

lantis1008, 2017-09-26:
Thanks. 
I would still like to verify the provided power table, but this is interesting.

lantis1008, 2017-09-29:
@brainslayer do you have any additional information from bin walks etc?

BrainSlayer, 2017-10-01:
nope. never tried to reverse engineer the firmware