kaloz/mwlwifi #220: Test request to users who get wifi issues

wongsyrone, 2017-09-30:
These days, one of LEDE developer, hauke, replaces the old compat-wireless with new backports-4.14-rc2. After 14 hours testing with latest hostapd git HEAD, I have no issues with the previously reported host command timed out on WRT3200ACM. It would be better if others can test this on more devices.

I talked with him and asked when it will be pushed to trunk, he said "I plan to get this into the master branch soon, would be nice if it gets some testing before so that I do not break so many devices."

You can find the most recent version of this patch series in his staging
branch:
https://git.lede-project.org/?p=lede/hauke/staging.git;a=shortlog;h=refs/heads/mac80211

It runs normally, no host command timed out and so far so good.

```
root@LEDE:~# dmesg 
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.52 (wong@wong-Arch) (gcc version 7.2.0 (LEDE GCC 7.2.0 r5606+50-6f16cd166e) ) #0 SMP Sat Sep 30 03:47:05 2017
[    0.000000] CPU: ARMv7 Processor [414fc091] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Linksys WRT3200ACM
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat c0663d00, node_mem_map dfbf9000
[    0.000000]   Normal zone: 1024 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] percpu: Embedded 13 pages/cpu @dfbce000 s22796 r8192 d22260 u53248
[    0.000000] pcpu-alloc: s22796 r8192 d22260 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 130048
[    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/mtdblock8 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,128K(u_env),256K(s_env),256K@8064K(devinfo),1920K@8320K(sysdiag),80m@10m(kernel),74m@16m(rootfs),80m@90m(alt_kernel),74m@96m(alt_rootfs),160m@10m(ubifs),-@170m(syscfg)
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 512728K/524288K available (4663K kernel code, 177K rwdata, 1240K rodata, 284K init, 306K bss, 11560K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0496178   (4665 kB)
[    0.000000]       .init : 0xc05f1000 - 0xc0638000   ( 284 kB)
[    0.000000]       .data : 0xc0638000 - 0xc0664600   ( 178 kB)
[    0.000000]        .bss : 0xc0664600 - 0xc06b10a0   ( 307 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] mvebu_mbus: [Firmware Warn]: deprecated mbus-mvebu Device Tree, suspend/resume will not work
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 D prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 Coherent cache controller enabled, 16 ways, 1024 kB
[    0.000000] L2C-310 Coherent: CACHE_ID 0x410054c9, AUX_CTRL 0x56070001
[    0.000000] Selected CPU frequency (16) unsupported
[    0.000000] Switching to timer-based delay loop, resolution 40ns
[    0.000003] sched_clock: 32 bits at 25MHz, resolution 40ns, wraps every 85899345900ns
[    0.000009] clocksource: armada_370_xp_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 76450417870 ns
[    0.000112] Calibrating local timer... 933.11MHz.
[    0.060027] Calibrating delay loop (skipped), value calculated using timer frequency.. 50.00 BogoMIPS (lpj=250000)
[    0.060032] pid_max: default: 32768 minimum: 301
[    0.060070] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060074] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060271] CPU: Testing write buffer coherency: ok
[    0.060359] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.060370] Setting up static identity map for 0x8280 - 0x82d8
[    0.060429] mvebu-soc-id: MVEBU SoC ID=0x6820, Rev=0x4
[    0.060488] mvebu-pmsu: Initializing Power Management Service Unit
[    0.060895] Booting CPU 1
[... 442 lines cut ...]

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b3:35:6a
2g: disable
5g: enable
antenna: 4 4
irq number: 45
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000000
radio: disable
iobase0: e1080000
iobase1: e1300000
tx limit: 1024
rx limit: 16384

```

BTW, I am using https://w1.fi/cgit/hostap/commit/?id=ee522d27cf3e23cc7d42eeaa151aee97e708b18e

s-pimenta, 2017-10-01:
If you compile, I can test on my WRT3200ACM...

wongsyrone, 2017-10-01:
It contains many packages you might not need. I will upload if you don't care.

So far so good.

```
root@LEDE:~# uptime 
 20:05:08 up 1 day,  6:40,  load average: 0.00, 0.00, 0.00
```

wongsyrone, 2017-10-01:
OK. It's already in LEDE master branch: https://github.com/lede-project/source/commit/a8f63a0717f553e0a1b37ee9212fc4cb2a801426

closed 2017-10-01 by wongsyrone

BrainSlayer, 2017-10-01:
the current dd-wrt version published on my ftp site uses the latest mac80211 backport too (since early september)