kaloz/mwlwifi #221: Firmware Glitch - LEDE

rs-se, 2017-10-05:
Since switching to LEDE 5Ghz wifi one phy0 has gone from being solid 99.9% of the time to having slower than normal speeds.

Current log log shows a mwl firmware error.

Oct 04 20:52:10 WRT3200ACM kernel:  [98035.911327] mwifiex_sdio mmc0:0001:1: cmd_wait_q terminated: -512

rs-se, 2017-10-05:
root@WRT3200ACM:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170606
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b9:8d:d2
2g: disable
5g: enable
antenna: 4 4
irq number: 105
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0e00000
iobase1: e1080000
tx limit: 1024
rx limit: 16384

rs-se, 2017-10-05:
I discovered after posting this is related to "mwifiex" not related to phy0. 

Still investigating slow wifi speeds!

yuhhaurlin, 2017-10-05:
Sorry. It is not related to mwlwifi.

closed 2017-10-05 by yuhhaurlin

kubrickfr, 2017-10-05:
@rs-se Your driver version is outdated as you can see in the info you posted. You have 

> driver version: 10.3.4.0-20170606

It should read:
> driver version: 10.3.4.0-20170810