kaloz/mwlwifi #222: Slow 5Ghz Speeds - Latest Driver - LEDE

rs-se, 2017-10-05:
root@WRT3200ACM:~# cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/info

driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170606
firmware version: 0x09030007
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:b9:8d:d2
2g: disable
5g: enable
antenna: 4 4
irq number: 105
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0e00000
iobase1: e1080000
tx limit: 1024
rx limit: 16384

Have no logs of why it is happening, occurring on LEDE only on all channels.

aaron1959, 2017-10-05:
I'm not sure but I really thought that driver was no longer being used? this is current as far as I know.
driver name: mwlwifi
chip type: 88W8964
hw version: 7
driver version: 10.3.4.0-20170810
firmware version: 0x09030008
power table loaded from dts: no
firmware region code: 0x10
mac address: 60:38:e0:13:64:fa
2g: enable
5g: enable
antenna: 4 4
irq number: 46
ap macid support: 0000ffff
sta macid support: 00010000
macid used: 00000001
radio: enable
iobase0: e0c00000
iobase1: e0e80000
tx limit: 1024
rx limit: 16384

yuhhaurlin, 2017-10-05:
What do you mean? The driver is running with good throughput!

closed 2017-10-05 by yuhhaurlin

thagabe, 2017-10-05:
@rs-se 
Use 20170810
Are you using release lede or master branch? Kernel? 
If you'd like you can use one of my self-compiled images for rango

rs-se, 2017-10-05:
Hi Gabe,

I am using LEDE release 17.0.1.2 with latest opkg updates.

I am on phone right now, I can give your images a try. Throughout isn't a
problem  stability and constant speed has been the issue.

<https://github.com/kaloz/mwlwifi/issues/222#issuecomment-334365072>, or mute
the thread

yuhhaurlin, 2017-10-08:
Please use the latest driver and show me what problem you encounter.

You can get LEDE image from here:
https://davidc502sis.dynamic-dns.net/